morrownr/USB-WiFi #416: Testing the wifi6 conf

fs30000, 2024-03-30:
Hello guys,

So far i've been running the AP with wifi5. I decided to try the wifi6 conf here.

**Details:**
Chinese mini pc
MT7922 card with 2 crappy antennas inside the pc itself.
Rocky Linux 9.3
Kernel 6.8.2-1.el9.elrepo.x86_64

**Clients:**
Oneplus 11 phone
Laptop with intel AX201

**Results** with iperf, 3 meters from AP, direct line:

O11
Normal mode:  Half of times 33 MB/s, sometimes 40 MB/s, one or two times, 60-70 MB/S
Reverse mode: Almost always 75 MB/s

AX201
Normal mode: Most times 38-37 MB/s, other times around 42 MB/s
Reverse mode: Most times: 45-50 MB/s

So, reverse mode is great, but sending results are kind of unstable. Anyway, no biggie for me, as this isn't my main AP. I got a xiaomi AX1800 (white model) for that. Amazing router for 30€!  I think this card is capable of 2x 80Mhz streams, but hostapd errors when setting [VHT160-80PLUS80]. Wifi analyzer on the phone shows the AP runs at 80 Mhz. Where is the 160Mhz?!

Anyway, this post if just for reporting. hostapd conf file here:

````interface=wlp4s0
bridge=br0
driver=nl80211

ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

ssid=AP1
wpa_passphrase=PASSWORDHERE
country_code=PT

auth_algs=3
macaddr_acl=0
ignore_broadcast_ssid=0
wpa=2

##### needed for wifi 6
wpa_key_mgmt=SAE
ieee80211w=2
wpa_pairwise=CCMP CCMP-256
rsn_pairwise=CCMP CCMP-256

ieee80211d=1
ieee80211h=1

hw_mode=a
channel=36

############ 2.4Ghz
ieee80211n=1
wmm_enabled=1

ht_capab=[RXLDPC][HT40+][SHORT-GI-20][SHORT-GI-40][GF][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40][SUBEAMFORMEE]

############ 5Ghz
ieee80211ac=1

vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

vht_capab=[RXLDPC][HT40+][VHT160][SHORT-GI-20][SHORT-GI-40][SHORT-GI-80][SHORT-GI-160][TX-STBC][RX-STBC12][MAX-AMSDU-7935][DSSS_CCK-40][MAX-MPDU-11454][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][MAX-A-MPDU-LEN-EXP7]

################ AX
ieee80211ax=1
he_oper_chwidth=1
he_oper_centr_freq_seg0_idx=42

he_bss_color=42
````

morrownr, 2024-03-31:
Hi @fs30000 

> I think this card is capable of 2x 80Mhz streams, but hostapd errors when setting [VHT160-80PLUS80]. Wifi analyzer on the phone shows the AP runs at 80 Mhz. Where is the 160Mhz?!

Evidently the driver supports 160 MHz in managed mode but only 80 MHz in AP mode. I have a mt7922 based PCIe card in my main dev box but have never tested AP mode. It will do 160 in managed mode and I regularly see 1+ Gbps with iperf3 on the connection to my main wifi router. Faster than 1 Gb ethernet. Cool. However, a user mentioned sometime in the last few months that this driver only supports 80 for AP mode. I don't know if the intent is to expand to 160 at some point.

morrownr, 2024-03-31:
Regarding hostapd. My guide to get hostapd working with WiFi 6:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Upgrade_hostapd.md

The guide is for RasPiOS, which is Debian based. 

The following is what I used and it seemed to work well but if you see any needed changes, let me know.

```
CONFIG_DRIVER_HOSTAP=y
CONFIG_DRIVER_NL80211=y
CONFIG_LIBNL32=y
CONFIG_RSN_PREAUTH=y
CONFIG_EAP=y
CONFIG_ERP=y
CONFIG_EAP_MD5=y
CONFIG_EAP_TLS=y
CONFIG_EAP_MSCHAPV2=y
CONFIG_EAP_PEAP=y
CONFIG_EAP_GTC=y
CONFIG_EAP_TTLS=y
CONFIG_IPV6=y

CONFIG_IEEE80211R=y
CONFIG_WNM=y
CONFIG_IEEE80211AC=y
CONFIG_IEEE80211AX=y
CONFIG_SAE=y
CONFIG_SAE_PK=y
CONFIG_DEBUG_FILE=y
CONFIG_NO_RANDOM_POOL=y
CONFIG_GETRANDOM=y
CONFIG_ELOOP_EPOLL=y
CONFIG_ACS=y
CONFIG_NO_TKIP=y
```

morrownr, 2024-03-31:
Notes:

> ieee80211h=1

Only needed for DFS channels in AP mode. Not sure if the card supports DFS channels so this may not be needed.

You only need to mandate WPA3 for WiFi 6e. For WiFi 6, you can use WPA 2 or mixed.

fs30000, 2024-04-01:
Thanks for your input @morrownr. 80Mhz in AP mode makes sense. Still a nice upgrade from my anterior wifi5 config. Even more signal! BTW, WPA3 is also a nice addition. More secure yes?

castillofrancodamian, 2024-04-01:
Does anyone know how to configure it to broadcast at 20 MHz?

fs30000, 2024-04-01:
> Does anyone know how to configure it to broadcast at 20 MHz?

Just run it at 2.4g i guess. Possible conf:

````bridge=br0
driver=nl80211

ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

ssid=AP1
wpa_passphrase=PASSWORDHERE
country_code=PT

auth_algs=3
macaddr_acl=0
ignore_broadcast_ssid=0
wpa=2

##### needed for wifi 6
wpa_key_mgmt=WPA-PSK WPA-EAP
ieee80211w=1

ieee80211d=1
ieee80211h=1

hw_mode=g
channel=0

############ 2.4Ghz
ieee80211n=1
wmm_enabled=1

ht_capab=[RXLDPC][SHORT-GI-20][GF][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40][SUBEAMFORMEE]````

morrownr, 2024-04-01:
> Does anyone know how to configure it to broadcast at 20 MHz?

Yes, I do. What band and N, AC or AX?

Post a hostpad.conf and I try to offer some advice.

castillofrancodamian, 2024-04-01:
> Just run it at 2.4g i guess. Possible conf:

That is not a solution because I am looking to broadcast in the 5 GHz band and with all its advantages.

I tried changing the following lines:

```
ssid=myPI-WiFi6
wpa_passphrase=myPW1234
channel=36
chanlist=36
country_code=US
vht_oper_chwidth=1 (Wi-Fi 5 section)
vht_oper_centr_freq_seg0_idx=42 (Wi-Fi 5 section)
he_oper_chwidth=1 (Wi-Fi 6 section)
he_oper_centr_freq_seg0_idx=42 (Wi-Fi 6 section)
```

To this:

```
ssid=FDC
wpa_passphrase=1111111111
channel=165
chanlist=165
country_code=AR
#vht_oper_chwidth=1 (Wi-Fi 5 section)
vht_oper_centr_freq_seg0_idx=165 (Wi-Fi 5 section)
#he_oper_chwidth=1 (Wi-Fi 6 section)
he_oper_centr_freq_seg0_idx=165 (Wi-Fi 6 section)
```

I get this:

```
[franco@pc ~]$ sudo systemctl status hostapd
● hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: disabled)
     Active: active (running) since Mon 2024-04-01 19:06:36 -03; 1s ago
   Main PID: 2302 (hostapd)
      Tasks: 1 (limit: 9128)
     Memory: 1.0M (peak: 1.1M)
        CPU: 31ms
     CGroup: /system.slice/hostapd.service
             └─2302 /usr/bin/hostapd /etc/hostapd/hostapd.conf

abr 01 19:06:36 pc systemd[1]: Started Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
abr 01 19:06:37 pc hostapd[2302]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
abr 01 19:06:37 pc hostapd[2302]: Frequency 5845 (secondary) not allowed for AP mode, flags: 0x1
abr 01 19:06:37 pc hostapd[2302]: Frequency 5845 (secondary) not allowed for AP mode, flags: 0x1
abr 01 19:06:37 pc hostapd[2302]: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
abr 01 19:06:37 pc hostapd[2302]: HT40 channel pair (165, 161) not allowed
abr 01 19:06:37 pc hostapd[2302]: Fallback to 20 MHz
abr 01 19:06:37 pc hostapd[2302]: wlan0: interface state HT_SCAN->ENABLED
abr 01 19:06:37 pc hostapd[2302]: wlan0: AP-ENABLED
```

The client shows that it is connected at 20 MHs, but when viewing the `hostapd` status, it outputs at 20 MHz due to a configuration error.

Now if I try to change the following lines to use in another channel in which a channel width greater than 20 MHz can be used, as in 36, I get another result:

```
ssid=myPI-WiFi6
wpa_passphrase=myPW1234
country_code=US
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
he_oper_chwidth=1
he_oper_centr_freq_seg0_idx=42
```

To this:

```
ssid=FDC
wpa_passphrase=1111111111
country_code=AR
#vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=36
#he_oper_chwidth=1
he_oper_centr_freq_seg0_idx=36
```

I get this:

```
[franco@pc ~]$ sudo systemctl status hostapd
○ hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: disabled)
     Active: inactive (dead) since Mon 2024-04-01 19:17:42 -03; 2s ago
   Duration: 299ms
    Process: 2634 ExecStart=/usr/bin/hostapd /etc/hostapd/hostapd.conf (code=exited, status=0/SUCCESS)
   Main PID: 2634 (code=exited, status=0/SUCCESS)
        CPU: 12ms

abr 01 19:17:42 pc hostapd[2634]: Could not set channel for kernel driver
abr 01 19:17:42 pc hostapd[2634]: Interface initialization failed
abr 01 19:17:42 pc hostapd[2634]: wlan0: interface state HT_SCAN->DISABLED
abr 01 19:17:42 pc hostapd[2634]: wlan0: AP-DISABLED
abr 01 19:17:42 pc hostapd[2634]: wlan0: interface state DISABLED->DISABLED
abr 01 19:17:42 pc hostapd[2634]: wlan0: AP-DISABLED
abr 01 19:17:42 pc hostapd[2634]: wlan0: CTRL-EVENT-TERMINATING
abr 01 19:17:42 pc hostapd[2634]: hostapd_free_hapd_data: Interface wlan0 wasn't started
abr 01 19:17:42 pc hostapd[2634]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
abr 01 19:17:42 pc systemd[1]: hostapd.service: Deactivated successfully.
```

The client does not see the new network.

castillofrancodamian, 2024-04-01:
> > Does anyone know how to configure it to broadcast at 20 MHz?
> 
> Yes, I do. What band and N, AC or AX?
> 
> Post a hostpad.conf and I try to offer some advice.

<details><summary>/etc/hostapd/hostapd-WiFi6.conf</summary>
<p>

```
# /etc/hostapd/hostapd-WiFi6.conf
# Documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf
# 2024-03-26
# Status: rc4, please test and report.
# Note: This hostapd.conf file is for WiFi 6, not WiFi 6e (6 GHz).

# Purpose: hostapd.conf example for WiFi 6 configuration
# - specific to mt7921au chipset based USB WiFi adapters
# - tested on RasPi4B with 64 bit RasPiOS, 2023-12-05
# - see https://github.com/morrownr/USB-WiFi for...
#   -- make sure mt7921 firmware files are installed/updated
#   -- you need hostap 2.10 or later for WiFi 6 support: $ hostapd -v
# - improvements are welcome

# SSID
ssid=FDC
# PASSPHRASE
wpa_passphrase=1111111111
# Band: a = 5GHz & 6GHz (a/n/ac/ax), g = 2Ghz (b/g/n)
hw_mode=a
# Channel availability varies by country
# https://en.wikipedia.org/wiki/List_of_WLAN_channels
# US:
# 2.4 GHz (1-11)
# 5 GHz ((36(42) and 149(155))
# 5 GHz DFS ((52(58), 100(106), 116(122), 132(138))
# 6 GHz (see op_class below)
# Channel
channel=36
chanlist=36
# The operating class is an indication of the 6 GHz channel size. An
# operating class of 134 indicates a 160 MHz channel. Likewise, 133
# denotes an 80 MHz channel, 132 indicates a 40 MHz channel, and 131
# indicates a channel size of 20 MHz.
# 133 signifies channel width: 80 MHz
#   channels 7, 23, 39, 55, 71, 87, 103, 119, 135, 151, 167, 183, 199, 215
# 134 signifies channel width: 160 MHz
#   channels 15, 47, 79, 111, 143, 175, 207
#op_class=133
# Country code
# https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
country_code=AR
# All environments
#country3=0x20
# Outdoor
#country3=0x4f
# Indoor environment only
#country3=0x49
# Advertises the country_code and the set of allowed channels and transmit power levels
ieee80211d=1
# Enables support for 5GHz DFS channels if supported (requires ieee80211d=1)
#ieee80211h=1

# Bridge interface
bridge=br0
# WiFi interface (wlan0 should be changed to your interface name)
interface=wlan0

# Set hostapd driver (nl80211 is used with all Linux mac80211 (in-kernel) and modern Realtek drivers)
driver=nl80211

# hostapd event logger configuration
#
# Module bitfield (ORed bitfield of modules that will be logged; -1 = all
# modules):
# bit 0 (1) = IEEE 802.11
# bit 1 (2) = IEEE 802.1X
# bit 2 (4) = RADIUS
# bit 3 (8) = WPA
# bit 4 (16) = driver interface
# bit 6 (64) = MLME
#
# Levels (minimum value for logged events):
#  0 = verbose debugging
#  1 = debugging
#  2 = informational messages
#  3 = notification
#  4 = warning
#
logger_syslog=1
logger_syslog_level=2

# Control interface
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

# Various settings
beacon_int=100
dtim_period=2
multicast_to_unicast=0
#tx_queue_data2_aifs=1
#tx_queue_data2_cwmin=7
#tx_queue_data2_cwmax=15
#tx_queue_data2_burst=3.0
tx_queue_data2_burst=2.0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=0
skip_inactivity_poll=1
preamble=1

# Security
# authentication algorithms (3 = both are allowed, use 3 for SAE)
auth_algs=3
# MAC address -based authentication (0 = accept unless in deny list)
macaddr_acl=0
# hidden SSID (1 = hidden)
ignore_broadcast_ssid=0
# Opportunistic Key Caching (1 = enabled)
okc=1
# Enable WPA. (2 is required for WPA2, mixed and WPA3)
wpa=2
# Pairwise cipher
wpa_pairwise=CCMP CCMP-256
rsn_pairwise=CCMP CCMP-256
#
# Activate only one of the following 3 sections
#
# 1. WPA2 Personal
#wpa_key_mgmt=WPA-PSK WPA-PSK-SHA256
#
# 2. WPA2/WPA3 Transitional (mixed)
#wpa_key_mgmt=SAE WPA-PSK WPA-PSK-SHA256
#ieee80211w=1
#sae_require_mfp=1
#
# 3. WPA3 Personal (required for WiFi 6e - 6 GHz)
wpa_key_mgmt=SAE
ieee80211w=2

# WiFi 4 (band 2.4 Ghz)
ieee80211n=1
wmm_enabled=1
#
# generic setting - 20 MHz channel width
#ht_capab=[SHORT-GI-20]
#
# generic setting - 40 MHz channel width (if allowed)
#ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40]
#
# mt7921au  (HT capabilities 0x9ff)
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
#

# WiFi 5 (5 GHz)
ieee80211ac=1
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
#vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=36
#
# generic setting
#vht_capab=[SHORT-GI-80]
#
# mt7921au (VHT capabilities 339071b2)
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]
#

# WiFi 6 (bands 2.4 Ghz, 5 GHz and 6 GHz)
ieee80211ax=1
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
#he_oper_chwidth=1
# HE center channel (chan + 6)
he_oper_centr_freq_seg0_idx=36
# used by clients to discern the source of interference
# each AP in your area needs to use a different number
# allowed: 1-63
he_bss_color=42
#he_default_pe_duration=4
#he_rts_threshold=1023
#he_mu_edca_qos_info_param_count=0
#he_mu_edca_qos_info_q_ack=0
#he_mu_edca_qos_info_queue_request=0
#he_mu_edca_qos_info_txop_request=0
#he_mu_edca_ac_be_aifsn=8
#he_mu_edca_ac_be_aci=0
#he_mu_edca_ac_be_ecwmin=9
#he_mu_edca_ac_be_ecwmax=10
#he_mu_edca_ac_be_timer=255
#he_mu_edca_ac_bk_aifsn=15
#he_mu_edca_ac_bk_aci=1
#he_mu_edca_ac_bk_ecwmin=9
#he_mu_edca_ac_bk_ecwmax=10
#he_mu_edca_ac_bk_timer=255
#he_mu_edca_ac_vi_ecwmin=5
#he_mu_edca_ac_vi_ecwmax=7
#he_mu_edca_ac_vi_aifsn=5
#he_mu_edca_ac_vi_aci=2
#he_mu_edca_ac_vi_timer=255
#he_mu_edca_ac_vo_aifsn=5
#he_mu_edca_ac_vo_aci=3
#he_mu_edca_ac_vo_ecwmin=5
#he_mu_edca_ac_vo_ecwmax=7
#he_mu_edca_ac_vo_timer=255

# End of hostapd-WiFi6.conf
```

</p>
</details>

morrownr, 2024-04-01:
Change the ht_capab line to:
ht_capab=[LDPC][SHORT-GI-20][TX-STBC][RX-STBC1][MAX-AMSDU-7935]

Then delete or comment all lines after the above.

castillofrancodamian, 2024-04-01:
@morrownr I have changed the line `ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935 ]` to `ht_capab=[LDPC][SHORT-GI-20][TX-STBC][RX-STBC1][MAX-AMSDU-7935]` and now the 20 MHz channels do work.

morrownr, 2024-04-01:
Muy bien