morrownr/USB-WiFi #418: RT8812AU How to enable 802.11v and 802.11k

Totix92, 2024-04-07:
The drivers on Windows activate it, on Linux (Raspberry Pi OS) they don't, how do I activate them? if they can be activated.

morrownr, 2024-04-07:
Which driver are you using?

Totix92, 2024-04-07:
> Which driver are you using?

This: https://github.com/morrownr/8812au-20210820

morrownr, 2024-04-07:
I grepped around in the code and found some lines that suggest it is supported. I did not find any switches to turn it on or off. Maybe someone else that uses this will have more information.

As much as I like my rtl8812au based adapter, I have resigned myself to the fact that Mediatek just makes better usb wifi drivers than does Realtek so most of my adapters these days use Mediatek chips.

Totix92, 2024-04-07:
I am aware, I had this wifi dongle stored, if I had to buy a new one I would get a Mediatek.

morrownr, 2024-04-07:
There is an effort to add rtl8812au support to rtw88. Can't say when it will be ready but maybe there is hope. I'll post in the repo you are using when and if it needs to be tested.

Totix92, 2024-04-08:
Thanks, can't wait to try it.