morrownr/USB-WiFi #422: mt7921au: Client Mode WDS/4addr not working

ntzb, 2024-04-19:
Hi,
to not mix the AP mode issue at https://github.com/morrownr/USB-WiFi/issues/392 I'm starting a new issue.
I tried to activate 4addr on regular client mode, but it refused working.
my device is [FU-AX1800](https://fenvi.com/product_detail_44.html), with latest firmware, and chip mt7921au.

what I tried:
1. multiple OS (debian, arch, clear linux) on a celeron mini pc.
2. take down the interface before doing any changes `ip link set dev wlx90de80d6883f down`
3. setting 4addr on the interface itself: `iw dev wlx90de80d6883f set 4addr on`
4. setting 4addr on a new pseudo interface: `iw dev wlx90de80d6883f interface add myNewIface type managed 4addr on`

in all cases I got:
`command failed: Operation not supported (-95)`

@morrownr since you successfully got it to work, can you please share your full env and steps taken?

ntzb, 2024-04-29:
@morrownr any help will be appreciated

morrownr, 2024-04-29:
@ntzb 

I'm watching but my inability to dig deeper into this issue is due to some health issues in my family so my time is llimited. Hopefully, at some point, I have more time. There are some pretty smart people that stop by here not necessarily on a daily basis as this is not a social media site but over time some help should come along.

ntzb, 2024-04-30:
of course, sorry.
all the best

ntzb, 2024-05-13:
decided to give it another go with OpenWRT, same thing:
```
daemon.notice hostapd: Set new config for phy phy0:
daemon.notice wpa_supplicant[1635]: Set new config for phy phy0
daemon.notice netifd: radio0 (7858): sh: out of range
daemon.notice wpa_supplicant[1635]: Set new config for phy phy0
daemon.notice hostapd: Set new config for phy phy0:
daemon.notice wpa_supplicant[1635]: Set new config for phy phy0
daemon.notice wpa_supplicant[1635]: Failed to create device phy0-sta0: Operation not supported
daemon.err wpa_supplicant[1635]: Could not read interface phy0-sta0 flags: No such device
dacmon.notice upa_supplicant[1635]: n180211: Driver does not support authentication/association
daemon.notice wpa_supplicant[1635]: n180211: deinit ifname=phy0-sta0 disabled_11b_rates=0
daemon.err wpa_supplicant[1635]: Could not read interface phy0-sta0 flags: No such device
daemon.err wpa_supplicant[1635]: phy0-sta0: Failed to initialize driver interface
```
can the x86 hardware itself block the ability to use 4addr?
all of my tries were done on the same PC

bjlockie, 2024-05-13:
`rfkill`maybe?

bjlockie, 2024-05-13:
```
$ sudo rfkill
ID TYPE      DEVICE              SOFT      HARD
 0 bluetooth acer-bluetooth   blocked unblocked
 1 wlan      phy0           unblocked unblocked
```

aleyowl, 2025-01-20:
I have the same issue with mt7921 wifi card. 
```
sudo  iw dev wlan0 set 4addr on
command failed: Operation not supported (-95)
```
also rfkill output is
```
rfkill list all
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
```