morrownr/USB-WiFi #424: News: OpenWrt One WiFi 6 router board

morrownr, 2024-04-22:
Earlier this year I learned that Banana Pi and OpenWRT were teaming up to release a low cost wifi 6 router. The board is designed by OpenWRT and will be producted and marketed by Banana Pi. Here is an article to update us on the status:

https://www.cnx-software.com/2024/04/17/first-openwrt-one-router-board-samples-auction-openwrt-summit/

```
OpenWrt One router specifications:

SoC – MediaTek MT7981B (Filogic 820) dual-core Cortex-A53 processor @ 1.3 GHz
System Memory – 1GB DDR4

Storage
128 MB SPI NAND flash for U-boot and Linux
4 MB SPI NOR flash for write-protected (by default) recovery bootloader (reflashing can be enabled with a jumper)
Two types of flash devices are used to make the board almost unbrickable
M.2 2242/2230 socket for NVMe SSD (PCIe gen 2 x1)

Networking
2.5GbE RJ45 port
Gigabit Ethernet RJ45 port
Dual-band WiFI 6 via MediaTek MT7976C (2×2 2.4 GHz + 3×3/2×2 + zero-wait DFS 5Ghz)
3x MMCX antenna connectors

USB
1x USB 2.0 Type-A host port
USB Type-C (device, console) port using Holtek HT42B534-2 UART to USB chip

Expansion – MikroBUS socket for expansion modules
Debugging – Console via USB-C port or 3-pin header, 10-pin JTAG/SWD header for main SoC

Misc
Reset and User buttons
Boot select switch: NAND (regular) or NOR (recovery)
2x PWM LEDs, 2x Ethernet LED (GPIO driven)
EM6324 External hardware watchdog
NXP PCF8563TS (I2C) RTC with battery backup holder for CR1220 coin-cell

Power Supply
12V USB-PD on USB-C port (might have changed to up to 15V)
Optional 802.3at/af PoE via RT5040 module
Dimensions – 148 x 100.5 mm compatible with Banana Pi BPI-R4 case design
Certifications – FCC/EC/RoHS compliance
```

My WiFi 6 router is based on the MediaTek MT7981B SoC. I run OpenWRT on it. It works well. OpenWRT One will have several cool features that mine does not have.

@morrownr

morrownr, 2024-04-27:
Here is an iperf3 test with my little WiFi 6 router that uses the same SoC  as OpenWRT One:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.127 port 34398 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   128 MBytes  1.08 Gbits/sec    0    991 KBytes       
[  5]   1.00-2.00   sec   130 MBytes  1.09 Gbits/sec    0    991 KBytes       
[  5]   2.00-3.00   sec   135 MBytes  1.13 Gbits/sec    0    991 KBytes       
[  5]   3.00-4.00   sec   130 MBytes  1.09 Gbits/sec    0    991 KBytes       
[  5]   4.00-5.00   sec   130 MBytes  1.09 Gbits/sec    0    991 KBytes       
[  5]   5.00-6.00   sec   131 MBytes  1.10 Gbits/sec    0    991 KBytes       
[  5]   6.00-7.00   sec   130 MBytes  1.09 Gbits/sec    0    991 KBytes       
[  5]   7.00-8.00   sec   130 MBytes  1.09 Gbits/sec    0    991 KBytes       
[  5]   8.00-9.00   sec   132 MBytes  1.11 Gbits/sec    0    991 KBytes       
[  5]   9.00-10.00  sec   130 MBytes  1.09 Gbits/sec    0    991 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.28 GBytes  1.10 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  1.27 GBytes  1.09 Gbits/sec                  receiver

```

My client system has a PCIe card with a mt7922 chip in it. Channel is 116. Channel width is 160.

Yup, wireless faster than 1 Gb ethernet.

morrownr, 2024-04-27:
Here is a ping test with my little WiFi 6 router that uses the same SoC as OpenWRT One:

```
$ ping 192.168.1.1
PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.
64 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=0.663 ms
64 bytes from 192.168.1.1: icmp_seq=2 ttl=64 time=0.658 ms
64 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=0.665 ms
64 bytes from 192.168.1.1: icmp_seq=4 ttl=64 time=9.83 ms
64 bytes from 192.168.1.1: icmp_seq=5 ttl=64 time=0.734 ms
64 bytes from 192.168.1.1: icmp_seq=6 ttl=64 time=5.53 ms
64 bytes from 192.168.1.1: icmp_seq=7 ttl=64 time=0.656 ms
64 bytes from 192.168.1.1: icmp_seq=8 ttl=64 time=0.696 ms
64 bytes from 192.168.1.1: icmp_seq=9 ttl=64 time=0.674 ms
...

Average ping time: 4.803

```

amisix, 2024-04-27:
Yeah, that's hot stuff. Thanks for the heads up, looks like something I'd
be interested in. Love me some custom Openwrt hardware where I don't have
to fight a poorly designed USB 3 bus on my RPis.

morrownr, 2024-04-27:
@amisix 

How is it going? I get what you are saying. The RasPi's are fun but some of the hardware choices have made for challenges. 

I follow OpenWRTm especially if I am looking for something new. By last fall, I had keyed in on the SoC and found a router that would get the job done. Mine is more limited than the one Banana Pt and OpenWRT are building but it does get the job done and I am surprised how stable it is. When you look at OpenWRT One, they have added some very handy capabilities that I have never seen in a home router... clock with battery back-up. What? It is designed to be opened up. In fact, I think you have to buy the board, the case and a power supply separately and put it together yourself... kinda like a Pi.

@morrownr

amisix, 2024-04-27:
Things are good. How are things w/you?

I'm still messing about with these routers, built a unique one from a 4g/5g
Waveshare RPi CM4 carrier board but replaced the m.2 4g/5g cell module with
an m.2 Alfa AWM27921U (USB bus) from Rockland. It's a very unique card
imo..

I'm sold. Designed to be built/opened up, count me in. The RTC with battery
backup is rare.. the Waveshare carrier board i used has it interestingly
enough, but you also have to have an RPi CM4 as opposed to the Openwrt
offering.. I'm looking forward to hearing more.

morrownr, 2024-04-28:
@amisix 

> Things are good. How are things w/you?

Things are going slow here, Hopefully things improve.

> I'm looking forward to hearing more.

You always have interesting things going. I'm looking forward to hearing more about this board as well. I hope this project is a success and they follow on with a tri band WiFi 7 board.

fakemanhk, 2024-06-11:
> Love me some custom Openwrt hardware where I don't have to fight a poorly designed USB 3 bus on my RPis.

My tests on RPi4B is showing me horrible experience with the USB3, I cannot imagine how bad it is, now when I need USB I just go to use my NanoPi R4S.

> Earlier this year I learned that Banana Pi and OpenWRT were teaming up to release a low cost wifi 6 router. The board is designed by OpenWRT and will be producted and marketed by Banana Pi

These days BananaPi is doing a lot Mediatek related products, I also have their BPI-R4 (yeah I need 10Gbps) and the spec is really good. The OpenWrt One is interesting but I had enough MT7981 based device (GL-INET MT3000 I liked most) so I would probably skip this one, however if the price is OK I would recommend it to friends.

joopdo, 2024-11-22:
It seems that the OpenWRT One is out, looking forward to see actual benchmarks with the device!

morrownr, 2024-11-22:
Hey @joopdo 

Do you have a link?

joopdo, 2024-11-22:
Not yet. I haven't gotten the device myself.

morrownr, 2024-11-22:
Where did you hear the news?

fakemanhk, 2024-11-23:
OpenWrt One has been on sale in AliExpress for quite some time already

morrownr, 2024-11-23:
https://www.aliexpress.us/item/3256807683891053.html

morrownr, 2024-11-23:
@fakemanhk 

Thanks for the heads up. I hope it is a success and we see a good reasonably priced OpenWRT Two that is a WiFi 7 router.