morrownr/USB-WiFi #425: Raspberry Pi 4 (32bit) WiFi Adapter AWUS036AXML

FWCPI, 2024-04-23:
I recently bought the new WiFi adapter from Alfa (AWUS036AXML). My Raspberry Pi 4 is running the latest kali linux image (kali-linux-2024.1-raspberry-pi-armhf.img.xz).
The adapter is recognized when you enter lsusb, but it is not possible to use the adapter as an interface.
Drivers are installed.

If I have understood correctly so far, the adapter only works with kernel version 5.18+. The Kali linux image comes with a kernel v5.15.44-Re4son-v8l+.

Is there a way to get the Adapter to run or to update the kernel? Or is it currently not possible to use the adapter with the kali image for raspberry pi 4? 
In that case the only solution would be buying a new adapter or a raspberry pi 5?

Thanks for the help.

morrownr, 2024-04-23:
Hi @FWCPI 

> If I have understood correctly so far, the adapter only works with kernel version 5.18+. The Kali linux image comes with a kernel v5.15.44-Re4son-v8l+.

You have understood correctly. It seems to me that a user recently mentioned a new Kali for RasPi4B that uses a newer kernel but I don't remember the details. There should be a new one as that kernel 5.15 is getting dated at this point. I don't have experience with Kali on RasPi but at times I keep a X86_64 version on a test system. It seems to me that almost all of the apps that a Kali user would want to use will work on the RasPiOS 64 bit if installed. The current RasPiOS 64 bit uses kernel 6.6 and works well with your new adapter.

> Is there a way to get the Adapter to run or to update the kernel? Or is it currently not possible to use the adapter with the kali image for raspberry pi 4?

You might ask in a Kali forum about a newer version of Kali for ARM that would include kernel 6.1 or later. There will be many users in the Kali forums that also use the AXML and other adapters based on the mt7921au chipset. There is also the option of the Alfa ACM as it uses a somewhat older chipset (AC1200 class) that will work with the 5.15 kernel. If it were me making the decision, I would burn the latest (2024-03-15) 64 bit version of the RasPiOS, get the adapter going and then install the apps you need... and then start work.

Lastly, keep in mind that your Pi4B is 64 bit so I would use a 64 bit distro.

Let me know if you have additional questions.

@morrownr

FWCPI, 2024-04-25:
Hi @morrownr 

Thanks for the quick response.
I tested the adapter with RasPiOS64 and it worked plug&play as expected.
I need to wait for a new Kali ARM build with newer Kernel Version to get it work with my setup.

Thanks for the advice to just use RasPiOS and get the needed tools, I hadn't thought of that.

Issue solved, i am closing the Issue. 
Thanks!

closed 2024-04-25 by FWCPI

morrownr, 2024-04-25:
Good. I hope you enjoy the adapter and I hope Kali updates their RasPi distro soon.

Something you might want to learn is how to check the firmware version. The firmware goes with the driver to run the adapter. On the Main Meni, see the menu item about firmware. The document will tell you how to check your firmware version and then how to update it if you want to do so.

Cheers