morrownr/USB-WiFi #426: PCIe device doc - Intel AX210 monitor mode

thewade, 2024-04-23:
Regarding your 'PCIe WiFi card information for Linux' page.

The Intel AX210 does support monitor mode with one caveat, it only works 2.4/5 GHz and does not work 6GHz.

This is because Intel uses Location Aware Regulatory to enable support for 6GHz.

While in client mode if you run iw reg get you can see the detected countries and channels. It does this by scanning nearby SSIDs and checking the country info in the beacons as part of 802.11d. 

However in monitor mode it does not work and the country remains UNSET and the channels are blocked.

morrownr, 2024-04-24:
Hi @thewade 

> Regarding your 'PCIe WiFi card information for Linux' page.

Do we have one those pages? I have no problem with us having one but someone else would need to maintain it as I have way too much stuff to do as it is. Can you point me to the page?

thewade, 2024-04-24:
I was refering to https://github.com/morrownr/USB-WiFi/blob/main/home/PCIe_WiFi_Devices.md

If you prefer I can create a pull request updating it.

morrownr, 2024-04-24:
@thewade 

I had forgotten all about that document. It really should be updated.

> If you prefer I can create a pull request updating it.

That would be great. Once your initial PR is merged, I'll look at creating a Main Menu item so users know it is there.

Something else that would be going above and beyond is if you could make a short guide for folks to know how to do a PR for the documents on this site. I get a lot of help but I really need more help and if there was a little guide for users to follow on the Main Menu, I think more users would be inclined to help keep things current.

Regards