morrownr/USB-WiFi #427: Review: EDUP EP-AX1672 (WiFi 6e, mt7921au chipset)

morrownr, 2024-04-24:
Review: (2024-04-24) I now have one of these adapters. It was ordered from the above Amazon link as shown in the Plug and Play list. Here is a quick and dirty initial review:

I tested this adapter for 3 days in client (managed) mode. I pushed it hard with iperf3. The client system is using Debian 12. The adapter was plug and play. I updated the driver firmware per the Firmware guide in the Main Menu. I could not find any problems. The adapter was fast and very stable. I did not notice any thermal related issues even though I pushed it at max WiFi 6 throughput for extended periods. The case of the adapter does have vent holes. When you combine well done vent holes with a chip that runs cool to begin with, there are no thermal problems. I need to do more testing but right now, indications are that it has better than average range and the antennas are removable. I consider that to be a really good feature. As I have time, I will test it with some Alfa antennas to see what happens. For the last 10 days, the adapter has been running in AP mode using my AP mode guide here on the Main Menu. I am using the WiFi 6 hostapd.conf example. AP mode performance has been very good. It is very stable with 5 GHz WiFi 6 AP mode. It has an LED that serves as a powered up indicator similar to what the above Alfa adapters do. The LED is a little brighter than what Alfa uses so remember that if you will be using it in a room where you sleep. At the time of this review, I see an average rating of 4.1 for the Amazon reviews but I think Linux users can ignore that number. When I read the Amazon reviews, it appears that all of the lower rated reviews are from Windows users that are complaining about finding a driver. As Linux users, we don't worry about that since the adapter is plug and play on Linux as long as the kernel you are using is kernel 5.19 or later. My testing is with kernel 6.6 and I have not found any problems that would affect daily use by regular desktop or laptop users. When I ordered this adapter, there was a 20% off clickable coupon so my final price was only about 26 USD. My overall opinion is that this adapter is a good adapter far and the price is good. Most Linux user running modern Linux distros such as Debian 12 or Ubuntu 24.04 or the latest fedora, etc. should find this adapter to be a problem free experience.

@morrownr

jsvnsk, 2024-04-30:
environment:

OS: Ubuntu 22.04.4 LTS
kernel: 6.5.0-28-generic
Wi-Fi adapter: EDUP EP-AX1672
Wi-Fi adapter has been updated with the latest available mt7921u driver from:
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek

----------------------------------------------------------------------------------------------------

issue description: 

1. Wi-Fi adapter occasionally drops out and remains undiscoverable and inoperable until a reboot occurs
2. Wi-Fi adapter dropout seems to occur daily at 22:40

----------------------------------------------------------------------------------------------------

troubleshooting:

1. disabled NetworkManager WiFi Power Saving:
sudo -e /etc/NetworkManager/conf.d/default-wifi-powersave-on.conf

changed "wifi.powersave = 3" to "wifi.powersave = 2"
"wifi.powersave = 3" was the original value (powersave enabled)
"wifi.powersave = 2" is the new changed value (powersave disabled)

2. then restarted the NetworkManager service:
sudo systemctl restart NetworkManager

----------------------------------------------------------------------------------------------------

results:

after disabling powersaving, adapter did not dropout at 22:40

instead, noted the following from journalctl logs:

Apr 29 22:40:37 {some host name goes here} wpa_supplicant[937]: {some NIC name goes here}: WPA: Group rekeying completed with {some MAC address goes here} [GTK=CCMP]

NOTE: {some MAC address goes here} is the MAC address for a Google Nest Wi-Fi Pro mesh router

----------------------------------------------------------------------------------------------------

summary:

if your EDUP EP-AX1672 Wi-Fi adapter drops out and remains undiscoverable and inoperable, try disabling powersaving

----------------------------------------------------------------------------------------------------

trnhx001, 2024-05-05:
Hello, thanks for your good review.
I am having one now (MT7921AU). I wonder if this is really Wifi6E or just Wifi6? Is it working on 6Ghz like and really Tri-band like  ALFA AWUS036AXML (MT7921AUN)?

morrownr, 2024-05-05:
Hi @trnhx001 

> I wonder if this is really Wifi6E or just Wifi6?

I cannot say for sure as I do not have an AP that supports 6 GHz yet. However, the mt7921au chip has been capable of 6 GHz from the beginning and is not something that I think makers would turn off. Maybe someone  else can confirm.

> Also the speed for me on 5Ghz is kinda low, do you experience the same thing?

iperf3 tests here show up to around 650 Mbps on the 5 GHz band using WiFi 6. I have seen no performance problems. If you could provide some details, we might help:

Install `wavemon` and start it. Here is an example for Debian:

$ sudo apt install wavemon
$ wavemon

What does it show for the following?

link quality
signal level
channel
channel width

trnhx001, 2024-05-06:
Hi @morrownr 
I have it plugged in my raspberry pi 5 soft router as an AP.  I did set my channel to 161 to avoid neighbor wifis, and it was probably the problem. 
Sorry, I am all new to this. I am just connecting my iPhone to it and use Speedtest site, nothing fancy. It seems like when I changed it back to 36. It went up to ~400 Mbps. It was only ~150 Mbps on 161.

I installed wavemon on my Raspberry Pi, but saw nothing like you mentioned.

morrownr, 2024-05-09:
@trnhx001 

> It went up to ~400 Mbps. It was only ~150 Mbps on 161.

It sounds like channel 161 was causing you to be limited to a 40 Mhz channel width. This all depends on what channels are allowed in your country. Do we need to explore this further?

trnhx001, 2024-05-09:
@morrownr 
That's interesting. 161 is allowed in my country, but probably something wrong here. I think can do more research on my side.
Thank you for the guidance. We do not need to explore this further.

morrownr, 2024-05-09:
> 161 is allowed in my country...

There is more to it than that. Channel 161 may be allowed but are all channels that would allow it to work with 80 Mhz channel width allowed... that is the issue. Try channel 149 or 100 if allowed in your country.

jsvnsk, 2024-05-16:
@morrownr thanks -- here's my conclusion after testing this device over the past several weeks:

- when using WPA3, the EDUP EP-AX1672 AX30000 Wi-Fi adapter drops out 24 hours after receiving an initial daily GTK rekeying event issued by the WAP (in other words, the very first GTK rekeying event succeeds, but the very next one 24 hours later consistently causes the dropout)
- the WAP is a brand-new Google Nest Wi-Fi Pro mesh Wi-Fi 6E router which supports "WPA3 transition mode" and is configured with WPA3 support enabled and also with the 160 MHz channel available
- the dropout occurs when the client computer's Wi-Fi security configuration is set to "WPA3 Personal", regardless of whether Wi-Fi powersave is disabled
- the dropout does not occur when the client Wi-Fi security configuration is set to "WAP & WPA2 Personal"
- no WPA3 problems occur under this configuration scenario when using an older Realtek 802.11ac Wi-Fi adapter

morrownr, 2024-05-16:
@jsvnsk 

Thanks for the report.

> the AP is a brand-new Wi-Fi 6E-compatible Google Nest Wi-Fi Pro mesh router which supports "WPA3 transition mode" and is configured with WPA3 support enabled and also with the 160 MHz channel available

Can I assume that "WPA3 transition mode" is the same as "WPA3WPA2 mixed mode"?

You might want to keep an eye out for any firmware updates to the Google Nest device. In fact, you might want to do a search to see if others are seeing similar,

> mesh router...

I do not have a mesh capable router so my knowledge is lacking. Are you able to turn mesh off?

> the dropout occurs when the client computer's Wi-Fi security configuration is set to "WPA3 Personal", regardless of whether Wi-Fi powersave is disabled

After the next time it happens, run the following and post the results:

$ sudo dmesg | grep mt7

I'm going to set mine up to test for this problem... I can probably reconfigure within a couple of days. If I do not reply by sometime early next week, ping me here. Sometimes I get very busy.

jsvnsk, 2024-05-17:
> @morrownr 
> 
> Can I assume that "WPA3 transition mode" is the same as "WPA3WPA2 mixed mode"?
> 
WPA3 transition mode allows an AP to work with both WPA2 and WPA3-capable connections, which is conceptually similar to WPA/WPA2 mixed mode but technically different.
> 
> You might want to keep an eye out for any firmware updates to the Google Nest device. In fact, you might want to do a search to see if others are seeing similar,
> 
OK will do
>
> mesh router...
> 
> I do not have a mesh capable router so my knowledge is lacking. Are you able to turn mesh off?
> 
Possibly could try that. A new residential Google Fiber home installation comes with a fiber jack and two Google Nest Wi-Fi Pro mesh routers. The fiber jack is installed in the basement and is connected to the Google broadband fiber optic cable. The two mesh routers act as wireless APs that work together to provide expanded wireless coverage throughout the residence. In this case, the one of the routers is in the basement physically connected to the fiber jack via an ethernet connection, and the other router is in an upstairs office. Devices can connect to either router wirelessly or through an ethernet cable, and it is also possible to add additional mesh routers to further expand the wireless coverage area, if desired.
> 
> > the dropout occurs when the client computer's Wi-Fi security configuration is set to "WPA3 Personal", regardless of whether Wi-Fi powersave is disabled
> 
> After the next time it happens, run the following and post the results:
> 
> $ sudo dmesg | grep mt7
> 
OK I will retest and send the dmesg results if the rekeying causes a dropout again (ETA circa 2024-05-19 Sunday morning CDST or some time thereafter).
> 
> I'm going to set mine up to test for this problem... I can probably reconfigure within a couple of days. If I do not reply by sometime early next week, ping me here. Sometimes I get very busy.
> 
Cool -- thanks !!!

jsvnsk, 2024-05-19:
Hi @morrownr --- here's the requested dmesg results (along with some related logs):

sudo dmesg | grep mt7
 
```
[   23.172416] usbcore: registered new interface driver mt7921u
[   23.572134] mt7921u 4-5:1.0: HW/SW Version: 0x8a108a10, Build Time: 20240219110958a
[   23.584965] mt7921u 4-5:1.0: WM Firmware Version: ____010000, Build Time: 20240219111038
[   25.245055] mt7921u 4-5:1.0 wlxadapterNameGoesHere: renamed from wlan0
```

journalctl --since "2024-05-18" | grep wlx

`May 18 08:53:09 hostNameGoesHere kernel: wlxadapterNameGoesHere: deauthenticated from APMACAddressGoesHere (Reason: 16=GROUP_KEY_HANDSHAKE_TIMEOUT)`

iwconfig before deauth:

```
wlxadapterNameGoesHere  IEEE 802.11  ESSID:"SSIDGoesHere"  
          Mode:Managed  Frequency:5.745 GHz  Access Point: APMACAddressGoesHere
          Bit Rate=432.3 Mb/s   Tx-Power=3 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=42/70  Signal level=-68 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:5  Invalid misc:1   Missed beacon:0
```

iwconfig after deauth:

```
wlxadapterNameGoesHere  IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=3 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
```

As noted, the Wi-Fi adapter dropout only seems to occur when the Wi-Fi client security is set to WPA3 and coincides with a GTK rekeying event initiated by the AP (Google Nest Wi-Fi Pro mesh router).

thanks!

morrownr, 2024-05-19:
@jsvnsk 

> WM Firmware Version: ____010000, Build Time: 20240219111038

I have set up my Edup 1672 for testing. I see that you are on fimrware files that are a little newer than mine. I change a lot for testing so I will test for a while with the files I have installed and then change to the firmware you are using. I am on kernel 6.6. Remind me which kernel you are using.

I have set my wifi router to WPA3 and channel width 160 (even though this adapter can only do 80. I am trying to get as close as possible to your setup.

I'd like to talk you into using `iw` and `ip` instead of iwconfig as it is depreciated. For example:

ip a
iw dev

Let me see what I see.

@morrownr

morrownr, 2024-05-19:
I set the GTK rekeying for every 10 minutes so if the adapter starts driving me crazy... we will know.

Here is my iperf3 output for the adapter:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.169 port 45388 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  76.4 MBytes   641 Mbits/sec    0    970 KBytes       
[  5]   1.00-2.00   sec  77.5 MBytes   650 Mbits/sec    0    970 KBytes       
[  5]   2.00-3.00   sec  76.2 MBytes   640 Mbits/sec    0    970 KBytes       
[  5]   3.00-4.00   sec  78.8 MBytes   661 Mbits/sec    0    970 KBytes       
[  5]   4.00-5.00   sec  76.2 MBytes   640 Mbits/sec    0    970 KBytes       
[  5]   5.00-6.00   sec  78.8 MBytes   661 Mbits/sec    0    970 KBytes       
[  5]   6.00-7.00   sec  77.5 MBytes   650 Mbits/sec    0    970 KBytes       
[  5]   7.00-8.00   sec  78.8 MBytes   661 Mbits/sec    0    970 KBytes       
[  5]   8.00-9.00   sec  76.2 MBytes   640 Mbits/sec    0    970 KBytes       
[  5]   9.00-10.00  sec  76.2 MBytes   640 Mbits/sec    0    970 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   773 MBytes   648 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   770 MBytes   646 Mbits/sec                  receiver

iperf Done.

```

jsvnsk, 2024-05-19:
> @jsvnsk
>
>
> I have set up my Edup 1672 for testing. I see that you are on fimrware files that are a little newer than mine. I change a lot for testing so I will test for a while with the files I have [...]
my current kernel is 6.5.0-35-generic
>
> I have set my wifi router to WPA3 and channel width 160 (even though this adapter can only do 80. I am trying to get as close as possible to your setup.
> 
> I'd like to talk you into using `iw` and `ip` instead of iwconfig as it is depreciated. For example:
> 
> ip a iw dev
>
yup; old habits die hard :)
```
 3: wlxAdapterNameGoesHere: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether adapterMACAddressGoesHere brd ff:ff:ff:ff:ff:ff
    inet 192.168.86.10/27 brd 192.168.86.31 scope global noprefixroute wlxAdapterNameGoesHere
       valid_lft forever preferred_lft forever
```
```
iw dev
phy#0
	Interface wlxAdapterNameGoesHere
		ifindex 3
		wdev 0x1
		addr adapterMACAddressGoesHere
		ssid SSIDGoesHere
		type managed
		channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```
> Let me see what I see.
> 
> @morrownr
thanks!

jsvnsk, 2024-05-19:
> I set the GTK rekeying for every 10 minutes so if the adapter starts driving me crazy... we will know.
> 
Google uses their Google Home app for managing the AP and I have not yet found a way to change the default keying interval from 24 hrs

my iperf3 test is about 1/3rd the bitrate you are getting (not unsurprising given my distance from the AP and being on a different floor of the house)

```
iperf3 -c 192.168.86.10 -p 8888
Connecting to host 192.168.86.10, port 8888
[  5] local 192.168.86.4 port 34392 connected to 192.168.86.10 port 8888
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  26.0 MBytes   218 Mbits/sec    0   1.42 MBytes       
[  5]   1.00-2.00   sec  28.8 MBytes   241 Mbits/sec    0   2.88 MBytes       
[  5]   2.00-3.00   sec  28.8 MBytes   241 Mbits/sec    0   3.18 MBytes       
[  5]   3.00-4.00   sec  27.5 MBytes   231 Mbits/sec    0   3.18 MBytes       
[  5]   4.00-5.00   sec  28.8 MBytes   241 Mbits/sec    0   3.18 MBytes       
[  5]   5.00-6.00   sec  28.8 MBytes   241 Mbits/sec    0   3.18 MBytes       
[  5]   6.00-7.00   sec  26.2 MBytes   221 Mbits/sec    0   3.18 MBytes       
[  5]   7.00-8.00   sec  27.5 MBytes   231 Mbits/sec    0   3.18 MBytes       
[  5]   8.00-9.00   sec  22.5 MBytes   189 Mbits/sec    0   3.18 MBytes       
[  5]   9.00-10.00  sec  25.0 MBytes   210 Mbits/sec    0   3.18 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   270 MBytes   226 Mbits/sec    0             sender
[  5]   0.00-10.13  sec   269 MBytes   223 Mbits/sec                  receiver
```

thanks again !

morrownr, 2024-05-21:
@jsvnsk 

I went ahead and reupgraded mt firmware to the level you are using. I did not see any problems with the old firmware over a few days time.

Do you have the results of?

sudo dmesg | grep mt7

jsvnsk, 2024-05-21:
@morrownr 

Yes -- here it is:
> sudo dmesg | grep mt7

```
[   23.559933] usbcore: registered new interface driver mt7921u
[   23.611928] mt7921u 4-5:1.0: HW/SW Version: 0x8a108a10, Build Time: 20240219110958a
[   23.624882] mt7921u 4-5:1.0: WM Firmware Version: ____010000, Build Time: 20240219111038
[   25.244397] mt7921u 4-5:1.0 wlxadapterNameGoesHere: renamed from wlan0
```
thanks!

morrownr, 2024-05-22:
@jsvnsk 

Well, the log entries don't seem to indicate any problems. I am seeing the same.

I'll continue testing but I am not seeing the problem. Sometimes I wonder how we make wifi as dependable as it is because it has grown into something that is very complicated. Let me throw some ideas out.

Our client systems are very close to the same but our wifi router setups may be different and that is where I am going to say the problem is. This Edup adapter is tri-band. Tell me about your Google router? Is it dual band or tri band? Are all of the SSIDs using the same name? If they are using the same name, do you have the capability to change one to make the sames different?

jsvnsk, 2024-05-23:
@morrownr 
totally agree; so many new Wi-Fi options to juggle these days :)
yes - the Google Nest Wi-Fi Pro routers support tri-band
yes - both of the routers use the same SSID
unfortunately the Google Home app is the only interface Google offers to manage the router settings and it does not have an option for creating unique SSIDs
one option would be to turn off one of the routers -- that way the network could be be isolated to using a single AP
I can give that a try and see if it yields anything
thanks again !

morrownr, 2024-05-23:
> I can give that a try and see if it yields anything

Good idea.

I'll keep eyes open. A lot of message traffic ends up here. You might just set WPA2 and keep an eye on the product forum. We don't know what is at fault but we can figure it out.

morrownr, 2024-05-23:
Just thought of something. Do you have the ability to turn the 6 GHz band off?

jsvnsk, 2024-05-23:
@morrownr

thanks -- unfortunately the only settable advanced networking options available via the Google Home app are:

WPA3 (on or off)
160 MHz channel (on or off)
UPnP (on or off)
IPv6 (on or off)

the current settings are:
WPA3 on
160 MHz channel on (prior testing scenarios with this off did not yield anything new)
UPnP off
IPv6 off

so, there is no option for turning off the 6 GHz band it seems

another angle I plan to explore is to replace wpa_supplicant and NetworkManager with iwd

after testing those scenarios I will post more findings week after the holiday

thanks!

morrownr, 2024-05-23:
> another angle I plan to explore is to replace wpa_supplicant and NetworkManager with iwd

I don't remember if you said which distro you are using... something to think about is that many Linux distros can be run from a usb flash drive if you burn an iso to a flash drive. You could boot Ubuntu 24.04 (just an example) and see if this happens without having to mess with installing iwd as Ubuntu 24.04 uses iwd.

Something else... we only looked for problems with the mt7921u driver and its firmware in the log but the log could show what the problem is and our focus was too narrow:

$ sudo dmesg

> after testing those scenarios I will post more findings next week after the holiday

Looking forward to hearing what you find.

Can you give me the exact name and model number of your Google device? It might help me know when I run onto something that might matter to this situation.

Comment for others that may be reading this thread:

It is extremely likely that this issue has nothing to do with this specific adapter (EDUP EP-AX1672) but would apply to all adapter using the mt7921au chipset and mt7921u driver. If fact, it could be that many other adapters being used on Linux would show the same thing and keep in mind that the world of WiFi is not perfect. It will take time for all of the rough edges having to do with WPA3, WiFi 6, WiFi 7 and the 6 GHz band to be worked out... this statement not only applies to Linux but to all modern operating systems. 

There are many devices out there that do not support WPA3. An example is the very popular Ruku streaming devices. I have two, one of which is a very recent model. It simply does not support WPA3 but then WPA3 does not provide much additional protection in a home setting anyway. What I do is use the 2.4 GHz band for the Roku devices, devices needing maximum range and all other IoT devices. I set the 2.4 MHz band to WPA2 AES as it will provide all of the protection needed for what I use on the band and it provides more than enough bandwidth for what I use it for.

I do set the 5 GHz band to WPA3 as my home network runs on it. In a few places around this site, I do mention that Linux users will probably be happier in the long run if they seek out WiFi APs/routers that have the ability to use different names for the different bands as that allows you to set things up for maximum performance and it makes troubleshooting easier.

jsvnsk, 2024-05-23:
@morrownr 

OS: Ubuntu 22.04.4
product: Google Nest Wi-Fi Pro router
model: G6ZUC
link: https://support.google.com/googlenest/answer/6280668?hl=en#zippy=%2Cnest-wifi-pro-router

good idea -- I have a Ventoy flash drive with a couple of ISOs for some other Linux distros that I can test with

also agree that this may not necessarily be an issue with this particular adapter as it could just as easily be something masked by the "walled garden" limitations of the Google Home app

thanks!

jsvnsk, 2024-05-23:
@morrownr 
gotta admit that is pretty cool but...
as a Google subscriber I technically don't "own" these devices, and so any potentially-destructive firmware mods to their routers is probably a bridge too far
it is good to know that option exists, though !

morrownr, 2024-05-23:
Got it.

joopdo, 2024-05-29:
@morrownr If I may ask, how many AP's can I create with the EDUP EP-AX1672? Any insights on range, compared to Alfa?

morrownr, 2024-05-29:
> how many AP's can I create with the EDUP EP-AX1672?

I really need you to explain this question. Are you asking how many clients can connect to an AP created with this adapter or is the question really now many APs can be created. 

> Any insights on range, compared to Alfa?

I have an Alfa AXML. It is not one of Alfa's max power models. I wish they would make a max power adapter for WiFi 6 or 7. Anyway, I did a quick and dirty comparison in AP mode and the range seemed to be similar. I have to admit that I took the Alfa antennas and put them on the Edup for testing. It is good have removable antennas. I'd have to do a good test to show exactly what we have in this case. I have used both adapters in AP mode a lot with my AP guide on the Main Menu, Both have performed very well. You won't get the range you will see on a good wifi router that is specifically designed and tuned for being an AP but very good nonetheless.

joopdo, 2024-05-29:
>  I really need you to explain this question. Are you asking how many clients can connect to an AP created with this adapter or is the question really now many APs can be created.

Apologies for the cryptic messaging for my question, Morrownr! 

I don't own this device (yet). For my home setup, I'm looking for a nice Wifi 6e adapter. Saw that you added the EDUP. What I am trying to achieve in OpenWRT, is that I want to setup two Wifi Networks (AP's). One for LAN and one for IOT. (2 APs). We recently had some back and forth about the "AWUS036AXML" which has a maximum ap  #{ AP, P2P-GO } <= 1 . Curious if the EDUP allows for more then one! 

Thank you for your insights as always!

morrownr, 2024-05-29:
> which has a maximum ap #{ AP, P2P-GO } <= 1

I see. The above is a driver issue, not an adapter thing. That means that all adapters with the mt7921au chipset will be the same,..for now. We either need to convince the Mediatek devs to work on this or get someone else that has the time and knowledge to work on it and submit a patch.

> one for IOT. (2 APs)

Sounds like dual band. You could go with the Edup or Alfa for the WiFi 6e and maybe a WiFi 4 adapter for IoT and 2.4 MHz band. The Plug and Play List has some Ralink RT5370 based WiFi 4 adapters that are pretty cheap and they can handle up to 8 APs:

```
valid interface combinations:
		 * #{ AP, mesh point } <= 8,
		   total <= 8, #channels <= 1
```

I have a couple of rt5370 based adapters. Do you want me to do an AP range check for you?

fakemanhk, 2024-06-11:
> AP mode performance has been very good. It is very stable with 5 GHz WiFi 6 AP mode

Were you trying this out on normal Linux distro, or something like OpenWrt? I found that my MT7921au one crashes a bit often under OpenWrt.

morrownr, 2024-06-11:
@fakemanhk 

> Were you trying this out on normal Linux distro, or something like OpenWrt?

I was using the March release of the Raspberry Pi OS 64 bit. The hardware was a RasPi4B. The setup was per my AP Mode guide on the Main Menu. It is a guide to setting up a bridged wireless access point. Scatter-Gather was turned off. mt76 supports one module parameter. I can point you to it if you like.

> I found that my MT7921au one crashes a bit often under OpenWrt.

You might just give shutting off Scatter-Gather a try. See:

https://github.com/morrownr/7612u

Yes, I know that says it is a mt7612u repo but shutting Scatter-Gather is the same with your chip. Go down to Known Issue 2.

Let me know if this helps. It is likely caused by specific usb3 hub chips and/or their driver.

morrownr, 2024-06-11:
Oh, adding the module parameter in OpenWRT might be different.

Hey @bjlockie can you help? 

I see the following:

https://forum.openwrt.org/t/adding-module-parameter-solved/70191

fakemanhk, 2024-06-11:
> You might just give shutting off Scatter-Gather a try. See:

Tried, but still not stable.
Later when I have more time to configure desktop Linux to become a router I will see if there's any difference.

mjschipper, 2024-06-24:
Hi @morrownr, thanks for the review, it helped me choose this device for my OpenWrt NanoPi R4S router project. I want to use this on my router as a failover wifi ap for when i have blackouts. My main access point (Ubiquity AP) doesnt have battery backup attached so i am wanting to setup the AX1672 for emergencies. The idea is that i'll write a script to monitor the main access points IP and enable the AX1672 in AP mode when the main access point goes down then disable itself when the main AP is back up. 

I'm pretty new to OpenWrt and USB wifi adapters, do you think this is achievable? Does OpenWrt have a builtin feature to do what im thinking? otherwise i'll work on a script i guess. :)

I also saw the 'mwan3 (Multi WAN load balancing/failover)' package but i'm not sure it fits this use case. happy to be corrected.

morrownr, 2024-06-24:
> I want to use this on my router as a failover wifi ap for when i have blackouts.

I have tested this Edup adapter with OpenWRT. The package (driver) to install is:

kmod-mt7921u

I have two APC UPS. I did a quick search and found the following::

https://forum.openwrt.org/t/apc-ups-control-in-openwrt/131358

It might have some info that could help you. There are OpenWRT packages for UPSs.

mjschipper, 2024-06-24:
Thanks for that info. The UPS is basically a 12v battery bank to run the Nanopi. I got ChatGPT to write me a basic solution already. I dont have the NanoPi or Wifi Adapter yet but will test once it arrives. Looks pretty good. Here it is if you are interested: https://github.com/mjschipper/openwrt-wififailover

bjlockie, 2024-08-31:
I just got one, I love the detachable antennas.
`$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.186 port 52254 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  32.2 MBytes   270 Mbits/sec    0   1.25 MBytes       
[  5]   1.00-2.00   sec  38.9 MBytes   326 Mbits/sec    0   1.96 MBytes       
[  5]   2.00-3.00   sec  47.9 MBytes   402 Mbits/sec    0   2.33 MBytes       
[  5]   3.00-4.00   sec  50.0 MBytes   419 Mbits/sec    0   2.98 MBytes       
[  5]   4.00-5.00   sec  51.6 MBytes   433 Mbits/sec    0   2.98 MBytes       
[  5]   5.00-6.00   sec  46.1 MBytes   387 Mbits/sec    0   2.98 MBytes       
[  5]   6.00-7.00   sec  49.6 MBytes   416 Mbits/sec    0   2.98 MBytes       
[  5]   7.00-8.00   sec  48.5 MBytes   407 Mbits/sec    0   2.98 MBytes       
[  5]   8.00-9.00   sec  43.1 MBytes   362 Mbits/sec    0   3.33 MBytes       
[  5]   9.00-10.00  sec  47.5 MBytes   398 Mbits/sec    0   3.33 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   456 MBytes   382 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   453 MBytes   380 Mbits/sec                  receiver`

[image]
[image]

morrownr, 2024-08-31:
@bjlockie 

> I just got one, I love the detachable antennas.

I tested quite a bit with mine. Those antennas seem to be good quality as the range is pretty good. I also like the vent holes. I've run long full speed iperf3 tests and it maintains good temperatures. 

Here is an iperf3 test:

The Edup adapter like yours is in a RasPi4B in AP mode.

Channel 36 (light traffic)
iperf3 was ran from a connected client.
WiFi 6 mode.
Band 2 (5 GHz)

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.127 port 37902 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  67.5 MBytes   566 Mbits/sec    7    585 KBytes       
[  5]   1.00-2.00   sec  70.6 MBytes   592 Mbits/sec    0    669 KBytes       
[  5]   2.00-3.00   sec  71.2 MBytes   598 Mbits/sec    0    745 KBytes       
[  5]   3.00-4.00   sec  72.5 MBytes   608 Mbits/sec    0    813 KBytes       
[  5]   4.00-5.00   sec  73.8 MBytes   619 Mbits/sec    0    860 KBytes       
[  5]   5.00-6.00   sec  71.2 MBytes   598 Mbits/sec    0    860 KBytes       
[  5]   6.00-7.00   sec  73.8 MBytes   619 Mbits/sec    0    901 KBytes       
[  5]   7.00-8.00   sec  72.5 MBytes   608 Mbits/sec    0    929 KBytes       
[  5]   8.00-9.00   sec  76.2 MBytes   640 Mbits/sec    0    929 KBytes       
[  5]   9.00-10.00  sec  71.2 MBytes   598 Mbits/sec    0    929 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   721 MBytes   604 Mbits/sec    7             sender
[  5]   0.00-10.00  sec   718 MBytes   602 Mbits/sec                  receiver

iperf Done.

```

I think overall it is as good of a USB WiFi adapter as I have ever run on Linux.

feiyax, 2024-09-02:
I just got the same adapter but having a hard time understanding how does it work with 6GHz. 

When i looked at `iw list`, it has `(disabled)` after all the 6 GHz channels. Does it mean the support isn't enabled at Linux driver level? Mediatek's firmware level? or even at hardware/efuse level?

`iwlist channel` and `iw list` outputs attached: [edup_ax3000.typescript.txt](https://github.com/user-attachments/files/16841868/edup_ax3000.typescript.txt)

morrownr, 2024-09-05:
Many distros have you set the country during installation but some do not. Glad you figured it out. How 6 GHz working?

joopdo, 2024-12-24:
I have this adapter installed in a FriendlyElec R4S. Every morning I would find it disappeared! 
After lot's of debugging, a simple:
`echo 'on' > /sys/bus/usb/devices/usb8/power/control`

Solved it for me. Now it's stable and up 24/7.

morrownr, 2024-12-26:
Hi @joopdo 

I've been thinking about starting a FAQ document to go on the Main Menu and this might be something to add. Do you have any links so I can better understand what is being done?

joopdo, 2024-12-28:
I think, reading up on it, that it's just another way to turn power saving off, something you already discussed [here](https://github.com/morrownr/USB-WiFi/issues/427#issuecomment-2103064862).

kphanipavan, 2025-06-15:
https://github.com/morrownr/USB-WiFi/issues/384#issuecomment-2973462883

Commented there, but will move my query here since I have the EDUP.

Bought the EP-AX1672 Adapter. Planned to use it in AP mode with my Raspberry Pi 5 running RasPiOS 64 bit. Installed and updated the firmware as per [7](https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md). AP mode works well but clients cannot ping each other. `ap_isolate` is set to `0` in hostapd config. Do I have to configure anything else for inter STA communication to work??

I found one possible way to bypass the possible driver limitation here: [unix stackexchange/bridging](https://unix.stackexchange.com/a/745961/488963). But I want to try to get this working...

TIA
~ Pvn

morrownr, 2025-06-16:
Hi @kphanipavan 

> but will move my query here since I have the EDUP.

This would probably get more attention as a new bug report as there does not seem to be anything special with the Edup adapter that would make it be the only adapter with this problem.

> AP mode works well but clients cannot ping each other. ap_isolate is set to 0 in hostapd config. Do I have to configure anything else for inter STA communication to work??

I do not recall ever having run into this issue.

I don't think this is a driver or adapter issue. What are you using for networking?

bjlockie, 2025-06-16:
Is there any chance there are 2 dhcp servers?

kphanipavan, 2025-06-17:
Hey @morrownr,

> I don't think this is a driver or adapter issue

You are right! I changed to the older firmware provided by raspi os repos, dated Feb, 2024. STAs can communicate now, surprisingly.

> What are you using for networking?

Um, RPi 5 connected to eth as uplink, AX1672 in AP mode for hardware. IPTables for network forwarding, Pi-Hole for DNS and DHCP, HostAPd for AP mode in software, managed by [this script](https://github.com/garywill/linux-router).

> Is there any chance there are 2 dhcp servers?

Verified only one dhcp server, running under Pi-Hole. Its working now, will test with the new firmware to check if the issue pops up again.

Thank you for the replies.

morrownr, 2025-06-17:
@kphanipavan 

> You are right! I changed to the older firmware provided by raspi os repos, dated Feb, 2024. STAs can communicate now, surprisingly.

This is not good. I guess the firmware may have a bug in the current version. Which version is not working? There was a very recent release.

FYI: I filed this bug with RasPi recently.

https://github.com/raspberrypi/linux/issues/6853

> Thank you for the replies.

Sure.

morrownr, 2026-02-02:
Hi @Vijayir2023 

> What is the maximum number of clients that can connect when the module is used in AP mode under Linux OS?

16

> Also, do we have information regarding its regulatory certifications, specifically the countries where it is approved?

We do.

> Additionally, does the module support the following pre‑certifications: FCC, IC, CE, RED, EU, China, India, and Korea radio approvals

I can confirm FCC and EU but I can't find the box mine came in so maybe you can check on the Edup website.

Vijayir2023, 2026-02-03:
On security standards does it supports: **WPA2‑Personal/Enterprise, WPA3‑Personal/Enterprise** ??