morrownr/USB-WiFi #428: Alfa AWUS036ACU false chipset

Majdoddin, 2024-04-25:
this device seems to have Chipset	Realtek RTL8812BU, not mt7921aun

source: https://alfa-network.eu/awus036acu

Majdoddin, 2024-05-02:
sorry, I was confused. 
I bought this device this weak and it has chipset RTL88x2bu (same series as RTL8812BU?). This can be checked from [ALFA](https://docs.alfa.com.tw/Support/Linux/RTL8812BU/) (check Linux Hardware). 
Not a big issue, because this chip uses the same in-kernel driver as rtl8812bu: check [here](https://linux-hardware.org/?id=usb:2357-012d) and [here](https://linux-hardware.org/index.php?id=usb:0bda-b812). 

I should mention that the in-kernel driver is available from kernel >=6.2, meaning current Debian stable (kernel 6.1) does not have it. I had to backport-upgrade my kernel to get it run.

morrownr, 2024-05-02:
Debian 12 has kernel 6.6 LTS available. That is what I use. It works well. 

rtw88 does not yet support USB3 for the rtl8812bu so speed is limited but it is fast enough to not be a problem for most people. Hopefully that can be fixed this year.

s888li, 2024-09-06:
i bought a AWUS036ACU, not working with kali 2023.2 x64. Does Anyone have the same issue ?

bjlockie, 2024-09-06:
> i bought a AWUS036ACU, not working with kali 2023.2 x64. Does Anyone have the same issue ?

What kernel is kali 2023.2?

morrownr, 2024-09-06:
Hi @s888li 

> not working with kali 2023.2 x64

Before I give you some options, tell me what your intended purpose is with the adapter. Generally those using Kali are looking for an adapter that works well with monitor mode.