morrownr/USB-WiFi #429: Is dual band AP mode possible with single USB adapter?

bdantas, 2024-04-26:
Hi Nick. My home wireless router is an old ThinkPad laptop running Tiny Core Linux, hostapd, and dnsmasq.

I use two Alfa AWUS036ACHM (MediaTek mt7610u chipset) adapters on said laptop: one for 5GHz AP on wlan1, one for a 2.4 GHz AP on wlan2. This requires two instances of hostapd.

I'm trying to simplify my setup. Is it possible to create the 5GHz and 2.4GHz on one single AWUS036ACHM adapter? If not, is there a similar adapter that would support this (just out of curiosity)? 

P.S. I'm hesitant to switch away from AWUS036ACHM because it has great range, is fast, and both adapters have worked hard for years without ever causing me a single problem. I selected this adapter based on information on your github page, thank you!

morrownr, 2024-04-26:
Hi @bdantas 

> My home wireless router is an old ThinkPad laptop running Tiny Core Linux, hostapd, and dnsmasq.

That is cool. I like to repurpose things also.

> This requires two instances of hostapd.

Are you starting hostapd twice or providing one instance with 2 .conf files?

>  Is it possible to create the 5GHz and 2.4GHz on one single AWUS036ACHM adapter?

No. The reason is that the adapter contains one radio. Dual band wifi routers have two radios. Tri band wifi routers have 3 radios.

> My guess is no, because each adapter presents a single interface in output of ifconfig, which leads me to believe that each adapter has a single radio.

I'd like to convince you to start using `iw` and `ip` instead of `ifconfig` as it has been depreciated for years. Try:

$ iw dev
$ ip a

Actually a single adapter can have more than one interface and share the resources of the adapter but all has to be in one band.

> If dual band AP on a single AWUS036ACHM is not possible, is there a similar adapter that would support this?

Not that I am aware of.

> I'm hesitant to switch away from AWUS036ACHM because it has great range, is fast, and both adapters have worked hard for years without ever causing me a single problem. I selected this adapter based on information on your github page, thank you!

I understand what you are saying about ACHM. It is a cool little adapter that just works and has great range. You are welcome.

If you could be specific about what is the cause of the need to simplify, if might help. I'm sure you know about my AP Mode guide on the Main Menu:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

While the document is not short, it is only long because of all the tips that are included. The actual setup for networking and hostapd is about as simply as it gets while still allowing for maximum performance. Other options for simplification would include running OpenWRT which is available for x86.

@morrownr

bdantas, 2024-04-26:
> Are you starting hostapd twice or providing one instance with 2 .conf files?
 
Starting it twice. I didn't realize it could be started once with 2 config files. That's already a nice simplification.

> No. The reason is that the adapter contains one radio.
 
That's what I figured. Thanks for confirming.

> I'd like to convince you to start using iw and ip instead of ifconfig as it has been depreciated for years.
 
Tiny Core Linux uses BusyBox (not coreutils). In BusyBox, `ifconfig` is alive and well. Nevertheless, I will try to familiarize myself with the newfangled utilities.

> If you could be specific about what is the cause of the need to simplify, if might help.
 
Just that I'm getting older. As I age, the age of the things I find myself tweaking/updating are no longer measured in months but in years. Older setups are easier to deal with if they are minimalistic. 
 
I'll take a look at your AP Mode guide. As always, thank you for being such a great resource and generous teacher.

dubhater, 2024-04-26:
I wonder if the upcoming wifi 7 devices will be able to do this. I think if they support multi link operation it's just a matter of having driver support.

morrownr, 2024-04-27:
@dubhater 

> I wonder if the upcoming wifi 7 devices will be able to do this. I think if they support multi link operation it's just a matter of having driver support.

I'm waiting for you to buy me an adapter so I can see. Oh yeah, none on the market yet...wait, wait ...

The Mediatek driver for usb wifi 7 has been in the kernel since 6.7. Do you have any idea what Realtek is doing for usb WiFi 7? I know exactly nothing.

dubhater, 2024-04-27:
> The Mediatek driver for usb wifi 7 has been in the kernel since 6.7. Do you have any idea what Realtek is doing for usb WiFi 7? I know exactly nothing.

I know nothing also. I expect they will do the same thing they did for RTL8852xU.

morrownr, 2024-04-28:
>  I expect they will do the same thing they did for RTL8852xU.

I hope not. My experience with the rtl8852xU drivers is not good and this method of supplying (or not supplying) drivers is only good for those working on embedded projects and maybe not even so good for that. Let's look at this Realtek out-of-kernel driver mess for USB.

Does Realtek release their out-of-kernel drivers to the public? No
Does Realtek keep their out-of-kernel drivers updated with newly released kernels in a timely manner? No
Can the public report problems to Realtek? No. For their in-kernel drivers for cards the answer is yes.
Are the Realtek out-of-kernel drivers easy for average computer users to use? Hell no.
Are Realtek out-of-kernel drivers missing features? Yes
Why would a Linux user buy a Realtek based WiFi 6 adapter if a Mediatek based WiFi 6 adapter is available? A lack of knowledge of the differences in the support and that is actually the main reason for this site... to get the knowledge out there to Linux users.

Realtek and Mediatek have guidance concerning how to support Linux with wifi drivers. Realtek follows that guidance for chips for cards but not for USB. Mediatek follows the guidance for all forms of wifi chips, including USB. The guidance for supporting wifi in Linux works for the development model that Linux uses. Linux uses a continuous dev model due to the makeup of the resources that are available. Realtek's USB support model is not compatible and only serves to cause problems.

Rant off.

whitslack, 2024-05-14:
> > Are you starting hostapd twice or providing one instance with 2 .conf files?
> 
> Starting it twice. I didn't realize it could be started once with 2 config files. That's already a nice simplification.

In fact, that's how you *must* do it (a single instance) if you want to support fast transition between bands without introducing any of the complicated, enterprise-level protocols that consumer-grade client kit rarely if ever supports. (Two instances of HostAPd aren't aware of each other by default, so they can't coordinate hand-off of clients from one to the other. By contrast, running two radios in a single HostAPd instance does allow clients to do fast roaming between the two, although you do still need to set a few configuration options to make it happen.)

fakemanhk, 2024-06-11:
> If dual band AP on a single AWUS036ACHM is not possible, is there a similar adapter that would support this?

Look for the keyword "Dual Band Dual Concurrent (DBDC)", however AFAIK there is no USB WiFi on market that can do this (which is normal, they are designed to be used as client but not AP)

> Dual band wifi routers have two radios. Tri band wifi routers have 3 radios.

For example MT7915 has DBDC so one single device can provide both 2.4GHz/5GHz simultaneously, however there is no such thing for USB WiFi.

dubhater, 2024-06-11:
Once upon a time there were some RTL8192DU devices, e.g. SAPIDO AU-5125, Planex GW-USFang300. I wish I could still buy one.