morrownr/USB-WiFi #430: Need Testers: New in-kernel (mac80211) driver for rtl8821au, rtl8811au and rtl8821ae chipsets.

morrownr, 2024-05-06:
New in-kernel (mac80211) driver for rtl8821au, rtl8811au and rtl8821ae chipsets.

Information about testing is located at:

https://github.com/morrownr/8821au-20210708/issues/133

That is Issue 133 in the 8821au repo here at this site. That issue will be used for testing and reporting. The first message shows the location of the driver. The supported chipsets have been popular with Linux users for several years.

The intent is to upstream the driver to the Linux kernel once it is ready. There is nothing like Plug and Play adapters that work well. Please help if you can. I have been testing managed and monitor mode today and the results are good. We need to test rigorously.

HOT: All devs currently only have adapters with rtl8811au chips. We need Linux users with rtl8821au and rtl8821ce chips.

@morrownr

morrownr, 2024-05-08:
Hi @fakemanhk 

> (USB ID 0bda:c811) should be qualified for testing?

We can certainly use your help testing but I need to give you a different link than the one above. You adapter uses the rtl8821cu chipset which uses a different driver than the one shown in the subject but it just so happens we already have an updated version of the driver for your chipset that needs testing also. The driver is located at:

https://github.com/lwfinger/rtw88

Follow the instructions in the README. 

Below is the location of the issue where you should report bugs or suggestions:

https://github.com/morrownr/8821cu-20210916/issues/115

> And which kernel version I need to use to test?

5.4 or later. If using 6.2 or later, you will need to blacklist the driver that is already in the kernel.

fakemanhk, 2024-05-12:
I found that I also have a few other Realtek based USB wifi at home (some are unknown brand), not sure if any testing needed for them?? I am listing the USB ID below:

1.  0bda:b812 (this should be 8812BU?)
2.  2357:011e (T2U Nano, probably this is the 8811au you want to test?)

morrownr, 2024-05-12:
@fakemanhk 

>2357:011e (T2U Nano, probably this is the 8811au you want to test?)

Yes, indeed. ID 2357:011e /* TP Link */

That is a rtl8811au chipset. We could use help testing the other one when the time is right but for now we need to get the new driver in the kernel. Go back to the first message and follow the link to get info on the location of the driver and to see what is going right now.

morrownr, 2024-05-20:
@fakemanhk 

How is the testing on the adapters going?

fakemanhk, 2024-06-10:
Are the 3 devices I list above using same rtw driver? I have installed it and when I plug one of the above 3 dongles it just works.

morrownr, 2024-06-10:
8821au = wifi plus bluetooth and uses the *8821au driver
8811au = wifi only and uses the *8821au driver
8821ae = wifi plus bluetooth and uses the *8821ae driver (this chip is for cards, not adapters)

fakemanhk, 2024-08-01:
Sorry forgot to reply this, previous laptop got some stupid problem which I wasn't able to test properly, just now I had an old Mac Mini which installed a fresh Peppermint Linux (with Debian backport 6.9.7)

The system:

`Linux macmini31 6.9.7+bpo-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.9.7-1~bpo12+1 (2024-07-03) x86_64 GNU/Linux`

After installation and modprobe command, it didn't load until I unplug/plug back the dongle, below are info of `T2U Nano v3`

```
Bus 002 Device 004: ID 2357:011e TP-Link AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2357 TP-Link
  idProduct          0x011e AC600 wireless Realtek RTL8811AU [Archer T2U Nano]
  bcdDevice            2.00
  iManufacturer           1 Realtek 
  iProduct                2 802.11ac WLAN Adapter 
  iSerial                 3 00e04c000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003c
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           6
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
[... 271 lines cut ...]
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
		* [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

fakemanhk, 2024-08-01:
Another dongle: Also TP-Link, this time `T2U v3`, using the same driver and it also works well:

USB information:
```
Bus 002 Device 005: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xc811 
  bcdDevice            2.00
  iManufacturer           1 Realtek
  iProduct                2 802.11ac NIC
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
[... 257 lines cut ...]
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
		* [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

fakemanhk, 2024-08-01:
Here is the unknown 2x2 AC1200 USB WiFi I bought from China long time ago, also working with this driver??

```
Bus 001 Device 009: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xb812 RTL88x2bu [AC1200 Techkey]
  bcdDevice            2.10
  iManufacturer           1 Realtek
  iProduct                2 USB3.0 802.11ac 1200M Adapter
  iSerial                 3 123456
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
[... 257 lines cut ...]
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
		* [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

morrownr, 2024-08-02:
>  Previously tested T2U Nano has GREEN light blinking but this T2U v3 doesn't.

A couple of adapters have temp code to make the LEDs work. The plan is to do permanent code that works for all rtw88 supported adapters. It is coming. Thanks for the report on this.

 > dmesg output, though it says USB3.0 but I plugged into other Linux machine with USB3 port and found that it only binds to USB 2.0 ports:

Does this 8822bu adapter support bluetooth? If so, it will be locked to USB2 speeds.

You can check the USB speed with the following command:

$ lsusb -t

fakemanhk, 2024-08-05:
> Does this 8822bu adapter support bluetooth? If so, it will be locked to USB2 speeds.
> 
> You can check the USB speed with the following command:
> 
> $ lsusb -t

Mine doesn't have bluetooth, but it's still running in USB2 mode:

`    |__ Port 3: Dev 5, If 0, Class=Vendor Specific Class, Driver=rtw_8822bu, 480M
`

morrownr, 2024-08-05:
@fakemanhk 

I happened to take another look at the info you provided:

```
[ 4040.643947] rtw_8822bu 1-1:1.0: Firmware version 27.2.0, H2C version 13
[ 4041.295409] rtw88_8822bu: disagrees about version of symbol rtw8822b_hw_spec
[ 4041.295418] rtw88_8822bu: Unknown symbol rtw8822b_hw_spec (err -22)
[ 4041.329598] rtw_8822bu 1-1:1.0 wlx1cbfcee41f90: renamed from wlan0
```

Note that the log is showing lines related to two drivers: rtw88_8822bu and rtw_8822bu

rtw88_8822bu is the in-kernel driver so that seems to indicate that you did not blacklist it. Those are conflicting drivers. You can't have both going.

fakemanhk, 2024-08-06:
> rtw88_8822bu is the in-kernel driver so that seems to indicate that you did not blacklist it. Those are conflicting drivers. You can't have both going.

Looks like it's compiled in, not module base so I wasn't able to blacklist it.
However without installing the driver you mentioned in the first post, my dongle never works, which implies the in kernel one is actually not functional?

morrownr, 2024-08-06:
@fakemanhk 

> Looks like it's compiled in, not module base so I wasn't able to blacklist it.

No distro for public consumption should be doing that. What distro and kernel is this? If they really are compiling in rtw88_8822bu, then you need to have a serious conversation with the leaders of the distro.

> which implies the in kernel one is actually not functional?

The USB drivers in the rtw88 in the mainline kernel were not great. The worse was rtw88_8821cu. We started on it and by early this year we send 5 patches to mainline. This helped greatly with that driver. That was in kernel 6.7. Work has continued. Numerous patches have been and continue to go in. They need to be sent in in a specific order. Once all of the existing patches are in and the new rtw88_8821au and rtw88_8812au are in, you should see reasonable performance. 

We can't fix your distro so maybe switching distros is something to look at.

morrownr, 2024-08-06:
> Even my blacklist command can't stop it.

How are you doing the blacklist?

EugeneS120, 2024-08-29:
> New in-kernel (mac80211) driver for rtl8821au, rtl8811au and rtl8821ae chipsets.

<details>
<summary>mini PC with RTL8821AE</summary>
        
 lspci -d 10ec:8821 -vv
        02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821AE 802.11ac PCIe Wireless Network Adapter
        Subsystem: AzureWave Device 2161
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 41
        Region 0: I/O ports at e000 [size=256]
        Region 2: Memory at f7100000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: <access denied>
        Kernel driver in use: rtl8821ae
        Kernel modules: rtl8821ae
        

</details>

Card work extremely slow.
Seems card type - mini PCIe. How I can test new driver?

morrownr, 2024-08-29:
@EugeneS120 

I made a mistake in the title of this issue. I corrected it today. rtl8821ae should have read rtl8812au. While there is an in-kernel driver for the rtl8821ae, it is not part of this project that is designed to go into rtw88. Maybe someday the rtl8821ae support can be moved to rtw88 and improved.

EugeneS120, 2024-08-31:
> @EugeneS120
>
> I made a mistake in the title of this issue. I corrected it today. rtl8821ae should have read rtl8812au. While there is an in-kernel driver for the rtl8821ae, it is not part of this [...]

Thanks.
Btw, there is a desktop with RTL8812AU (Comfast, internal antenna) nearby and the speed is stable over 100 Mbit/s. At the same time RTL8821AE shows ~2  Mbit/s ;(