morrownr/USB-WiFi #431: First Sighting: mt7925 m.2 card (WiFi 7)

morrownr, 2024-05-09:
There may be other cards with the mt7925/27 chipsets but this is the first one I have seen for sale:

https://zfishtek.com/index.php/product/mt7925-wireless-lan-card/

This is NOT a recommendation for the above product. What this message is about is to point out that the mt792x chipsets are flowing to product makers. Cards almost always come to market before USB adapters. That is just the way it is so now that we see cards on the market, it probably means that adapters will be available soon. It is hard to say how soon but sometime this year is a reasonable expectation. 

The driver for the mt792x chipsets has been in the Linux kernel since kernel 6.7.

If anyone gets a card with a mt792x chipset, please give us a report.

@morrownr

ilikenwf, 2024-05-14:
I found a few on aliexpress and have one coming to play with.

morrownr, 2024-05-14:
Hi @ilikenwf 

> I found a few on aliexpress and have one coming to play with.

I am looking forward to your report. None of us out here in the real world know what kind of shape the mt7925 driver is in so any word would be appreciated.

Thanks

ilikenwf, 2024-05-14:
I'm just taking a $30 risk because my QCNFA765 keeps having persistent bugs even despite the fixes and I'm kinda tired of it...I'll take dysfunctional wifi 7 with functional 6E over dysfunctional 6E...

ilikenwf, 2024-05-14:
...and I'm sure once this chip is perfected the price will 2x-4x.

morrownr, 2024-05-16:
FYI: A user report indicates the mt7925 driver does not support the mt7927 chip that you may see. It would be wise to get cards with the mt7925 chip and not the mt7927 until we can sort this out.

ilikenwf, 2024-05-16:
Here is the one I have coming:

https://www.aliexpress.us/item/3256806768193339.html

morrownr, 2024-05-16:
>  Haven't had my coffee yet.

I am working on my coffee. This is a coffee kind of day.

I am finding more info...see next msg.

morrownr, 2024-05-16:
MT7925

Wi-Fi 7, WPA3, 2.4/5/6GHz, 4.5Gbps,
DBDC, BW 160MHz, 4K QAM, MLO,
BT 5.3 LE, MRC, MRU, HDT, 6 nm
FCC ID: RAS-MT7925B22M

MT7927

Wi-Fi 7, WPA3, 2.4/5/6GHz, 6.5Gbps,
DBDC, BW 320MHz, 4K QAM, MLO,
BT 5.3 LE, MRC, MRU, HDT, 6 nm
FCC ID: RAS-MT7927 • AMD RZ738

It appears that the only difference between the chips is that the mt7925 supports 160 Mhz channel width and the mt7927 supports 320 MHz channel width.

Evidently support for the mt7927 has not been added yet.

ilikenwf, 2024-05-16:
I have not seen any mt7827s in my travels so far.

morrownr, 2024-05-16:
> I have not seen any mt7827s in my travels so far.

mt7927

https://zfishtek.com/index.php/product/mediatek-mt7927-wireless-lan-card/

mt7925

https://zfishtek.com/index.php/product/mt7925-wireless-lan-card/

ilikenwf, 2024-05-29:
`1b:00.0 Network controller: MEDIATEK Corp. Device 0717`

So far it just works...I don't have a 7 AP, though.

morrownr, 2024-05-29:
Three words I like to hear... "it just works".

morrownr, 2024-05-29:
I have noticed a couple of mt7925 M.2 cards on Amazon US this week so I may grab one soon so I can test along with you.

We are talking about a new driver that includes code for a new generation of wireless so it will have bugs. I have been noticing patches going in. It isn't just the Mediatek WiFi 7 drivers but Intel, Qualcomm and Realtek as well. I still have no idea what Realtek's USB strategy is for WiFi 7. Since this site is primarily about USB WiFi, I watch it more closely and I can usually pick up some hints prior to product introduction but not with Realtek this time. Mediatek's driver support plan is well known as USB and PCIe support for WiFi 7 is in the mt7925u and mt7925e drivers that have been in the kernel since 6.7. I expect mt7927 support soon.

Keep us posted as you continue testing.

patrakov, 2024-06-02:
> So far it just works...

Please provide a definition of "works", i.e., what exactly has been tested. Client mode, or AP mode? Does WDS work? Can more than one SSID be created? Is the advertisement of the DBDC capability true?

ilikenwf, 2024-06-02:
Well, I'm using it on a client device and so I haven't tried setting up a network on it, though if it weren't for my openwrt devices that would be a killer idea if it were slightly more reliable.

Right now, on kernel 6.9, I find that after running for 12-24 hours I eventually need to reboot to get it running properly. I'm assuming some kernel module issue or perhaps a buffer issue somewhere, but things get slow and then eventually despite having a valid IP, the connection becomes useless - not sure if due to latency/speed or if something else is happening. While I can restart the networking service or reset the card, only a real reboot seems to fix it.

morrownr, 2024-06-02:
@ilikenwf 

> if it weren't for my openwrt devices that would be a killer idea if it were slightly more reliable...

I just checked and it appears the mt7925 driver is already in the current stable version of OpenWRT:

kmod-mt792x-usb

I think they named it that because it will cover the 7925 and 7927 chips as the 7927 code goes in.

What you seeing in the log?

$ sudo dmesg | grep mt7

I'm a little hesitant to jump all over making a bug report on this driver right now because 6.9 seems to have had some stack issues that need to be resolved before we blame individual drivers.

ilikenwf, 2024-06-02:
[   10.224161] mt7925e 0000:1b:00.0: ASIC revision: 79250000
[   10.303401] mt7925e 0000:1b:00.0: HW/SW Version: 0x8a108a10, Build Time: 20231227093012a
[   10.647100] mt7925e 0000:1b:00.0: WM Firmware Version: ____000000, Build Time: 20231227093232

morrownr, 2024-06-03:
Well, that is clean. I'll see about getting my own little card soon so that I can play along.

Muh87, 2024-06-07:
Can anyone who has MT7925 share the output of "iw list" ?
I want to know if 160 MHz is allowed for AP mode .
Thanks in advance .

morrownr, 2024-06-07:
@ilikenwf 

> Right now, on kernel 6.9, I find that after running for 12-24 hours I eventually need to reboot to get it running properly. I'm assuming some kernel module issue or perhaps a buffer issue somewhere, [...]

The stack is undergoing a lot of work at the moment so folks like you out there on the bleeding edge are helping figure out what needs to be fixed but I am curious in this case if turning off Scatter/Gather would help:

If the result of the following command is Y, then it is on:

grep [[:alnum:]] /sys/module/mt76_usb/parameters/*

To turn it off:

sudo -i
echo "options mt76_usb disable_usb_sg=1" >> /etc/modprobe.d/mt76_usb.conf
exit
sudo reboot

patrakov, 2024-06-07:
@morrownr I think your previous comment is for a different card. There is no USB WiFi involved in @ilikenwf's setup.

ilikenwf, 2024-06-07:
Yes, my card is PCIe

morrownr, 2024-06-07:
> Yes, my card is PCIe

Serious lack of coffee around here this morning. What was I thinking?

dudez7, 2024-06-08:
FYI @ [A Quick Look at MSI's New Roamii Mesh Systems and WiFi 7 USB Dongle](https://www.techpowerup.com/323191/a-quick-look-at-msis-new-roamii-mesh-systems-and-wifi-7-usb-dongle)

""_MSI was also showing off the BE6500 WiFi 7 USB adapter which is a tri-band dongle and the first of its kind that we've seen. It supports speeds of up to 2880 Mbps on the 6 and 5 GHz bands and 688 Mbps on the 2.4 GHz band, assuming you have a suitable WiFi 7 router or access point. It also offers support for MLO and interestingly enough, it also has built in drives, which means you don't need to download drivers to get it up and running. Finally MSI is promising support for Windows 10 and 11, as well as Linux. So far Windows 10 has lacked proper support for WiFi 7 devices, so this is an interesting development._""

Guess it should be based on mt7925?

morrownr, 2024-06-08:
> Guess it could be based on mt7925?

Well, we only have two companies that produce usb wifi chips these days. It has to be either a Mediatek or Realtek chip. The details do appear to match the mt7925 but I have no details at all on what Realtek is up to regarding usb wifi. The fact that Linux support is mentioned by MSI could be a big hint that it is indeed the Mediatek chip.

I'll do some snooping around to see what I can find. I have already found the below link:

https://www.tp-link.com/us/home-networking/usb-adapter/archer-tbe400uh/

TP-Link using a Mediatek chip? Something just froze over if it is true.

morrownr, 2024-06-08:
FYI: One thing I do know about Realtek and WiFi 7 support for Linux is that Realtek cannot use the same technology as their out-of-kernel drivers have used as there is code in the Linux kernel that will stop Linux wifi 7 drivers that are not fully compliant with modern standards. So things can't be business as usual concerning Realtek and usb wifi 7. They could add usb support to rtw89 but I have not seen any indications of that so far. My best guess right now is that Realtek has simply fallen way behind Mediatek with usb wifi on Linux and maybe on other platforms as well.

bjlockie, 2024-06-08:
> Well, we only have two companies that produce usb wifi chips these days. It has to be either a Mediatek or Realtek chip. The details do appear to match the mt7925 but I have no details at all on [...]

TP-Link is suspect.
I would bet it is Realtek.
What is the latest kernel they can support with an old-tech driver?
All they have to do is support one kernel and they can claim Linux support.
I have a TP-Link router and it has the crappiest GUI ever.
I'm using OpenWRT on a Raspberry Pi and the TP-Link I just use for the wireless 6 AP. :-)
I don't trust TP-Link as a company.
I thought of an idea, maybe the will use Mediatek and claim Linux support and then switch to Realtek but keep the same product number.

morrownr, 2024-06-09:
@bjlockie 

> TP-Link is suspect.

Agree. I was not advocating this TP-Link product.

> What is the latest kernel they can support with an old-tech driver?

I don't remember but this is not new. It was several years ago.

> I don't trust TP-Link as a company.

Me either.Too many years of seeing them do things that make it clear that they do not care about Linux.

> I thought of an idea, maybe the will use Mediatek and claim Linux support and then switch to Realtek but keep the same product number.

Now hold on. You did not think of that. You are reading it from the TP-Link playbook.

morrownr, 2024-06-10:
@ilikenwf 

Would you mind posting the result of `iw list` for your mt7925?

I'd be curious to see what capabilities it is reporting. I think @Muh87 had ask back up the thread so I think he wants to know as well.

ilikenwf, 2024-06-11:
Once my laptop gets back to me I can - the backlight died.

Muh87, 2024-06-11:
> @ilikenwf
> 
> Would you mind posting the result of `iw list` for your mt7925?
> 
> I'd be curious to see what capabilities it is reporting. I think @Muh87 had ask back up the thread so I think he wants to know as well.

Thanks for the reminder .

joshschmelzle, 2024-06-11:
MT7925B22M:

```
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
00:15.1 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #1 (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:16.3 Serial controller: Intel Corporation Sunrise Point-LP Active Management Technology - SOL (rev 21)
00:17.0 SATA controller: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #2 (rev f1)
00:1c.3 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #4 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point LPC/eSPI Controller (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (4) I219-LM (rev 21)
01:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS522A PCI Express Card Reader (rev 01)
02:00.0 Network controller: MEDIATEK Corp. Device 7925
03:00.0 Non-Volatile memory controller: Silicon Motion, Inc. SM2263EN/SM2263XT (DRAM-less) NVMe SSD Controllers (rev 03)
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 427 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 1 usec (0x03)
		HT TX/RX MCS rate indexes supported: 0-15
[... 857 lines cut ...]
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
		* [ BEACON_RATE_HT ]: HT beacon rate setting
		* [ BEACON_RATE_VHT ]: VHT beacon rate setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ ACK_SIGNAL_SUPPORT ]: ack signal level support
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
		* [ POWERED_ADDR_CHANGE ]: can change MAC address while up
```

morrownr, 2024-06-11:
Hi @joshschmelzle 

Thanks for posting. What is your opinion of the chip/driver so far?

Muh87, 2024-06-13:
Thank you so much for the reply .
I see it supports 160 MHz as an access point , hopefully it would be possible to get 2400 MBPS when connected to it .

morrownr, 2024-06-13:
> I see it supports 160 MHz as an access point , hopefully it would be possible to get 2400 MBPS when connected to it .

I order an M.2 mt7925 card yesterday so I should have one soon. We'll see what speeds we get. My router is only WiFi 6 but it can push packets fairly well. For testing AP mode, my RasPi4B is setup and runs as a dual band AP but does not support M.2 so I'll have to figure out how to test AP mode.

lmcarneiro, 2024-06-17:
Hi everyone, I bought a Legion 9i and it came with the mt7927 card. I read in this repo that it's basically the same card as the mt7925 but has higher bandwidth. Should the mt7927 card work with the mt7925 driver? If not, is updating the driver to include support for mt7927 a big change? Sorry if this isn't the place to ask these questions, I just want to use my dumb laptop without attachments!

morrownr, 2024-06-17:
Hi @lmcarneiro 

> Should the mt7927 card work with the mt7925 driver?

Not yet. Support for the mt7927 has not been merged.

> If not, is updating the driver to include support for mt7927 a big change?

It is something that Mediatek is going to have to do. Be patient.

 > Also, if there's any diagnostics I can do to help out let me know.

Just use a supported usb wifi adapter until the driver is available. It is like having a toy that you can't play with.

morrownr, 2024-06-29:
My M.2 mt7925 arrived last week. I installed it in a Mini PC that I have. The distro is Ubuntu 24.04. The card came to life. Connection was uneventful. Use has been uneventful in managed mode. I do not have a WiFi 7 / 6 GHz capable AP yet so have not tested that. Nothing bad to report regarding wifi at this point.

Bluetooth, on the other hand, did not come up so I investigated. The firmware and driver were in place in Ubuntu 24.04. Further investigation led to the answer. The vid-pid for the bluetooth capability was simply not in the kernel yet but I found a patch that has already gone in so help is already on the way there. I have a PCIe card with the mt7922 chip and the bluetooth works perfectly with it so I would expect the same with the mt7925.

I will continue to monitor linux-wireless for patches that add mt7927 support. Hopefully that comes soon.

@morrownr

patrakov, 2024-06-30:
Does 160 MHz channel width work? Do you have any speed test results?

morrownr, 2024-06-30:
@patrakov 

> Does 160 MHz channel width work? 

I can test that, I have a WiFi 6 dual band router that is capable of 160 MHz. It is currently offline. Give me a change to finish the testing that I am doing and I work on this.

> Do you have any speed test results?

I just did a quick test with iperf3 with a WiFi 5 dual band router that can handle 80 MHz channel width. It showed 645 Mbps. That is what I would expect. I do have a PCIe card with the mt7922 chip that is in my main dev box that gives me over 1 Gbps with my WiFi 6 router so that is what I will expect from this mt7925 card... I'll get some results posted as able.

morrownr, 2024-06-30:
@patrakov 

Here is an initial test with WiFi 6, 160 MHz:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.212 port 43230 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  99.5 MBytes   834 Mbits/sec    0    970 KBytes       
[  5]   1.00-2.00   sec   130 MBytes  1.09 Gbits/sec    0    970 KBytes       
[  5]   2.00-3.00   sec   127 MBytes  1.07 Gbits/sec    0    970 KBytes       
[  5]   3.00-4.00   sec   127 MBytes  1.06 Gbits/sec    0    970 KBytes       
[  5]   4.00-5.00   sec   130 MBytes  1.09 Gbits/sec    0    970 KBytes       
[  5]   5.00-6.00   sec   125 MBytes  1.05 Gbits/sec    0    970 KBytes       
[  5]   6.00-7.00   sec  66.2 MBytes   556 Mbits/sec    2    970 KBytes       
[  5]   7.00-8.00   sec  40.2 MBytes   338 Mbits/sec    3   1.41 KBytes       
[  5]   8.00-9.00   sec  39.5 MBytes   331 Mbits/sec    2   1.03 MBytes       
[  5]   9.00-10.00  sec  43.4 MBytes   364 Mbits/sec    2   1.06 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   928 MBytes   778 Mbits/sec    9             sender
[  5]   0.00-10.05  sec   925 MBytes   772 Mbits/sec                  receiver

```

It is not clear to me why we saw the drop in the last part of the test. I'll have to continue testing. It could be a result of a lot of things. What I do know is that the mt7925 driver is not perfect yet. It can't be because these modern WiFi 6 and 7 drivers are incredibly complex. It is mind bending to me. Like Dorthy once said to Todo, I don't think we are in Kansas anymore.

I think it was last week when I saw a series of 48 patches go through for this driver. I think it was mostly to add capability but I did not have time to investigate all 48 patches. So, for now, my testing shows solid stable managed mode performance on my WiFi 5 router. We did not get the performance I was looking for with the WiFi 6 router as I was expecting sustained performance above 1 Gbps.

I think it is reasonable most Linux users wanting WiFi 7 to expect the mt7925 driver to start solidifying by around time for this years LTS kernel which looks like it will be 6.12.

morrownr, 2024-07-04:
Update:

I saw another large patch set for the mt7925 flow into linux-wireless yesterday. It consist of 29 patches and adds support for MLO. The addition of MLO support may signal that mt7927 support will be coming soon.

Testing:

I tested with Linux Mint 22 beta today. WiFi works, BT not yet. It appears that anything based on Ubuntu 24.04 should be good to go with the mt7925 chip. Will the recent large patches work their way back into Ubuntu's kernel 6.8? I do not know but there is a good change that it could happen. I'd say the lone patch to get BT going is a really high probability.

I'll forecast that any distro based on this year's LTS or later will be in really good shape with cards and adapters based on the mt7925 and mt7927 chips. It look like the LTS for this year will be 6.12 but I don't make that call so we will see.

upsangel, 2024-07-05:
I got my MT7925 from taobao for about 15USD. Inserted into my N100 miniPC but nothing show up (lspci).  Checked the M.2 slot only supports CNVI. I guess the MT7925 is a NGFF instead of CNVI?

morrownr, 2024-07-05:
@bjlockie @upsangel 

I don't have much experience with these little cards. What is the difference between cnvi and ngff?

bjlockie, 2024-07-05:
It is my understanding that Intel created CNvi so they could offload some of the processing from the M2 card to the CPU so they could make cheaper M2 cards.
Pretty stupid idea IMO. :-)

morrownr, 2024-07-05:
> so they could make cheaper M2 cards.

Ouch. Sounds like the Windows Modem thing from back in the 90's. I guess I was lucky that my Mini PC has a ngff slot and the mt7925 card I got is a ngff card. We really don't need this crap.

upsangel, 2024-07-06:
> How do you know if an m2 slot only supports cnvi? I have a laptop that has a cnvi Intel card that I want to replace with a mediatek.

I ask the manufacturer for a specs manual and found it says M.2 （cnvi)

I guess there's no way can tell from the appearance, and yes if it's an Intel CPU higher chance it's cnvi only. But I can't rule out no cnvi exists on AMD platform though.

upsangel, 2024-07-06:
> > so they could make cheaper M2 cards.
> 
> Ouch. Sounds like the Windows Modem thing from back in the 90's. I guess I was lucky that my Mini PC has a ngff slot and the mt7925 card I got is a ngff card. We really don't need this crap.

Good on you! What platform is your mini pc? I hv ordered a minipcie to m.2m2 adapter hopefully it works.

morrownr, 2024-07-06:
> What platform is your mini pc?

AMD. It is a Beelink SER 5 Max 5800H. Been looking for a while. I am not a gamer and this thing is advertised as a game box but I figured I could detune it and I was able to detune it. That keeps the fans quiet. It came with a WiFi 6 Intel card. These days it can be somewhat hard to figure out what you are getting. A couple of years ago I got my wife a new laptop with Intel CPU and it came with a mt7921 chip card. Mediatek has an agreement with AMD so AMD rebrands and sells Mediatek based cards to computer makers.

I'm still evaluating the Mini PC so don't take what I did as a recommendation.

upsangel, 2024-07-06:
> AMD. It is a Beelink SER 5 Max 5800H

I got my N100 with 4 x 2.5GbE ports but no any WIFI. Did a quick research and find these:

WIFI7
M.2 NGFF: MT7925

WIFI6E
MINIPCIE: AX210 (MPE-AXE3000H)  
M.2 CNVIO2: AX211
M.2 CNVIO2: AX411

Is MT7925 the most promising for AP mode on Linux? I saw some sources saying Intel's card AP mode is no hope.

fs30000, 2024-07-06:
> Is MT7925 the most promising for AP mode on Linux? I saw some sources saying Intel's card AP mode is no hope.

Yeah, you can forget AP mode with intel cards. Only 2.4G works on them. Due to some legalilties, just google the kernel issues pages about it.

On my side, i'm battling to have this mt7925 working on a banana pi R4. Running debian instead of openwrt. I need a full linux system. I need to apply those recent MLO patches. Using this card while i don't catch of of those new wifi7 cards from sinovoip. Rare species... Hardly seen around :)

upsangel, 2024-07-06:
> On my side, i'm battling to have this mt7925 working on a banana pi R4. 

Do you mean running MT7925 on vanilla Linux is even more challenging than on Openwrt? I am with perception that Linux kernel and driver support is more updated than Openwrt.

> Using this card while i don't catch of of those new wifi7 cards from sinovoip. Rare species... Hardly seen around :)

Do you mean BPI-R4-NIC-BE14? I recently come across this selling at about 70USB in China: https://m.tb.cn/h.ghODKMN?tk=adLd3Z6xznW HU0854 「BPI-R4-NIC-BE14 香蕉派开发板Banana Pi BPI-R4 wifi7模块」
Though it's ~5 times pricy than MT7925, would it be a good / better option for Banana?

fs30000, 2024-07-06:
> > On my side, i'm battling to have this mt7925 working on a banana pi R4.
> 
> Do you mean running MT7925 on vanilla Linux is even more challenging than on Openwrt? I am with perception that Linux kernel and driver support is more updated than Openwrt.

Yeah, you're right, those MLO patches are very recent, probably openwrt doesn't have them, but then, when the drivers come available to openwrt, configuring the card as an AP should be much easier in openwrt than doing it manually in vanilla linux. I haven't used openwrt tough, please clarify me if i'm wrong.

> > Using this card while i don't catch of of those new wifi7 cards from sinovoip. Rare species... Hardly seen around :)
> 
> I recently come across this BPI-R4-NIC-BE14 selling at about 70USB in China: https://m.tb.cn/h.ghODKMN?tk=adLd3Z6xznW HU0854 「BPI-R4-NIC-BE14 香蕉派开发板Banana Pi BPI-R4 wifi7模块」 Though it's ~5 times pricy than MT7925, would it be a good / better option for Banana?

Yeah, that one. does taoao have it in stock? I can't check, even after translating the page. Never used taobao, just aliexpress. 
Supposedly, that card is a beast. The real deal. Plus the fact R4 has 2 sfp+ ports, wow... Still the sfp+ are suffering some kind of speed problems, probably going to get fixed with time.

upsangel, 2024-07-06:
> Yeah, that one. does taoao have it in stock? I can't check, even after translating the page. Never used taobao, just aliexpress. Supposedly, that card is a beast. The real deal. Plus the fact R4 has [...]

Should be in stock because the page says 9 was sold (but no any review yet) I have purchased many different SBC boards from this seller and should be a reliable one (though I am not assuming any accountability).

R4 has been on market for a while but I still hesitate to get one. My major concern is not sure how to make the best use of it. I know it's fast with two fibre ports, but my current setup is still as the majority: RJ45 for all intranet devices. Hong Kong internet is fast but the fastest is 2.5GbE, that won't make use of the SPF+. That makes the R4 only for research purpose to me. I hope you can share some of your experience so I will pull the trigger.

Back to the BPI-R4-NIC-BE14, if you can't get one via your channel, I am glad to help. Shipping of small gadget from taobao to Hong Kong is nearly nothing, then from Hong Kong to your place shouldn't break a bank (ball park less than 10USD air to most part of the world).

fs30000, 2024-07-06:
Yeah, some were already sold. But went out of stock pretty quickly. Seriously, i doubt that seller has any, just like the sellers in aliexpress. I'll wait for the new batch, besides, new batch means more hardware fixes i think.

For me, R4 means faster wifi, which always gets handy in some situations. Plus the fact it has a nvme slot, as i want to throw backups in there as well. Thing is, apparently when a nvme disk is inserted, the sfp+ ports go down, due to some extra resistors installed on the board. They are making new batches with fixes around that. Still, it is possible to tape some "lanes" on the disk to make it work without problems. I haven't tested this yet.

Be aware, the RJ45 ports are only gigabit, but you can use sfp modules for RJ45 tough. Oh, and also bound 2 RJ45 ports if your switch allows it.

I'll report later how it goes about the R4 of course.

fhteagle, 2024-07-06:
So if I am correctly reading the iw list output:

```

valid interface combinations:
		 * #{ managed, P2P-client } <= 2, #{ AP, P2P-GO } <= 1,
		   total <= 2, #channels <= 2

```

This means the card is capable of making a single AP on two channels simultaneously? As in, it can be a DBDC and/or MLO provider?

@joshschmelzle  -  Once you get the new MLO patches tjhat just landed a couple of days ago in play, can you check the iw list output again and see if this part of the output changes at all? Thanks.

MLO is the one feature I am really waiting for to make a move from Wifi 6 to 7. Finally got my Wifi 6 setup all stable throughout the house, so of course it is time to irritate my girl by starting again to tinker with things lol.

patrakov, 2024-07-06:
@fhteagle you are misreading.

The card can provide an AP on one channel while being connected as a client to another channel. Or, it can connect as a client to two different APs on different channels. But it cannot act as an AP on two channels at once.

morrownr, 2024-07-06:
@fhteagle 

@patrakov is correct. The following is the clue:

#{ AP, P2P-GO } <= 1

You can't have more than one interface in AP mode. 

Adding 2 AP capability has been on our wishlist. Not sure how difficult that would be but the Mediatek devs are neck deep in work on mt7925 and probably on mt7927 support as well... not to mention other Linux drivers they work. Mediatek has been bringing a lot of new SoCs and wireless chips online over the last few months and it is still ongoing. It appears that several SoCs they make are Linux only so they have to get it right and it is their Linux devs that have to do it.

fs30000, 2024-07-06:
Alright guys, i managed to have this card working on the banana pi r4 with some effort :) However i wonder if there are configs tweaks to make this better. Kernel 6.10-rc. Compiled latest hostapd.

 iw list: 

````Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 4
        max scan IEs length: 427 bytes
        max # sched scan SSIDs: 10
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 1 usec (0x03)
                HT TX/RX MCS rate indexes supported: 0-15
                HE Iftypes: managed
                        HE MAC Capabilities (0x08011a000040):
                                +HTC HE Supported
                                Trigger Frame MAC Padding Duration: 2
                                OM Control
                                Maximum A-MPDU Length Exponent: 3
                                A-MSDU in A-MPDU
                        HE PHY Capabilities: (0x2270ce126dc0b3064e3f00):
                                HE40/2.4GHz
                                242 tone RUs/2.4GHz
                                Device Class: 1
                                LDPC Coding in Payload
                                HE SU PPDU with 1x HE-LTF and 0.8us GI
                                NDP with 4x HE-LTF and 3.2us GI
                                STBC Tx <= 80MHz
                                STBC Rx <= 80MHz
                                Full Bandwidth UL MU-MIMO
                                Partial Bandwidth UL MU-MIMO
                                DCM Max Constellation: 2
[... 916 lines cut ...]
###################################################### WIFI 6
# WiFi 6 (bands 2.4 Ghz, 5 GHz and 6 GHz)
ieee80211ax=1
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
he_oper_chwidth=1
# HE center channel (chan + 6)
he_oper_centr_freq_seg0_idx=42

#he_oper_centr_freq_seg0_idx=39
# used by clients to discern the source of interference
# each AP in your area needs to use a different number
# allowed: 1-63
he_bss_color=42

he_su_beamformer=1
he_su_beamformee=1
he_mu_beamformer=1
he_6ghz_max_ampdu_len_exp=7

########################### WIFI 7
#ieee80211be=1
#eht_su_beamformer=1
#eht_su_beamformee=1
#eht_mu_beamformer=1
````

morrownr, 2024-07-06:
@fs30000 

> However i wonder if there are configs tweaks to make this better.

Probably. I'll try to help but tell me what band you are trying to enable and is this for WiFi 6 or 7? If it is for WiFi 6 or 7, I think you are missing some things.

[171 more replies]