morrownr/USB-WiFi #432: AP mode on a RP5, preferibly within OpenWRT

RicardoJCMarques, 2024-05-14:
Long story short I'm trying to make a portable router nas type thing with an RP5. I'm getting a crash course on Linux drivers and stuff as I get used to an OS as barebone as OpenWRT and while I got sad that I can only get like 802.11g to work with the integrated adapter I'm considering buying an adapter in case I need more speed.

And that's where I'm getting overwhelmed with all the information.

I get that mt7921aun = great. mt7612u = good. What about the rtl88x2? In the context of an Access Point? Can all the rtl88x2's even be APs?

The thing is in my country I can buy TP-Link Archer T3U Mini or Plus reasonably cheap (seem to have either the 8812 or the 8822, right?). A bump to something with a mt7612 is about 2.5-3x as expensive and I need to order it from another country and a mt7921 is about 4.5-5x, also from another country.
Is it worth aiming higher? The Netgears do look less stabby to Airport security so that'd be a specific plus I guess....

PS: link 5 in the chipsets page is dead.

morrownr, 2024-05-14:
Hi @RicardoJCMarques 

> And that's where I'm getting overwhelmed with all the information.

Well, you you are between a rock and hard place as the saying goes.

OpenWRT is a good foundation for what you are trying to do but... and this is a BIG but, getting Realtek drivers going on OpenWRT is a real challenge unless you are an expert. I understand what you are saying about the availability of certain products in certain parts of the world. I have lived in several countries and know this first hand. I would suggest talking to as many local computer stores as you can to see what they can get for you. Any decently made adapter with the following chipsets will make this project much easier:

mt7610u
mt7612u
mt7921au

The drivers for those chips are supported by OpenWRT and all you have to do is install them and worry about other things. If you need me to help you search, I can do what I can do. Do you have any in-country mail order stores that you use? If so, give me the link and I will take a look.

RicardoJCMarques, 2024-05-14:
Heyo,

Define expert... I'm a linux driver "beginner" but I'd also consider myself a fast learner xD

There's a price comparator in Portugal called KuantoKusta. That'll give you an idea of everything that isn't available here. If you tell me that I should really, really, aim for the Netgears I can ask the shops where I buy my stuff if they can get any for me. Although I expect price to be more expensive than ordering it from Spain or Germany... Only the Netgear A7000 shows up and that's realtek too right?

Generally D-Link and Tp-Link are the easiest to find. Some Asus. All the Realtek stuff... Netgear used to be around a bit but I'm guessing a large distributor in the country stopped stocking them so now no one sells them explicitly. No Alfa Networking gear at all. Stuff like that only ordering in. I never order things from Asia because customs suck here. The only places outside the EU I frequently buy from are like Digikey and Mouser to get electronics components. They seem to have a few cheap commoditized dongles but nothing noteworthy.

I had assumed that as long as there's a package for a specific chipset I could figure it out? It may not be plug&play but as long as there's someone, somewhere on the internet that has written about it it'd find it. Is it trickier than just having a driver package show up on the LuCI manager?

morrownr, 2024-05-15:
> Define expert... I'm a linux driver "beginner" but I'd also consider myself a fast learner xD

You are welcome to try. There is help somewhere on the OpenWRT website. Keep in mind that you will have to do it again every time you upgrade version of OpenWRT. 

> If you tell me that I should really, really, aim for the Netgears...

I am not saying that.

> ordering it from Spain or Germany...

https://www.varia-store.com/de/produkt/689425-awus036achm-802-11ac-wifi-range-boost-usb-adapter.html
https://www.varia-store.com/de/produkt/689267-awus036acm-802-11ac-mimo-dualband-wlan-usb-adapter-mit-2-4-5-ghz.html
https://www.varia-store.com/de/produkt/693575-alfa-awus036axm-wifi-6-6e-2x2-tri-band-bluetooth-v5-2-usb-adapter.html

If Germany is an option, I see products that are available. You would have to judge for yourself if you want to do business with that retailer.

You could contact Alfa directly at their web site to get info about dealers. However, there are far more products available that use the Mediatek chips.

> Is it trickier than just having a driver package show up on the LuCI manager?

Like I said, you are welcome to compile your own but you will learn that in the end, it may not be worth it. It is complicated to compile those out-of-kernel drivers for OpenWRT and it is something you have to do again with each upgrade. How are your cross compiling skills?

Give me links to some online stores in Spain or Germany that you feel comfortable doing business with and I will help you find some adapters to consider.

morrownr, 2024-05-15:
Another possibility:

https://www.pollin.de/p/allnet-wlan-stick-all-wa1200ac-1200-mbit-s-741314

RicardoJCMarques, 2024-05-15:
Oh, it involves compiling my own stuff? I'm not there yet and don't have time to invest into getting comfortable... +20-40€ isn't worth that headache. That's a set of skills I want but it'll probably be a few years before I can look at that.

Yeah because of the NAS part I'm still going over how to get timely upgrades easily. Anything being packages would also be less than ideal. Even Attended Sysupgrade is giving me trouble with partition sizes for all the packages and docker and what not. PINN would help me not have to create a single do all OS but I don't think they'll ever put up with OpenWRT shenanigans and

Just mentioned the Netgears since it's been around and general reviews seemed favorable, even if they're not covering my use case. Alfa looks nice.

I kinda just order some stuff from Amazon when I can't find it here. Though I'm getting slightly tired of their support. I don't know if google's algorithms are trustworthy enough to find me good stores in other countries... Don't really trust it here lol. I'll give those a look around thanks.
Talking to Alfa might be a good bet too. Shame Rokland doesn't deliver here.

If I find devices with Mediatek chipsets, can I assume they'll have around the same features from the drivers? Even if the actual components may vary in how they're put together? Or does the manufacturer have a lot of control with just the hardware?

One thing I'm quite confused with is the driver packages on the software manager. Are they different from regular drivers I see around for usb adapters? I see a few chipset numbers there and don't understand what's up...
I think I installed the drivers for a usb-ethernet adapter from there? Bit blurry these last few days.

And thanks for taking the time to answer my endless barrage of questions. <3

bjlockie, 2024-05-15:
AFAIK, any adapter with a supposed chipset will be good.
There is a potential problem with usb-ids (from lsusb) but that can usually be fixed in the confirmation of the distro.

I don't understand the question about regular drivers and packages.
Do you mean a driver compiled into the kernel vs a driver as a package?
The package driver is to make the kernel size smaller by not including it for everyone.

morrownr, 2024-05-15:
@bjlockie 

There are some terminology issues at play here. I the terminology issue is coming about because @RicardoJCMarques mentioned that he would prefer to use OpenWRT as his operating system. OpenWRT, if you are not familiar with it, is very different than regular desktop distros because many WiFi router have extremely limited storage and RAM. I'll talking about 8-16 Mb of storage and 32-128 Mb of RAM. Everything that goes into OpenWRT has to be trimmed to a minimum and that includes the kernel. You do not include drivers that you do not need in the kernel and usb drivers are never needed... but can be installed if you do want to use a usb adapter.

OpenWRT is similar to Debian and other distros in that it has a package manager that can download and install packages from the internet and install the contends. The packages can include utilities, apps and drivers. You can install what you want as long as your hardware has the storage. RasPi4B and RasPi 5 have massive storage and RAM compared to most routers so no problem adding usb adapter drivers.

OpenWRT has pre-build drivers that are included in packages for several usb wifi chipsets. It looks like all Mediatek drivers are available. To install the mt7610u driver, the package is:

kmod-mt76x0u

You can even install the mt7925 driver and we don't even have adapters on the market yet:

kmod-mt792x-usb

At this point, I have either helped or confused things further. I did not address Realtek support so here goes.

At OpenWRT, you will find some of the Mediatek devs involved in the organization. Do you see any Realtek devs involved? Not that I am aware of. Also, note that mt792x driver is in the current OpenWRT that runs kernel 5.15. Well, that driver did not enter the mainstream kernel until 6.7... so who backported it to work on OpenWRT? The Mediatek devs.

Compiling Realtek drivers for OpenWRT becomes complicated for many reasons. You will need to set up an environment to cross-compile. That is, compile the driver on your desktop for the kernel/cpu in your hardware that will run the compilation and then you have to copy the driver into OpenWRT... and you have to do this every time you want to update OpenWRT. This just makes one want to run screaming naked into the cold dark night!

The bottom line is that if you want to run a usn adapter with OpenWRT, get an adapter with a Mediatek chip. Period.

RicardoJCMarques, 2024-05-15:
Hum, think that settles it. No realtek for me. Worst case scenario I should be able to get Alfa or Netgear from Amazon. I'll look around a bit for a store in Spain or something or just order some next time I get more stuff from Amazon in a couple of weeks.
For now I'll focus on dealing with the nvme drives on openwrt and docker and I'll pop back over once I have updates...

I do have one final question for now: outside of the context of OpenWRT, are the basic realtek chipsets fine for client and maybe AP mode? Like with basic Raspbian or something? Or even then they can be problematic? <3

morrownr, 2024-05-15:
> are the basic realtek chipsets fine for client and maybe AP mode?

Some are better than others for client and AP mode. We would need to talk specifics like what you have available to buy. It is important to avoid multi-state adapters and all of the Realtek WiFi 6 adapters that I am aware of are multi-state. Driver support for Realtek WiFi 6 usb is really bad so I would recommend staying completely away from Realtek's WiFi 6 chips. For WiFi 5, it depends on the chip.

> Like with basic Raspbian or something?

I have a guide on the Main Menu for setting up a Bridged Wireless Access Point with RasPiOS 64 bit on a RasPi4B. I would still recommend Mediatek chips but some Realtek chips can work reasonably well in this type of setup. We can discuss the details.

> Or even then they can be problematic? 

Progress is being made with the rtw88 in-kernel driver series but you need to be careful what chip you use.

If you are making an Amazon order in the next 2 weeks or so, once you have some adapters picked out, send me the links and I will give you my analysis.

@morrownr