morrownr/USB-WiFi #444: Alfa wireless cards from official distributer

Getic-Wifi, 2024-05-30:
It’s easy to buy an original Alpha wireless card from an official distributor at competitive prices.

- Fast delivery all over the world 
- 14+ years of experience on the market and hundreds of thousands of satisfied customers.
-  some of the lowest prices on the Internet
 
[Alfa-wifi-usb-adapters](https://www.getic.com/shop/alfa-wi-fi-usb-adapters)

Best regards,
Getic

bjlockie, 2024-05-30:
Shipping seems expensive but maybe it's good if you live in Europe.

morrownr, 2024-05-31:
@Getic-Wifi 

It is good to see your post as users here often ask about Alfa retailers. I understand how much of a challenge worldwide shipping is as I have lived in multiple countries. I have your site listed in the Plug and Play List for one adapter but have no problem adding your site to other adapters. One thing to understand is that if I get information from users that they had a bad experience with Getic, I will remove references to your site. The links here to your site will include direct links to products and those products will be the adapters that are currently Plug and Play.

Users:

After looking at the Getic page in top post, here is a list of the Alfa adapters that use good quality in-kernel drivers:

Alfa USB Adapter AWUS036ACM
Alfa USB Adapter AWUS036NHA
Alfa USB Adapter AWUS036ACHM
Alfa USB Adapter AWUS036AXM
Alfa USB Adapter AWUS036AXML

Note: the adapter that is best for a specific user case depends on the use case. Ask if you have questions.

As improvements to the rtw88 drivers are made this year, additional adapters could be added to this list.

Regards,

@morrownr

Getic-Wifi, 2024-05-31:
Yes, the company and warehouse are located in Europe.
And in Europe, delivery prices start from 3-5-7 euros.
Considering the product's often best price on the market, this comes out to be the best deal.

Getic-Wifi, 2024-05-31:
@morrownr 
Thank you,
adding links to our website for the products we have (Alfa) would be very kind of you.
Getic, as the official distributor of Alfa, guarantees the quality of the products it offers to its customers.