morrownr/USB-WiFi #448: Panda PAU0D

unhandyandy, 2024-06-02:
Works on Ubuntu 24.04 out of the box, using the driver mt76x2u provided by linux-modules-6.8.0-31-generic.

One mistake I made: installing linux-modules-iwlwifi-generic which lead to module disagreements and failure to load.  After uninstalling that, it worked fine.

morrownr, 2024-06-02:
@unhandyandy 

> Works on Ubuntu 24.04 out of the box...

As it should. The mt7612u driver has been in the kernel since 2018 and is very solid and feature rich. The ChromeOS even includes it these days so Chromebooks and other devices based on the ChromeOS make this adapter plug and play.

> One mistake I made: installing linux-modules-iwlwifi-generic which lead to module disagreements and failure to load. 

I'm curious what made you think loading the iwlwifi driver would help with this adapter? I'm always looking for ways to improve the docs around here.

Would you mind doing a short review of this adapter as you have time? In the Plug and Play List, there is no review yet. Other users would appreciate it.

bjlockie, 2024-06-02:
I'm curious why iwlwifi caused conflicts.

morrownr, 2024-06-03:
In theory there should be no conflict. I'd test to see if I can figure something out but I have exactly zero cards that use tat driver.

unhandyandy, 2024-06-03:
Maybe the mere presence of two copies of the same driver triggers warnings?

The mistake I made, I think, was in not rebooting after connecting the device.  The nm-applet didn't see the device, and I thought that was because the correct driver wasn't available.

morrownr, 2024-06-03:
> Maybe the mere presence of two copies of the same driver triggers warnings?

I recently redid some testing to see what drivers support more than one adapter with the same chipset in the same machine.

The Mediatek drivers had no problems. I tested adapters with the mt7610u, mt7612u and mt7921au chipsets. The rtw88 drivers had no problems either. I tested 8821/11cu and 8812bu chipset. It appears that maybe all in-kernel, standards compliant drivers work well. Keep in mind that this testing was one instance of the driver supporting more than one adapter/module which is how things should work. The Realtek out-of-kernel drivers are notorious for not being about to support more than one adapter with the same chip in the same machine.

> I think, was in not rebooting after connecting the device. 

In theory, you should not have needed to reboot for your Panda adapter to work. It is new so maybe some of the usb pins need to break in or clean themselves off. This happens.

unhandyandy, 2024-06-03:
How do I do a review?

morrownr, 2024-06-03:
> How do I do a review?

I don't have a template so it is really up to you. Use the adapter for a while and tell us your experience. Make sure you explain what modes you are using... managed (client), AP, monitor, P2P, etc. Does it seem to have good range? How do you like the included cable? Does it have good speed? Does it get hot?

You can look in the Plug and Play List as some of the adapters have reviews by users. That could give you some ideas.

Users that are looking to buy appreciate what current users have to say. You can post the review here and I will move it into the Plug and Play List if you approve of me doing so.

unhandyandy, 2024-06-03:
OK, but how do I upload it?  Do I make a pull request?

morrownr, 2024-06-03:
Nothing so complicated. Just paste it here and I'll move it to where in goes in the Plug and Play List.