morrownr/USB-WiFi #449: Warning: Use caution if you are considering buying a USB WiFi adapter or module with a Realtek WiFi 6 chip!!!

morrownr, 2024-06-02:
Hello All,

I have seen the messages from users having problems with Realtek based WiFi 6 USB WiFi 6 adapters increase to the point that I think I need to add warning around this site for users to see. As will see below, my strong recommendation is for Linux USB adapter and module users to AVOID products that include Realtek chips.

Why?

Realtek still does not provide USB support for WiFi 6 that is in accordance with established Linux Wireless Standards. All they are providing out-of-kernel drivers that are based on some of their own ideas which means these drivers will never be allowing into the Linux kernel. Even if we don't care about standards, the Linux WiFi 6 out-of-kernel drivers are BAD. I worked on the Realtek out-of-kernel for the rtl8852/32au chip after it was released. I wanted to get it at least to the point that managed mode would work reasonably well for users but I was never able to make that happen. The driver is JUNK. I have a driver up here for the rtl8852/32bu chip but I have warnings posted. The only mode that I can say that works reasonably well is managed mode. The driver as a whole is JUNK. The only way I will consider future work on Realtek WiFi 6 is if it is on USB support for the rtw89 driver. Not sure when that can happen so YOU HAVE BEEN WARNED!!!

Realtek USB support on Linux is BAD. If you want to buy a USB adapter or module that is supported with Linux Wireless Standards drivers, see the `Plug and Play List` on the Main Menu. On the Main Menu is also the `USB WiFi chipset information for Linux` item which gives good information about supported chipset. The information to help you make good decision is here... use it!

Regards,

@morrownr

morrownr, 2024-06-14:
Here is an example of an adapter that Linux users want to avoid:

https://us-store.msi.com/Wireless-Adapter/AXE5400-WiFi-USB-adapter?limit=24

If you scroll down, you can see that the company is listing Linux as supported. Beware! The only support for this chip currently is an out-of-kernel poor quality, non-standards compliant  driver. It could be years before a good quality, standards compliant driver support is available for this adapter and other Realtek WiFi 6 adapters.

The adapter is classified as AXE5400. Mediatek does not make an AXE5400 class chip, only Realtek does. Mediatek goes directly from AXE3000 to BE6500.

Linux users beware! Be informed so you can avoid deceptively advertised products such as the above.

smoke, 2024-11-20:
I only wish I have seen this several hours ago! Thank you @morrownr

morrownr, 2024-11-20:
@smoke 

What did you order?

smoke, 2024-11-20:
> That should be a single-state adapter so that is one good thing to be said. It is the only single-state adapter with the rtl8852cu chip that I am aware of. If you are interested, I am working on a driver for the rtl8852cu but it is private for testing. Let me know if you want me to put you in the private repo so you can test? I still recommend the Mediatek based adapters but I do try to help the community.

Yes, please I am interested in testing. I use Gentoo and I don't mind building stuff from sources. I like helping the community as well.

morrownr, 2024-11-20:
@smoke 

Expect an email with an invitation. See you there.

By the way. I have and use an Edup  AX-1672. Really good adapter. It also has the mt7921au chip like your Edup AX-1673. I think the main difference in the adapters is that yours is a little smaller.

Jerome-Marichez, 2024-12-02:
Hmm i think that's why my  NightHawk AX1800 7500 Netgear don't work then in linux.

Still thanks for your job.

Hope one day theses issue will be fixed one day.

morrownr, 2024-12-03:
@Jerome-Marichez 

The warning is to help. There are many good adapters available that work very with Linux. 

However, if you have an adapter based on a Realtek WiFi 6 chip, I'm currently working on new drivers for the rtl8852bu and rtl8852cu. They are the less-than-optimal Realtek out-of-kernel drivers but I am trying. What chip does your adapter have? If you don't know, run and the results of the following:

$ lsusb

morrownr, 2024-12-03:
@smoke 

Did you get the Edup 1673 yet? If not, give us a report after it arrives.

Andy-45, 2024-12-03:
@Jerome-Marichez

the NightHawk AX1800 7500 should be a Mediatek adapter, it shares the driver with Netgear A8000,
look at the workaround on the Main Menu for A8000

Netgear driver INF file for Windows "mtkwl6eux.inf" -> 

> [MediaTek.NTAMD64.10.0]
> ; DisplayName                   Section                     DeviceID
> ;Mediatek A8000 entry
> %A8000.DeviceDescEXA%       = MTKWLAN_DEFAULT.ndi,    USB\VID_0E8D&PID_7961
> %A8000.DeviceDescEXA%       = MTKWLAN_DEFAULT.ndi,    USB\VID_0E8D&PID_7961&REV_0100
> %A8000.DeviceDescEXA%       = MTKWLAN_DEFAULT.ndi,    USB\VID_0E8D&PID_7961&MI_00
> %A8000.DeviceDescEXA%       = MTKWLAN_DEFAULT.ndi,    USB\VID_0E8D&PID_7961&MI_03
> 
> [Netgear7500.NTAMD64.10.0]
> ; DisplayName                   Section                     DeviceID
> ;Netgear A7500 entry
> %A7500.DeviceDesc7500%       	= NETGEARWLAN7500_DEFAULT.ndi,    USB\VID_0846&PID_9065
> %A7500.DeviceDesc7500%       	= NETGEARWLAN7500_DEFAULT.ndi,    USB\VID_0846&PID_9065&MI_00
> %A7500.DeviceDesc7500%       	= NETGEARWLAN7500_DEFAULT.ndi,    USB\VID_0846&PID_9065&MI_03
> %A7500.DeviceDesc7500%       	= NETGEARWLAN7500_DEFAULT.ndi,    USB\VID_0846&PID_9065&REV_0100

smoke, 2024-12-09:
> @smoke 
> 
> Did you get the Edup 1673 yet? If not, give us a report after it arrives.

It arrived today, it was a real plug and play with the kernel I am using!
I didn't run any tests, but the internet felt nice and snappy, no lags, no hickups, everything smooth and streamlined.

Moreover I love the form-factor of the device and the fact it did extend at a glance noticeably the range and quality of wifi, compared to the built-in AX201 with Realtek chipset.

Thanks again for the instructions you maintain, amazingly helpful!!!

DoomsDay007, 2025-02-07:
> Here is an example of an adapter that Linux users want to avoid:
>
> https://us-store.msi.com/Wireless-Adapter/AXE5400-WiFi-USB-adapter?limit=24
>
> If you scroll down, you can see that the company is [...]

@morrownr Dang it, I got bamboozled! 😱🤯 I just bought this exact USB WiFi adapter. Returning to Amazon now lol. Thank you for the heads up. I only wish I had seen this before I had spent so much time trying to get it work. 🥲 I appreciate you making the list of working adapters, and will chose a different one from there. Thank you so much for your efforts!

morrownr, 2025-02-08:
@DoomsDay007 

>  I only wish I had seen this before I had spent so much time trying to get it work.

Not the first time I have heard that. 

> Ended up going with the EDUP EP-AX1672...

That is a really solid adapter. I have one. 

I have fun maintaining The Plug and Play List and I am astounded at how many people stop by this site. It averages about 25,000 hits per week. I think there are more Linux users out there than we realize.

morrownr, 2025-09-05:
Hi @Fadedlimes 

Hopefully I can help. Run the following command and paste the results in a reply:

$ lsusb

Fadedlimes, 2025-09-05:
Hi, thanks for the reply! Here are the results of the command:
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 258a:010c BY Tech Gaming Keyboard
Bus 001 Device 003: ID 3554:fa09 CX 2.4G Wireless Receiver
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 005: ID 1532:00b9 Razer USA, Ltd Razer Basilisk V3 X HyperSpeed
Bus 001 Device 006: ID 1c75:af90 Arturia MiniFuse 2
Bus 001 Device 008: ID 6557:1621 Emtec USB DISK 2.0
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 2563:0526 ShenZhen ShanWan Technology Co., Ltd. Android Gamepad
Bus 003 Device 003: ID 05ac:12a8 Apple, Inc. iPhone 5/5C/5S/6/SE/7/8/X/XR
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 002: ID 0846:9065 NetGear, Inc. Wireless_Device

```

Fadedlimes, 2025-09-06:
> Is there another way to recover your receipt? There are a lot of good adapters that work really well on Linux, the you have may take going down a long rabbit hole.

Unfortunately not, I guess I will just live with my mistake, thanks anyways!

dubhater, 2025-09-06:
The .inf file in the Windows driver may tell you what chip it is. It's a text file.

The file name of the firmware included in the Windows driver is also a good clue.

Fadedlimes, 2025-09-06:
> The .inf file in the Windows driver may tell you what chip it is. It's a text file.
> 
> The file name of the firmware included in the Windows driver is also a good clue.

`[Settings]

RSSI = -80

Debugmode=0

[Service]
ServiceName=MediatekSwitchUSB
ServiceDisplay=MediatekSwitch for USB

[MT7612U]

DeviceID_1=USB\VID_0E8D&PID_7961
DeviceID_2=USB\VID_0846&PID_9060
DeviceID_3=USB\VID_0846&PID_9065

[LOCKUSB2]

DeviceID_1=PCI\VEN_1B6F
DeviceID_2=PCI\VEN_1B21`

Heres whats in the .ini file.

dubhater, 2025-09-06:
That's .ini, you need .inf.

a5a5aa555oo, 2025-09-06:
@dubhater 

This is the Windows driver for A7500, maybe it uses a MT7961U chip?

[Data1.tar.gz](https://github.com/user-attachments/files/22192127/Data1.tar.gz)

Fadedlimes, 2025-09-06:
> That's .ini, you need .inf.

Ah well there is only the EXE and that INI file in the folder.

a5a5aa555oo, 2025-09-06:
@Fadedlimes 

You can use 7-Zip to extract that EXE file, then extract the DATA1.cab and you will get the .inf files.

Fadedlimes, 2025-09-06:
@a5a5aa555oo alright I’ll do that when I’m back from my walk.

morrownr, 2025-09-07:
@Fadedlimes 

What distro and version are you using?

And what kernel are you using:

$ uname -r

@dubhater @a5a5aa555oo 

> It only includes the firmware for MT7921, so that must be the chip. They call it MT7961 in the firmware file name for some reason.

That is normal for Mediatek. The mt7921x chips are part of the mt7961 family. Given what you are showing, the chip is a mt7921au that uses the mt7921u driver and the mt7961 firmware files.

We can test that with:

Hotplug automation using udev.

Open a terminal: Ctrl + Alt + T

Create a file called /etc/udev/rules.d/90-usb-0846:9065-mt7921u.rules

sudo nano /etc/udev/rules.d/90-usb-0846:9065-mt7921u.rules

Note: you can change nano to the text editor of your choice in the above command.

Copy the below lines and paste them into the above file that you are creating:

ACTION=="add", \
	SUBSYSTEM=="usb", \
	ENV{ID_VENDOR_ID}=="0846", \
	ENV{ID_MODEL_ID}=="9065", \
	RUN+="/usr/sbin/modprobe mt7921u", \
	RUN+="/bin/sh -c 'echo 0846 9065 > /sys/bus/usb/drivers/mt7921u/new_id'"

Save file: Ctrl + O, Ctrl +X

Reboot:

sudo reboot

--- end udev automation

I do need to know the distro, version and kernel version to know for sure that the above test will work. If it does work, I can send a patch toward Felix as soon as my dev box is rebuilt. Early on after the mt7921au chip was released, some adapter makers released some adapters with the chip and the adapters were advertized as AX1800, however, the chip is actually capable of AXE3000 operation. It might have had to do with Win10 not having any 6 GHz support.

Fadedlimes, 2025-09-07:
> [@Fadedlimes](https://github.com/Fadedlimes)
> 
> Okay, so Fedora 42 with kernel 6.15. That should most definitely have the mt7921u driver along with the mt7961 firmware. If we are correct about the chipset, then a quick test by you will confirm it:
> 
> ```
> Hotplug automation using udev.
> 
> Open a terminal: Ctrl + Alt + T
> 
> Create a file called /etc/udev/rules.d/90-usb-0846:9065-mt7921u.rules
> 
> sudo nano /etc/udev/rules.d/90-usb-0846:9065-mt7921u.rules
> 
> Note: you can change nano to the text editor of your choice in the above command.
> 
> Copy the below lines and paste them into the above file that you are creating:
> 
> ACTION=="add",
> SUBSYSTEM=="usb",
> ENV{ID_VENDOR_ID}=="0846",
> ENV{ID_MODEL_ID}=="9065",
> RUN+="/usr/sbin/modprobe mt7921u",
> RUN+="/bin/sh -c 'echo 0846 9065 > /sys/bus/usb/drivers/mt7921u/new_id'"
> 
> Save file: Ctrl + O, Ctrl +X
> 
> Reboot:
> 
> sudo reboot
> ```
> 
> After the reboot, you should have a wireless interface and should be able to connect to an AP... if we are correct about the chipset. Let us know what the test shows.

Holy crap it worked! I can see all the APs around me now 😄. Thank you so much for your help!!

Fadedlimes, 2025-09-07:
> [@Fadedlimes](https://github.com/Fadedlimes)
> 
> You are welcome. It can take putting multiple minds working together to make a difference and [@dubhater](https://github.com/dubhater) [@a5a5aa555oo](https://github.com/a5a5aa555oo) have added a lot to the effort to make USB WiFi support much better on Linux.

Yes! Thanks to both @dubhater and @a5a5aa555oo as well :)!

> Is your AP/wifi router capable of band 4 (6 GHz)? The reason that I ask is that even though Netgear may have advertized the A7500 as AX1800, it very well may be capable of AXE3000. The windows driver for it may be limited to AX1800 but the Linux driver never was limited.

The page for my modem from my ISP says its tri-band capable of 2.4/5/6 GHz.

morrownr, 2025-09-07:
@Fadedlimes 

> The page for my modem from my ISP says its tri-band capable of 2.4/5/6 GHz.

This part can be a challenge but I'm sure we can figure it out.

Here is what I would do to check if band 4 is working: I would make sure that each band uses a different SSID (name) so that I know which band I am connected to:

my2ghzband
my5ghzband
my6ghzband

Something like that. Many routers have a setting that causes the router to use the same name for each band. That needs to be turned off so that you can set each band with its own name. Once that is complete, you can then connect to and test band 4 (6 GHz) and let us know if it is working and if so, how well it is doing... and you can check the other bands to see how well they are doing also.

morrownr, 2025-09-07:
@Fadedlimes 

> Unfortunately it seems to not want to work on band 4...

It may be that Netgear turned the band 4 support off in hardware. A lot of chips are designed where that is possible. How about we put that issue on the shelf to ponder later.

Let's make sure you are on the latest firmware: (distros do a bad job of this):

I have a guide for upgrading the firmware:

Go to the Main Menu:

https://github.com/morrownr/USB-WiFi

See menu item 7 Firmware.

This document provides a lot of information about firmware including how to check the version of your existinf firmware and how to update to the latest firmware from the official Linux repo. Look for Section 3 MT7921.

After finished and rebooted, could you run the following command and paste the results:

$ iw dev

Let me know how it goes.

Fadedlimes, 2025-09-07:
> After finished and rebooted, could you run the following command and paste the results:
> 
> $ iw dev
> 
> Let me know how it goes.

After following section 3 and rebooting and running the command I get:

```
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr 54:07:7d:91:28:0f
                type P2P-device
                txpower 3.00 dBm
        Interface wlp1s0f0u3
                ifindex 3
                wdev 0x1
                addr 02:2f:35:fd:30:59
                ssid WIFI-2FD8
                type managed
                channel 157 (5785 MHz), width: 80 MHz, center1: 5775 MHz
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0

```
Hopefully I did it right.

Fadedlimes, 2025-09-16:
> As a follow on: If you are interested in providing your experience with the adapter and making a recommendation as to whether it should be added to The Plug and Play List, I would appreciate it.

I would say yes, now that it actually works as intended. The adapter has been serving me well and I haven't had any further issues with it 😃.

Always happy to be helpful!

morrownr, 2025-09-18:
@Fadedlimes 

I went ahead and added the Netgear A7500 to The Plug and Play List in the AX1800 section. If you have time to review and check my work, please so.

Glad this adapter is working out for you. It is interesting how this adapter managed to steer clean of those of us who try to support Linux.

oGiestas, 2025-09-19:
> [@Fadedlimes](https://github.com/Fadedlimes)
>
> I went ahead and added the Netgear A7500 to The Plug and Play List in the AX1800 section. If you have time to review and check my work, please so.
>
> Glad [...]

I have this Netgear A7500 working perfectly in windows. However I just recently installed debian 12 on another disk for work and it does not have a driver for this. Is there a workaround to get this to work?

morrownr, 2025-09-19:
>  Is there a workaround to get this to work?

Yes, it was spelled out earlier in this thread but... here is a link to The Plug and Play List:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

Go to the AX1800 section and see the Netgear A7500. The temp workaround is there. The patch that does away with the need for the workaround has gone in so within a few weeks, the workarounds should not be necessary.