morrownr/USB-WiFi #453: RTS CTS in Bandwidth Signaling and Block ACK in channel bonding

serial115200, 2024-06-07:
Hello guys

[image]

I have some doubts about why the sniffer can only capture one rts and cts packet.In theory I should capture multiple RTS/CTS.

> when an 802.11ac device sends an RTS, (1) this initiating device has to verify that the 80-MHz channel is 
clear in its vicinity, (2) the RTS is normally sent in an 802.11a Physical Protocol Data Unit (PPDU) format, and (3) 
the basic 802.11a transmission, which is 20 MHz wide, is replicated another three times to fill the 80 MHz (or 
another seven times to fill 160 MHz). Then every nearby device, regardless of whether it is an 802.11a/n/ac device, 
receives an RTS that the device can understand on its primary channel. And every device that hears the RTS has 
its virtual carrier sense set to busy (see Figure 3a). To make the protocol robust, the replication bandwidth of the 
RTS is reported inside the 802.11a PPDU.

Block ack only captured one, was it only one or four but the driver merged it? Is this related to hardware implementation?

thanks

morrownr, 2024-06-10:
Hi @serial115200 

Can you provide a little information about why you posted this message? Are you using a specific adapter with a specific driver? What is your setup and what are you trying to do?

Thanks

serial115200, 2024-06-11:
hi 
I am a newbie learning Wi-Fi, and found that there is a discrepancy between the results of packet capture and the book, and want to get some help from here.

    OReilly-802.11ac_A Survival Guide.pdf
    802.11ac-The Fifth Generation of Wi-Fi (2014.03).pdf

The device and driver version:

  COMFAST CF-953AX 
  kernel 5.19.0-43

Wireshark wifidump (Wi-Fi remote capture) will setup monitor mode and capture packet.

I suspect that the driver or firmware merged multiple RTS/CTS packages. and Duplicate block ack will be drop.maybe this issue needs to be consulted by the openwifi developers

Can you share some captured Wi-Fi packets.

Thanks