morrownr/USB-WiFi #455: New mt7921au based adapter - Fenvi FU-AX1801D - need confirmation

morrownr, 2024-06-14:
I am throwing out the below information but I do not have the adapter nor do I have reports from anyone to confirm that it uses the mt7921au chip. Evidence points this direction. I went to the Fenvi website and downloaded the windows driver and it does show to be a mediatek mt7921 driver so the information indicating this to be true may indeed be true. The links below have not been verified as sellers you can trust yet.

Fenvi FU-AX1801D

https://www.ebay.com/itm/196415159770

https://www.aliexpress.us/item/3256806897869983.html

It anyone buys the Fenvi FU-AX1801D and would like to verify the chip and vendor, please do so. A review would be nice also. I am looking for information to help make a decision as to whether this adapter should be in the Plug and Play List.

@morrownr

Engino00, 2024-06-23:
Hi morrown, 
I'm going to buy a Fenvi FU-AX1801D (if available in my country) and I'll let you know if the mt7921 of such adapter is "plug-and-play" on linux. Besides this, I notice on the "USB WiFi adapters with Linux out-of-kernel drivers" list that you had a terrible experience with the RTL8832AU/RTL8852AU driver but maybe the Fenvi brand might have a solution. I accidentaly notice that the Fenvi adapter FU-AX1800P, listed on their website, has a driver for linux despite having a RTL8832AU driver. By clicking on the model, you should be able to download it. Here you can find the list of their wifi usb adapters: " https://cn.fenvi.com/drive.html?id=3 ". I don't have a FU-AX1800P so I cannot confirm if it works but maybe you can test it and possibly improve it. I hope you can find this useful.

morrownr, 2024-06-23:
Hi @Engino00 

It is good to hear from you. 

> I'm going to buy a Fenvi FU-AX1801D (if available in my country)

I have lived in various countries, possibly even yours, so I understand that it can be difficult to get specific products in some parts of the world. With a little effort it can be possible to find a source for adapters that work well. That is why I maintain the Plug and Play List with plenty of information. You may not find this specific Fenvi adapter but you probably can find a source for another adapter with the mt7921au chip.

> I accidentaly notice that the Fenvi adapter FU-AX1800P, listed on their website, has a driver for linux despite having a RTL8832AU driver.

I checked. That is the source for the driver that I had so much trouble with. It is a really bad driver. 

This site is really about getting information to Linux users. The Main Menu is:

https://github.com/morrownr/USB-WiFi

Menu item 1 is an attempt to educate Linux users regarding USB WiFi. You will see me say that if sellers say they have Linux support, it is probably based on an out-of-kernel driver that probably won't even compile on your kernel. Will your see posted drivers for adapters that use the mt7610u, mt7612u, mt7921au or mt7925 chips? No. Why? There is no driver for the sellers to post. The driver is located in the Linux kernel where it should be. The driver is maintained in the Linux kernel as it should be.

General rule of thumb: If a seller is posting a Linux driver, run away as fast as you can. Linux is not Windows and posting drivers is a Windows thing. Linux development is near continuous and driver have to have near continuous updates and that can only be accomplished with in-kernel drivers. Realtek does make out-of-kernel drivers for usb wifi adapters but most uses are better off staying away from them as they are problematic.

If you find a local source for an adapter that you think is good, send me a link and I'll give you my opinion.

Saludos,

@morrownr

Engino00, 2024-06-23:
Thank you for sharing your experience. I'll be aware of these deceptive drivers and their sources. If I find something worthwhile I'll let you know. Apart from that, I have just bought a FU-AX1801D from Aliexpress hoping that it's genuine. I'll tear it down to check the components and I'll test it on a raspberry pi 5 (making sure its kernel version is at least 6.1).
Thanks again for your help.

morrownr, 2024-06-28:
Looking forward to your report. The currently posted RasPiOS, 03-15-24, uses kernel 6.6. I have had no problems with this release of RasPiOS and mt7921 based usb adapters.

> I'll tear it down to check the components...

Just run the following, it will tell you what chip is inside by showing the vip/pid:

$ lsusb

Snuupy, 2024-07-04:
Why did Fenvi make another one? What's different compared to the FU-AX1800?

morrownr, 2024-07-04:
@Snuupy 

> Why did Fenvi make another one?

Somebody at Fenvi knows.

> What's different compared to the FU-AX1800?

I think @Engino00 is going to let us know. We can see that the case is a little different.

Snuupy, 2024-07-04:
https://www.newegg.com/fenvi-fu-ax1800p-usb-3-0/p/0XM-00JK-000B1 Newegg shows this adapter with a RTL8832AU

Engino00, 2024-07-04:
> What's different compared to the FU-AX1800?

I bought both the FU-AX1800 and FU-AX1801D. 
I will run some speed and range tests (plus some extras) to compare them.

morrownr, 2024-07-04:
@Engino00 

> I will run some speed and range tests (plus some extras) to compare them.

That is great. Looking forward to your report.

Engino00, 2024-07-07:
Comparison between the WiFi USB adapters Fenvi AX1800 and Fenvi AX1801D running on a Raspberry Pi 5 (kernel version: 6.6.31+rpt-rpi-2712):

In practical terms, these two wifi usb adapters are very similar. 
They share the same controller (Mediatek MT7921U) and the wifi performances are comparable, but not identical.

Indeed, there are differences.
After my tests, I concluded that the Fenvi AX1800 is better than the AX1801D.
The AX1800 has a better wifi speed and range even at long distances and through high-density obstacles.
Despite after hours of tests where neither adapeters showed connectivity loss, it seems that for the AX1800 mantaining a stable internet speed over time is easier and didn't experienced sigificant speed drops.

Another difference regards the antenna design. Both the adapeters have two antennas, but I feel that for the AX1800 it's easier to place and adjust the direction of the antennas than the AX1801D which suffers of larger internet speed drops even when changing its position and the direction of its antennas by a very small degree. However, this disadvantage isn't noticeable when performing normal internet tasks like browsing the web, watching videos and downloading files. Generally the web experince feels the same for both adapters.

Finally the cost. The AX1800 is a clear winner even in this category, costing about 10% to 20% less than the AX1801D.

In conclusion, I raccomend the Fenvi AX1800 over the AX1801D.

For more infos and to see my tests results, you can check the summary in the .txt files I'll attach later.

[Comparison_Fenvi_AX1800_vs_AX1801D.zip](https://github.com/user-attachments/files/16120571/Comparison_Fenvi_AX1800_vs_AX1801D.zip)

Engino00, 2024-07-07:
Anyway, the AX1801D, as for the AX1800, is compatible with linux and it's plug-and-play. In my case, the raspberry pi automatically started using it without downloading any staff.

morrownr, 2024-07-07:
@Engino00 

Good report. I will start working on making an entry for the AX1801D as long as you think it should go in the Plug and Play List. I understand that the AX1800 may be better in some categories but since users from all over the world look at this site, it can be good to include good adapters that work well as it can be hard to find some products in some places.

I like to include a picture and a link to a known good retailer and since you just ordered it, maybe the link to your retailer.

FYI: There are other adapters that use the mt7921au chip but I am picky. I will not include an adapter in the List if I can't include one known good retailer or if their is a question about it causing problems in some situations. Linux users need a good location where they can get information and I hope to keep this site going as long as I can... I get a lot of help. This is not something one person can do. I do appreciate your help.

Engino00, 2024-07-07:
Thank you @morrownr .

I bought both the Fenvi modules via Aliexpress. The Fenvi AX1800 is still sold, and there are plenty of shops offering it. On the other hand, the AX1801D is quite difficult to find.

For the AX1800:
https://aliexpress.com/item/1005006439886497.html
https://aliexpress.com/item/1005006808568845.html
https://aliexpress.com/item/1005006871124629.html
(the name of the shop of the first link is called "fenvi Official Store" but never trust "official" in anything from aliexpress)

For the AX1801D I only found these:
https://aliexpress.com/item/1005007076639821.html
https://aliexpress.com/item/1005007097614925.html
https://aliexpress.com/item/1005007088760987.html
(The only visible difference between the AX1801D and the AX1800P is that the AX1801D has a yellow writing instead of a white one)

The AX1801D module is also available on SHEIN, but at triple the price, and I do not recommend anyone buy electronic items from them. Furthermore, there are no results on Temu and Amazon.

Tomorrow I'll send some photos and maybe some interior shots of the AX1801D.

Engino00, 2024-07-10:
Here I've attached some photos of the AX1801D.
Inside there's the MT7921AUN, a bunch of capacitors for filtering and stability, diodes, a crystal oscillator and two ceramic biplexers for each antenna.
[Fenvi_FU-AX1801D_pictures.zip](https://github.com/user-attachments/files/16165131/Fenvi_FU-AX1801D_pictures.zip)

morrownr, 2024-07-10:
@Engino00 

I started adding a section to the Plug and Play List for the AX1801D. It is at the bottom of the mt7921au section. Take a look and tell me what you think should be added or fixed.

frodri, 2024-08-23:
@morrownr 

Device: Raspberry Pi 4
OS: Raspberry PI OS on kernel 6.6.31+rpt-rpi-v8.

When used in AP mode, the Fenvi AX1800 seems to have some issues as far as txpower is concerned. `iw dev` shows 3 dBm regardless of any `iw set txpower` changes, and its range suffers greatly on environments with relatively low ambient interference (e.g.: office with air conditioning hum).

Snuupy, 2024-08-23:
@frodri I noticed on aliexpress the specs listed are max 20 dBm in some listings so I'm not surprised the signal is weaker.

morrownr, 2024-08-23:
Hi @frodri 

> iw dev shows 3 dBm

This is incorrect output. Several Mediatek drivers show this same problem. It is a cosmetic issue and I have no idea why it has not been fixed.

> its range suffers greatly

We can work on finding out why your AP mode range is not meeting your expectations. Is that some you want to pursue?

I do feel the need to add a couple of things: USB WiFi adapters are not optimized for AP mode. They are capable of doing several things, including AP mode but are not optimized for AP mode. Wifi routers and AP's are optimized for AP mode because that is what they do. 

Another thing to note is that adapter makers are basically stuck as far as txpower is concerned. Laws and regulations govern the maximum txpwr an adapter can use according to laws and regulations. Most modern adapters these days have fixed settings for txpwr. Range is basically a function of the quality of the amps, antennas and band in use. Good quality antennas and amps can make a big difference but they are expensive. Don't expect good quality amps and antennas from a cheap adapter. The maker of the adapter controls what amps and antennas are used, not the chip maker so range varies based on the individual product. All things equal, cheap adapter means less range.

I have helped people get the most out of their adapters and will try if you are interested...but I cannot work miracles. 

Hope this helps.

Regards

frodri, 2024-08-23:
Thanks for taking the time to respond, @morrownr. I'm not expecting miracles out of this Fenvi adapter, but if I can get it to work reliably, that's enough for me.

I've been running a jerry-rigged AP with an ancient Edimax USB adapter for a while now, and I wanted to see if I could upgrade the adapter to something with an in-kernel driver this time around. I did some testing with a Wifi analyzer after you mentioned the issue with the iw dev readings, and while I was able to see the txpower adjustments take effect, some of the gear in my other room was still disconnecting despite the power changes.

I'm doubting txpower is the issue now, but the question remains: what else could be causing the disconnects? Old drivers, perhaps?

morrownr, 2024-08-23:
@frodri 

> I'm doubting txpower is the issue now, but the question remains: what else could be causing the disconnects? 

It could be a lot of things.

Are any of your client systems running Linux? If so, try installing `wavemon`. It can help us monitor some things regarding the signal.

Exact which Fenvi is this? Fenvi FU-AX1800 or Fenvi FU-AX1801D

There was a user of a Fenvi FU-AX1800 early this week that had posted about the chip running hot in AP mode. He removed the cover and stuck a little RasPi heatsink on the chip and it then ran cool enough to be stable. I hope this is not the case with your adapter. Users of this adapter last year gave no indication along these lines and based on their recommendations I added the adapter to the Plug and Play List. Let's no go the route of pulling the cover off yet. Get `wavemon` going first and lets see what is going on.

frodri, 2024-08-24:
@morrownr 

My adapter's a FU-AX1800. I can't say I have spare mini-heatsinks lying around, though.

I don't have a Linux client running at the moment, but I think I can have my old laptop boot off an USB drive to diagnose this. I'll report back once I get `wavemon` up and running.

frodri, 2024-08-30:
@morrownr

I've set up my old laptop with Debian 12 and `wavemon`. It's 2.4GHz only, though - would that be an issue?

morrownr, 2024-08-30:
@frodri 

That may help depending on the problem.

On your client system, open a terminal and run `wavemon`. It is a good little utility to watch while doing things that have caused the drops in the past. You may need to move the terminate to one corner and reduce the size and possible even reduce the size of the windows for perhaps your browser if that is what you were using so you can see both at the same time.

If you have a drop, the information on the terminal should disappear. We would like to know your signal-level and link-quality numbers that you are seeing.

If you are using the client that only supports 2.4 then your Fenvi in AP mode will need to be using 2.4.

morrownr, 2024-08-30:
@frodri 

This may not have anything to do with your problem but I just noticed new firmware for your Fenvi adapter (mt7921au) just came into linux-wireless today. It should be posted and available for download at some point next week. Note the Firmware menu entry on the Main Menu. Firmware updates do not list what was fixed but this was a rapid up from the older files of about 2 months ago so this could be an urgent issue. You would need to update the 2 firmware files on your system that is running the Fenvi adapter in AP mode.

frodri, 2024-08-31:
@morrownr 

I moved one of the devices from the other room (a smart clock) to monitor when it starts to drop its connection.

Signal level typically stays at a steady -30dbm, and the link quality stays at 100% for the most part.  However, when the smart clock disconnects, it seems to sync up with both the signal level and link quality numbers going up and down like an EKG - and until those numbers level out, the clock is not able to reconnect.

I will keep an eye out for the release of those firmware files so I can give them a shot.

madushan1000, 2024-09-09:
I bought one of these(AX1801D) for very cheap(~5EUR) on aliexpress.
I can only get about ~500Mbits/s even if I'm very close to the dongle when it's on AP mode while using a wifi 6 client. 
Is the advertised ~1200Mbit/s only available on wifi 6e? 
Also is is supposed to support 160Mhz channels and MU-MIMO? 

`iw` shows this on kernel `6.8.0-41-generic`, but I'm not sure if that's a software issue.
```
Supported Channel Width: neither 160 nor 80+80
```

morrownr, 2024-09-09:
Hi @madushan1000 

> Is the advertised ~1200Mbit/s only available on wifi 6e?

1200 Mbps is available with WiFi 6. The 1200 Mbps is the link speed, not the transfer speed. This is half-duplex stuff so take half of the link speed and there you have about what you max transfer speed should be... so 500 Mbps transfer speed is really good.

> Also is it supposed to support 160Mhz channels

The mt7921au chip supports 80 MHz channel width.

> and MU-MIMO?

Haven't checked.

Overall, not sure that I see a problem. You could post your hostapd.conf so we can check if there is anything wrong but with the speed you are showing there couldn't be much wrong.

madushan1000, 2024-09-09:
@morrownr Ah that sounds great,  thanks for the explainer :) 
For my use-case 500 Mbps is more than enough. I was just curious if I can get higher speeds.

morrownr, 2024-09-09:
>  I was just curious if I can get higher speeds.

You would have to do some testing to see if another channel would be faster due to less traffic. I regularly get about 600 Mbps transfer on my mt7921au based usb wifi adapter AP Mode but there is very little congestion on the channel I use.

Remember that WiFi 6 is also available on band 1 (2.4 GHz). I was tweaking this last week in my hostapd.conf WiFi 6 guide that is on the Main Menu under the AP Mode section. That reminds me that I need to go finish editing that document. Anyway, band 1 can give you longer range so it really depends on what you need.

morrownr, 2024-09-09:
Oh, the MU-MIMO question. About 3-4 years ago, I did some testing and research. What I found was that you can see just a little help if the client is just the right amount of distance from the AP. I think it was in the 22-30 feet distance. It is of no help if farther away or if closer. The help was very marginal. This is based on real world testing with multiple adapters and AP's.

Basically my opinion is that MU-Mimo is mostly useful for marketing departments so I do not pay attention to it anymore.

madushan1000, 2024-09-09:
Interesting stuff! I will play around with the adapter a little bit more and report back if I find anything new. Thank you :)

frodri, 2024-09-15:
@morrownr 

I have another update for the FU-AX1800. I've updated its firmware to the new linux-wireless drivers you mentioned, and they seem to address the connection drops when in AP mode, judging from my hostapd logs over the last couple of days. I will continue to monitor for issues until the end of the week, but if that goes well, I don't see any remaining problems for the FU-AX1800.

frodri, 2024-11-10:
@morrownr 

Additional update from over a month of usage in AP mode: 

The FU-AX1800 dongle seems to work fine with the new linux-firmware drivers with one major caveat - the dongle needs to be plugged into USB2. Otherwise, the connection drops for the smart clock seem to return once again if I try something like running a Fast.com network speed check. I can confirm, though, that the message logs that appear the Debian post in issue #508 (https://github.com/morrownr/USB-WiFi/issues/508) seem to sync up with the instances where the connection drops on my end. I haven't done much BIOS troubleshooting on it since I got a MT7612 dongle for my AP, but if I were to dig deeper into what's going on, that's where I'd start.

Here's the hardware specs I was using during my FU-AX1800 test, in case it helps:
SBC: Raspberry Pi 4
OS: OpenWRT 23.05.05
Additional Ethernet Port: TP-Link UE300 on USB3 Port (RTL8153)

morrownr, 2024-11-10:
@frodri 

Run the following and tell me what you get?

grep [[:alnum:]] /sys/module/mt76_usb/parameters/*

I just tested it on my RasPi4B AP but I run RasPiOS and OpenWRT so hopefully it works.

FYI: I do use OpenWRT, just not on my Pi's.

Also, how many devices do you have plugged into usb ports? And what are they?

frodri, 2024-11-10:
@morrownr 

Sure. I still have the SD card I set up during the FU-AX1800 test, so if you need more info, let me know.

```
// Note: disable_usb_sg is manually set to 1
$ grep [[:alnum:]] /sys/module/mt76_usb/parameters/*
Y
```

I only used those 2 devices during my trial: the FU-AX1800 (connected using a USB 3 cable extension), and the TP-Link UE300.

frodri, 2024-11-12:
> Have you tried without the extension cable?

The dongle's too big. It gets in the way of the Pi's Ethernet jack if I plug it in directly.

I did try using two of my USB3 hubs instead of the extension cable. Didn't work with them either.

morrownr, 2025-10-26:
@sjain882 

This is probably not an issue with the adapter or its driver. This is likely an issue with Linux Mint. I have multiple adapters that are similar to yours in that they use the mt7921 chip and the mt7921u driver. I have not seen problems with WPA3 for a few years but your version of Linux Mint appears somewhat dated. What version of Ubuntu is your Linux Mint based on?

morrownr, 2025-10-27:
If your AP/router is set to WPA3 only and your client connects, then WPA3 is in use whether the Mint interface is offering the option or not. It have been a couple of years since I have used the Ubuntu based Mint and I don't remember if this was an issue.

sjain882, 2025-11-01:
Thanks for the reply. Unfortunately my ASUS router does not have the ability to show a device's security type.

However, I ran `sudo wpa_cli status` which reported `key_mgmt=SAE`, so I can confirm this adapter works with WPA3 out of the box on Linux, and the Linux Mint GUI simply reports it incorrectly.

exkc, 2026-08-18:
i also got mine 

see https://github.com/morrownr/USB-WiFi/issues/726

i didnt see the issue in here so i  made a new issue here