morrownr/USB-WiFi #456: News: Plug and Play List Updated

morrownr, 2024-06-15:
I try to update the Plug and Play List every 2 months or so and I completed the last update yesterday. I usually do some reading and research while updating the list so that I keep my perspective up to date. Here are some of my thoughts:

We still don't have any new adapters based on the new mt7925 chipset. The mt7925 is a new WiFi 7 chipset that supports 160 Mhz channel width and is tri-band. It will be available for USB and PCIe. I recently ordered a little M.2 card so as to check out the driver that has been in the kernel since kernel 6.7. I'll give a report on that when able. I have noticed recent patches of great length going into the kernel in an effort to continue to add capability to this new driver. I am also see bug fixes come in. One thing to note here is that Mediatek has also released the mt7927 chipset... and M.2 cards are also available for this chipset... but Linux users need to know that support for the mt7927 is NOT in the kernel yet. The mt7927 supports 320 MHz channel width and will be available for USB and PCIe.

We have numerous good quality adapters available for Linux users depending on your needs and budget. The Plug and Play List has adapters from numerous brands:

Alfa
Panda
Edup
Fenvi
Netgear
ALLNET
PIX-LINK
BrosTrend
ANDDEAR
ZyXEL
Asus
Zibo
Various Generic brands

There are also adapters available at many prices. There are single band adapters that work well that are very cheap. I have found it to be very handy to keep a single band adapter that is plug and play in my toolkit just in case you need it. Things happen. 

I hope everyone enjoys the latest update to the Plug and Play List and as usual, let me remind you that your feedback and recommended additions/deletions are welcomed and needed.

@morrownr

DelanyLove, 2024-06-25:
I recently bought Alfa AWUS036ACS and it is worst worst experience. 1st i was having issues with drivers and then available networks were not showing up. After solving all the issue now it breaks when it try different wifi attacks. I am considering to buy Fenvi. Please let me know if it is worth to buy it ? 

Source - I posted my issue in reddit and moderator provided me your link.

bjlockie, 2024-06-25:
The Alfa AWUS036ACS has a Realtek chipset and I can't say enough BAD things about Realtek USB and Linux.
Someone might know which driver for the RTL8811AU is best but I would get an adapter with a Mediatek chipset on the plug&play list (https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md).
What distro/kernel do you have?

DelanyLove, 2024-06-25:
I am using latest Kernel I think 6.8.11amd

morrownr, 2024-06-25:
@DelanyLove 

An effort is underway to add in-kernel driver support for the 8821/11au chip that is in your adapter. Testing is underway so I am recommending that you help us test by going to the following location:

https://github.com/morrownr/8821au-20210708/issues/133

Read the first message as it gives instructions and the location of the repo that has the driver. Let me know if you have any problems.

morrownr, 2024-07-11:
@DelanyLove 

Checking on this issue. Were you able to test the driver I posted about above? The intention is to upstream it to Linux mainline soon?

Also, that 6.8.11 kernel may be one that had some serious wifi stack problems. This came up recently with 6.8. It is stuff that we can't fix with driver changes and I'm not sure if this has been fixed. It was a high priority issue so you might check to see if a new kernel is available for your distro.