morrownr/USB-WiFi #458: How do RTL8822CU and RTL8822BU behave with rtw_switch_usb_mode=1 in a USB 2 port?

dubhater, 2024-06-20:
RTL8812AU and RTL8814AU drivers try to switch to USB 3 mode only once when plugged into a USB 2 port. What about RTL8822CU and RTL8822BU?

If they all behave the same it would simplify things for rtw88.

morrownr, 2024-06-20:
I assume you are asking about the vendor drivers? Right? I guess it would have to be the vendor drivers as rtw88 does not have the ability to detect and switch to usb3 mode unless I have missed it.

I have adapters with the 8812bu chip so I can check that. Are you saying that you set the module parameter `rtw_switch_usb_mode` to 1 and tried to enter usb3 mode in a usb2 port? Okay, after looking at that question, it seems not to make sense.

What I can say is that it seems to me that all of the usb3 capable Realtek WiFi 5 vendor drivers seem to operate the same way. If you need to me to do a specific test for rtl88x2bu, let me know what to do.

dubhater, 2024-06-21:
Yes, the vendor drivers.

> Are you saying that you set the module parameter rtw_switch_usb_mode to 1 and tried to enter usb3 mode in a usb2 port?

That is correct. The idea is to see if rtw_switch_usb_mode=1 can be made the default. The driver has to behave well even when the device is plugged into a USB 2 port.

I was expecting 8812au/8814au to get stuck in a loop trying to switch to USB 3 over and over, but they don't. They try to switch once, the device disappears, it comes back still USB 2, of course, and that's it. They work like normal after that.

The code for switching RTL8822CU and RTL8822BU is different and I don't have a device to test. The specific test:
- load the driver with rtw_switch_usb_mode=1
- plug the device into a USB 2 port 
- monitor dmesg

morrownr, 2024-06-21:
Sorry for the confusion. I was not fully understanding what you wanted.

> The code for switching RTL8822CU and RTL8822BU is different and I don't have a device to test. The specific test:

I'll do the test tomorrow on the 8812bu using the 88x2bu driver. I don't have a 8822cu based adapter so maybe someone that does will stop by.

FYI: You might compare the code for the 76x2u in the mt76 driver series as it auto-switches very well. There might be an idea there.

morrownr, 2024-06-21:
Test results:

Adapter: rtl8812bu based.
Driver: 88x2bu (the 88x2bu vendor drive in this site)
Distro: clean Ubuntu 24.04

> The idea is to see if rtw_switch_usb_mode=1 can be made the default. The driver has to behave well even when the device is plugged into a USB 2 port.

I did a clean compilation, set the module parameter to 1 and reboot with adapter in a usb3 port. Ops normal as expected.

I then pulled the adapter out of the usb3 port and put it into a usb2 port. No wifi interface and ugly dmesg. I then rebooted the system with the adapter in the usb2 port. Samr thing. No wifi interface and ugly dmesg... 

[   30.087381] rtl88x2bu 1-3.3:1.0: Runtime PM usage count underflow!
[   30.108043] usb 1-3.3: USB disconnect, device number 15
[   30.799628] usb 1-3.3: new high-speed USB device number 16 using xhci_hcd
[   30.883937] usb 1-3.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[   30.883954] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   30.883962] usb 1-3.3: Product: 802.11ac NIC
[   30.883968] usb 1-3.3: Manufacturer: Realtek
[   30.883973] usb 1-3.3: SerialNumber: 123456
[   32.881003] rtl88x2bu 1-3.3:1.0: Runtime PM usage count underflow!
[   32.924055] usb 1-3.3: USB disconnect, device number 16
[   33.616631] usb 1-3.3: new high-speed USB device number 17 using xhci_hcd
[   33.701586] usb 1-3.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[   33.701601] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   33.701608] usb 1-3.3: Product: 802.11ac NIC
[   33.701613] usb 1-3.3: Manufacturer: Realtek
[   33.701618] usb 1-3.3: SerialNumber: 123456
[   35.700374] rtl88x2bu 1-3.3:1.0: Runtime PM usage count underflow!
[   35.711673] systemd-journald[400]: Time jumped backwards, rotating.
[   35.740036] usb 1-3.3: USB disconnect, device number 17
[   36.429309] usb 1-3.3: new high-speed USB device number 18 using xhci_hcd
[   36.509919] usb 1-3.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[   36.509932] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   36.509939] usb 1-3.3: Product: 802.11ac NIC
[   36.509945] usb 1-3.3: Manufacturer: Realtek
[   36.509949] usb 1-3.3: SerialNumber: 123456
[   38.509367] rtl88x2bu 1-3.3:1.0: Runtime PM usage count underflow!
[   38.556046] usb 1-3.3: USB disconnect, device number 18
[   39.243317] usb 1-3.3: new high-speed USB device number 19 using xhci_hcd
[   39.324050] usb 1-3.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[   39.324070] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   39.324077] usb 1-3.3: Product: 802.11ac NIC
[   39.324083] usb 1-3.3: Manufacturer: Realtek
[   39.324088] usb 1-3.3: SerialNumber: 123456
[   41.329753] rtl88x2bu 1-3.3:1.0: Runtime PM usage count underflow!
[   41.372043] usb 1-3.3: USB disconnect, device number 19
[   42.063622] usb 1-3.3: new high-speed USB device number 20 using xhci_hcd
[   42.148003] usb 1-3.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[   42.148020] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   42.148028] usb 1-3.3: Product: 802.11ac NIC
[   42.148033] usb 1-3.3: Manufacturer: Realtek
[   42.148038] usb 1-3.3: SerialNumber: 123456
[   44.153004] rtl88x2bu 1-3.3:1.0: Runtime PM usage count underflow!
[   44.188041] usb 1-3.3: USB disconnect, device number 20
[   44.875592] usb 1-3.3: new high-speed USB device number 21 using xhci_hcd
[   44.959079] usb 1-3.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[   44.959094] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   44.959101] usb 1-3.3: Product: 802.11ac NIC
[   44.959106] usb 1-3.3: Manufacturer: Realtek
[   44.959111] usb 1-3.3: SerialNumber: 123456
[   46.959127] rtl88x2bu 1-3.3:1.0: Runtime PM usage count underflow!
[   47.004059] usb 1-3.3: USB disconnect, device number 21
[   47.696633] usb 1-3.3: new high-speed USB device number 22 using xhci_hcd
[   47.780465] usb 1-3.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[   47.780482] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   47.780489] usb 1-3.3: Product: 802.11ac NIC
[   47.780495] usb 1-3.3: Manufacturer: Realtek
[   47.780500] usb 1-3.3: SerialNumber: 123456
[   49.777005] rtl88x2bu 1-3.3:1.0: Runtime PM usage count underflow!
[   49.820056] usb 1-3.3: USB disconnect, device number 22
[   50.511618] usb 1-3.3: new high-speed USB device number 23 using xhci_hcd
[   50.592552] usb 1-3.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[   50.592565] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   50.592572] usb 1-3.3: Product: 802.11ac NIC
[   50.592577] usb 1-3.3: Manufacturer: Realtek
[   50.592582] usb 1-3.3: SerialNumber: 123456
[   52.607005] rtl88x2bu 1-3.3:1.0: Runtime PM usage count underflow!
[   52.636046] usb 1-3.3: USB disconnect, device number 23
[   53.327625] usb 1-3.3: new high-speed USB device number 24 using xhci_hcd
[   53.408836] usb 1-3.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[   53.408851] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   53.408858] usb 1-3.3: Product: 802.11ac NIC
[   53.408863] usb 1-3.3: Manufacturer: Realtek
[   53.408868] usb 1-3.3: SerialNumber: 123456
[... cut ...]
[   56.143617] usb 1-3.3: new high-speed USB device number 25 using xhci_hcd
[   56.227956] usb 1-3.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[   56.227971] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   56.227978] usb 1-3.3: Product: 802.11ac NIC
[   56.227983] usb 1-3.3: Manufacturer: Realtek
[   56.227988] usb 1-3.3: SerialNumber: 123456
[   58.237756] rtl88x2bu 1-3.3:1.0: Runtime PM usage count underflow!
[   58.268046] usb 1-3.3: USB disconnect, device number 25

morrownr, 2024-06-21:
Additional testing with 7612u based adapter:

There is no module parameter to set. The driver is in-kernel. I placed the adapter in usb2 and usb3 ports and checked with `lsusb -t`. When the adapter was in a usb2 port, it was in usb 2 mode and when it a usb3 port, it was in usb3 mode. I plugged and replugged many times. Totally automatic. Just works.

The mt7612u driver is AC1200 class just like the 8812bu. It resides in the mt76 series of in-kernel drivers.

dubhater, 2024-06-21:
Thanks for testing. Can you try it again with this patch, please?

```diff
diff --git a/hal/halmac/halmac_88xx/halmac_usb_88xx.c b/hal/halmac/halmac_88xx/halmac_usb_88xx.c
index f87ad99..be46a1c 100644
--- a/hal/halmac/halmac_88xx/halmac_usb_88xx.c
+++ b/hal/halmac/halmac_88xx/halmac_usb_88xx.c
@@ -434,6 +434,7 @@ set_usb_mode_88xx(struct halmac_adapter *adapter, enum halmac_usb_mode mode)
 
 	/* check if HW supports usb2_usb3 switch */
 	usb_tmp = HALMAC_REG_R32(REG_PAD_CTRL2);
+	pr_err("usb_tmp 0x%08x\n", usb_tmp);
 	if (0 == (BIT_GET_USB23_SW_MODE_V1(usb_tmp) |
 	    (usb_tmp & BIT_USB3_USB2_TRANSITION))) {
 		PLTFM_MSG_ERR("[ERR]u2/u3 switch\n");
```

It will print "usb_tmp 0x________" every time it tries to switch. The first two messages with usb_tmp are important.

dubhater, 2024-06-21:
Just to be sure, these are the first two instances of usb_tmp? I'm wondering because device number 84 is pretty high.

morrownr, 2024-06-21:
They are the LAST two instances. I should have made that clear. Sorry. Would you like to see the first two instances?

dubhater, 2024-06-21:
Oh, yes, please. The first two are important.

morrownr, 2024-06-21:
First instance:

[    4.745718] usb_tmp 0x00100000
[    4.754434] rtl88x2bu 1-3.1:1.0: Runtime PM usage count underflow!
[    4.754550] usbcore: registered new interface driver rtl88x2bu

Second instance (all others instances show the same value as below for usb_tmp):

[    7.385960] usb_tmp 0x00080400
[    7.394688] rtl88x2bu 1-3.1:1.0: Runtime PM usage count underflow!

dubhater, 2024-06-21:
That looks promising. Just to have a complete picture, what do you get when you plug it into a USB 3 port?

morrownr, 2024-06-21:
[    3.357428] usbcore: registered new interface driver rtl88x2bu
[    5.051626] rtl88x2bu 4-2.2:1.0 wlx347de4402370: renamed from wlan0

And, of course, a wifi interface.

dubhater, 2024-06-21:
No usb_tmp? It looks like you rebooted. Did you forget rtw_switch_usb_mode=1 ?

morrownr, 2024-06-22:
> No usb_tmp? 

None. I started over with a cold boot and made sure usb_tmp is not there by running the following: (adapter is in a usb3 port and is in usb3 mode)

$ sudo dmesg | grep usb_tmp

>  It looks like you rebooted. Did you forget rtw_switch_usb_mode=1 ?

My installation script installs a file that makes the module settings persistent. I double checked.

This time I started cold. Booted. Checked `sudo dmesg | grep usb_tmp` and see nothing. I ran `lsusb -t` and the the adapter in usb3 mode. The adapter is in a usb3 port.

I'll be glad to run specific steps exactly as you need them to be run. Just tell me what you want.

I have extra 8812bu adapters so maybe we should get Scotty to beam me over to your place. You buy me a beer and I give you an adapter.

dubhater, 2024-06-22:
I checked the code again and I can only conclude that you loaded a version of the module which doesn't have the patch above. I don't see how else it could switch to USB 3 mode without printing the usb_tmp.

> I have extra 8812bu adapters so maybe we should get Scotty to beam me over to your place. You buy me a beer and I give you an adapter.

That would be nice...

morrownr, 2024-06-22:
> I checked the code again and I can only conclude that you loaded a version of the module which doesn't have the patch above.

Interesting. I had not checked the logic so you are saying that with the usb_tmp code added, we should see usb_tmp in the log at all times? Right? 

Let me start over and see what I can see.

dubhater, 2024-06-22:
You should see usb_tmp in the log every time it tries to switch from USB 2 to USB 3 mode.

morrownr, 2024-06-22:
Understand.

I think I may have figure this out. I am breaking in a new system. A little Mini PC. What I determined was that for the usb3 port that I was using would not reset to usb2 unless I did a shutdown and removed power for some period of seconds. Interesting.

New test results:

Totally cold boot. Checked to make sure port was using usb2. Rechecked and reinstalled the driver. Reboot.

$ sudo dmesg | grep usb_tmp

[    4.355805] usb_tmp 0x00100000

Now lsusb -t is showing the adapter in usb3 mode:

Port 002: Dev 003, If 0, Class=Vendor Specific Class, Driver=rtl88x2bu, 5000M

Since any previous results I have posted may be suspect, is there anything you want me to test again?

dubhater, 2024-06-22:
Thank you. I implemented USB 3 support in rtw88. You should see it in a few days.

> Since any previous results I have posted may be suspect, is there anything you want me to test again?

I think it's okay. The previous results make sense.

morrownr, 2024-06-22:
> Thank you. I implemented USB 3 support in rtw88. You should see it in a few days.

I am looking forward to testing it. The lack of detection and selection of USB3 mode has been a hole that needs to be filled. Your work is appreciated.

ntzb, 2024-06-22:
@dubhater can you link the changes you're pushing?
I didn't understand if they are more than setting `rtw_switch_usb_mode = 1`

morrownr, 2024-06-22:
@ntzb 

This discussion is about working on the rtw88 in-kernel drivers. This is a side issue to get usb3 going.

To see the current action:

https://github.com/morrownr/8821au-20210708/issues/133

That has to do with a new rtl8821/11au driver. More are planning. What other Realtek based adapters do you have?

The patches @dubhater is talking about will show up in the downstream rtw88 repo we are using for this project. You can find the link to the repo by reading the first message in the link I posted above.

closed 2024-06-24 by morrownr