morrownr/USB-WiFi #461: Fourth EDUP EP-AX1672 is actually an EDUP AX5400M which is a RTL8832cu

ddrown, 2024-06-26:
On the [plug and play page](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md), there's a list of four EDUP EP-AX1672 links, but the fourth one is not using the mediatek chipset, but uses the Realtek RTL8832cu chipset instead:

Amazon - 25 USD - EDUP EP-AX1672 - https://www.amazon.com/AX5400M-802-11AX-Tri-Band-Antennas-Wireless/dp/B0C77GBBY6

This should probably be removed from the plug and play list

Thanks for collecting all this useful wifi info!

bjlockie, 2024-06-26:
That link goes to a different device. :-(
Maybe they kept the link but changed the product.
This link goes to the 1672 which is a Mediatek device:
https://www.amazon.com/EDUP-Wireless-802-11AX-Tri-Band-Compatible/dp/B0CVVWNSH2

Maybe there should be a general warning to verify the model and chipset before purchasing anything.

morrownr, 2024-06-26:
@ddrown 

I have removed the problematic link. Please check and confirm that I took down the right one.

I really appreciate you and the others that report issues as it makes the list much better.

@morrownr

morrownr, 2024-06-26:
@bjlockie 

> Maybe there should be a general warning to verify the model and chipset before purchasing anything.

Suggest as to where it should go and how it should read?

ddrown, 2024-06-26:
> I have removed the problematic link. Please check and confirm that I took down the right one.

Looks good to me, the other three mention mtk7921au as the chipset on their product pages

morrownr, 2024-06-26:
> Looks good to me, the other three mention mtk7921au as the chipset on their product pages

I actually use far more than an ad listing the chipset or even saying Linux support. I don't make mistakes very often but it happens. What happens more often is ads being changed in between the times when I go over the list. Hopefully this list very helpful to Linux users.

morrownr, 2024-06-26:
For what it is worth, I have one of the Edup EP-AX1672 adapters and the more I use it, the more I like it. It is single state so no problems related to an onboard windows driver. I has good ventilation and the chips run cool anyway so you can abuse it without it getting warm. The range is good. AP mode, monitor mode and P2P are very good but that is because of the driver. The LED also works. It is a power on indicator. I could be wrong but it sure looks like Edup aimed this product at Linux users.

ddrown, 2024-06-26:
I have an old USB 802.11n adapter that I've been wanting to upgrade, so I put an order in for the EP-AX1672. I have some other Mediatek wifi hardware, and it's worked well.

I also have the EDUP AX5400, and both the Linux and Windows drivers aren't very good.

morrownr, 2024-06-26:
> I also have the EDUP AX5400, and both the Linux and Windows drivers aren't very good.

Oh, I am aware of that. See: #449 

I took another look at that link that you suggested that I delete. I think @bjlockie is correct. I know Linux does not have an adapter with good in-kernel drivers that is AX5400 so that would have chased me away. In fact, Mediatek is skipped from AXE300 to BE6500. In fact, I had ordered a little mt7925 based M.2 recently and it came today. Hopefully I have time tomorrow to install it and see for myself how that new WiFi 7 in-kernel driver is coming along. mt7925 is supposed to have a usb version available so hopefully we see some adapters at some point this year... not that I have a WiFi 7 AP but it will be good to see how the driver is coming along.

Linux USB Wireless has come a LONG way in the last few years. We are actually getting drivers in the KERNEL before product is even available. Now if Realtek would wake up and support USB.

ddrown, 2024-06-27:
My AX1672 arrived today and I can confirm it works well in Linux. I experimented with creating a 6GHz AP, but it looks like all the 6GHz channels are marked "No IR" for the US region. 5GHz AP worked fine, though.

morrownr, 2024-06-30:
> My AX1672 arrived today and I can confirm it works well in Linux.

Good to hear.

> 5GHz AP worked fine, though.

Yes, I have a WiFi 6, 5 GHz band sample hostapd.conf on the Main Menu. It works really well. I was working last fall to see if I could get 6 GHz band going and was not able to figure out exactly what the problem was and I have been busy since. Users in other places in the world such as the EU seem to have no problem. In fact, it seems to be a US regulatory problem. If you run onto the problem, let me know.

ddrown, 2024-06-30:
> Yes, I have a WiFi 6, 5 GHz band sample hostapd.conf on the Main Menu. It works really well. I was working last fall to see if I could get 6 GHz band going and was not able to figure out exactly [...]

The problem is the "No Initial Radiation" flag (No IR) on all the 6GHz channels.

```
$ iw phy phy0 channels
...
Band 4:
	* 5955 MHz [1] 
	  Maximum TX power: 12.0 dBm
	  No IR
	  Channel widths: 20MHz
...
```

I suspect they're set this way because the US regulations are around outdoor vs indoor transmitters. My understanding is that outdoor 6GHz is only permitted through Automated Frequency Coordination (AFC). I suspect the driver developers are being cautious because they can't detect if their device is outside or inside.

```
$ iw reg get
global
country US: DFS-FCC
...
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
...
```

morrownr, 2024-06-30:
> The problem is the "No Initial Radiation" flag (No IR) on all the 6GHz channels.

Yes, I am aware of that.

> I suspect they're set this way because the US regulations are around outdoor vs indoor transmitters.

That is what I was thinking when I moved on to other projects but I can see some issues with outdoor vs indoor regulation.

>  My understanding is that outdoor 6GHz is only permitted through Automated Frequency Coordination (AFC).

This is new to me. Can you elaborate?

>  I suspect the driver developers are being cautious because they can't detect if their device is outside or inside.

I guess the next step is to add some eyes and AI capability to our adapters... that would do the trick and then our adapters would cost a couple of thousand bucks. Something is not adding up to me. Our adapters are obviously capable of 6 GHz AP mode as I have heard too many people report as much and it does appear to be a regulatory thing in the US that is causing the problem... how do we resolve it is where I am stuck.