morrownr/USB-WiFi #462: [Help]: Fenvi AX1800 slow on OpenWRT, Fast on Ubuntu.

austonpramodh, 2024-06-26:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux OpenWrt 5.15.150 #0 SMP Fri Mar 22 22:09:42 2024 mips GNU/Linux

### lsusb

Bus 002 Device 003: ID 0e8d:7961 MediaTek Inc. Wireless_Device

### rfkill

rfkill: not found

### dkms

```text
dkms: not found
```

### iw

```text
phy#3
	Interface phy3-sta0
		ifindex 19
		wdev 0x300000003
		addr 90:de:80:47:92:22
		type managed
		channel 40 (5200 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
phy#0
	Interface wlan0
		ifindex 6
		wdev 0x1
		addr 94:83:c4:11:96:c9
		type managed
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

### What happened?

I am using Fenvi AX1800(MT7921AU). It seems like its running slow on OpenWRT router, but runs fast on my PC(Ubuntu-24.04).

I tested the USB port if that was the issue, but it's not. Tested with USB3 SSD, got around 103MBps

[image]

[image]
[image]

austonpramodh, 2024-06-27:
Update: 
Got access to another router with OpenWRT. It gets the same 200mbps speed.  I am assuming this could be a driver issue. 
[image]

But speeds are not going above 240mbps.

morrownr, 2024-06-28:
Hi @austonpramodh 

Sorry for the delay. Been busy.

A quick look and the first thing that comes to mind is that the USB port(s) are running in USB2 mode. Recommend you go to the OpenWRT command line and run:

lsusb -t

Things that I noticed:

- I see a bitrate of 1200 Mbps so that is what we are looking for.
- Your sig.noise reading is really low. How close to the AP are you? Being too close can cause problems.

wgetge, 2024-07-01:
You should check the MT1300 CPU usuage. I don't think MT7621 support HWNAT for USB Device. So most likely CPU bottlenecked.

austonpramodh, 2024-07-04:
Hi @morrownr,

Sorry for the delayed response. I did had disconnected this setup.

I have connected it again, Here:

[image]

It looks like its running at 5000M

austonpramodh, 2024-07-04:
@wgetge 

I tested it out, It could be the issue. Here

[image]

kissmorning, 2024-09-06:
I have the same problem, did you find a solution? I tried about 5 openwrt usb3.0 supported routers, the speed does not exceed 200mbps

morrownr, 2024-09-06:
@kissmorning 

Remember that OpenWRT has to be able to support routers with very limited resources so drivers need to be installed if they are needed.

In other words, you may have required drivers installed for USB2 but do you have the drivers installed for USB3?

kissmorning, 2024-09-06:
https://openwrt.org/toh/xiaomi/mir3g
I am using this device, I have 1gb internet, I get internet in client mode with the internal wifi to avoid loss, I distribute internet in master mode with mt7612's usb wifi, but I cannot exceed 200mbps speed, could this be due to processor bottleneck?

morrownr, 2024-09-06:
@kissmorning 

I guess you did not understand my question:

> ...you may have required drivers installed for USB2 but do you have the drivers installed for USB3?

That question relates to OpenWRT. How do you install the drivers in OpenWRT? ssh or Luci?

What I am trying to get at is whether you installed the USB3 drivers and have you checked to see if the adapter is in USB3 mode. The USB port on your router may be a USB3 capable port but if only USB2 drivers are installed, it will operate in USB2 mode.

kissmorning, 2024-09-06:
Thank you for your quick response. USB 3.0 drivers are enabled, I get high speeds when I connect an SSD to the USB output.@austonpramodh I have a similar problem with

austonpramodh, 2024-09-07:
@kissmorning I am not 100% sure if mine was more of a CPU bottleneck problem, Have you checked the CPU load when read/writing heavily?

Alex-Denysiuk, 2026-03-12:
I have the exact same adapter (ID 0e8d:7961) and have spent a lot of time debugging this exact issue on OpenWrt.

The harsh truth is: this is a hardcoded limitation set by the Chinese manufacturer (often sold as Fenvi AX1800 or similar unbranded dongles), and there is nothing we can do to fix it via software.

They have locked the TX power limit to a maximum of 3 dBm directly in the device's EEPROM/firmware. No matter what software tweaks you try—changing the regulatory domain (country code to US/UA/etc.), updating the mt76 driver, or forcing iw commands—the hardware simply will not transmit above 3 dBm.

At 3 dBm, the signal is incredibly weak, which leads to massive packet loss (often 50-100%) and constant disconnects/re-associations. The only practical workaround to get a somewhat stable connection out of this dongle is to force the channel width down to 20 MHz (HT20) and disable 80/40 MHz widths to concentrate the weak signal, but the range will still be terrible.

Ultimately, we can't bypass this 3 dBm limit.