morrownr/USB-WiFi #463: Issue with hostapd config AWUS036AXM

tiix95, 2024-06-27:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux xwing 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

### lsusb

Bus 002 Device 002: ID 0e8d:7961 MediaTek Inc. Wireless_Device

### rfkill

1: hci1: Bluetooth 	Soft blocked: no 	Hard blocked: no 2: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no 3: hci0: Bluetooth 	Soft blocked: no 	Hard blocked: no 4: phy1: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
N/A
```

### iw

```text
phy#1
	Interface wifi-alfa
		ifindex 5
		wdev 0x100000001
		addr 00:c0:ca:b5:61:17
		type managed
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
phy#0
	Interface wifi-board
		ifindex 4
		wdev 0x1
		addr dc:a6:32:1c:f1:fa
		type managed
		channel 34 (5170 MHz), width: 20 MHz, center1: 5170 MHz
[0] 15:44:12 root @ xwing /etc/hostapd #
```

### What happened?

Hello,

I can't manage to make hostapd start with your hostapd-Wifi6e.conf file (also tried wifi6 and 5) using brand new AWUS036AXM on a raspberry pi 4B. Here is the output :

`[1] 15:45:52 root @ xwing /etc/hostapd # hostapd  -d -P /run/hostapd.pid /etc/hostapd/6e.conf
random: getrandom() support available
Configuration file: /etc/hostapd/6e.conf
ctrl_interface_group=0
nl80211: TDLS supported
nl80211: TDLS external setup
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:10
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supported cipher 00-0f-ac:13
nl80211: Supported cipher 00-0f-ac:11
nl80211: Supported cipher 00-0f-ac:12
nl80211: Using driver-based off-channel TX
[... 208 lines cut ...]

ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=39
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]

ieee80211ax=1
he_oper_chwidth=1
he_oper_centr_freq_seg0_idx=39
he_bss_color=128
he_default_pe_duration=4
he_rts_threshold=1023
he_mu_edca_qos_info_param_count=0
he_mu_edca_qos_info_q_ack=0
he_mu_edca_qos_info_queue_request=0
he_mu_edca_qos_info_txop_request=0
he_mu_edca_ac_be_aifsn=8
he_mu_edca_ac_be_aci=0
he_mu_edca_ac_vo_ecwmin=9
he_mu_edca_ac_vo_ecwmax=10
`

Could you please help ? I can be available for more tests.

Thanks a lot, for this and for all your git, this is amazing job !

morrownr, 2024-06-28:
> I can't manage to make hostapd start with your hostapd-Wifi6e.conf file...

I recommend that you stick with `hostapd-Wifi6.conf` for now. Using it for WiFi 6 on the 5 GHz band is well tested at this point. Notice that `hostapd-Wifi6e.conf` shows `alpha` for status. I have not had time to investigate why I had problems with the 6 GHz band but I think it has to do with some capabilities of the wifi stack and maybe hostapd that needed to be put in place. Users have reported that 6 GHz band is working in some countries such as DE and FR. The rest of the EU is probably working also. It really seems to be related to the country.

If you have time, it would be nice to have a review on the AXM.

tiix95, 2024-06-29:
Hi,

I will try with Wifi6 configuration more deeply and update here, thanks. I did not see the alpha status for Wifi6e.

I'd be glad to help with a review on the AXM, could you please tell me which kind of tests will be helpful for you ? 

All I can say for now is that it is really plug-and play, tested on Linux debian 12 (kernel 6.1.20) and on a raspberry pi 4 (latest stable kernel on raspbian). Also, tell me where you want me to put the review if not in this issue :)

morrownr, 2024-07-05:
Any progress?

> I'd be glad to help with a review on the AXM, could you please tell me which kind of tests will be helpful for you ?

You could look in the Plug and Play List document as several of the adapters there have reviews. Pick out one that you think is well done and use it for a template. These adapters are used for many things so explain the modes you are using and if you like the results. There are managed (client), AP, monitor, and P2P modes. What do you think of the range? Does it appear to be well built? What distro are you using it with? Have you had any problems?

> Also, tell me where you want me to put the review if not in this issue :)

You can post the review here and I can move it to the Plug and Play List if that is okay with you.

morrownr, 2024-07-11:
@tiix95 

Just checking. Were you able to get the AP going with WiFi 6 on the 5 GHz band?

As Dorothy once said: "I don't think we are in Kansas anymore Toto." Back in the days of single-band WiFi 4, setting up an AP was simple but it can take a little work these days

cyp0633, 2024-07-31:
I think it might be associated with BSS coloring (from [this post](https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg64824.html)).

I commented out the `he_bss_color=128` line. The AP worked normally, at least for now.