morrownr/USB-WiFi #464: using Archer T2U Plus with RPi for creating hotspot

hbadshahp, 2024-07-03:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

### lsusb

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 001 Device 007: ID 04d9:0006 Holtek Semiconductor, Inc. Wired Keyboard (78/79 key) [RPI Wired Keyboard 5] Bus 001 Device 006: ID 05e3:0610 Genesys Logic, Inc. Hub Bus 001 Device 005: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse Bus 001 Device 004: ID 0781:5567 SanDisk Corp. Cruzer Blade Bus 001 Device 003: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU] Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### rfkill

0: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
rtl8821au/5.12.5.2, 6.6.31+rpt-rpi-v8, aarch64: installed
```

### iw

```text
phy#0
	Interface wlx242fd0d9e22c
		ifindex 4
		wdev 0x1
		addr 24:2f:d0:d9:e2:2c
		type managed
		txpower -100.00 dBm
```

### What happened?

I have followed all the steps as per suggestion to configure my Tp-link Archer T2U Plus AC600, so that I will use it for as a hotspot system. Firstly I successfully installed the driver using https://github.com/morrownr/8821au-20210708?tab=readme-ov-file. Then I followed the steps suggested in https://github.com/morrownr/USB WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md for  WiFi4 configuration. But after completing the whole steps I am not able to see hotspots on other device to connect.

So I run the below command 
sudo journalctl -xeu hostapd.service
░░ Support: https://www.debian.org/support
░░ 
░░ The unit hostapd.service has entered the 'failed' state with result 'exit-code'.
Jul 02 01:52:21 raspberrypi systemd[1]: Failed to start hostapd.service - Access point and authentication server for Wi-Fi and Ethernet.
░░ Subject: A start job for unit hostapd.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit hostapd.service has finished with a failure.
░░ 
░░ The job identifier is 5314 and the job result is failed.
Jul 02 01:52:23 raspberrypi systemd[1]: hostapd.service: Scheduled restart job, restart counter is at 56.
░░ Subject: Automatic restarting of a unit has been scheduled
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ Automatic restarting of the unit hostapd.service has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Jul 02 01:52:23 raspberrypi systemd[1]: Stopped hostapd.service - Access point and authentication server for Wi-Fi and Ethernet.
░░ Subject: A stop job for unit hostapd.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A stop job for unit hostapd.service has finished.
░░ 
░░ The job identifier is 5395 and the job result is done.
Jul 02 01:52:23 raspberrypi systemd[1]: Starting hostapd.service - Access point and authentication server for Wi-Fi and Ethernet...
░░ Subject: A start job for unit hostapd.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit hostapd.service has begun execution.
░░ 
░░ The job identifier is 5395.
Jul 02 01:52:29 raspberrypi systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An ExecStart= process belonging to unit hostapd.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
Jul 02 01:52:29 raspberrypi systemd[1]: hostapd.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit hostapd.service has entered the 'failed' state with result 'exit-code'.
Jul 02 01:52:29 raspberrypi systemd[1]: Failed to start hostapd.service - Access point and authentication server for Wi-Fi and Ethernet.
░░ Subject: A start job for unit hostapd.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit hostapd.service has finished with a failure.
░░ 
░░ The job identifier is 5395 and the job result is failed.

Kindly suggest me where i am wrong. Thank you in advance.

morrownr, 2024-07-03:
Setting up a hotspot with an adapter can be very useful and educational but for maximum performance, it can be a little complicated at first. The AP_Mode/Bridged_Wireless_Access_Point.md guide has a lot of details but there are other things involved. This may not be a quick fix but rather an investigation that may take a few days.

A good place to start:

Run in the driver directory:

$ sudo edit-options.sh

Post the results of the `options` line so that we can see how the driver is configured.

Next, do you see the hostapd.log file? It should show up in your home directory.

morrownr, 2024-07-11:
@hbadshahp 

Checking on you. How is this problem progressing?

The rtl8811au chip is really good and it makes a good AP. It is normal for someone starting out with hostapd to make a little mistake here and there.