morrownr/USB-WiFi #465: News: New USB WiFi 6e Adapter from Panda plus some other popular Panda products

morrownr, 2024-07-11:
Panda Wireless® PAU0F AXE3000 Tri Band (2.4/5 / 6 GHz) WiFi 6E USB 3.0 Adapter

https://www.amazon.com/Panda-Wireless%C2%AE-PAU0F-AXE3000-Adapter/dp/B0D972VY9B

While the ad does not mention the chipset, it has to be the mt7921au which means as long as Linux users are using kernel 5.18 or later, the adapter should be plug and play. Panda has a long history is providing usb wifi and bluetooth adapters that are plug and play with Linux. In fact, I do not remember Panda ever releasing a wifi or bluetooth product that has not been plug and play with Linux. I will likely add this adapter to the Plug and Play List soon as Panda is one of only two usb adapter makers that I trust. If anyone purchases this adapter and would like to provide a review, it would be welcome.

-----

While I am talking about Panda, let highlight some products that are known good quality and are popular:

Panda Wireless® PAU0B AC600 Dual Band (2.4GHz and 5GHz) Wireless N USB Adapter W/High Gain Antenna

https://www.amazon.com/Panda-Wireless-PAU0B-Adapter-Antenna/dp/B08NPX2X4Z

The above Panda dual band AC600 adapter sells well to Linux users and the reviews from Linux users are consistently good. Users report good range if you are looking for an adapter that has good range. It uses the mt7610u in-kernel driver which is a full feature driver that is very stable and even supports features like Active Monitor Mode. It should work well for Kali users as well as most regular Linux users that are looking for a solid USB WiFi adapter.

-----

Panda Ultra WiFi (b/g/n) 150Mbps Wireless-N 2.4GHz USB Adapter (nano)

https://www.amazon.com/Panda-Ultra-150Mbps-Wireless-Adapter/dp/B00762YNMG

I have one of these adapters. I bought it in 2014. Yes, 10 years ago. It still sells well and it works well. It is of course, plug and play. I haven't been able to break it. 

-----

Panda Bluetooth 4.0 USB Nano Adapter

https://www.amazon.com/Panda-Bluetooth-4-0-Nano-Adapter/dp/B00BCU4TZE

As many of you know if you stop by this site, I recommend Linux users to buy USB WiFi adapters that are wifi only. I've seen too many problems over the years with usb wifi adapters that have wifi and bluetooth. The above adapter is a standalone usb bluetooth adapter that is plug and play with Linux. It sells well to Linux users and has good reviews.

-----

Let me know if you have any questions.

@morrownr

lukejenkins, 2024-07-26:
I ordered a PAU0F and can confirm that it is mt7921au based.

morrownr, 2024-07-26:
@lukejenkins 

Thanks for the report. When you have time, a report would be helpful to others. I think you know by now that maintaining the Plug and Play List is a challenge... but it is rewarding when it helps Linux users.

lukejenkins, 2024-08-05:
Here you go: https://github.com/morrownr/USB-WiFi/pull/479

I know that this project is very hard work and you don't get thanked enough for it, so thank you!

Barbs56, 2025-05-19:
@morrownr 

I'm building a wifi beacon decoding prototype for a government project. I have specialized upstream amplification and antennas (18 db net) and need a simple USB interface to handle the sniffing. The interface must be small (like a thumbdrive). I had success using an Alfa device with a Realtek chip and rtw88 firmware, but found that the increased gain could not be offset via firmware nor could I get around an apparent minimum signal strength packet reporting limit (or some related setting in the device firmware that prevented any signals below -70 dB to be decoded. 

I am trying the PAU0B you mentioned, above. Do you know if any open source driver/firmware allows me to set the input gain and reduce/remove any minimum RSSI limits? (or, not have the limits to begin with)

morrownr, 2025-05-19:
Hi @Barbs56 

Let me  pass some contact information to you as they may be able to help you better than me:

https://wireless.docs.kernel.org/en/latest/en/users/drivers/mediatek.html

That is the Mediatek official page on kernel.org. Scroll down and you will see email address. I would recommend an email to the addresses that end with mediatek.com as they are the Mediatek employees. Hints: These are very busy people. Explain your problem as directly as possible and talk their language. Do not say Panda XXX adapter as they may have no idea about an end product but rather say the adapter uses the mt7921au chipset. That they understand.

Also, if he has time, @dubhater , might be of assistance as far as rtw88 is concerned. He is not a Realtek employee but is very knowledgeable. The offer of a consultation fee might help. The Realtek USB driver devs do not work via the public linux-wireless Linux support channels like they should so contacting them is basically impossible unless you buy chips from them in 10k lots.

Good luck.

morrownr, 2025-12-09:
It may be that there is an issue with power save. Try this temp solution to see if turning off power save helps:

Open a Terminal interface: Ctrl + Alt + T

Run:

iw dev

...to see what your wireless interface name is. Then use the wireless interface name below where you see wlan0.

Test to see if power_save is causing the issue:

Run the following command (remember to change wlan0 to the name of your wireless interface):

iw wlan0 set power_save off

Let the system go into power saving and then bring it back and see what happens.

morrownr, 2025-12-09:
> Interface wlx9cefd5f630ed

iw wlx9cefd5f630ed set power_save off

or

sudo iw wlx9cefd5f630ed set power_save off

depending on your distro and where they locate iw.

Once that is run, just let the system go to sleep and then later activate it and see how it goes.

kedonley, 2025-12-09:
No Change.  No wireless connection when I returned.  tried to select a different access point, it would not connect.  Tried two different USB plugs, neither worked.  Plugged it into a USB 2.0 plug instead of a 3.0 and will see if there is a difference.

morrownr, 2026-04-05:
Hi @renzonuez23-commits 

I have both adapters. I should have added a good review of the Panda 0B by now but I haven't due to higher priorities so I'll give you short overview:

The Alfa ACHM and the Panda 0B use the same mt7610u chip which is an AC600 class dual-band chip. Both have outstanding range. The Alfa does have longer range but it is very close. I have seen a few ACHM users over the years report a strange sound coming from the ACHM but that was never resolved and I think Alfa would just replace the adapters if this was a problem. I've never had any problems myself with either adapter and I have seen zero bad reports on the Panda adapter.

If you decide to get the Panda adapter, you may want to get a USB extension cable so that the adapter can be placed in an optimal location.

I need to direct you to an issue that I opened in December becuase I was seeing some problems, basically all over the place in monitor mode and we need to get to the bottom of it:

https://github.com/morrownr/USB-WiFi/issues/682

Right now, with modern 6.18 or later kernels, the only driver/adapter chip that is fully performing is the mt7925u which is a WiFi 7 tri-band adapter. The only adapters currently available with this cchip are very expensive as you can see in The Plug and Play List. Are we working with the kernel devs to fix the problems. Yes, you can see some of the efforts in the above link.

If you want, I could made a recommendation for a guy on a budget that would likely work out in the longer term while you learn.

renzonuez23-commits, 2026-04-05:
@morrownr , Thanks for the quick reply! I thoroughly read most of the details about the adapters I'm interested in, and I found this ACHM one that I liked. But on your Plug and Play list, I found this Panda PAU0B, and I noticed it has the same chipset and, according to your opinion, also works perfectly. That's why I asked if it was necessary to spend more money or not, since both are excellent but the difference lies in the brand of each adapter. I'm also open to any further information you might have. Cheers!