morrownr/USB-WiFi #467: Don't know what adapter to choose

Sofialover6, 2024-07-17:
I use Kali nethunter on my laptop it runs windows 10 and 11 I need a adapter that's 2.4 and 5 and it needs to have exceptional range and great speeds in you're opinion what do you think is the best wifi adapter that I can get that's affordable has 2.4 and 5 it's long range had has fast speeds and I rather it be plug and use and it be smaller but tell me what you think is the best in general so I won't have to buy another .

morrownr, 2024-07-17:
Hi @Sofialover6 

The usb wifi adapter that is closest to your wishlist is the ALFA AWUS036ACHM. It is shown in the Plug and Play List. You have to scroll down to the mt7610u section. It can be had for around $40.

- I rather it be plug and play - The ACHM is plug and play on any desktop/laptop distro of Linux that is using kernel 4.18 or later which is basically everything at this point. It is also plug and play on Windows 10/11. I very rarely boot the Windows partitian on one of my test systems but I can double check if you want.

- it needs to have exceptional range - I have done testing. I have not found an adapter that has greater range than the ACHM. It is the champ of WiFi 5 adapter when it comes to range.

- and great speeds - it really depends on what you mean by great speeds. This is an AC600 capable adapter so not the latest greatest speeds but an adapter that meets all of your wishlist is not made.

- and it be smaller - it depends on what you mean by smaller. The ACHM is not a big adapter but it is not a nano adapter either. It does come with an extension cable and a clip which makes it rather flexible.

If you are doing monitor mode work, my opinion is that currently there is no better adapter than the ACHM.

Let me ask @ZerBea to consult on the issue if he has time.

Sofialover6, 2024-07-17:
> Hi @Sofialover6
>
> The usb wifi adapter that is closest to your wishlist is the ALFA AWUS036ACHM. It is shown in the Plug and Play List. You have to scroll down to the mt7610u section. It can be had [...]

What about ALFA AWUS036ACU AC1200

morrownr, 2024-07-17:
The Alfa ACU has a Realtek chip. Avoid Realtek for what you are doing. Stick with mt7610u. mt7612u and mt7921au chips. 

There are not that many options when it comes to long range in a WiFi 5 adapter so let me know if you are willing compromise.

Sofialover6, 2024-07-17:
> The Alfa ACU has a Realtek chip. Avoid Realtek for what you are doing. Stick with mt7610u. mt7612u and mt7921au chips.
> 
> There are not that many options when it comes to long range in a WiFi 5 adapter so let me know if you are willing compromise.

I don't think I want to compromise any I mean it can be a smaller wifi adapter or may less range if you think that's the best for my needs then I'll get that one but could you give me a link to where I should get it for the best price

Sofialover6, 2024-07-17:
Also I'm just getting into is so do you have any recommendations for websites or apps or anything to help me learn Kali nethunter/coding/hacking in general cause I'm new

ZerBea, 2024-07-17:
@morrownr 
I can't give a recommendation because I neither use Kali (Nethunter) nor Windows.

morrownr, 2024-07-17:
Advice: Take small steps to learn what you are trying to learn. I am not the one to give you good links but some experts do stop by here. Give it time and you should get more info.

My expertise is more with AP mode but I have learned monitor mode enough to help users.

Here are some alternate adapter selections for you to look at:

EDUP EP-AX1672
ALFA AWUS036ACM
PANDA - PAU0B

> could you give me a link to where I should get it for the best price...

The following in the link to the Main Menu of this site, Menu item 2 has information and links to buy numerous known good adapters. The adapters in the Plug and Play List are known good adapters. Some are better at some things than others so it is good to ask like you are doing. You probably should read menu items 1, 2 and 3:

https://github.com/morrownr/USB-WiFi

If the Plug and Play List does not have a link to a retailer you want to use then let us know what country you are in and what adapters ypu want to check out and we can try to help.

Sofialover6, 2024-07-17:
> Advice: Take small steps to learn what you are trying to learn. I am not the one to give you good links but some experts do stop by here. Give it time and you should get more info.
> 
> My expertise is more with AP mode but I have learned monitor mode enough to help users.
> 
> Here are some alternate adapter selections for you to look at:
> 
> EDUP EP-AX1672 ALFA AWUS036ACM PANDA - PAU0B
> 
> > could you give me a link to where I should get it for the best price...
> 
> The following is the link to the Main Menu of this site, Menu item 2 has information and links to buy numerous known good adapters. The adapters in the Plug and Play List are known good adapters. Some are better at some things than others so it is good to ask like you are doing. You probably should read menu items 1, 2 and 3:
> 
> https://github.com/morrownr/USB-WiFi
> 
> If the Plug and Play List does not have a link to a retailer you want to use then let us know what country you are in and what adapters ypu want to check out and we can try to help.

Well thanks I appreciate the help I'n youre opinion just give me you're Best recommendation for the adapter that I need that meets my specifications that's affordable and then I'll close this issue

morrownr, 2024-07-17:
@Sofialover6 

@ZerBea has some good points. I need to be out of the office for a few hours but will do some research when I get back.

Sofialover6, 2024-07-17:
> @Sofialover6
> 
> @ZerBea has some good points. I need to be out of the office for a few hours but will do some research when I get back.

Ok well just give me you're best recommendation I want to buy one that's works great with my laptop and if I got Kali nethunter on a phone it would work great to and it needs to meet the prerequisites/requirements I stated before 2.4/5 long range fast speed good quality in general etc

ZerBea, 2024-07-17:
The ALFA AWUS036ACHM is working as expected.
Long range only in combination with a good antenna (panel or dish),
To cover the entire frequency range (tx and rx) it is mandatory to set wireless regulatory settings.

Sofialover6, 2024-07-17:
By antenna you mean the one that comes attached to it right and I don't know what tx and rx is but do you think that adapter would be good for me

ZerBea, 2024-07-17:
Please take a look at this antennas:
dish 2.4 GHz
https://github.com/ZerBea/hcxdumptool/wiki/Penetration-testing-system-5
panel 2.4 / 5GHz:
https://github.com/ZerBea/hcxdumptool/wiki/Penetration-testing-system-3
panel 2.4 GHz:
https://github.com/ZerBea/hcxdumptool/wiki/Penetration-testing-system-4

TX = transmit
RX = receive

Some information about the device:
```
$ lsusb
ID 0e8d:7610 MediaTek Inc. WiFi

$ sudo hcxdumptool -I wlp48s0f4u2u3

Requesting physical interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0   3 00c0cab37c17 00c0cab37c17 * wlp48s0f4u2u3    mt76x0u (NETLINK)

available frequencies: frequency [channel] tx-power of Regulatory Domain: IN

  2412 [  1] 3.0 dBm	  2417 [  2] 3.0 dBm	  2422 [  3] 3.0 dBm	  2427 [  4] 3.0 dBm
  2432 [  5] 3.0 dBm	  2437 [  6] 3.0 dBm	  2442 [  7] 3.0 dBm	  2447 [  8] 3.0 dBm
  2452 [  9] 3.0 dBm	  2457 [ 10] 3.0 dBm	  2462 [ 11] 3.0 dBm	  2467 [ 12] 3.0 dBm
  2472 [ 13] 3.0 dBm	  2484 [ 14] disabled	  5180 [ 36] 16.0 dBm	  5200 [ 40] 16.0 dBm
  5220 [ 44] 16.0 dBm	  5240 [ 48] 16.0 dBm	  5260 [ 52] 16.0 dBm	  5280 [ 56] 15.0 dBm
  5300 [ 60] 15.0 dBm	  5320 [ 64] 15.0 dBm	  5500 [100] 13.0 dBm	  5520 [104] 12.0 dBm
  5540 [108] 12.0 dBm	  5560 [112] 12.0 dBm	  5580 [116] 12.0 dBm	  5600 [120] 12.0 dBm
  5620 [124] 12.0 dBm	  5640 [128] 12.0 dBm	  5660 [132] 12.0 dBm	  5680 [136] 12.0 dBm
  5700 [140] 13.0 dBm	  5720 [144] 13.0 dBm	  5745 [149] 13.0 dBm	  5765 [153] 13.0 dBm
  5785 [157] 13.0 dBm	  5805 [161] 13.0 dBm	  5825 [165] 14.0 dBm	  5845 [169] 6.0 dBm
  5865 [173] 6.0 dBm	  5885 [177] disabled

$ time hcxdumptool -c 10a --exitoneapol=7 -w test.pcapng --bpf=target.bpfc

95 packet(s) captured
1 SHB written to pcapng dumpfile
1 IDB written to pcapng dumpfile
1 ECB written to pcapng dumpfile
18 EPB written to pcapng dumpfile

exit on EAPOL M1M2
bye-bye

real	0m7,512s
user	0m0,000s
sys	0m0,015s
```

Sofialover6, 2024-07-17:
Well do I need those I mean how far is the range without those

ZerBea, 2024-07-17:
Please note that transmit power and frequency range depend on the wireless regulatory settings
```
E.g. US:
available frequencies: frequency [channel] tx-power of Regulatory Domain: US

  2412 [  1] 3.0 dBm	  2417 [  2] 3.0 dBm	  2422 [  3] 3.0 dBm	  2427 [  4] 3.0 dBm
  2432 [  5] 3.0 dBm	  2437 [  6] 3.0 dBm	  2442 [  7] 3.0 dBm	  2447 [  8] 3.0 dBm
  2452 [  9] 3.0 dBm	  2457 [ 10] 3.0 dBm	  2462 [ 11] 3.0 dBm	  2467 [ 12] disabled
  2472 [ 13] disabled	  2484 [ 14] disabled	  5180 [ 36] 16.0 dBm	  5200 [ 40] 16.0 dBm
  5220 [ 44] 16.0 dBm	  5240 [ 48] 16.0 dBm	  5260 [ 52] 16.0 dBm	  5280 [ 56] 15.0 dBm
  5300 [ 60] 15.0 dBm	  5320 [ 64] 15.0 dBm	  5500 [100] 13.0 dBm	  5520 [104] 12.0 dBm
  5540 [108] 12.0 dBm	  5560 [112] 12.0 dBm	  5580 [116] 12.0 dBm	  5600 [120] 12.0 dBm
  5620 [124] 12.0 dBm	  5640 [128] 12.0 dBm	  5660 [132] 12.0 dBm	  5680 [136] 12.0 dBm
  5700 [140] 13.0 dBm	  5720 [144] 13.0 dBm	  5745 [149] 13.0 dBm	  5765 [153] 13.0 dBm
  5785 [157] 13.0 dBm	  5805 [161] 13.0 dBm	  5825 [165] 14.0 dBm	  5845 [169] 6.0 dBm
  5865 [173] 6.0 dBm	  5885 [177] 16.0 dBm
```

e.g. AU
```
available frequencies: frequency [channel] tx-power of Regulatory Domain: AU

  2412 [  1] 3.0 dBm	  2417 [  2] 3.0 dBm	  2422 [  3] 3.0 dBm	  2427 [  4] 3.0 dBm
  2432 [  5] 3.0 dBm	  2437 [  6] 3.0 dBm	  2442 [  7] 3.0 dBm	  2447 [  8] 3.0 dBm
  2452 [  9] 3.0 dBm	  2457 [ 10] 3.0 dBm	  2462 [ 11] 3.0 dBm	  2467 [ 12] 3.0 dBm
  2472 [ 13] 3.0 dBm	  2484 [ 14] disabled	  5180 [ 36] 16.0 dBm	  5200 [ 40] 16.0 dBm
  5220 [ 44] 16.0 dBm	  5240 [ 48] 16.0 dBm	  5260 [ 52] 16.0 dBm	  5280 [ 56] 15.0 dBm
  5300 [ 60] 15.0 dBm	  5320 [ 64] 15.0 dBm	  5500 [100] 13.0 dBm	  5520 [104] 12.0 dBm
  5540 [108] 12.0 dBm	  5560 [112] 12.0 dBm	  5580 [116] 12.0 dBm	  5600 [120] disabled
  5620 [124] disabled	  5640 [128] disabled	  5660 [132] 12.0 dBm	  5680 [136] 12.0 dBm
  5700 [140] 13.0 dBm	  5720 [144] 13.0 dBm	  5745 [149] 13.0 dBm	  5765 [153] 13.0 dBm
  5785 [157] 13.0 dBm	  5805 [161] 13.0 dBm	  5825 [165] 14.0 dBm	  5845 [169] 6.0 dBm
  5865 [173] 6.0 dBm	  5885 [177] disabled
```

Don't trust the transmit power values. Due to a driver bug, they are not correct shown.

ZerBea, 2024-07-17:
Some explanations about the range:
https://www.centurylink.com/home/help/internet/wireless/which-frequency-should-you-use.html

Sofialover6, 2024-07-17:
How long is the range with out any of that

ZerBea, 2024-07-17:
Nobody can tell you that, because it depend on several screws:
obstacles (rain, walls, trees, indoor outdoor)
bandwidth (20MHz, 40 MHz)
frequency (2.4 GHz, 5GHz)
antenna

https://www.pinterest.de/pin/an-illustrative-example-of-wifi-range-in-a-typical-home--372321094185337115/

Sofialover6, 2024-07-17:
Okay but do you think the The ALFA AWUS036ACHM is good for me just to plug into my laptop or phone running Kali nethunter

ZerBea, 2024-07-17:
For me (Arch Linux and OpenWRT) it is ok.
If your Kali Linux is well configured it will be ok, too.
Don't expect a range more than 150m (2.4GHz) and 50m (5Ghz) with the stock antenna.
Expect several problems if your Kali Linux is unconfigured (e.g running services that interfere with the task).

https://github.com/ZerBea/hcxdumptool/discussions/418
https://github.com/ZerBea/hcxdumptool/discussions/419
https://github.com/ZerBea/hcxdumptool/discussions/406

Sofialover6, 2024-07-17:
Well what's the cheapest smallest antenna that will increase range alought

morrownr, 2024-07-17:
I am back and it seems I am behind in this conversation...

> Long range only in combination with a good antenna (panel or dish),

@ZerBea  Hmmm. Maybe we can argue a little and it will be good information for @Sofialover6 .

I consider the ACHM to be long range with the stock antenna. I have test results to backup what I am saying:

https://github.com/morrownr/USB-WiFi/blob/main/home/Performance_Comparison.md

That is an older test but gives you an idea. I need to do a modern test with more adapters but for now, this test shows what I am talking about. The ACHM and ACH walk away from the pack in this test. Both have outstanding range with stock antennas but I will concede that a good flat or round antenna could improve range further but then it would be directional and I do not think that it what the OP is looking for, at least until he learns more.

Sofialover6, 2024-07-17:
Yeah I probably do need something more beginner friendly but to the point where I won't have to buy something in awhile so I'll let y'all decide on what I should get

morrownr, 2024-07-17:
@Sofialover6 

> do you think the The ALFA AWUS036ACHM is good for me just to plug into my laptop or phone running Kali nethunter...

Hold on. Let me read some more about Kali Nethunter. What I have learned so far is that it is designed to be installed on phones and it based on Android. Android is Linux but I must read more to see what support it contains for usb wifi adapters.

I shall return.

morrownr, 2024-07-17:
@Sofialover6 

I have read the following about what adapters that are supported by Nethunter:

https://www.kali.org/docs/nethunter/wireless-cards/

The mt7610u is listed as supported so the Alfa ACHM should work well.

Now, I understand that you are a beginner but I need you to understand that Nethunter is not a beginner project in my opinion. It also requires you to have a phone that you no longer need to use as a phone and Nethunter has a ROM for it. I have a couple of old phones and this might be a cool project so I think I will take a look at this.

As a beginner, you might want to start with something that is easier to setup... like the regular X86_64/AMD64 version of Kali. It would be best for it to go on a laptop/desktop/Mini PC as the only OS. Kali can run on very old hardware... as long as the laptop is 64 bit, it would be good so find an old system and let's get started.

I think the Alfa ACHM, Alfa ACM or the Panda adapter I mentioned would work well with the ACHM at the top of my list.

FYI: Please don't close this issue as people like to read them.

Sofialover6, 2024-07-17:
Oh so the The ALFA AWUS036ACHM and yes I won't close the issue

ZerBea, 2024-07-21:
Do not update to kernel 6.9.10:
https://github.com/ZerBea/hcxdumptool/discussions/465