morrownr/USB-WiFi #469: Warning (2nd) : Use caution if you are considering buying a USB WiFi adapter or module with a Realtek WiFi 6 chip!!!

morrownr, 2024-07-20:
Hello All,

Some Linux users are making the mistake of buying USB WiFi adapters based on the Realtek WiFi 6 chipsets. I have been seeing a steady increase in problem report messages from users of Realtek based WiFi 6 USB WiFi 6 wifi adapters. It is to the point that I think I need to add more warnings around this site for users to see. As you will see below, my strong recommendation has been and continues to be for Linux USB wifi adapter and module users to AVOID products that include Realtek WiFi 6 USB chips. This includes the following chips:

rtl8852/32au
rtl8852/32bu
rtl8852/32cu

Why?

Realtek still does not provide USB support for WiFi 6 that is in accordance with established Linux Wireless Standards. All they are providing is out-of-kernel drivers that are based on some of their own ideas which means these drivers will never be allowed into the Linux kernel. Even if we don't care about standards, the Linux WiFi 6 out-of-kernel drivers are BAD. I worked on the Realtek out-of-kernel for the rtl8852/32au chip after it was released. I wanted to get it at least to the point that managed mode would work reasonably well for users but I was never able to make that happen. The driver is JUNK. I have a driver up here for the rtl8852/32bu chip but I have warnings posted. The only mode that I can say that works reasonably well is managed mode. The driver as a whole is JUNK.

Realtek USB support on Linux is not good but is particular BAD for WiFi 6. If you want to buy a USB adapter or module that is supported with Linux Wireless Standards compliant drivers, see the Plug and Play List on the Main Menu. On the Main Menu is also USB WiFi chipset information for Linux which gives good information about supported chipsets. The information to help you make good decision is here... use it!

One last additional issue to consider is that all adapters that I am aware of the use Realtek WiFi 6 chips are multi-state. This means that they have a windows driver onboard. This is another potential problem area that Linux users should consider (and avoid).

To repeat, I strongly recommend that Linux users avoid USB WiFi adapters and modules based on Realtek WiFi 6 chips. It could be years before well supported, standards compliant drivers are available...if it ever happens Your happiness will likely be far better if you use the Plug and Play List and pick out one of the well supported adapters. If you have questions, please ask.

Regards,

@morrownr

bjlockie, 2024-07-20:
I suspect alot of users buy Realtek adapters first and then come here when they have problems.
Realtek adapters seem to more available/cheaper.

morrownr, 2024-07-20:
@bjlockie 

> I suspect a lot of users buy Realtek adapters first and then come here when they have problems.

I can't argue with your point. It would be wise to come here and be informed first.

> Realtek adapters seem to more available/cheaper.

I certainly agree that Realtek based adapters are more widely available but I will add that at times when I help individual users look on sites that are acceptable to them, they find out that Mediatek based adapters really are available to them. It is common for users to not understand what to look for which is one of the reasons I start this site.

Cheaper? I'm not so sure about that. Two of the six adapters that are WiFi 6e in The Plug and Play List can be found for less than 5 usd. The mt7921au section is The Plug and Play List is maturing nicely. As time passes, I continue to add new adapters and cull adapters that are proving to not be as trouble free as we thought for one reason or another. The mt7612u and mt7610u sections have plenty of good adapters these days as well. Linux users have MANY very solid modern choices these days at various price points. There is no reason for Linux users to be buying usb wifi adapters that are going to cause them a lot of frustration.

micahkane, 2024-07-31:
you knowthe most frustrating part is the wifi adapters i purchased where recommended to me by the people who i am currently taking some learning courses with. But i am ordering one from the PnP list. And will be next week until i receive it. I did manage to get them into monitor mode following the updates  here from morrownr, but it doesnt scan . i get no data coming back to me. Is anyone else having that problem after installing the updates?. it be nice to get this one running until i get my other.

morrownr, 2024-07-31:
@micahkane 

> the most frustrating part is the wifi adapters i purchased where recommended to me by the people who i am currently taking some learning courses with.

I understand this. I've been using Linux since '94 and usb wifi adapters since they were available. As Linux distros have become more and more user friendly for general users, I have seen many Linux users struggle getting usb wifi adapters that are easy to use and well supported. That is why I started this site a few years ago. I do not make any money from this site but I thought there was a need to help. Things have been getting better from a driver perspective over the last 6 years but the information was not out there.

I think we are going to see a very big jump forward with usb wifi adapter support in the next few months and into next year as new, high quality drivers are added to the kernel and existing drivers are improved greatly. The one area that is of concern to me, which is the reason for this and other messages about Realtek WiFi 6 chip based adapters, is that I do not see the big improvements helping that specific adapters and Linux users need to know to avoid them.

If you do not mind sharing what your problems have been with the existing adapter, maybe we can help you get some use from the adapter. Also, now I am curious what you ordered if you don't mind sharing that information. Keep in mind, there are some serious experts and driver developers that stop by this site at times so a lot of good information can be had over time.

Cheers

morrownr, 2024-08-14:
@micahkane 

Did you get the new adapter?