morrownr/USB-WiFi #471: News: New firmware files available for the mt7921 chips (mt7921au, mt7921, mt7921k).

morrownr, 2024-07-23:
Updated firmware files are available for adapters and cards with the subject chips. A guide to updating or installing the new firmware is available here at:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

If you are experiencing any problems, it might be a good idea to update the firmware files.

Advice: We are in a period of very heavy development of the Linux WiFi stack. This is needed to support many new modern capabilities such as WiFi 7. This work comes with some breakage but things will stabilize. If you experience wifi problems and you are using a bleeding edge kernel such as 6.9 or 6.10, my recommendation is that you back off to a kernel that is working. Be advised that this modernization work may break older drivers such as mt7612u and mt7610u

Note: Recently M.2 cards with the new mt7925 chip became available for sale so I bought one to test. I wanted to see how the mt7925 in-kernel driver is doing. I am currently only using it in managed mode but wifi is working well with WiFi 5 and WiFi 6 APs. I can't test WiFi 7 yet because I do not have a WiFi 7 capable AP/WiFi router yet, Bluetooth was not working as bluetooth is supported on cards. I quickly chased this down to the lack of the needed vip/pid in the kernel so I was going to submit a patch but someone beat me to it.

Badger101, 2024-07-26:
Thank you for the update - do you know if there is a changelog?

morrownr, 2024-07-26:
> do you know if there is a changelog?

I can't remember ever seeing a changelog with firmware. The firmware contains code that companies keep private in order to keep information out of competitors hands so they are ultra protective with firmware.