morrownr/USB-WiFi #473: Panda PAU0B adapter help

Anakinjh, 2024-07-29:
I have been trying to get my [Panda wireless PAU0B](https://www.amazon.com/gp/aw/d/B08NPX2X4Z?psc=1&ref=ppx_pop_mob_b_asin_title) adapter to work on OpenWrt. I have been trying to make a small router/ extender for my home network and be able to travel with it. 

Under section 7, I followed sub-section 4, the one with specific OpenWrt instructions. I was able to use `scp -O [NAME_OF_FILE]` to copy the mentioned files into my router in /lib/firmware/mediatek and I ran an ls to make sure they were there. I rebooted but I still am having issues I believe. 

Running the command `ifconfig` doesn't show any wlan interfaces at first. The guide I was following said that `ifconfig wlan1 up` should bring up the interface for the adapter, and then wlan0 should already be up. Mine is not, and wlan1 gives me a "device not found error"

Did I miss something installing the firmware? Did I somehow miss the driver install?

I've been following this [guide by NetworkChuck](https://youtu.be/jlHWnKVpygw?si=BjUeLT0r0P59goVX) and I feel confident that my configurations are right, but I must be missing something

morrownr, 2024-07-29:
Hi @Anakinjh 

OpenWRT cannot afford to waste space installing drivers that may not be used. You will need to install the driver for any and all usb wifi adapters that are to be used with OpenWRT. In this case, simply install the following driver:

kmod-mt76x0u

Are you using Luci?

Anything, in addition to the driver, that is required, such as the firmware, will also be installed at the same time automatically. Using your adapter and others based on the Mediatek mt7610u, mt7612u, mt7921au and mt7925 chips is easy and fun.

Good luck. Let us know how well it works.

Anakinjh, 2024-07-31:
I will try this again this week, I'll likely reflash the card in my Pi, as the configuration is pretty simple and I can do that rather quickly. I believe however I installed that package in particular, and didn't see any results, but they all meld together in my mind at this point. 

I'll reach out if I need more guidance, thank you

morrownr, 2024-07-31:
@Anakinjh 

> I don't know the exact chip my adapters use and I was struggling to find out.

We can teach you how to do that. Try this:

$ lsusb

That will give you a list of the ID's for usb devices in your system. I just ran it and I see the following:

Bus 006 Device 024: ID 0e8d:7961 MediaTek Inc. Wireless_Device

Sometimes you may need to select more than one if it is not obvious which line is for your adapter. I then do a search engine look on:

ID 0e8d:7961

With the results of the search, I read. It becomes apparent soon that this device uses the Mediatek mt7921au chipset. This is indeed an AXE3000 class usb wifi chip. It is the Edup adapter listed in the Plug and Play List in the AXE3000 section.

> Also, do you know the purpose of the button on the body of the device? I was unable to find the function anywhere, including Panda's own website?

I do not. I have Panda adapters but have never needed to contact Panda but I have heard from many Panda customers who have contacted them via their website and Panda appears to be quite supportive if customers have questions. Panda is an interesting little company. I do not think that Panda has ever released an adapter that is not plug and play on Linux. The adapter you have gets really good reviews from Linux users and I know the mt7610u chipset that it contains is a really good chipset for Linux so I suspect that you will be happy.

I run OpenWRT here on 2 routers, not on a Pi though, so if you have questions, I maybe able to help.

Anakinjh, 2024-08-01:
My real issue is I need a lot of help, I don't feel good asking for it. 

I ran the `lsusb` command and the output relevant to the adapter was

Bus 1 Device 3: ID 0e8d:7610 MediaTek WiFi 

It seems like it uses the 7610u chipset which was the reason o picked this one, after being directed to this repo.