morrownr/USB-WiFi #474: Opinion: Best WiFi Adapters for Kali Linux

morrownr, 2024-07-31:
I have been running this site for the last several years in an attempt to help Linux users make good decisions about what USB WiFi adapters will be meet their needs. I have used USB WiFi adapters since they were first available. I have maintained some out-of-kernel drivers here at this site for those who have adapters that use out-of-kernel drivers. I have worked hard to maintain those drivers but am only too aware that Realtek's out-of-kernel drivers are not good drivers... especially for normal users that do not compile drivers on a regular basis.

Why am I making this post? There is something that really annoys me. It is the amount of BAD information on the internet regarding USB WiFi adapters. I am not talking about discussions on reddit or the like, I am talking about articles and ads that say they have a list of the best USB WiFi adapters for Linux... and Kali Linux is often the target audience. The below link is an example but there are hundreds more:

https://www.stationx.net/best-wifi-adapters-for-kali-linux/

I only agree with one, yes one, of the adapters in that list. The Alfa AWUS036ACHM is a really good adapter for Linux in general and especially for Kali. The other adapters on that list either depend on Realtek out-of-kernel drivers or are overpriced or both. 

The best adapter for Kali Linux depends on what you intend to do with it. Do you need a single band, dual band or tri band adapter? Do you need an adapter with long range? Do you need Active Monitor support (that is a capability within monitor mode)? Do you need to travel with the adapter? What size of adapter do you need?

As you can see, there is no one size fits all when it comes to USB WiFi adapters, even for Kali Linux. If you are looking to buy an adapter and the person answering your question does not turn around and ask you several questions, like those above, then you are asking the wrong person.

My recommendation for Kali Linux is for you to first go to the Plug and Play List here at this site. Read the information and look at the adapters. Follow the links to see what the reviews are like. Make a list of a few adapters and come here and ask for input. Provide answers to some of the above questions and you should get some good advice. Keep in mind that there are hundreds of articles like I used as an example above. I have yet to find one that is worth reading. They contain BAD information.

At this moment, there is not one single Realtek chip based adapter that I think is good for use with Kali. None. However, this may change as we are currently seeing dramatic improvements to the Linux in-kernel rtw88 driver. An effort to improve the drivers in rtw88 was started late last year. Several patches have been pushed since then. I am expecting 2 new drivers to be sent in any day now. rtw88 drivers for the rtl8821/11au and rtl8812au chips. My testing with a Alfa ACS has caused me to say that this new rtw88 rtl8821/11au driver will be the best USB WiFi driver in the kernel for Realtek USB WiFi once it is in the kernel.

However, the Mediatek drivers are still the best. No Realtek driver supports Active Monitor. If you need that, get an Alfa AWUS036ACHM or Alfa AWUS036ACM. If you need a triband adapter, get the EDUP EP-AX1672. There are many good adapter for general use and many good for use with Kali Linux, just be really careful not to fall for the bad or dated information that is all over the internet. Take your time and research before buying.

Regards,

@morrownr

bjlockie, 2024-07-31:
If you only have time to ask one question then ask if it is a Mediatek chip.

ZerBea, 2024-08-02:
You can add this one (ASUS USB-AC51) to your list
```
$ lsusb
ID 0b05:17d1 ASUSTek Computer, Inc. AC51 802.11a/b/g/n/ac Wireless Adapter [Mediatek MT7610U]
```
I love it:
https://github.com/ZerBea/hcxdumptool/wiki/Penetration-testing-system-6

morrownr, 2024-09-09:
@ZerBea 

I finally added a new list on the Main Menu for adapters for those looking for good monitor mode support. It says it is for Kali but we know there is more involved than just Kali.

Here is a direct link:

https://github.com/morrownr/USB-WiFi/blob/main/home/Recommended_Adapters_for_Kali_Linux.md

If you find time to take a look, I would value any additions or deletions that you would make. I assume adding the ASUS AC-51 would be an addition. Your input is very welcome.

Also, if you have an explanation as to why adapters that are only supported with Realout out-of-kernel drivers are not good, that would be welcome as many people read this site. It is so common for me to get questions and bug reports from users new to Kali that have bought an adapter on one of the many "Best USB WiFi Adapter for Kali" list that are all over the internet and then I have to tell them that they are unlikely to be happy with their new adapter. Therefore I'm trying to get something out there in that space. Maybe it will help.