morrownr/USB-WiFi #475: 6GHz in AP mode on Netgear A8000

ffund, 2024-08-01:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux nyu-rpi5-01 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

### lsusb

Bus 002 Device 002: ID 0846:9060 NetGear, Inc. Wireless_Device

### rfkill

0: hci0: Bluetooth 	Soft blocked: no 	Hard blocked: no 1: phy0: Wireless LAN 	Soft blocked: no 	Hard blocked: no 2: phy1: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
sudo: dkms: command not found
```

### iw

```text
phy#1
	Interface wlan1
		ifindex 4
		wdev 0x100000001
		addr 54:07:7d:7e:ab:45
		type managed
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

### What happened?

Hi! thanks for creating this wonderful resource.

I am using a Netgear A800 on Linux kernel 6.6 - 

```
driver: mt7921u
version: 6.6.31+rpt-rpi-2712
firmware-version: ____010000-20240716151027
```

and hostapd version `v2.12-devel-hostap_2_11-16-g92374d59d` (compiled with the config described [here](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Upgrade_hostapd.md).)

I haven't been able to get it to use 6GHz in AP mode. Relevant part of the `hostapd` config is 

```
hw_mode=a
channel=37
chanlist=37
op_class=134
country_code=US
country3=0x49
ieee80211d=1
```

The error message I get is

```
SAE: Enabling SAE H2E on 6 GHz
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Frequency 6135 (primary) not allowed for AP mode, flags: 0x13803 NO-IR
Primary frequency not allowed
wlan1: IEEE 802.11 Configured channel (37) or frequency (6135) (secondary_channel=1) not found from the channel list of the current mode (2) IEEE 802.11a
wlan1: IEEE 802.11 Hardware does not support configured channel
Could not select hw_mode and channel. (-3)
wlan1: interface state COUNTRY_UPDATE->NO_IR
wlan1: AP-NO_IR
```

(I have tried a few different 6GHz frequencies with the same result.)

I also see in `iw list` that it shows "no IR" for *all* the 6GHz channels - 

```
		Frequencies:
			* 5955 MHz [1] (12.0 dBm) (no IR)
			* 5975 MHz [5] (12.0 dBm) (no IR)
			* 5995 MHz [9] (12.0 dBm) (no IR)
			* 6015 MHz [13] (12.0 dBm) (no IR)
			* 6035 MHz [17] (12.0 dBm) (no IR)
			* 6055 MHz [21] (12.0 dBm) (no IR)
			* 6075 MHz [25] (12.0 dBm) (no IR)
			* 6095 MHz [29] (12.0 dBm) (no IR)
			* 6115 MHz [33] (12.0 dBm) (no IR)
			* 6135 MHz [37] (12.0 dBm) (no IR)
			* 6155 MHz [41] (12.0 dBm) (no IR)
			* 6175 MHz [45] (12.0 dBm) (no IR)
			* 6195 MHz [49] (12.0 dBm) (no IR)
			* 6215 MHz [53] (12.0 dBm) (no IR)
			* 6235 MHz [57] (12.0 dBm) (no IR)
			* 6255 MHz [61] (12.0 dBm) (no IR)
			* 6275 MHz [65] (12.0 dBm) (no IR)
			* 6295 MHz [69] (12.0 dBm) (no IR)
			* 6315 MHz [73] (12.0 dBm) (no IR)
			* 6335 MHz [77] (12.0 dBm) (no IR)
			* 6355 MHz [81] (12.0 dBm) (no IR)
			* 6375 MHz [85] (12.0 dBm) (no IR)
			* 6395 MHz [89] (12.0 dBm) (no IR)
			* 6415 MHz [93] (12.0 dBm) (no IR)
			* 6435 MHz [97] (12.0 dBm) (no IR)
			* 6455 MHz [101] (12.0 dBm) (no IR)
			* 6475 MHz [105] (12.0 dBm) (no IR)
			* 6495 MHz [109] (12.0 dBm) (no IR)
			* 6515 MHz [113] (12.0 dBm) (no IR)
			* 6535 MHz [117] (12.0 dBm) (no IR)
			* 6555 MHz [121] (12.0 dBm) (no IR)
			* 6575 MHz [125] (12.0 dBm) (no IR)
			* 6595 MHz [129] (12.0 dBm) (no IR)
			* 6615 MHz [133] (12.0 dBm) (no IR)
			* 6635 MHz [137] (12.0 dBm) (no IR)
			* 6655 MHz [141] (12.0 dBm) (no IR)
			* 6675 MHz [145] (12.0 dBm) (no IR)
			* 6695 MHz [149] (12.0 dBm) (no IR)
			* 6715 MHz [153] (12.0 dBm) (no IR)
			* 6735 MHz [157] (12.0 dBm) (no IR)
			* 6755 MHz [161] (12.0 dBm) (no IR)
			* 6775 MHz [165] (12.0 dBm) (no IR)
			* 6795 MHz [169] (12.0 dBm) (no IR)
			* 6815 MHz [173] (12.0 dBm) (no IR)
			* 6835 MHz [177] (12.0 dBm) (no IR)
			* 6855 MHz [181] (12.0 dBm) (no IR)
			* 6875 MHz [185] (12.0 dBm) (no IR)
			* 6895 MHz [189] (12.0 dBm) (no IR)
			* 6915 MHz [193] (12.0 dBm) (no IR)
			* 6935 MHz [197] (12.0 dBm) (no IR)
			* 6955 MHz [201] (12.0 dBm) (no IR)
			* 6975 MHz [205] (12.0 dBm) (no IR)
			* 6995 MHz [209] (12.0 dBm) (no IR)
			* 7015 MHz [213] (12.0 dBm) (no IR)
			* 7035 MHz [217] (12.0 dBm) (no IR)
			* 7055 MHz [221] (12.0 dBm) (no IR)
			* 7075 MHz [225] (12.0 dBm) (no IR)
			* 7095 MHz [229] (12.0 dBm) (no IR)
			* 7115 MHz [233] (12.0 dBm) (no IR)
```

Here's what I see for `iw reg get` - 

```
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)
```

Do you know if it is possible to use this device in AP mode on 6 GHz? (If not, do you know if there is another one that does support AP mode on 6 GHz?)

Thanks!

fakemanhk, 2024-08-02:
In your hostapd.conf, you don't need to config something like `ieee80211ax=1`? How about `ht_capab` & `vht_capab`? I guess you have to tell it?

morrownr, 2024-08-02:
With your full hostapd.conf, we are more or less guessing.

> Do you know if it is possible to use this device in AP mode on 6 GHz?

It is. The users that I am aware of that are using adapters with the mt7921au chip are mostly in the EU, with country codes such as FR and DE. I have done some digging but I'm been too busy to figure out why this is not working in the US. This is why my example hostapd.conf got WiFi 6 is only using 5 Ghz:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6.conf

You might take a look at the following issue:

https://github.com/morrownr/USB-WiFi/issues/431

Near the bottom you will see information posted from a user that is using a mt7925 chip based M.2 card to setup an ap with hostapd. The information is detailed and appears to work but that is the next generation of chip and no usb wifi adapters are available yet even the driver is in the kernel and has been since 6.7.

ffund, 2024-08-02:
Sorry about that, I added the full hostapd conf below.

Thanks for the tip about this adapter working in FR and DE, that is very useful to know. And the pointer to the related issue is also very useful. I appreciate your help!

```
ssid=edgenet
wpa_passphrase=myPW1234
hw_mode=a
channel=37
chanlist=37
op_class=134
country_code=US
country3=0x49
ieee80211d=1

interface=wlan1

driver=nl80211

logger_syslog=1
logger_syslog_level=2

ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

tx_queue_data2_burst=2.0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=0
skip_inactivity_poll=1
preamble=1

auth_algs=3
macaddr_acl=0
ignore_broadcast_ssid=0
okc=1
wpa=2
wpa_pairwise=CCMP CCMP-256
rsn_pairwise=CCMP CCMP-256
wpa_key_mgmt=SAE
ieee80211w=2

ieee80211n=1
wmm_enabled=1
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]

ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][BF-ANTENNA-4][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

ieee80211ax=1
he_oper_chwidth=1
he_oper_centr_freq_seg0_idx=42
he_bss_color=37
```

morrownr, 2024-08-02:
@ffund 

> vht_oper_centr_freq_seg0_idx=42

> he_oper_centr_freq_seg0_idx=42

I do not think these settings are correct for band 4 (6 GHz). I think the number should be 39. You might want to look at my WiFi 7 example since it is for band 4:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi7.conf

That WiFi 7 guide has not been tested by me yet, which is why the status is listed as alpha. I think it will work or is close.

This WiFi 6 and 7 stuff is compliated. Remember back in the simple WiFi 4 days?

fakemanhk, 2024-08-04:
I guess OP can try to find a machine with OpenWrt and create AP with 6GHz
then get the config and compare.