morrownr/USB-WiFi #476: ALFA AWUS036AXML

KaliAssistant, 2024-08-02:
I'm fed up with the Realtek junk. I've decided to purchase the ALFA AWUS036AXML, which is based on the MediaTek MT7921AUN. If I run it on my laptop with Kali 6.8.11-amd64 and on a Raspberry Pi 5B with Kali 6.1.73-v8+ (aarch64), are there any driver or other issues I need to be aware of? (run hcxdumptool)

ZerBea, 2024-08-02:
I recommend an ALFA AWUS036ACHM and OpenWRT 23.05.
Absolutely no problems running this combination.

ZerBea, 2024-08-02:
Based on your request, I tested Linux kernel 6.11-rc1.
Now the driver crashed immediately after the AXML has been plugged in:
https://bugzilla.kernel.org/show_bug.cgi?id=219040#c9

KaliAssistant, 2024-08-02:
ok, I see.

ZerBea, 2024-08-02:
On the bright side, the patch that fixed monitor mode and packet injection on mt76 devices has been implemented now:
My test device (ALFA AWUS036ACM) is working as expected. There is no longer need to patch the kernel.
```
$ sudo hcxdumptool -i wlp5s0f4u2  --rcascan=active
...
168 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
24 PROBERESPONSE(s) captured
```

Simon566, 2024-08-03:
im running 6.11.3 on x86 (odroid h3) , im not crashing on plugin , but very soon after i start AP mode. are you running client mode ?

my message always says something like "tx urb failed" 👍

ZerBea, 2024-08-03:
@Simon566 
I would like to run monitor mode, but the USB subsystem crashed before I can set monitor mode.
Running kernel 5.15 (OpenWRT 23.05) I can set monitor mode. But after awhile, the device stops working, especially when running under heavy load.

BTW:
As of today I have decided not to make any more bug reports on https://bugzilla.kernel.org/
Also I have closed all my open bug reports.
https://github.com/ZerBea/hcxdumptool/discussions/469

morrownr, 2024-08-03:
@ZerBea 

> As of today I have decided not to make any more bug reports on https://bugzilla.kernel.org/

I understand. Bugzilla is not very effective for wireless reports. I think the only option is sending bug reports directly to linux-wireless.

ZerBea, 2024-08-03:
@morrownr 
I agree.
Sending bug reports directly to Linux wireless is fine but unfortunately it's easy to lose track of things.

Bugzilla would be much better ( a central point to collect bugs). If you encountered a bug you can first take a look there. If it has been reported you can add your solution.

morrownr, 2024-08-03:
@ZerBea 

I agree with everything you said. I'm sure there is a way to work toward a solution, I just don't know what it is short of contacting Linus.

ZerBea, 2024-08-03:
I almost think the kernel hackers do not realize/know that we are the first point of contact in case of driver issues (ap mode, managed mode, monitor mode).
Nearly 20% of the reported issues are related to drivers (and 75% to KALI users and 5 % hcxdumptool bugs).

bjlockie, 2024-08-03:
Maybe try turning off scatter-gather.

KaliAssistant, 2024-08-06:
@ZerBea  Is the mt76 driver ok in kernel 6.1.73 ?
My raspberry pi 5B:
```
uname --kernel-release
6.1.73-v8+
```

ZerBea, 2024-08-06:
@KaliAssistant 
I don't know.
6.1.73 is an older kernel. Latest longterm release is 6.1.103

KaliAssistant, 2024-08-06:
okay

KaliAssistant, 2024-08-07:
@ZerBea
I know that fixing the Linux driver for the ALFA AWUS036AXML will take some time, so I've decided to temporarily give up on MediaTek and 802.11AX, and switch back to using the ALFA AWUS036ACH v.2 (Type-C) with the RTL8812AU chipset. It has two RP-SMA antennas and a Type-C interface. Do you recommend it?

ZerBea, 2024-08-07:
I don't have an ALFA AWUS036ACH v.2,
Maybe someone else can answer the question.

morrownr, 2024-08-07:
@KaliAssistant 

I have an Alfa ACH v1. The only difference is the usb plug on the adapter but otherwise the same. What is your question?

KaliAssistant, 2024-08-07:
@morrownr 

I'm just afraid that there will be problems with the hardware

morrownr, 2024-08-07:
Do you have one or are you looking to buy one? And what is your use case?

KaliAssistant, 2024-08-07:
yes , I'm looking to buy one, and use moniter mode & injection

morrownr, 2024-08-07:
When it comes to monitor mode, I really can't recommend anything that uses a Realtek chip because the driver support is not where it should be.  For monitor mode work I generally recommend the Alfa ACHM or Alfa ACM. The ACHM has very long range but is only an AC600 class adapter. The ACM is a AC1200 class adapter. Both support Active monitor mode and various other things that the Realtek drivers simply do not support and probably won't for a long time, if ever. I have both and they are good but I am mostly an AP mode expert, not monitor mode.

I think @ZerBea has one or both of those adapters so maybe he can be more detailed for monitor mode.

KaliAssistant, 2024-08-07:
Okay

mahdi2001pop, 2024-08-15:
Hello brother
I want to buy a Wi-Fi adapter
I use Kali Linux as a default on Windows 11
I don't care about the speed because my internet does not exceed 50 mbps
But I want it to work without problems with Kali Linux to test Wi-Fi penetration
I am thinking of alfa awus036achm and alfa awus036axml
Which is better
Thank you brother

mahdi2001pop, 2024-08-15:
> When it comes to monitor mode, I really can't recommend anything that uses a Realtek chip because the driver support is not where it should be. For monitor mode work I generally recommend the Alfa [...]

Hello brother
I want to buy a Wi-Fi adapter
I use Kali Linux as a default on Windows 11
I don't care about the speed because my internet does not exceed 50 mbps
But I want it to work without problems with Kali Linux to test Wi-Fi penetration
I am thinking of alfa awus036achm and alfa awus036axml
Which is better
Thank you brother

morrownr, 2024-08-15:
Hi @mahdi2001pop 

awus036achm

I think the achm is the single best usb adapter for use with monitor mode/Kali, even when using a VM. It supports Active monitor and basically anything else you need for Kali. It also has range that cannot be beaten in a dual band adapter. It is not the latest technology but it is almost like it was designed for Kali.

mahdi2001pop, 2024-08-15:
> Hi @mahdi2001pop
>
> awus036achm
>
> I think the achm is the single best usb adapter for use with monitor mode/Kali, even when using a VM. It supports Active monitor and basically anything else you need [...]

Thank you very much ❤

morrownr, 2024-08-17:
> Are there several versions of alpha awus036achm?

Alfa, unlike TP-Link, does not have a history of multiple versions of an adapter so there is only one version.

If you go to the Plug and Play List and scroll down:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

...you will see a section on the achm. It shows several dealers but you can email Alfa and they can point you to other dealers.

Cheers

mahdi2001pop, 2024-08-18:
> > هل هناك عدة إصدارات من alpha awus036achm؟
> 
> على عكس TP-Link، لا تمتلك شركة Alfa تاريخًا لإصدارات متعددة من المحول، لذا يوجد إصدار واحد فقط.
> 
> إذا ذهبت إلى قائمة التوصيل والتشغيل وانتقلت إلى الأسفل:
> 
> https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md
> 
> ...ستشاهد قسمًا على موقع achm. يعرض هذا القسم العديد من التجار، ولكن يمكنك إرسال بريد إلكتروني إلى Alfa حتى يتمكنوا من توجيهك إلى تجار آخرين.
> 
> هتافات

Thank you for helping people.

morrownr, 2024-08-18:
> Thank you for helping people.

You are welcome. Please let us know how it goes. This site depends on reports from people around the world. The Plug and Play List changes based on user reports.

mahdi2001pop, 2024-08-19:
I will receive achm from Amazon to Iraq within 20 days.
Then I will talk about my experience💌

fhteagle, 2024-09-08:
I have Alfa AXML running as AP on a RPi3B ArchLinux ARM aarch64 host. Its the most stable mt7921 in my collection. I am running hostapd compiled from git though.

Super-Nova646, 2025-01-01:
> I have Alfa AXML running as AP on a RPi3B ArchLinux ARM aarch64 host. Its the most stable mt7921 in my collection. I am running hostapd compiled from git though.

AXML is good adapter for evil twin attack?

morrownr, 2025-01-01:
Right now I can't recommend the Alfa AXML or AXM. Both have a serious problem that has something to do with bluetooth. We are trying to figure out what is causing this. The adapters in the Plug and Play List are not affected. There is also a `Recommended for Kali` list right under the Plug and Play List on the Main Menu.

morrownr, 2025-01-08:
@andyzot 

> Wish I'd seen this before spending my money...

I know the feeling. I do my best with this site to keep current information available. Things do change. The AXML and AXM were in the Plug and Play List until last year when it became apparent that there was a problem that could not be worked around with something simple. This is a strange problem in that only the 2 Alfa adapters and one from Comfast have the problem. Most adapters with the mt7921au chipset are not affected. I have an AXML and AXM and I am working to solve the problem. I am in contact with people. I also have a couple more adapters with the same chipset and I don't see the problem at all. The Edup 1672 is an example of an adapter where the problem does not show up. The adapters with the problem were very early products to come to market.

> I've spent hours today trying to get the thing working.

If you can go back to a kernel as early as 6.1, you may not see the problem. This is a problem that first appeared about this time one year ago with kernel 6.6 but we could work around it by simply deleting the Bluetooth firmware file. As time has progressed, this no longer works. I cannot make either Alfa adapter work with the latest Ubuntu... 24.10 with kernel 6.11. I suspect the problem may have something to do with efuse information in the eeprom but I do not know this for sure right now. Hopefully we figure this out soon.

andyzot, 2025-01-08:
Hi and thanks for the update. Really appreciate the work you're doing. I had the same issue with an AWUS1900 under Kali, but it worked fine under Parrot OS, maybe there's something in that, but probably Parrot is slightly behind Kali as far as dev and updates are concerned, just guessing, I'm a complete novice. All the best and Happy New Year!

ninjape, 2025-01-10:
I have a new alfa AXML and kali with 6.11.2-amd64 and with the drivers from the alfa website, I cannot see the adapter. I think I managed to see it once out of 50 attempts. So I would say it is not working. Any ideas?

andyzot, 2025-01-10:
I've been advised that it's definitely not working but being worked on, I guess it's just a waiting game with fingers crossed.

ZerBea, 2025-01-10:
The AXML works with a few tricks (e.g USB2 instead of USB3 and disabled BT), but the value for money is not good as a simple  test shows:

$ sudo hcxdumptool --rcascan=active --tot=1 -c 6a

AWUS036AXML 68.58€ (amazon.de):
2607 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
1007 PROBERESPONSE(s) captured

TP-Link TL-WN822N 12.49€ (amazon.de):
2912 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
1159 PROBERESPONSE(s) captured

andyzot, 2025-01-10:
Really annoying, as so many historical posts recommend it. 
But thanks to all who are working on this.

ZerBea, 2025-01-10:
Here are some real tests instead of High gloss images:
https://github.com/ZerBea/hcxdumptool/discussions/361

Super-Nova646, 2025-01-10:
> The AXML works with a few tricks (e.g USB2 instead of USB3 and disabled BT), but the value for money is not good as a simple test shows:
> 
> $ sudo hcxdumptool --rcascan=active --tot=1 -c 6a
> 
> AWUS036AXML 68.58€ (amazon.de): ID 0e8d:7961 MediaTek Inc. Wireless_Device 2607 Packet(s) captured by kernel 0 Packet(s) dropped by kernel 1007 PROBERESPONSE(s) captured
> 
> TP-Link TL-WN822N 12.49€ (amazon.de): ID 2357:0108 TP-Link TL-WN822N Version 4 RTL8192EU 2912 Packet(s) captured by kernel 0 Packet(s) dropped by kernel 1159 PROBERESPONSE(s) captured

Can you tell me how can make AXML work tell me the steps

ZerBea, 2025-01-10:
Use one of the the supported Linux kernel:
```
$ uname -r
6.6.70-1-lts
```

or
```
$ uname -r
6.12.8-1
```

Make sure you're running latest firmware:
`[18615.766176] mt7921u 1-9.3:1.3: WM Firmware Version: ____010000, Build Time: 20241106151045`

Disable BT stack:
`echo "install btusb /bin/false" >> /etc/modprobe.d/local-dontload.conf`

reboot or unload btusb

connect the AXML to an USB2 port or a USB2 hub

Now dmesg should print this:
```
[20844.686881] usb 1-9.3: new high-speed USB device number 12 using xhci_hcd
[20844.818337] usb 1-9.3: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[20844.818343] usb 1-9.3: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[20844.818346] usb 1-9.3: Product: Wireless_Device
[20844.818347] usb 1-9.3: Manufacturer: MediaTek Inc.
[20844.818349] usb 1-9.3: SerialNumber: 000000000
[20845.018645] usb 1-9.3: reset high-speed USB device number 12 using xhci_hcd
[20845.159101] mt7921u 1-9.3:1.3: HW/SW Version: 0x8a108a10, Build Time: 20241106151007a

[20845.420854] mt7921u 1-9.3:1.3: WM Firmware Version: ____010000, Build Time: 20241106151045
[20847.020857] mt7921u 1-9.3:1.3 wlp22s0f0u9u3i3: renamed from wlan0
```

test monitor mode and packet injection:
```
$ sudo hcxdumptool --rcascan=active --tot=1 -c 1a
...
^C
132 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
57 PROBERESPONSE(s) captured

exit on sigterm
```
$ sudo dmesg
```
[21146.046337] mt7921u 1-9.3:1.3 wlp22s0f0u9u3i3: entered promiscuous mode
[21153.630136] mt7921u 1-9.3:1.3 wlp22s0f0u9u3i3: left promiscuous mode
```

Monitor mode (device entered  promiscuous mode) and packet injection (57 PROBERESPONSE(s) captured) is working!

Super-Nova646, 2025-01-10:
> Use one of the the supported Linux kernel:
>
> ```
> $ uname -r
> 6.6.70-1-lts
> ```
>
> or
>
> ```
> $ uname -r
> 6.12.8-1
> ```
>
> Make sure you're running latest firmware: `[18615.766176] mt7921u 1-9.3:1.3: WM Firmware [...]

why tx-power 3dBm only can I increase to 20 or 30 dBm?

ZerBea, 2025-01-10:
Because the driver shows faulty values.
Neither trust TX power values nor RSSI values - never ever!

Super-Nova646, 2025-01-10:
So can you tell me how much the true tx-power of AXML and also I have problem in evil twin attack that when I enter into the fake AP suddenly exit before I write the password into fake page and range signal for the AP short not like alfa NHA

ZerBea, 2025-01-10:
The relative TX power:
```
$ hcxdumptool -I wlp22s0f0u9u3i3

Requesting physical interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  8  11 00c0cab57576 00c0cab57576 * wlp22s0f0u9u3i3  mt7921u (NETLINK)

available frequencies: frequency [channel] tx-power of Regulatory Domain: US

  2412 [  1] 30.0 dBm	  2417 [  2] 30.0 dBm	  2422 [  3] 30.0 dBm	  2427 [  4] 30.0 dBm
  2432 [  5] 30.0 dBm	  2437 [  6] 30.0 dBm	  2442 [  7] 30.0 dBm	  2447 [  8] 30.0 dBm
  2452 [  9] 30.0 dBm	  2457 [ 10] 30.0 dBm	  2462 [ 11] 30.0 dBm	  2467 [ 12] disabled
  2472 [ 13] disabled	  2484 [ 14] disabled	  5180 [ 36] 23.0 dBm	  5200 [ 40] 23.0 dBm
  5220 [ 44] 23.0 dBm	  5240 [ 48] 23.0 dBm	  5260 [ 52] 24.0 dBm	  5280 [ 56] 24.0 dBm
  5300 [ 60] 24.0 dBm	  5320 [ 64] 24.0 dBm	  5500 [100] 24.0 dBm	  5520 [104] 24.0 dBm
  5540 [108] 24.0 dBm	  5560 [112] 24.0 dBm	  5580 [116] 24.0 dBm	  5600 [120] 24.0 dBm
  5620 [124] 24.0 dBm	  5640 [128] 24.0 dBm	  5660 [132] 24.0 dBm	  5680 [136] 24.0 dBm
  5700 [140] 24.0 dBm	  5720 [144] 24.0 dBm	  5745 [149] 30.0 dBm	  5765 [153] 30.0 dBm
  5785 [157] 30.0 dBm	  5805 [161] 30.0 dBm	  5825 [165] 30.0 dBm	  5845 [169] 27.0 dBm
  5865 [173] 27.0 dBm	  5885 [177] 27.0 dBm	  5955 [  1] 12.0 dBm	  5975 [  5] 12.0 dBm
  5995 [  9] 12.0 dBm	  6015 [ 13] 12.0 dBm	  6035 [ 17] 12.0 dBm	  6055 [ 21] 12.0 dBm
  6075 [ 25] 12.0 dBm	  6095 [ 29] 12.0 dBm	  6115 [ 33] 12.0 dBm	  6135 [ 37] 12.0 dBm
  6155 [ 41] 12.0 dBm	  6175 [ 45] 12.0 dBm	  6195 [ 49] 12.0 dBm	  6215 [ 53] 12.0 dBm
  6235 [ 57] 12.0 dBm	  6255 [ 61] 12.0 dBm	  6275 [ 65] 12.0 dBm	  6295 [ 69] 12.0 dBm
  6315 [ 73] 12.0 dBm	  6335 [ 77] 12.0 dBm	  6355 [ 81] 12.0 dBm	  6375 [ 85] 12.0 dBm
  6395 [ 89] 12.0 dBm	  6415 [ 93] 12.0 dBm	  6435 [ 97] 12.0 dBm	  6455 [101] 12.0 dBm
  6475 [105] 12.0 dBm	  6495 [109] 12.0 dBm	  6515 [113] 12.0 dBm	  6535 [117] 12.0 dBm
  6555 [121] 12.0 dBm	  6575 [125] 12.0 dBm	  6595 [129] 12.0 dBm	  6615 [133] 12.0 dBm
  6635 [137] 12.0 dBm	  6655 [141] 12.0 dBm	  6675 [145] 12.0 dBm	  6695 [149] 12.0 dBm
  6715 [153] 12.0 dBm	  6735 [157] 12.0 dBm	  6755 [161] 12.0 dBm	  6775 [165] 12.0 dBm
  6795 [169] 12.0 dBm	  6815 [173] 12.0 dBm	  6835 [177] 12.0 dBm	  6855 [181] 12.0 dBm
  6875 [185] 12.0 dBm	  6895 [189] 12.0 dBm	  6915 [193] 12.0 dBm	  6935 [197] 12.0 dBm
  6955 [201] 12.0 dBm	  6975 [205] 12.0 dBm	  6995 [209] 12.0 dBm	  7015 [213] 12.0 dBm
  7035 [217] 12.0 dBm	  7055 [221] 12.0 dBm	  7075 [225] 12.0 dBm	  7095 [229] 12.0 dBm
  7115 [233] 12.0 dBm
```

I can't answer the second part of your question, because I don't run phishing attacks like "Evil Twin".

Super-Nova646, 2025-01-10:
Last question what is the best kernel for alfa nha?
Thanks for helping

ZerBea, 2025-01-10:
Latest longterm 6.6.71 or latest stable 6.12.9.
https://www.kernel.org/

apiscode, 2025-01-26:
Kernel 6.8.11 available in Kali Linux 2024.3 works for me as well. I'm using Kali Linux for ARM on VMware fusion on my MacBook and x86_64 version in VMware Workstation.

nikolozk, 2025-01-28:
I hade same problem.
For AWUS036AXML You must update drivers.
I have Kali Linux with 6.11.2 kernel on VMware Workstation.

go to https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek
download: 
BT_RAM_CODE_MT7961_1a_2_hdr.bin
WIFI_MT7961_patch_mcu_1a_2_hdr.bin
WIFI_RAM_CODE_MT7961_1a.bin

then open bash and copy this files to /lib/firmware/mediatek directory.
type:
sudo cp BT_RAM_CODE_MT7961_1a_2_hdr.bin /lib/firmware/mediatek
sudo cp WIFI_MT7961_patch_mcu_1a_2_hdr.bin /lib/firmware/mediatek
sudo cp WIFI_RAM_CODE_MT7961_1a.bin /lib/firmware/mediatek

after this just restart your Kali Linux.
for me it worked perfectly no crashes.

Dread-Fox, 2025-02-28:
Does this mean we can't use Bluetooth while using this adapter?

morrownr, 2025-02-28:
> Does this mean we can't use Bluetooth while using this adapter?

Are you talking about the Bluetooth in the adapter or another Bluetooth in your system?

The answer should not matter soon. I think we should see this problem fixed soon. New firmware was released in November that contained part of the fix. Now a couple of patches are headed into the kernel so hopefully soon, this will no longer be an issue. This has been a confusing issue that took a long time to fix. It was a change in the Bluetooth system that messed this adapter up and even shut down wifi.

andyzot, 2025-02-28:
Great news & thanks to all involved

morrownr, 2025-02-28:
@Super-Nova646 

> why tx-power 3dBm only can I increase to 20 or 30 dBm?

I have good news on this issue as well.

Go this the following link:

https://patchwork.kernel.org/project/linux-wireless/list/?series=&submitter=&state=&q=mt79&archive=&delegate=

Scroll down until you find the following series of patches:

	wifi: mt76: fix returned txpower for mt7921 and mt7925

You can open and read the patches.

2024 was a tough year for WiFi but I think we are seeing a light at the end of the tunnel.

pepitosbazzegutisbazzeguti, 2025-06-15:
Hi. We are in mid 2025: this card is working better/flawlessy or it still has problems? Now hcxdumptool works flawless or there are still problems? Thanks!

pepitosbazzegutisbazzeguti, 2025-06-15:
Thanks, fhteagle. Anyone knowing about hoe it works with hcxdumptool?

ZerBea, 2025-06-15:
> Hi. We are in mid 2025: this card is working better/flawlessy or it still has problems? Now hcxdumptool works flawless or there are still problems? Thanks!

Please make up your own mind:

```
$ uname -r
6.15.2-arch1-1

$ lsusb
ID 0e8d:7961 MediaTek Inc. Wireless_Device

$ sudo dmesg
...
[   68.116074] usb 1-9.3: new high-speed USB device number 8 using xhci_hcd
[   68.248208] usb 1-9.3: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[   68.248216] usb 1-9.3: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[   68.248219] usb 1-9.3: Product: Wireless_Device
[   68.248222] usb 1-9.3: Manufacturer: MediaTek Inc.
[   68.248224] usb 1-9.3: SerialNumber: 000000000
[   68.403730] Bluetooth: Core ver 2.22
[   68.403756] NET: Registered PF_BLUETOOTH protocol family
[   68.403757] Bluetooth: HCI device and connection manager initialized
[   68.403762] Bluetooth: HCI socket layer initialized
[   68.403764] Bluetooth: L2CAP socket layer initialized
[   68.403768] Bluetooth: SCO socket layer initialized
[   68.442285] usbcore: registered new interface driver btusb
[   68.444930] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20241106151414
[   71.536816] Bluetooth: hci0: Device setup in 3021975 usecs
[   71.536824] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   73.587868] Bluetooth: hci0: Opcode 0x0c03 failed: -110
[   75.636150] Bluetooth: hci0: Failed to read MSFT supported features (-110)
[   77.684206] Bluetooth: hci0: AOSP get vendor capabilities (-110)
[   77.763373] usb 1-9.3: reset high-speed USB device number 8 using xhci_hcd
[   77.860127] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20241106151414
[   77.884251] usbcore: registered new interface driver mt7921u
[   77.893895] mt7921u 1-9.3:1.3: HW/SW Version: 0x8a108a10, Build Time: 20241106151007a
[   78.005179] Bluetooth: hci0: Device setup in 143807 usecs
[   78.005187] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   78.147131] mt7921u 1-9.3:1.3: WM Firmware Version: ____010000, Build Time: 20241106151045
[   78.232603] Bluetooth: hci0: AOSP extensions version v1.00
[   78.232610] Bluetooth: hci0: AOSP quality report is supported
[   79.755177] mt7921u 1-9.3:1.3 wlp22s0f0u9u3i3: renamed from wlan0
[  126.502584] mt7921u 1-9.3:1.3 wlp22s0f0u9u3i3: entered promiscuous mode
[  566.119579] mt7921u 1-9.3:1.3 wlp22s0f0u9u3i3: left promiscuous mode

$ sudo hcxdumptool --rcascan=active -F
...
^C
1182 Packet(s) captured by kernel
8 Packet(s) dropped by kernel
exit on sigterm

$ sudo hcxdumptool -c 48b,1a,11a,6a
...
^C
4281 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on sigterm

$ hcxpcapngtool 20250615164922-wlp22s0f0u9u3i3.pcapng
hcxpcapngtool 6.3.5-70-gce0daae reading from 20250615164922-wlp22s0f0u9u3i3.pcapng...

summary capture file
--------------------
file name................................: 20250615164922-wlp22s0f0u9u3i3.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 6.15.2-arch1-1
application..............................: hcxdumptool 6.3.5-147-g6cb43de
interface name...........................: wlp22s0f0u9u3i3
interface vendor.........................: 00c0ca
openSSL version..........................: 1.1
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: 1100aae06ec3 (incremented on every new client)
MAC CLIENT...............................: b0febdcf2b73
REPLAYCOUNT..............................: 62554
ANONCE...................................: 3909c20dd8881b247c1e9ac0e2d572b112d07fb1d46e6903b3a35b0098b01cb5
SNONCE...................................: 4b26c35c76963092a0bcbab0b4f1472bbb555dd7b9ef12e5a04ee3be558a632f
timestamp minimum (timestamp)............: 15.06.2025 14:49:27 (1749998967)
timestamp maximum (timestamp)............: 15.06.2025 14:50:13 (1749999013)
duration of the dump tool (seconds)......: 45
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianness (capture system)..............: little endian
packets inside...........................: 93
packets received on 2.4 GHz..............: 83
packets received on 5 GHz................: 9
ESSID (total unique).....................: 18
BEACON (total)...........................: 18
BEACON on 2.4 GHz channel (from IE_TAG)..: 1 6 11 
PROBEREQUEST (undirected)................: 1
PROBERESPONSE (total)....................: 13
AUTHENTICATION (total)...................: 1
AUTHENTICATION (OPEN SYSTEM).............: 1
ASSOCIATIONREQUEST (total)...............: 2
ASSOCIATIONREQUEST (PSK).................: 2
EAPOL messages (total)...................: 50
EAPOL RSN messages.......................: 50
EAPOLTIME gap (measured maximum msec)....: 21993
EAPOL ANONCE error corrections (NC)......: not detected
EAPOL M1 messages (total)................: 44
EAPOL M2 messages (total)................: 3
EAPOL M3 messages (total)................: 1
EAPOL M4 messages (total)................: 2
EAPOL M4 messages (zeroed NONCE).........: 2
EAPOL pairs (total)......................: 4
EAPOL pairs (best).......................: 1
EAPOL ROGUE pairs........................: 1
EAPOL M12E2 (challenge - ANONCE from M1).: 1
RSN PMKID (total)........................: 19
RSN PMKID (best).........................: 1
RSN PMKID ROGUE..........................: 1

frequency statistics from radiotap header (frequency: received packets)
-----------------------------------------------------------------------
 2412: 29	 2437: 8	 2462: 46	 5240: 9	

Information: no hashes written to hash files

session summary
---------------
processed pcapng files................: 1
```

Test target has been attacked successfully.
Range in 5GHz is poor, but that's physically determined if compared to 2.4 Ghz.

pepitosbazzegutisbazzeguti, 2025-06-16:
> Range in 5GHz is poor, but that's physically determined if compared to 2.4 Ghz.

So, it's better to use hcxdumptool on 2.4 Ghz than 5 Ghz, correct?

ZerBea, 2025-06-16:
hcxdumptool's and hcxlabtool's (hcxm2hunter) main target are weak CLIENTs. 
For sure both tools can do old school DEAUTHENTICATION attacks and PMKID attacks on APs too, but that is definitely not the goal.
That is mentioned in README.md section "Warning":
```
- hcxdumptool is able to capture handshakes from non-connected clients. (Only one single M2 from the client is required. Use hcxpcapngtool to convert them to a format Hashcat or JtR understands.)
- hcxdumptool is able to capture handshakes from 5/6GHz clients on 2.4GHz. (Only one single M2 from the client is required. Use hcxpcapngtool to convert to a format Hashcat or JtR understands.)
```

You don't need an expensive premium adapter to run these attacks, because target CLIENT will try to connect to hcxdumpool/hcxlabtool. This attack is called AP-LESS attack and it works like a charm (if you know what you are doing):
https://wpa-sec.stanev.org/?search=0050c74f1ce9

A cheap TP-Link TL-WN722N(v2/3) (external antenna connector), TL-WN-822N (external antenna) or TL-WN823N (on board antenna) is more than enough to run these attacks.

Jarvis785, 2025-08-09:
> I hade same problem. For AWUS036AXML You must update drivers. I have Kali Linux with 6.11.2 kernel on VMware Workstation.
> 
> go to https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek download: BT_RAM_CODE_MT7961_1a_2_hdr.bin WIFI_MT7961_patch_mcu_1a_2_hdr.bin WIFI_RAM_CODE_MT7961_1a.bin
> 
> then open bash and copy this files to /lib/firmware/mediatek directory. type: sudo cp BT_RAM_CODE_MT7961_1a_2_hdr.bin /lib/firmware/mediatek sudo cp WIFI_MT7961_patch_mcu_1a_2_hdr.bin /lib/firmware/mediatek sudo cp WIFI_RAM_CODE_MT7961_1a.bin /lib/firmware/mediatek
> 
> after this just restart your Kali Linux. for me it worked perfectly no crashes.
> 
> P.s disable BT it crashes system. open terminal and type: echo "install btusb /bin/false" >> /etc/modprobe.d/local-dontload.conf

Hi @nikolozk @ZerBea @morrownr 
monitor mode and packet capture works for me, but
even after performing the above steps, VIF is not working on airgeddon.
while evil twin attack AP is hosted fine but deauth is not working for me

could anyone help please !!

Windows 11 host
vamware pro 17
usb 3.0 pass through 
Kali 2025.2
kernel 6.12.33+kali-amd64

[3 more replies]