morrownr/USB-WiFi #477: [Help]:

Simon566, 2024-08-03:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux odroid-h3 6.10.3 #2 SMP PREEMPT_DYNAMIC Sat Aug  3 13:32:57 CEST 2024 x86_64 GNU/Linux

### lsusb

Bus 002 Device 002: ID 0e8d:7961 MediaTek Inc. Wireless_Device Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver Bus 001 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### rfkill

not installed

### dkms

```text
dkms isnt installed
```

### iw

```text
before the crash:

phy#3
        Interface wlx00c0cab38118
                ifindex 22
                wdev 0x300000001
                addr 00:c0:ca:b3:81:18
                ssid iot22
                type AP
                channel 6 (2437 MHz), width: 40 MHz, center1: 2447 MHz
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
phy#1
        Interface wlx24050f9d9b4b
                ifindex 20
                wdev 0x100000001
                addr 24:05:0f:9d:9b:4b
                ssid guest22
                type AP
                channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
                txpower 20.00 dBm
```

### What happened?

hostapd is crashing when running a AP on mt7921u device and the client triggers a speed check (openspeedtest.com f.e.)

---

[Sa Aug  3 19:13:52 2024] Bluetooth: hci0: Failed to load firmware file (-2)
[Sa Aug  3 19:13:52 2024] Bluetooth: hci0: Failed to set up firmware (-2)
[Sa Aug  3 19:13:52 2024] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[Sa Aug  3 19:13:52 2024] usbcore: registered new interface driver mt7921u
[Sa Aug  3 19:13:52 2024] mt7921u 2-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20240716150944a

[Sa Aug  3 19:13:52 2024] mt7921u 2-2:1.3: WM Firmware Version: ____010000, Build Time: 20240716151027
[Sa Aug  3 19:13:54 2024] mt7921u 2-2:1.3 wlx00c0cab38118: renamed from wlan0
[Sa Aug  3 19:14:16 2024] br-iot: port 2(wlx00c0cab38118) entered blocking state
[Sa Aug  3 19:14:16 2024] br-iot: port 2(wlx00c0cab38118) entered disabled state
[Sa Aug  3 19:14:16 2024] mt7921u 2-2:1.3 wlx00c0cab38118: entered allmulticast mode
[Sa Aug  3 19:14:16 2024] mt7921u 2-2:1.3 wlx00c0cab38118: entered promiscuous mode
[Sa Aug  3 19:14:17 2024] br-iot: port 2(wlx00c0cab38118) entered blocking state
[Sa Aug  3 19:14:17 2024] br-iot: port 2(wlx00c0cab38118) entered forwarding state
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71
[Sa Aug  3 19:15:47 2024] mt7921u 2-2:1.3: tx urb failed: -71

morrownr, 2024-08-06:
@Simon566 

There are some problems in the wifi stack currently. It happens when modernization happens like it is right now. Things that you don't know exist break things that you need. Patches are being worked.

My recommendation is for you back off to an older kernel for now. Maybe 6.1 or 6.6 and hopefully the things that broke this haven't made it back that far. Things started getting messy with 6.9 and it is not just with your mt7921u driver.

Simon566, 2024-08-07:
@morrownr 

which bugs are existing in 6.10 in wifi stack to look out for ?

im coming from kernel 6.1. half a year ago and had the same issues there. im a bit disappointed by ALFA / kernel , i have this adapter since they appeared on the market and couldnt use it up till now ( my only use is AP mode). No virtual AP and all these crashes happening .....

regards,
Simon