morrownr/USB-WiFi #478: [Help]: Alfa AWUS036AXM device not recognized

laubblaeser, 2024-08-05:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux ferdux 6.10.2-1-default #1 SMP PREEMPT_DYNAMIC Mon Jul 29 08:51:47 UTC 2024 (65a34e2) x86_64 x86_64 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 003 Device 002: ID 1b1c:1b36 Corsair Corsair Gaming K70 LUX Keyboard  Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 004 Device 010: ID 0fce:720d Sony Ericsson Mobile Communications AB XQ-CT54 Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 005 Device 002: ID 1b1c:1b2f Corsair Sabre RGB [CH-9303011-XX] Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

### rfkill

EMPTY OUTPUT

### dkms

```text
EMPTY OUTPUT
```

### iw

```text
EMPTY OUTPUT
```

### What happened?

The OS doesn't recognize the USB WiFi dongle at all. After plugging it in, I can run `lsusb`, which takes about 60-90 seconds until it returns anything and then the listed USB ports are "empty". 

There's no problem with the stick itself as I've tested it on a Windows machine where it gets recognized and works fine after installing the drivers. The stick was recognized exactly *once* after letting it plugged in when shutting down the PC from Windows and letting it reboot into Linux. Once I removed the stick and put it in *any* USB port of my PC again, it is not recognized again.

So, I can imagine that this is a really weird issue but it's worth a shot: Has anyone had this problem or a similar one? What can I do to track down the root of this issue?

## System info

Operating System: openSUSE Tumbleweed 20240801
KDE Plasma Version: 6.1.3
KDE Frameworks Version: 6.4.0
Qt Version: 6.7.2
Kernel Version: 6.10.2-1-default (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 3700X 8-Core Processor
Memory: 31.3 GiB of RAM
Graphics Processor: AMD Radeon RX 7800 XT

bjlockie, 2024-08-05:
In a terminal type `sudo dmesg` after you plug in the Alfa.

morrownr, 2024-08-05:
Hi @laubblaeser 

My thoughts after looking over the info you posted:

This is not a driver or firmware issue as usb adapters will show with `lsusb` even if the driver and firmware are not there. In fact, a lot of things can be ruled out at this point. Until `lsusb` can see the adapter, nothing else matters.

Is there a problem with the usb port? Maybe you should try a different port if the system has multiple ports.

Do you have another Linux system to test? 

Is it possible that some bios usb settings could be causing the problem?

Can you backup your kernel to an earlier version like say 6.6? We have been seeing a LOT of problems related to wifi lately with all of the modernization that is going on.

You can always burn a flash drive with a Ubuntu 24.04 image and boot to the desktop to test. That would allow you to test a different distro and kernel without installing anything.

Cheers

laubblaeser, 2024-08-05:
> In a terminal type `sudo dmesg` after you plug in the Alfa.

Sure, here's the output:

```
[ 6554.969849] [   T7445] usb 3-5: new high-speed USB device number 4 using xhci_hcd
[ 6555.112994] [   T7445] usb 3-5: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[ 6555.113002] [   T7445] usb 3-5: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[ 6555.113007] [   T7445] usb 3-5: Product: Wireless_Device
[ 6555.113010] [   T7445] usb 3-5: Manufacturer: MediaTek Inc.
[ 6555.113013] [   T7445] usb 3-5: SerialNumber: 000000000
[ 6555.271753] [  T19523] Bluetooth: Core ver 2.22
[ 6555.271780] [  T19523] NET: Registered PF_BLUETOOTH protocol family
[ 6555.271782] [  T19523] Bluetooth: HCI device and connection manager initialized
[ 6555.271786] [  T19523] Bluetooth: HCI socket layer initialized
[ 6555.271788] [  T19523] Bluetooth: L2CAP socket layer initialized
[ 6555.271792] [  T19523] Bluetooth: SCO socket layer initialized
[ 6555.299105] [  T19525] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[ 6555.299248] [  T19525] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[ 6555.299403] [  T19525] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[ 6555.304767] [  T19520] usbcore: registered new interface driver btusb
[ 6555.338858] [    T191] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20240219111427
[ 6559.015966] [    T191] Bluetooth: hci0: Device setup in 3624245 usecs
[ 6559.015972] [    T191] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
```

laubblaeser, 2024-08-05:
> This is not a driver or firmware issue as usb adapters will show with `lsusb` even if the driver and firmware are not there. In fact, a lot of things can be ruled out at this point. Until `lsusb` can see the adapter, nothing else matters.

Yeah, I figured as much but thought it's worth a shot asking here if anyone else ever experienced such a problem.

> Do you have another Linux system to test?
> 
> Is it possible that some bios usb settings could be causing the problem?

Well, only other systems with Tumbleweed on them. 😄 Certainly worth a shot to try the USB dongle on another device just to rule out that the hardware isn't failing or BIOS settings aren't blocking anything. But to be honest that shouldn't be the case as it works on the same machine under Windows without any issues.

morrownr, 2024-08-05:
> New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00

Interesting that this line shows in the log but not with `lsusb`.  That test with the Ubuntu 24.04 flash drive may give us an idea.

morrownr, 2024-08-05:
> The dongle always lights up...

That just means it is getting power.

> I've tried 7 different ports on the machine so far.

Had to try it.

bjlockie, 2024-08-05:
Type `lsusb -t` and `lshw -c network`

laubblaeser, 2024-08-05:
So it just worked while I plugged it in and out multiple times in a row. Here's a bit more dmesg output showing the failed and successful attempts.

<details>
```
[ 6554.969849] [   T7445] usb 3-5: new high-speed USB device number 4 using xhci_hcd
[ 6555.112994] [   T7445] usb 3-5: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[ 6555.113002] [   T7445] usb 3-5: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[ 6555.113007] [   T7445] usb 3-5: Product: Wireless_Device
[ 6555.113010] [   T7445] usb 3-5: Manufacturer: MediaTek Inc.
[ 6555.113013] [   T7445] usb 3-5: SerialNumber: 000000000
[ 6555.271753] [  T19523] Bluetooth: Core ver 2.22
[ 6555.271780] [  T19523] NET: Registered PF_BLUETOOTH protocol family
[ 6555.271782] [  T19523] Bluetooth: HCI device and connection manager initialized
[ 6555.271786] [  T19523] Bluetooth: HCI socket layer initialized
[ 6555.271788] [  T19523] Bluetooth: L2CAP socket layer initialized
[ 6555.271792] [  T19523] Bluetooth: SCO socket layer initialized
[ 6555.299105] [  T19525] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[ 6555.299248] [  T19525] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[ 6555.299403] [  T19525] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[ 6555.304767] [  T19520] usbcore: registered new interface driver btusb
[ 6555.338858] [    T191] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20240219111427
[ 6559.015966] [    T191] Bluetooth: hci0: Device setup in 3624245 usecs
[ 6559.015972] [    T191] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[ 6561.103107] [    T191] Bluetooth: hci0: Opcode 0x0c03 failed: -110
[ 6563.239755] [    T191] Bluetooth: hci0: Failed to read MSFT supported features (-110)
[ 6565.373058] [    T191] Bluetooth: hci0: AOSP get vendor capabilities (-110)
[ 6565.512710] [  T19525] usb 3-5: reset high-speed USB device number 4 using xhci_hcd
[ 6570.829683] [  T19525] usb 3-5: device descriptor read/64, error -110
[ 6586.402851] [  T19525] usb 3-5: device descriptor read/64, error -110
[ 6586.639531] [  T19525] usb 3-5: reset high-speed USB device number 4 using xhci_hcd
[ 6591.946128] [  T19525] usb 3-5: device descriptor read/64, error -110
[ 6607.522600] [  T19525] usb 3-5: device descriptor read/64, error -110
[ 6607.755978] [  T19525] usb 3-5: reset high-speed USB device number 4 using xhci_hcd
[ 6612.945886] [  T19525] xhci_hcd 0000:05:00.3: Timeout while waiting for setup device command
[ 6618.279129] [  T19525] xhci_hcd 0000:05:00.3: Timeout while waiting for setup device command
[ 6618.489117] [  T19525] usb 3-5: device not accepting address 4, error -62
[ 6618.619150] [  T19525] usb 3-5: reset high-speed USB device number 4 using xhci_hcd
[ 6623.825764] [  T19525] xhci_hcd 0000:05:00.3: Timeout while waiting for setup device command
[ 6629.155746] [  T19525] xhci_hcd 0000:05:00.3: Timeout while waiting for setup device command
[ 6629.365654] [  T19525] usb 3-5: device not accepting address 4, error -62
[ 6629.365726] [    T635] usb 3-5: USB disconnect, device number 4
[ 6629.365860] [  T19525] mt7921u 3-5:1.3: probe with driver mt7921u failed with error -5
[ 6629.365908] [  T19525] usbcore: registered new interface driver mt7921u
[ 6629.367652] [      C1] retire_capture_urb: 147 callbacks suppressed
[ 6629.488990] [    T635] usb 3-5: new high-speed USB device number 5 using xhci_hcd
[ 6634.828948] [    T635] usb 3-5: device descriptor read/64, error -110
[ 6650.405454] [    T635] usb 3-5: device descriptor read/64, error -110
[ 6650.645407] [    T635] usb 3-5: new high-speed USB device number 6 using xhci_hcd
[ 6655.955401] [    T635] usb 3-5: device descriptor read/64, error -110
[ 6671.528521] [    T635] usb 3-5: device descriptor read/64, error -110
[ 6671.638594] [    T635] usb usb3-port5: attempt power cycle
[ 6672.058497] [    T635] usb 3-5: new high-speed USB device number 7 using xhci_hcd
[ 6677.158604] [    T635] xhci_hcd 0000:05:00.3: Timeout while waiting for setup device command
[ 6682.491773] [    T635] xhci_hcd 0000:05:00.3: Timeout while waiting for setup device command
[ 6682.701704] [    T635] usb 3-5: device not accepting address 7, error -62
[ 6682.831703] [    T635] usb 3-5: new high-speed USB device number 8 using xhci_hcd
[ 6688.034993] [    T635] xhci_hcd 0000:05:00.3: Timeout while waiting for setup device command
[ 6693.371618] [    T635] xhci_hcd 0000:05:00.3: Timeout while waiting for setup device command
[ 6693.581580] [    T635] usb 3-5: device not accepting address 8, error -62
[ 6693.581644] [    T635] usb usb3-port5: unable to enumerate USB device
[ 6693.690170] [  T19715] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[ 6693.690175] [  T19715] Bluetooth: BNEP filters: protocol multicast
[ 6693.690180] [  T19715] Bluetooth: BNEP socket layer initialized
[ 7792.935879] [  T19519] usb 3-5: new high-speed USB device number 9 using xhci_hcd
[ 7793.079498] [  T19519] usb 3-5: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[ 7793.079507] [  T19519] usb 3-5: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[ 7793.079511] [  T19519] usb 3-5: Product: Wireless_Device
[ 7793.079515] [  T19519] usb 3-5: Manufacturer: MediaTek Inc.
[ 7793.079518] [  T19519] usb 3-5: SerialNumber: 000000000
[ 7793.213856] [    T191] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20240219111427
[... 117 lines cut ...]
[... cut ...]
[ 8231.093582] [    T191] Bluetooth: hci0: Device setup in 3708430 usecs
[ 8231.093588] [    T191] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[ 8231.171385] [    T191] Bluetooth: hci0: AOSP extensions version v1.00
[ 8231.171393] [    T191] Bluetooth: hci0: AOSP quality report is supported
[ 8231.171505] [  T19697] Bluetooth: MGMT ver 1.22
```
</details>

It seems as if sometimes the stack doesn't properly load and some probing fails? Does this help in any way?

bjlockie, 2024-08-05:
> So it just worked while I plugged it in and out multiple times in a row. Here's a bit more dmesg output showing the failed and successful attempts.
> 
> It seems as if sometimes the stack doesn't properly load and some probing fails? Does this help in any way?
> 
> So this means that
> 
>     1. once the dongle is initialized properly, then there's no problem with running it in dual mode (BT and WiFi appear in the system tray, although so far I haven't tested using them both at the same time).
> 
>     2. the issue I experience seems to be kind of the norm but replugging the dongle multiple times at the same USB port sporadically allows it to be recognized after all. (Previously I switched around the various ports and didn't repeatedly try the same port.)

I would try booting a different kernel via USB.
My guess is there are kernel bugs. :-)

morrownr, 2024-08-05:
Is your cpu amd or intel?

laubblaeser, 2024-08-05:
> I would try booting a different kernel via USB.
> My guess is there are kernel bugs. :-)

Currently have to deal with data limits from a metered connection. Maybe it's really worth a shot at some point but maybe this regression (?) will be fixed at some point in the near future.

> Is your cpu amd or intel?

> Processors: 16 × AMD Ryzen 7 3700X 8-Core Processor

morrownr, 2024-08-05:
> AMD Ryzen 7 3700X 8-Core Processor

I have emailed a person that can provide a workaround for us to try. Standby.

FYI: This seems to only happen on AMD cpus. No idea why at this point... and it is only with the Alfa adapters... the AXM and ACML. This is the first time, if this is the issue i think it is, that I have seen a report from an AXM owner.

ZerBea, 2024-08-06:
@laubblaeser 
Issue reports on ALFA AWUS036AXM and AWUS036AXM rise and all are similar to your report, e.g.
I don't expect any improvement in the next time:
https://bugzilla.kernel.org/show_bug.cgi?id=219040

morrownr, 2024-08-06:
@laubblaeser 

The quick way to test and see if we are the right track is to delete the bluetooth firmware:

$ sudo rm /usr/lib/firmware/mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin*

Thanks for @ZerBea for helping out. Evidently this only shows up on AMD systems. Until very recently I only had Intel and ARM64 systems and had not ran into this issue. I have an AMD mini pc now but am not configured to test yet. This is really a mystery.

We really could use feedback at to where deleting the firmware as shown above is a good workaround.

laubblaeser, 2024-10-03:
Sorry for the late reply.

The workaround (removing the BT driver from the firmware folder) indeed provides a good result as a *workaround*. The dongle instantly connects in that case and WiFi works without any issue, also for prolonged time. Haven't tested for multiple day uptimes because that's not my use case but it does work without issues for multiple hours. It's stable as far as I can tell (tested with video streaming, video calls, big downloads).

Another remark on the usage in dual mode where Bluetooth stays enabled: It works after 2-3 replugs in most cases. Sometimes already on first try and sometimes after four tries. The BT driver seems to work fine at first but then causes issues as it drops its connection to the device. The devices (in my case a PS4 dual shock controller that works perfectly fine otherwise with a separate BT dongle) need to be reconnected afterwards. It's a hassle and really annoying. I hope this issue gets fixed at some point in the future because habing one dongle for both BT and WiFi would be nice. But for now there's not really anything we can do as it seems.

Hope this helps.

closed 2024-10-03 by laubblaeser

morrownr, 2024-10-20:
I noticed a patch that should fix this issue the other day. Hopefully it works.