morrownr/USB-WiFi #481: Chipset naming: MT7921AU vs. MTK7921AU

AnaMarkH, 2024-08-07:
I found the following WiFi USB stick:
**Delock Wi-Fi 6E Dualband WLAN USB Adapter AX3000 (2 x 1201 + 574 Mbps) (Prod. No. 12773)**
[https://www.delock.de/produkt/12773/merkmale.html?setLanguage=en](https://www.delock.de/produkt/12773/merkmale.html?setLanguage=en)

According to the technical details, it has the Mediatek MTK7921AU chipset.

Online search for "MTK7921AU" seems to give primarily european sites with one exception:
[https://www.amazon.com/EDUP-Wireless-802-11AX-Tri-Band-Compatible/dp/B0CZ82RM5L](https://www.amazon.com/EDUP-Wireless-802-11AX-Tri-Band-Compatible/dp/B0CZ82RM5L)  lists the EDUP AX3000M and also states MTK7921AU.

mediatek.com knows MT7921AU only; an MT**K**7921AU cannot be found.

Hence, I wonder, whether this is the same chipset than the MT7921AU (without the "K").

I would be good to add a note about the MTK7921AU to the site (even if it's just saying "There is also a Mediatek chipset named MTK7921AU; currently it is unclear to what extent it differs from the MT7921AU.").

bjlockie, 2024-08-07:
I've seen MTK used for MediaTek.
MT is also used.
Both are short forms for MediaTek.
I'd buy that adapter

morrownr, 2024-08-07:
@AnaMarkH 

I agree with what @bjlockie said.

I have one of those Edup adapters from the link you posted. I think it is excellent. I use it with an extension cable with pedestal so as to place it where I want.

EGrunzke, 2025-01-05:
I bought the `EDUP AX3000M` [from Amazon](https://www.amazon.com/EDUP-Wireless-802-11AX-Tri-Band-Compatible/dp/B0CZ82RM5L) as mentioned by OP. Plug and play for me on Ubuntu 24.04.1 LTS kernel 6.8.0! Plus it comes with a 256MB flash drive with the windows driver 🤣

morrownr, 2025-01-06:
Hi @EGrunzke 

Glad to hear you had success. Hopefully soon we will see adapters that use the following driver:

driver=mt7925u