morrownr/USB-WiFi #482: [ NEED HELP] mt7921u does not reconnect to AP after deauthorized from successful connection.

QTXLG, 2024-08-08:
Adapter: Fenvi AX1800 ( **MediaTek mt7921u** )
Device OS: official **Debian 12 Arm** (kernel 6.1.0-23)
Mode: Client, wpa_supplicant
Firmware: **HW/SW Version: 0x8a108a10, Build Time: 20240716150944a**

> The system lacks the firmware and regulatory.db, probably because it had been installed before I bought the adapter.Thereby I installed the firmware as well as regulatory.db on my own.

AP: **openwrt 23.05.04 wpad-openssl** ( not wpad-basic)

Encryption:
1. WPA2-EAP TLS 

Rekeying every hour for default openwrt settings and device's logs recorded `Rekeying PTK for STA 52:13:8e:92:de:65 but driver can't safely do that.` and afterwards `WPA: Key negotiation completed with 52:13:8e:92:de:65 [PTK=CCMP GTK=CCMP]`
It runs well until losing network for a failure of rekeying, mostly one day after connected. `wlx90de807391b5: deauthenticated from 52:13:8e:92:de:65 (Reason: 2=PREV_AUTH_NOT_VALID)`

2. WPA2-PSK

The access point deauthenticates all the clients at a static point every day, such as at 01:10, which varies according to the point AP was booted at. If losing network, its logs only records `wlx90de807391b5: deauthenticated from 52:13:8e:92:de:65 (Reason: 16=GROUP_KEY_HANDSHAKE_TIMEOUT)` and nothing further after that. Sometimes it can reconnect to AP but it must fail in two days after connected. 

All the other devices have no such issue. If the AP rebooted, mt7921u also loses signals but is able to reconnect automatically.

morrownr, 2024-08-11:
Are you seeing anything in the system log of the client that might give a hint?

QTXLG, 2024-08-12:
> Are you seeing anything in the system log of the client that might give a hint?

They are all that I think is a hint.

morrownr, 2024-08-12:
I run Debian 12 but not the ARM version.

You might think about using synaptic to install a later kernel like 6.6 if available for your system. Search on "linux-image".

That would give you an updated version of the driver.

This could by a wpa_supplicant issue also. Have you thought about compiling a new version?

QTXLG, 2024-08-13:
> You might think about using synaptic to install a later kernel like 6.6 if available for your system. Search on "linux-image".
> 
> That would give you an updated version of the driver.
> 
> This could by a wpa_supplicant issue also. Have you thought about compiling a new version?

Nope. I have no experience with compiling kernel but I am glad to take a try. Is there any way  that could help to confirm if it is caused by wpa_supplicant?

morrownr, 2024-08-13:
>  I have no experience with compiling kernel but I am glad to take a try.

I think I have a guide around here somewhere but before we go there... why don't you try updating to kernel 6.6 and see if that makes a difference.

QTXLG, 2024-08-14:
> > I have no experience with compiling kernel but I am glad to take a try.
> 
> why don't you try updating to kernel 6.6 and see if that makes a difference.

The device booting from a NOR flash on which initramfs and kernel are located limits kernel size to below 3MB. The official debian only provides up to kernel 6.1. I doubt it would get support of kernel 6.6 in future. It is too outdated.

reubenfarrelly, 2025-06-08:
I'm seeing the same thing with this hardware on a Raspberry Pi 5, using Raspberry Pi OS (latest with all updates):

```
[    5.197668] mt7921u 2-1:1.0: HW/SW Version: 0x8a108a10, Build Time: 20240219110958a
[    5.452895] mt7921u 2-1:1.0: WM Firmware Version: ____010000, Build Time: 20240219111038
```

Kernel:  Linux temp-1 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux

Literally the whole wireless interface stops working and requires a reboot (although the system is otherwise normal and responsive and if I plug in ethernet I can log on):

```
[   10.947350] wlan0: authenticate with 34:5d:a8:0b:4e:25 (local address=94:18:65:5e:d5:bd)
[   10.948590] wlan0: send auth to 34:5d:a8:0b:4e:25 (try 1/3)
[   10.952789] wlan0: authenticated
[   10.954761] wlan0: associate with 34:5d:a8:0b:4e:25 (try 1/3)
[   10.972985] wlan0: RX AssocResp from 34:5d:a8:0b:4e:25 (capab=0x11 status=0 aid=34)
[   10.987541] wlan0: associated
[68980.204634] wlan0: disassociated from 34:5d:a8:0b:4e:25 (Reason: 16=GROUP_KEY_HANDSHAKE_TIMEOUT)
```

The wlan0 interface retains an IP address and the system thinks that the wlan0 is connected and operational, but it isn't.

There's nothing logged, aside from these messages constantly written to the messages log:

```
Jun  8 11:50:48 temp-1 wpa_supplicant[695]: wlan0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-44 noise=9999 txrate=1200900
Jun  8 11:50:51 temp-1 wpa_supplicant[695]: wlan0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-43 noise=9999 txrate=1200900
Jun  8 11:50:54 temp-1 wpa_supplicant[695]: wlan0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-44 noise=9999 txrate=1200900
Jun  8 11:50:57 temp-1 wpa_supplicant[695]: wlan0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-43 noise=9999 txrate=1200900
Jun  8 11:51:00 temp-1 wpa_supplicant[695]: wlan0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=-43 noise=9999 txrate=1200900
```

...which seem to be erroneous but are present even when the system is working normally.

I'm going to assume that this is the same problem that the OP is seeing - @QTXLG can you please confirm if this matches your experiences?  If not then I will create a separate bug report for this.  But it seems like it might be the same issue.

I'm about to update the wireless firmware files to see if that fixes the problem.

morrownr, 2025-06-08:
Hi @reubenfarrelly 

The following issue might be one you should take a look at:

https://github.com/morrownr/USB-WiFi/issues/603

Patches went in for the mt7921 and mt7925 related to the issue as described:

https://patchwork.kernel.org/project/linux-wireless/patch/f23a72ba7a3c1ad38ba9e13bb54ef21d6ef44ffb.1748149855.git.deren.wu@mediatek.com/

That is the patch for mt7921. If you are into compiling, you might try the patch. It will at some point work its way back into the kernel 6.12

> As shipped the system uses this slightly old firmware version:

I'd say that the firmware is really old. WiFi 6 and 7 is pretty darn complicated and has led to regular firmware updates from all of the wireless makers. This old firmware really should be reported to RasPiOS. They need to do a better job of keeping their firmware up to date.

Neirop, 2025-08-01:
I have the same issue with OpenWrt SNAPSHOT r30608-6fa1775348, which include the patch mentioned.