morrownr/USB-WiFi #483: [EDUP EP-AX1672] Anyone tested this adapter with 6Ghz in AP mode?

gdluca, 2024-08-11:
I am looking at this interesting adapter to use as my main 6Ghz AP. I read your ( @morrownr ) test and it seems to be working really well for 5 Ghz wifi 6. Have you tried the 6 Ghz? Another thing I would like to know is if there are any issues(as in not all frequencies available) in setting JP as the iw reg country.

Thank you for any feedback.

morrownr, 2024-08-11:
Hi @gdluca 

>  I read your ( @morrownr ) test and it seems to be working really well for 5 Ghz wifi 6.

Yes, the example hostapd.conf that I have for WiFi 6 on the Main Menu continues to be rock solid here. Have I tried get get 6 GHz going? Yes but it was short lived as other projects too my time away from it. I have seen other users report support success with 6 GHz AP mode but I noted that they were all in the EU with country codes like FR and DE. I think the problem with using 6 GHz in AP mode in the US has to do with the regdb settings but have not had time to get to the bottom of it,

There is one report that might provide help. It is by @ibrtuyima and is located in the following issue near the bottom:

https://github.com/morrownr/USB-WiFi/issues/431

He is using a country code of US and AP mode with 6 GHz appears to be workings. This is not with a mt7921au chip but rather the next gen mt7925 chip but maybe there is some information there to be used. I have no report from a JP country code.

I do have a EDUP EP-AX1672 adapter and think it is a really good adapter. I also have a mt7925 chip based m.2 card that is working well in managed mode with Ubuntu 24.04 but have no tested AP mode yet.

I hope something in this reply helps.

@morrownr

gdluca, 2024-08-12:
Hi @morrownr,

thanks for the reply! Yes, it helped. I want to try anyway so I have ordered one EDUP EP-AX1672. Hopefully will have some feedback for the end of this month.

gdluca, 2024-08-18:
Hi @morrownr,

my EDUP EP-AX1672 arrived and it works really really well. Unfortunately, my wifi 6e client has issues (not related to wifi) and cannot test the 6Ghz for now. However, hostapd works just fine with country set to JP and 'iw list' shows all expected 6Ghz channels available so it should be fine. Quickly done a speedtest using my FTTH at home and I can easily achieve ~730Mbps download/upload in wifi 6 5Ghz. Way better than my old pcie wifi 5 adapter so I am satisfied!

Thank you for keeping this repo updated!

morrownr, 2024-08-18:
Hi @gdluca 

> my EDUP EP-AX1672 arrived and it works really really well.

That is good to hear. 

> However, hostapd works just fine with country set to JP and 'iw list' shows all expected 6Ghz channels available so it should be fine. 

That is a very good sign. There are some settings specific for band 4 (6 GHz) and if it is not clear from my 2 examples. come back and post you have in your hostapd.conf and tell me what you are trying to do. Maybe we can help.

> Quickly done a speedtest using my FTTH at home and I can easily achieve ~730Mbps download/upload in wifi 6 5Ghz

I see around 730-750 Mbps with iperf3 and WiFi 6 5GHz. This is an 80 MHz channel width adapter so that is about the maximum we should expect which is really fast. I have a PCIe card in one system that uses the mt7922 chip but it is capable of 160 MHz channel width and with WiFi 6 5 GHz I get over 1 Ghps. A lot of different from back in the WiFi 4 days.

> Thank you for keeping this repo updated!

You are welcome but the thanks goes to everyone that shares information here. I am basically the secretary that keeps things more or less in order and I get help with that. We don't always get it right but as time passes and more information is known, we correct things to ensure the best information is available. The Plug and Play List is a living document that changes regularly.

I look for to hearing from you when your AP is going in band 4.

jbosboom, 2024-12-10:
I have the EDUP EP-AX1672 and am located in the US.  The regdb does prevent this device from initiating radiation in the 6 GHz band, though no justification is provided and I cannot find one in the regulations.  Even if the no-IR were lifted, the kernel does not support power limitations varying with AP vs client and channel width, so only the minimum power would be available.

The relevant regdb commit is https://git.kernel.org/pub/scm/linux/kernel/git/wens/wireless-regdb.git/commit/?id=888da5f48030e45ce14f43a2bf7be07ec79b60d4 (GitHub mirror at https://github.com/beagleboard/mirror-wireless-regdb/commit/888da5f48030e45ce14f43a2bf7be07ec79b60d4).  The commit ends with "An IR (Initiate Radiation) restriction is mentioned in the 6 GHz specification.", but I don't see it either in the rulemaking document linked from the regdb or in the current text of 47 CFR 15.  But I am not a lawyer, so maybe it's in there somewhere.  As far as I know, the commit's description of kernel limitations regarding power limits is still accurate.

With respect to this particular device, an additional regulatory problem is that low-power/indoor access points "must employ a permanently attached integrated antenna" (47 CFR 15.407(a)(10)).  The provided antennas are unlikely to provide enough gain to be a practical problem complying with the power limits, and I suppose you could use epoxy to permanently attach them.

(I purchased this to be as a 2.4/5 access point to replace a Raspberry Pi 3B for improved signal quality (actual antennas vs just traces on the RPi board) and to free up the RPi for other uses, so these limitations don't bother me personally.)

morrownr, 2024-12-10:
@jbosboom 

Thanks for the post. I am wondering if posting a message to the current regdb maintainer as well as to linux-wireless would help clarify why we are seeing `no IR` on the 6 GHz band in the US. It is not clear to me either.

However, I am currently working in a private repo to bring the Realtek rtl8852cu vendor driver to life in a public repo. It supports 6 GHz and I have it working in 6 GHz in AP mode. The range is very disappointing. Given the higher frequency and lower txpwr limits, range is not good at all. With the same adapter in 5 GHz mode, the range and ability to penetrate walls is much better so I get much faster speeds. 

With that said, given the regdb that we have, it is not clear to me why the Realtek driver is allowing me to IR with their driver in the US. I am investigating.