morrownr/USB-WiFi #487: effect of 802.11ac vs. 802.11n on AWUS036ACHM lifespan

bdantas, 2024-08-14:
Hi, Nick. I use a GNU/Linux laptop as a wireless router in my home. hostapd creates the AP using an AWUS036ACHM, which is a nice adapter with very long range as you know.

hostapd was successfully configured to create a 802.11ac AP. Sadly, the adapter stopped working completely after slightly over a year being used this way. So I got a new adapter and "downgraded" the hostapd configuration to create a 802.11n AP.

Do you think it is probable that operating in 802.11ac AP mode is more taxing on a USB adapter's circuitry (and lifespan) than operating in 802.11n AP mode? Or is it more likely that it makes no difference and the device that stopped working was just a bad apple? I would value your thoughts on this--it will sway me to either stick to 802.11n or go back to operating at 802.11ac.

morrownr, 2024-08-14:
Hi @bdantas 

> Do you think it is probable that operating in 802.11ac AP mode is more taxing on a USB adapter's circuitry (and lifespan) than operating in 802.11n AP mode?

I really don't think this is case but I am not an EE. I have used my achm in very demanding use cases over the last few years and most of that was in 5 GHz with a flow of about 200 Mbps and it still works fine. 

What I have noticed over the years is that electronics can be hurt over the long term by less than stable power or even killed by lightning or a static discharge. One would think that this adapter being powered from a usb port might be protected somewhat and it may be but I have used ups and surge protectors with what appears to be success.

One of the reasons that I think the adapters based on the Mediatek chips seem to last well in heavier use cases is that the chips use less power. I have a little usb power meter and have test the power use of many adapters. The mt7610u, mt7612u and mt7921au use less power than their Realtek equivalents and, therefore, there is less heat involved which usually leads to longer life. Unfortunately there can be manufacturing problems that lead to DOA devices or devices that work but do not last. Generally speaking, if electronics are going to die, they will die in the first few months of operation.

I do not recall any others having problems with their achm. You might email the story to Alfa to see what they have to say.

@morrownr

morrownr, 2024-08-14:
No problem. Good luck and keep us posted.

I'm going to reactivate this issue if you don't mind as a lot of people just stop by to read and this could be helpful to some.

bdantas, 2024-08-16:
When using a USB adapter in AP mode at 5 GHz, can you think of any disadvantage in creating an AC access point rather than sticking with N?
 
I hope I don't sound like a Luddite. It's just that "there's no such thing as a free lunch" has proven to be a very reliable rule of thumb.

bdantas, 2024-09-07:
I guess nobody can think of disadvantages in using AC over N. Works for me :)

closed 2024-09-07 by bdantas