morrownr/USB-WiFi #488: smallest usb adapter with 802.11ac support and in-kernel driver?

bdantas, 2024-08-15:
Hello again. My household is GNU+Linux only and everybody has laptops, some of which only support 802.11n. I'd like to get USB adapters for those laptops that are lacking 802.11ac support. Trouble is, some family members do not want bulky USB adapters sticking out of their laptops all the time.

Of the adapters you know, what is the smallest one with 802.11ac support and in-kernel driver? For the "customers" I serve, form factor is king and performance is not a primary consideration.

Any guidance would be much appreciated.

bdantas, 2024-08-15:
Here are four candidates. Your thoughts? I just need something in the nano form factor that would work without hassle other than installing firmware. Oldest OS among the machines I support runs linux kernel version 6.1.0.
 
TP-Link: https://tinyurl.com/yrnkespn
D-Link: https://tinyurl.com/y6x7tb9h
Cudy: https://tinyurl.com/2p83zdrh
Edimax: https://tinyurl.com/y8d8h74n

bjlockie, 2024-08-15:
Have you considered upgrading the wifi cards in the laptops?
They might be easily upgradable.
A few screws and an m2 catd maybe.
I hate putting on the antenna leads thought but if you have 3 hands and a magnifying glass then you should be fine.

bdantas, 2024-08-15:
Hi bjlockie. You're right, I think upgrading the wifi cards is the way to go now that I've seen the lay of the land and all but excluded an easier (USB adapter) solution. Time to roll up my sleeves.

morrownr, 2024-08-15:
@bjlockie might have a good idea:

> Have you considered upgrading the wifi cards in the laptops?

You would need to do the research because you will need to know exactly what card slot your laptops have.

For what it is worth, here is a thread about m.2 cards and the new mt7925 chip. I got one and run it will Ubuntu 24.04. It is WiFi 7. It is plug and play with kernel 6.7 and later as far the wifi is concerned. The bt is not currently working but I checked and the new vid/pid is scheduled to go into kernel 6.12. Here is the thread:

https://github.com/morrownr/USB-WiFi/issues/431

Now back to your question:

I don't have pictures for all adapters in the Plug and Play List so it is easy to miss some. Let me see if I can help:

These are near nano and are 100% plug and play with the mt7610u driver:

https://www.amazon.com/Asus-Dualband-Wirel-AC600-USB-AC51/dp/B00T3DK154
https://www.amazon.com/Panda-Wireless-PAU0A-AC600-Adapter/dp/B07C9TYDR4

From your list:

I just cannot recommend anything by TP-Link and D-Link given their bad track record of doing things that make it hard on Linux users.

I recommend against the Cudy adapter because it is based on the rtl8821/11cu chip. It is supported in rtw88 since kernel 6.2 but only recently is that driver for this chip becoming usable. My history of maintaining an out-of-kernel driver for that chip and my work with the team that is improving rtw88 cause me to pause. That chip is a little problematic so I will probably never have a section for it in the Plug and Play List.

The Edimax adapter is probably the best of the ones in your list. It is based on the rtl8812bu chip which is a good chip and with recent improvements, the rtw88 driver is not bad. I'd say when the final patches are sent in for kernel 6.12, it will have a good in-kernel driver and in the meantime I do have a pretty good out-of-kernel driver for it here.

I have probably created more questions so ask away.

morrownr, 2024-08-15:
@bdantas 

> Time to roll up my sleeves.

Yes, if you go the direction of the little internal cards, you need to do research. The card will need to be for the slot your laptop has AND some laptops have been known to limit what cards can be used.

I mentioned cards with the mt7925 chip but I have a laptop with a M.2 ngff card that uses the mt7921 chip and it is rock solid. It is dual band but get the mt7921k card if you want tri-band. So:

M.2 card:
mt7925 (wifi 7 tri-band)
mt7921k (wifi 6 tri-band)
mt7921 (wifi 6 dual band)

Cheers

morrownr, 2024-08-15:
Speaking of the mt7925, I replaced the card in a new Mini PC I bought a few months ago. The new card connects to my wifi 6 dual band router and shows iperf3 readings of over 1 Gbps. It is in wifi 6, not 7 because my router can't do wifi and it is on 5 GHz because my router is only dual band but my router can do 160 MHz channel width so I can only imagine what the speed will be with a good Wifi 7 router.

Wild future ahead: There is also a new mt7927 chip that is also shown to support usb and is already available in m.2 card form and it does 320 MHz channel width. The driver for the mt7927 is NOT in the kernel yet so use cause going that direction for now.

bjlockie, 2024-08-15:
I forgot to mention I changed the nvme drive on my laptop and I am considering the wifi card (it is Intel and I like Mediatek :-)).
It was easy but some laptops are difficult.
Do a web search for videos and comments.
If you happen to have an Acer laptop, there is a forum specific to them.
Other manufacturers might have similar forums.

bdantas, 2024-08-15:
The laptops we use are all quite old (10+ years). I'm talking ThinkPad X230 and Sony Vaio E series. Therefore, m.2 form factor is not an option.
 
The form factor our laptops use is the Half Height MiniPCIe. I've put an Intel wireless card supporting 802.11ac in my ThinkPad X230 before and had trouble scanning for available APs (every other scan resulted in no usable APs because reported signal strengths were spuriously low), so I am slightly hesitant to put newfangled wireless cards in these old laptops again. Another hassle is that some of the Sony Vaios we own have the dreaded "wifi whitelist", which requires a lot of work to bypass (I have to painstakingly tape electrical tape over one of the wireless card's metal contacts). 

All things considered (form factor, risk, ease of setup, manufacturer shenanigans) I would be most comfortable giving the nano USB adapter from Edimax a try. I've compiled morrownr's out-of-kernel driver 88x2bu driver before and had no troubles. I'll report back.

bjlockie, 2024-08-15:
Are there any nano adapters by Realtek that might have future in kernel drivers.

morrownr, 2024-08-15:
> The form factor our laptops use is the Half Height MiniPCIe.

I'm pretty sure there are cards like that with the mt7612u chip available.

> Are there any nano adapters by Realtek that might have future in kernel drivers.

That Edimax in his list is one of them. The rtw88 rtw88_8822bu in-kernel driver has been in place since kernel 6.2 but there were some things like usb3 that weren't supported and performance was far less than my driver here but those problems are rapidly going away and should mostly be solved in kernel 6.12. For managed mode, it should work well.

morrownr, 2024-08-16:
@bdantas 

> I searched the web for "Edimax EW-7822ULC chipset" and could not find a definitive statement anywhere regarding the chipset in this device. Can you teach how you determined that?

Sure. I never use ads to determine the chip as ads are notorious for being wrong. I never call or email sellers as they are notorious for having support staff that won't even know what you are asking about. What I do is find the driver and see what chip it is for.

https://www.edimax.com/edimax/download/download/data/edimax/global/download/

home > wireless adapters > EW-7822ULC

In this case, since Edimax does post a Linux driver, I downloaded the Linux driver and extract it. I open the Makefile and see the following line:

CONFIG_RTL8822B = y

That tells me the chip is either a rtl8822bu or rtl8812bu but since no bluetooth is advertised, I know the chip is a rtl8812bu.

If no Linux driver is available, I will download the Windows driver and look around in the file with a hex editor. Sometimes there is documentation with the drivers that helps... sometimes not. I generally confirm by searching the net. I've been doing this while helping users here for some years so I know a lot of little things to look for so I can usually figure out the chip very fast.

> So does that mean that the nano Edimax adapter in my link above (the Edimax EW-7822ULC) will work well with linux 6.12 without having to install anything?

Probably so. A small team and I started working on the in-kernel rtw88 usb support late last year. While usb support for rtw88 had been in the kernel since 6.2, there were problems and missing features. Since that time, many patches have been sent into linux-wireless to improve things. Last week saw the biggest project so far in that new drivers for the popular rtl8812au and rtl8821/11au chips went in and should be in kernel 6.12 when it is ready. Right now 6.11 is in work so it will be a while before 6.12 is stable and in distros. While working on the various issues, many things have been improved for the rtl8812bu chip. I am testing an adapter with a rtl8812bu chip with the latest rtw_8822bu driver and it is looking good except for a performance issue but hopefully that can be fixed soon. This work is really making Realtek WiFi 5 usb viable but the key is that  Realtek is not doing this work. Mediatek does its work on usb support and users provide a patch here and there but things are really different and Mediatek does a far better job of supporting usb wireless. If you look around this site, you will see several places where I warn Linux users to stay away from usb wifi adapters that use the Realtek wifi 6 chips as that will likely lead to many years of pure aggrevation that is unnecessary because we have well supported wifi 6 chips by Realtek. 

> If so, where will the proprietary firmware be coming from?

On the Main Menu is a menu item that explains this:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

Firmware is part of the distro and as such is the responsibility of the distro. However, users can install or upgrade the firmware files if they wish.

Hopefully this helps.

bdantas, 2024-08-16:
Hi @morrownr
 
> Sure. I never use ads to determine the chip as ads are notorious for being wrong. I never call or email sellers as they are notorious for having support staff that won't even know what you are asking about. What I do is find the driver and see what chip it is for.
 
I see. That does seem like a reliable approach and not one I would have thought of. Thanks for sharing. It's a pity that this information is so difficult to obtain. Not even the vendor's [datasheet](https://www.edimax.com/edimax/mw/cufiles/files/download/datasheet/EW-7822ULC_Datasheet_EN.pdf) lists the device's chipset!

> Firmware is part of the distro and as such is the responsibility of the distro.
 
That's what I thought. I was just a bit confused because the 88x2bu driver available on your github page includes the firmware. So I guess when this same driver is distributed as an in-line kernel module, then the firmware is not included and it is the responsibility of the distro or user to obtain it.

bdantas, 2024-08-18:
I received the Edimax EW-7822ULC arrived, I compiled and installed the 88x2bu driver, and got the adapter to work. I also upgraded the hostapd settings in my router to run at 802.11ac.
 
Unfortunately, where I usually spend time in my house (around -60 dBm signal strenth) I get slower download speed (per speedtest.net) with this adapter than I do with my built-in wireless N card. Here are the speeds I'm getting, all in the same spot of the house, using the same laptop:

- internal half-height mini PCIe card (Atheros with AR9382 chipset): 80 Mbps
- Edimax EW-7822ULC (rtl8812bu): 50 Mbps
- Linksys AE6000 (mt7610u chipset): 130 Mbps
 
It seems I have not yet found a nano 802.11ac adapter that I can use or recommend :(

morrownr, 2024-08-18:
> Unfortunately, where I usually spend time in my house (around -60 dBm signal strenth)...

Oh, geez, I should have pinged you about the limited range of nano adapters. Sorry about that. I have never seen a true nano that has much range at all. The Linksys adapter, I have one, and it does pretty good but is not a true nano adapter.

If you need range and don't want something that may break off in a laptop, look at the Alfa ACHM. It comes with an extension cable that you would need to unplug if you get up and move but if range is an issue and an external adapter won't get the job done then the only options I see are moving the wifi router/AP to a better location or getting a better wifi router/AP.

bdantas, 2024-08-18:
> Oh, geez, I should have pinged you about the limited range of nano adapters. Sorry about that.
 
No worries. I learned a lot of new things, so the experience was an overall win.
 
I'm determined to upgrade my home to AC, so will press on even though I'm ditching the nano adapter idea. I'd like to ask for a recommendation that's off-topic in this thread, so I'll start a new one.