morrownr/USB-WiFi #490: recommendation for reliable half-height mini PCIe card with in-kernel driver and 802.11ac support?

bdantas, 2024-08-18:
This is a continuation of my quest to upgrade my home to Wi-Fi 5.
 
My wireless router is a repurposed laptop that creates a 802.11ac AP using a ALFA AWUS036ACHM USB adapter (MediaTek mt7610u chipset). The AP works like a charm but signal strength is only -60 dBm in some parts of the house.
 
My family members' laptops run GNU/Linux and use half-height mini PCIe cards than only support b/g/n. I'd like to upgrade all the machines to ac. Upgrading using a USB adapter is not going to be possible because folks here don't want anything bigger than nano, which performs worse than the b/g/n PCIe cards in many parts of the house.

It seems my options at this point are to move the AP, upgrade the AP, or upgrade the half-height mini PCIe cards. I would like to try the latter. I know your forte is USB, but I'm sure a lot of the knowledge crosses over.
 
Can you recommend a half-height mini PCIe card that performs well, supports Wi-Fi 5, and uses an in-kernel driver? I understand I'll have to fetch external, proprietary firmware and that's okay.

bdantas, 2024-08-18:
I'm considering [this one](https://www.aliexpress.com/i/3256804173168831.html?gatewayAdapt=4itemAdapt)
 
According to [this source](https://techinfodepot.shoutwiki.com/wiki/Mt76), the mt7612en chipset is supported by the in-kernel mt76 module.
 
The only missing piece before I purchase one to try is figuring out where to get the firmware. It is not at all obvious. Do you think one of the .bin files in [this list](http://repo.tinycorelinux.net/15.x/x86_64/tcz/firmware-mediatek.tcz.list) would work? The closest item in the list is "mt7610e.bin"

morrownr, 2024-08-18:
@bdantas 

Mediatek firmware is kept at the following official site: (this information is also on the Main Menu - Firmware)

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek

I'm pretty sure the files you would need are:

mt7662.bin
mt7662_rom_patch.bin

However, check to see if those files are already in your distro. I would be surprised if they are not.

Something to consider: If range is an issue, usb wifi is not optimized for range. Many people, including me, use usb wifi adapters for AP's but there are limitations. A usb wifi adapter, and the one you have is really good, cannot compete with a dedicated wifi router as they are optimized for AP mode.

I'd suggest a combination of the 7612e card you are looking at least for one laptop to test and also consider getting something like the following:

Cudy WR3000 v1

https://www.amazon.com/Cudy-802-11ax-Internet-Beamforming-WR3000/dp/B0BRK3CYY3

I always check to see what OpenWRT is supporting and this router is supported by OpenWRT. I run mine with OpenWRT but the factory firmware is adequate for most uses. You can always move to OpenWRT when the maker stops support. OpenWRT was working with Banana Pi to ship a new wifi router based on the same chipset as this one but I don't know if it is shipping yet. I think it is called OpenWRT One. Anyway, my point is that dedicated wifi routers can be had for low prices and if placed in the proper location, can dramatically improve range.

bdantas, 2024-08-19:
@morrownr 
 
> I'm pretty sure the files you would need are:
> 
> mt7662.bin mt7662_rom_patch.bin
> 
> However, check to see if those files are already in your distro. I would be surprised if they are not.
 
Thank you for this information. Yes, my distro provides a _firmware-mediatek_ package that contains these files. Now that I know this card's firmware will probably be easy to obtain, I went ahead and bought one of these to try. I'll report back when I finally get my hands on this card.
 
> Anyway, my point is that dedicated wifi routers can be had for low prices and if placed in the proper location, can dramatically improve range.
 
Thanks for the pointers. The idea of using a dedicated wifi router is one that I revisit often--I would certainly get better range and speed that way. However, I would miss the flexibility that comes from running a general purpose GNU+Linux distro (Tiny Core Linux x86_64 in my case) in my router. I'm a tinkerer and like my homebrew :)

Thanks for all your generous help.

bdantas, 2024-08-23:
I found another option: [Qualcomm Atheros QCA6174](https://www.amazon.com/dp/B0B15M3XGY)
 
Can anyone comment on relative merits of Mediatek mt7612en chipset (mt76 driver) vs. Atheros QCA6174 chipset (ath10k driver)?

bdantas, 2024-08-31:
One final update: I went with the mt7612en card. It's working like a charm. Reliable scanning. Typical download speed on my X230 ThinkPad increased from 80 Mbps (with my old AR9382) to 150 Mbps with this new card.

I can confirm that the firmware files are 

mt7662.bin
mt7662_rom_patch.bin

P.S. The only wrinkle I experienced is that the driver initially could not find mt7662_rom_patch.bin because my distro packages the file at /usr/local/lib/firmware/mediatek/mt7662_rom_patch.bin while the mt76x2e driver is hardwired to look for the file at /lib/firmware/mediatek/mt7662_rom_patch.bin. Putting the file in the expected location and rebooting fixes the issue.

closed 2024-08-31 by bdantas

morrownr, 2024-08-31:
Glad to hear this is working for you.

> I experienced is that the driver initially could not find mt7662_rom_patch.bin because my distro packages the file at /usr/local/lib/firmware/mediatek/mt7662_rom_patch.bin....

You need to report this to your distro and point out what needs to be changed. Your distro is putting the file in the wrong place.

Cheers

bdantas, 2024-08-31:
> You need to report this to your distro and point out what needs to be changed. Your distro is putting the file in the wrong place.
 
Yes, I'll make sure to do that.
 
Thanks again for all your help. I'm really enjoying finally having 802.11ac in my home :)