morrownr/USB-WiFi #491: New rtl8812/21 drivers backported to OpenWrt for testing

mj22226, 2024-08-25:
I had backported to OpenWrt for testing the recently rtl8812/21au patches  submitted to the linux kernel
https://github.com/morrownr/USB-WiFi/issues/484
Successfully tested 2 rtl8821au USB adapters D-Link DWA-171 A1 and Alfa AWUS036ACS, STA and AP are working, performance is average 240Mbps download 200Mbps upload.

```
root@OpenWrt:~# dmesg | grep rtw_8821au
[   12.155672] usbcore: registered new interface driver rtw_8821au
[  539.772363] rtw_8821au 1-1:1.0: Firmware version 42.4.0, H2C version 0
[  695.692112] rtw_8821au 1-1:1.0 phy0-ap0: entered allmulticast mode
[  695.693275] rtw_8821au 1-1:1.0 phy0-ap0: entered promiscuous mode
root@OpenWrt:~# iwinfo
phy0-ap0  ESSID: "AWUS036ACS"
          Access Point: 00:C0:CA:A4:00:00
          Mode: Master  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 20 dBm  Link Quality: 70/70
          Signal: -40 dBm  Noise: unknown
          Bit Rate: 433.3 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/b/g/n
          Hardware: USB 0BDA:0811 [Realtek RTL8821AU]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: no  PHY name: phy0
```

The rt8812 USB adapter (ID 20f4:805b) I've tested doesn't work yet, getting this error;

```
[   12.861706] rtw_8812au 5-1:1.0: Firmware version 52.14.0, H2C version 0
[   13.880270] rtw_8812au 5-1:1.0: failed to dump efuse logical map
[   13.886885] rtw_8812au 5-1:1.0: failed to setup chip efuse info
[   13.887419] rtw_8812au 5-1:1.0: failed to setup chip information
[   13.888250] rtw_8812au: probe of 5-1:1.0 failed with error -22
[   13.888914] usbcore: registered new interface driver rtw_8812au
```
I not sure if is a specific device issue or other devices are also affected.

patches : https://github.com/mj22226/openwrt/commits/rtw88-8821au/
or
[01-rtl8812-21.zip](https://github.com/user-attachments/files/16741595/01-rtl8812-21.zip)

morrownr, 2024-08-25:
Hi @mj22226 

Thanks for the report. Let me add @dubhater to this thread so that he sees the 8812au report.

Something to note: There are additional patches that are going in that supplement the primary 20 patches that Bitterblue Smith sent in. At least one is by Ping-ke, the Realtek wireless gatekeeper. Some help from Realtek was needed. There will be a V2 PATCH series as soon as it is ready. A lot of work is in progress.

dubhater, 2024-08-26:
Try this:

```diff
diff --git a/drivers/net/wireless/realtek/rtw88/rtw8821a.c b/drivers/net/wireless/realtek/rtw88/rtw8821a.c
index 0ff02ab27866..91068df8be4a 100644
--- a/drivers/net/wireless/realtek/rtw88/rtw8821a.c
+++ b/drivers/net/wireless/realtek/rtw88/rtw8821a.c
@@ -4023,7 +3996,7 @@ const struct rtw_chip_info rtw8812a_hw_spec = {
 	.rx_buf_desc_sz = 8,
 	.phy_efuse_size = 512,
 	.log_efuse_size = 512,
-	.ptct_efuse_size = 96 + 1, /* TODO or just 18? */
+	.ptct_efuse_size = 18,
 	.txff_size = 131072,
 	.rxff_size = 16128,
 	.rsvd_drv_pg_num = 9,
```

What is the host hardware?

mj22226, 2024-08-26:
> Try this:
>
> ```diff
> diff --git a/drivers/net/wireless/realtek/rtw88/rtw8821a.c b/drivers/net/wireless/realtek/rtw88/rtw8821a.c
> index 0ff02ab27866..91068df8be4a 100644
> --- [...]
@dubhater  I've tried but it didn't work, same error.

> What is the host hardware?

I've tested on different hardware x86, rockchip, rpi, mediatek, same error on all.

dubhater, 2024-08-26:
Can you reproduce it with a regular desktop distro?

mj22226, 2024-08-27:
> Can you reproduce it with a regular desktop distro?

on Ubuntu
```
[  864.770925] rtw_8812au 3-1.1:1.0: Firmware version 52.14.0, H2C version 0
[  865.141076] rtw_8812au 3-1.1:1.0: failed to dump efuse logical map
[  865.143500] rtw_8812au 3-1.1:1.0: failed to setup chip efuse info
[  865.143513] rtw_8812au 3-1.1:1.0: failed to setup chip information
[  865.143675] rtw_8812au: probe of 3-1.1:1.0 failed with error -22
[  865.143783] usbcore: registered new interface driver rtw_8812au
[  932.591838] rtw_8821au 3-1.1:1.0: Firmware version 42.4.0, H2C version 0
[  933.059932] rtw_8821au 3-1.1:1.0 wlx00c0caa4650a: renamed from wlan0

```
@dubhater are other realtek version of the rtl8812au firmware ?

dubhater, 2024-08-27:
52.14.0 is the latest. You can find older versions in older drivers, but I don't think they can help with this problem. At the time when "failed to dump efuse logical map" is printed, the firmware has not been used yet.

Have you tried this repository? https://github.com/lwfinger/rtw88/

mj22226, 2024-08-28:
> Have you tried this repository? https://github.com/lwfinger/rtw88/

Same error, definitely is a hardware issue with this specific USB adapter TRENDnet TEW-805UB V1.0R, I had seen similar issues with realtek USB ethernet adapters that do not work with a specific firmware version so this may be one of those cases.

dubhater, 2024-08-28:
To rule out firmware incompatibilities, you can use
* the patch above, or
* this driver, which uses the same firmware but is probably reading the efuse correctly: https://github.com/morrownr/8812au-20210820/

aarush67, 2025-03-28:
is the image with the drivers preinstalled made or not yet I am new I need this image