morrownr/USB-WiFi #492: Android devices have trouble reauthenticating with 802.11ac AP

bdantas, 2024-08-25:
I recently upgraded my AP's hostapd configuration from 802.11n to 802.11ac. Everything is working except for Android phones' ability to authenticate with the AP reliably. This issue was not present when AP was 802.11n.
 
The Android phones authenticate fine after a fresh boot. If we walk away from the house then return, roughly 60% of the time the phone is able to successfully authenticate. Other 40% of the time it tries to authenticate but fails silently (in WiFi section of settings, under SSID the phone says "Connecting..." then "Saved"). hostapd log does not seem to show anything related to the authentication attempt. The only fix is to reboot the phone. After reboot, phone is able always able to authenticate.

My router is an x86_64 GNU/Linux machine running hostapd 2.11. AP is created using an Alfa AWUS036ACHM usb adapter. Here is the problem-free 802.11n hostapd.conf I was using:

```
ssid=sweet_home
interface=wlan1
driver=nl80211
country_code=US
channel=44
ignore_broadcast_ssid=0
hw_mode=a
auth_algs=1
wpa=2
wpa_passphrase=8PasSwoRD
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
rsn_pairwise=CCMP

# N
ieee80211n=1
wmm_enabled=1
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40]
```
 
And here is the hostapd.conf I'm using now, which leads to Android devices sporadically being unable to authenticate:

```
ssid=sweet_home
interface=wlan1
driver=nl80211
country_code=US
channel=36
ignore_broadcast_ssid=0
hw_mode=a
auth_algs=1
wpa=2
wpa_passphrase=8PasSwoRD
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
rsn_pairwise=CCMP

# N
ieee80211n=1
wmm_enabled=1
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40]

# AC
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42 # should be channel+6
ieee80211ac=1
vht_capab=[SHORT-GI-80]
```

Does the config file need to be tweaked? Maybe the Android phones are caching something when they successfully authenticate with the 802.11ac AP that is fouling things up when the phones tries to authenticate again later?

bdantas, 2024-08-26:
I did more troubleshooting. "Forgetting" the SSID and re-entering the wifi password does not help. Toggling "Airplane Mode" on and off doesn't help. When the phone is unable to authenticate, only rebooting the phone helps. This affects both of the Android phones I've tested.

bdantas, 2024-08-26:
I edited these phones' wpa_supplicant settings (at /data/vendor/wifi/wpa/wpa_supplicant), changing `fast_reauth=1` to `fast_reauth=0` then rebooting the phones. Could just be a fluke, but I have not had another failed authentication after limited testing.

morrownr, 2024-09-03:
@bdantas 

Have you figured anything out about this?

I did see one problem:

> vht_oper_centr_freq_seg0_idx=42 # should be channel+6

Don't put the comment on the same line as the setting. Doing this used to be okay but not anymore. Also, what are you seeing in the hostapd log file?

bdantas, 2024-09-03:
No, I was not able to solve this. I will remove that comment from hostapd.conf, thanks for the tip. 
 
hostapd log file did not show anything I thought was helpful (although I may have missed something). Android phones' wpa_supplicant logs show only this--over and over--on the rare occasions when a phone cannot reauthenticate:

```
08-28 21:18:04.969  3153  3153 I wpa_supplicant: wlan0: Trying to associate with SSID 'bibliotheca_alexandrina'
08-28 21:18:08.074  3153  3153 I wpa_supplicant: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:c0:ca:b1:09:0c status_code=1
08-28 21:18:11.282  3153  3153 I wpa_supplicant: wlan0: Trying to associate with SSID 'bibliotheca_alexandrina'
08-28 21:18:14.383  3153  3153 I wpa_supplicant: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:c0:ca:b1:09:0c status_code=1
```
 
BTW I also have a 2.4 GHz 802.11n access point for my 2.4 GHz-only devices (e.g., network printer). The Android devices can authenticate and re-authenticate with this AP reliably. I'm happy with my workaround of using only this AP with our phones.

The only thing I'll be wondering about is whether you think this newbie mistake (comment on same line as setting)...
```
vht_oper_centr_freq_seg0_idx=42 # should be channel+6
```
...could have been responsible for select clients (Android phones) sometimes failing to authenticate.

bdantas, 2025-09-04:
> I've used Android phones myself and over the years they can dislike some wifi.
 
Thank you for sharing that. I'm glad I'm not the only one seeing this.

> I think the problem is at least partly because Android is SO far behind the rest of the Linux world with what kernel it is using.
 
You can say that again! I'm too embarrassed to say how old the linux kernel is in these phones, definitely >5 years old. But there's not much I can do about it (see my postscript below).

> You have it working so trying one thing at a time over time and we can probably figure it out.
 
Indeed. As soon as I'm convinced that I have a baseline of stability (I'm almost there), I'll start testing one small change at a time.

P.S. I'd upgrade the kernel in the phones but the various components of an Android device are so tightly integrated that I'm sure I'd screw something up. Another option would be to upgrade the phones' Android OS to a newer version, but such a project would require orders of magnitude more time than I can spare right now. Truth be told, if I had the requisite time to upgrade our Android version, I would rather use that time to bail out of Android altogether and switch to GNU/Linux phones (e.g., Librem 5 or PinePhone Pro). But this is just a pipe dream.