morrownr/USB-WiFi #494: New Realtek usb-wifi device RTL8912AU Wifi7

Kiwimannen, 2024-08-30:
Realtek RTL8912AU - only Windows drivers are available, providing some info from Linux and Windows

Linux:

```
lsusb -v
Bus 003 Device 004: ID 0bda:8912 Realtek Semiconductor Corp. 802.11be WLAN Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8912 
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11be WLAN Adapter
  iSerial                 3 XXXXXXXXXXX
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0051
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           9
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 802.11be WLAN Adapter
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
[... 175 lines cut ...]
    Bus Level Reset                             : Not Supported
    MAC Randomization                           : Not Supported
    Fast Transition                             : Supported
    MU-MIMO                                     : Supported
    Miracast Sink                               : Supported
    BSS Transition (802.11v)                    : Supported
    IHV Extensibility Module Configured         : Supported
    SAE Authentication                          : Supported
    SAE Hash-to-Element Authentication          : Supported
    WPA3 SUITE-B Authentication                 : Supported
    WPA3 SUITE-B FIPS Mode                      : Not Supported
    OWE Authentication                          : Supported
    FTM as Initiator                            : Not Supported
    MBO Support                                 : Supported
    Number of Tx Spatial Streams                : 2
    Number of Rx Spatial Streams                : 2
    Number of Concurrent Channels Supported     : 1
    P2P GO ports count                          : 1
    P2P Clients Port Count                      : 1
    P2P Max Mobile AP Clients                   : 8
    Max ANQP Service Advertisements Supported   : 0
    Maximum Number of Secondary STAs supported  : 0
    Number of MultiLink Connections supported   : 0
    Co-existence Support                        : Wi-Fi performance is maintained
```

morrownr, 2024-08-30:
@Kiwimannen 

Thanks for the post. I think the warnings to Linux users about Realtek WiFi 6 USB adapters that I continue to post apply to Realtek WiFi 7 also. 

Realtek historically has maintained 2 separate groups of wifi developers for Linux. The team working on internal chips works with linux-wireless to maintain and add Linux Standards Compliant drivers for the internal cards that are available. You can see their work in rtw88 and rtw89. On the other hand, the Realtek USB wifi group does not and never has worked with linux-wireless and only provides external drivers that are not fully standards compliant. Additionally, these external drivers can be hard to find, they are certainly not maintained at a centrally available repo. Then you throw in the fact that they are not regularly maintained and Realtek does not take bug reports from end users and you have a situation that does not end well for most Linux users. I was hoping that we would see better support from Realtek for WiFi 7 and maybe will happen. I have not seen any evidence of it so far.

alexmi256, 2024-09-01:
Good to see this here as I was hoping to use it on linux which doesn't seem possible at the moment.

morrownr, 2024-09-01:
> ...which doesn't seem possible at the moment.

As far as I know, it does not seem possible at the moment. Somebody will like post here if more info becomes available.

I am already using a WiFi 7 capable M.2 card. It is based on the Mediatek mt7925 chip. I installed it in my mini-pc a couple of months ago and then installed Ubuntu 24.04. The card just worked. The driver for the mt7925 chip was added to kernel 6.7 and Ubuntu 24.04 uses kernel 6.8. I tested with Windows 11 and it works well there also. I'm not a Windows user but my mini does have a Win11 partition so I can test if need be.

We should mt7925 based usb wifi adapters sometime this year. The support is already in kernel 6.7 and later.

frans-panken, 2024-09-18:
Are you also surprised that 802.11be is not among the radio types supported?

morrownr, 2024-09-18:
> I think the windows drivers are a work in progress like most drivers...

I was looking at the capabilities repo earlier in this thread. It appears the Windows driver is indeed a work in progress as I only see managed and P2P modes as supported. Monitor and AP modes are not supported and those modes are very popular with Linux users.

I still have not run into any information about a Linux driver for this Realtek chip. What I do know is that the Linux kernel has code in it to refuse to run WiFi 7 drivers that depend on the very old Wireless Extensions support. That means Realtek can't continue with their out-of-kernel driver model. They have to do it right.

I'm not so sure that I would agree that other WiFi 7 drivers are a work in progress. I have a little M.2 card that uses the mt7925 chip and it supports managed, AP, monitor, P2P and modes. I run it on a system with Ubuntu 24.04 and it worked immediately after I installed it and I have seen no problems at all in managed mode. I don't have a WiFi 7 capable wifi router yet but it does over 1 Gbps with my WiFi 6 dual band router.

What we need is for an adapter maker to release an adapter with the mt7925 chip. The chip is available for USB.

Kiwimannen, 2024-09-18:
I have the mt7925 M.2 too, but personally I would love to see a new "mesh point" capable usb-adapter, like the old MT7612U

(or if Mediatek could add it to existing MT7921AU driver - the EDUP AX3000M with MT7921AU is great)

morrownr, 2024-09-18:
@Kiwimannen 

>  I would love to see a new "mesh point" capable usb-adapter, like the old MT7612U...

Then I would make this known. See link below:

https://wireless.wiki.kernel.org/en/users/drivers/mediatek

I would use the email addresses that end with mediatek.com. A nice explanation of what you want along with a note that if this is not the right place for a feature request, can they provide the correct contact information?

Kiwimannen, 2024-09-18:
> I would use the email addresses that end with mediatek.com. A nice explanation of what you want along with a note that if this is not the right place for a feature request, can they provide the correct contact information?

Will do, thanks! :)

lukejenkins, 2024-09-18:
EDUP has a USB adapter based on the RTL8912AU posted to their page. Nothing interesting other than the driver download link seems to be a .1 of the .0 windows driver @Kiwimannen mentioned 3 weeks ago.

No I'm not brave enough to install it.

Kiwimannen, 2024-10-04:
Its interesting to see that in Windows, RTL8912AU share the same drivers with 8922:

https://oemdrivers.com/network-realtek-5002-24-117-1

And there is already Linux drivers for 8922 in RTW89.

morrownr, 2024-10-04:
Hi @Kiwimannen 

> Its interesting to see that in Windows, RTL8912AU share the same drivers with 8922:

This follows Realtek's naming scheme. Example: rtl8822b driver supports rtl8822b and rtl8812b chips. The 22 will have bluetooth support and the 12 will not.

> And there is already Linux drivers for 8922 in RTW89.

Yes, that is correct but rtw89 has zero USB support. It currently only supports internal cards. Realtek never added USB support to rtw88, it was added by a community member and improvements continue to come from additional community members. Realtek's USB support has been and continues to be abysmal. Adding USB support to rtw89 is something that Realtek needs to do but I have seen no evidence of it happening. This is why I have placed warning messages in several places on this site for Linux users to AVOID Realtek WiFi 6 and 7 USB WiFi adapters.  One of the warning messages is pinned, recommend that you read it.

Kiwimannen, 2024-10-04:
Yes, i have been looking for usb-support in rtw89 - but nothing yet. I knew when I got the RTL8912AU that it probably wouldnt get linux-support at all - im just using it to test wifi-speeds from OpenWRT in Windows at the moment.
I have read your warnings and have several other of your recommended usb-wifi  that work really good in linux :)

morrownr, 2024-10-04:
@Kiwimannen 

I do not know what Realtek's plans are for USB WiFi 7 support. What I do know is that if they come up with something, it cannot be like the previous out-of-kernel drivers that are not standards compliant. The kernel has code in it to detect if the drivers are not standards compliant starting with WiFi 7 so Realtek has no choice but to get it together. They have known this was coming for years.

If you can find an out-of-kernel driver for your adapter, I'd like to see it.

Kiwimannen, 2024-10-05:
I frequently look around for updated drivers and if i happen to stumble on something interesting i will be sure to let you know :)

duc5800, 2024-11-03:
> > Are you also surprised that 802.11be is not among the radio types supported?
> 
> I think the windows drivers are a work in progress like most drivers, there are windows drivers for the CF-983BE available at the Comfast website, but the newest drivers i have found are for a EDUP model - EP-BE1703 - with the same chip:
> 
> https://www.szedup.com/content/137
> 
> All models with the same chip can use available drivers.

I've just got an EDUP EP-BE1703S myself and when it's connected to my Wi-Fi 7 router Windows 11 24H2 shows "Protocol: Wi-Fi 6 (802.11ax)" in the Properties screen.
I can understand the "work in progress" thing, but if the main selling point of a product doesn't even work yet (Wi-Fi 7 in this case), then that's quite harsh, I think.

morrownr, 2024-11-03:
@duc5800 

> I can understand the "work in progress" thing, but if the main selling point of a product doesn't even work yet (Wi-Fi 7 in this case), then that's quite harsh, I think.

I am of the opinion that USB WiFi group at Realtek has an exceptionally poor driver testing program for Windows and Linux. This statement probably surprises nobody but I felt like saying it anyway.

qemu-gvm, 2024-11-08:
> I've just got an EDUP EP-BE1703S myself and when it's connected to my Wi-Fi 7 router Windows 11 24H2 shows "Protocol: Wi-Fi 6 (802.11ax)" in the Properties screen. I can understand the "work in [...]

I run into the same situation. I got a TP-LINK TBE400UH. Driver version is 5002.24.117.3. The Windows 24H2 shows "Protocol Wifi6".

duc5800, 2025-03-02:
Finally someone has come forward with a proper driver for Windows 11. ASUS now offers a driver version 5102.24.126.0 for download that appears to provide proper support for Wi-Fi 7, including MLO. It can be downloaded via the product webpage of their ROG USB-BE92.

Since ASUS decided to remove the hardware IDs of all competing products from the driver, it needs to be installed via the "Have Disk ..." method for other RTL8912AU devices. So far this driver works fine with my adapter made by EDUP.

You might have noticed that the "Aggregated link speed" for MLO is being shown wrongly in my screenshot. I think this must be related to my Wi-Fi 7 router, because I see the same with non-Realtek Wi-Fi 7 cards.

[image]

Kiwimannen, 2025-03-03:
Thanks, works on my Fenvi rtl8912au adapter too. (remember that you have to have Windows 11 24H2 to get real Wifi7 support in Windows)

Unfortunately, still no sightings of a linux driver for this chip - and for some strange reason it does not work in Windows on arm processor - actually, there are no usb-wifi drivers available for any adapter at all for Windows on arm.

No sightings of a MT7925U (usb) adapter either, so sadly no wifi7 on usb at all at the moment.

morrownr, 2025-03-03:
> Unfortunately, still no sightings of a linux driver for this chip...

I have no information about a Linux driver for the rtl8912au. What I do know is that the Linux kernel has had code in it for years that will stop any driver or app that uses the old Wireless Extensions (WEXT) support. I saw a case last week where a user with a mt7925 based M.2 card was trying to use Aircrack-ng. He wanted to know why some of the Aircrack-ng apps would not work. It was because those apps were using WEXT and were being stopped by the kernel. We are going to see this in other old apps. The apps need to be reworked to work with the modern stack.

What does this have to do with Realtek's Linux support for the rtl8912au? Realtek's USB WiFi support has continued to be based on out-of-kernel drivers that still depend to one degree of another on WEXT. Realtek has known for YEARS that WiFi 7 meant the end of their use of WEXT but who knows what they have or have not done for WiFi 7. Realtek's support for its internal cards, PCIe and M.2, has made great progress over the last few years because that Realtek WiFi team works with linux-wireless and does things the way they are supposed to be done. Working with linux-wireless not helps provide drivers that are standards compliant but it also gives us visibility and someone to communicate with when we have problems. On the other hand, the Realtek WiFi team that does USB support does not work with linux-wireless and who knows what they are doing. I will try to see if I can find any information.

> No sightings of a MT7925U (usb) adapter either, so sadly no wifi7 on usb for linux at all at the moment.

I had hoped that we would see USB adapters based on the mt7925 before now. It is normal that the chips for internal cards ship before USB chips but we have had M.2 and PCIe cards since the middle of last year. I am getting some indications that the USB chips are still not shipping. There is interest in bringing mt7925 based USB WiFi adapters to market and we would have product if the chips were shipping. Maybe it will happen soon. It is frustrating to see a driver in the Linux kernel for over one year and not have the hardware available.

morrownr, 2025-04-15:
> too bad there is no sightings of any linux driver.

This is a situation that calls for some humor:

Mediatek's WiFi 7 mt7925u chip has had a driver in the kernel for over a year but no adapters are available to buy yet. (I know of one in external testing). Then there is Realtek where adapters are available and no driver is available. I use a M.2 mt7925e card so I know the driver is good.

I can only speculate about why the Realtek rtl8912au driver is late. Several years ago the kernel devs includes code in mainline to stop any WEXT code from executing if the wireless is WiFi 7. All evidence points to this being added to push Realtek into modern times. Realtek divides their wireless driver support into 2 parts: USB and chips for cards. Over the last few years, the team that does the drivers for the cards has been great. They are working with linux-wireless and the drivers for the cards are in the kernel and are pretty good. On the other hand, the USB team seems to be lost. Their out-of-kernel WiFi 6 drivers still depend on WEXT and are just not what regular users need. The USB team may be moving over to making standards (mac80211) compliant drivers for WiFi 7 but who knows. Realtek really needs to get it together when it comes to Linux USB support. I've had to post a pinned message to users:

https://github.com/morrownr/USB-WiFi/issues/449

We do have properly supported WiFi 6 adapters and modules based on the Mediatek mt7921 chip but it would be nice if Realtek would get it together so that we have a some good competition going.

MR0000001, 2025-04-19:
Hi guys, i was thinking about getting one of those RTL8912AU devices at very cheap price on aliexpress, but this 0 support on linux is unsetting.

[image]

morrownr, 2025-04-19:
@MR0000001 

> but this 0 support on linux is unsetting.

Unfortunately, getting no information from Realtek about their USB support is normal. Realtek's Linux support for their chips for cards is really up to speed these days but USB is not. I try to keep with all things USB as best that I can and I do not see any evidence of Linux support for the rtl8912au... nothing, nada, zilch. We'll have WiFi 7 adapters based on the mt7925 soon.

MR0000001, 2025-04-19:
> [@MR0000001](https://github.com/MR0000001)
>
>
> Unfortunately, getting no information from Realtek about their USB support is normal. Realtek's Linux support for their chips for cards is really up to [...]

i hope the mt7925 devices will be cheap too.

morrownr, 2025-04-19:
> i hope the mt7925 devices will be cheap too.

I think that what we see with mt7921au adapters is probably pretty close to what we will see with the mt7925 adapters. There will be high, medium and low priced adapters. We'll just have to test and collect information so that the community will have the information to make decisions. I'd say by mid-year we will see adapters available and more available by the end of the year. It appears that I was off by a few months regarding when Mediatek was finally able to ship the USB chips. They are shipping now, at least to the tier one makers.

Let's hope that the makers continue what we see with the mt7921au adapters in that almost all of the adapters are single-state (no windows driver onboard), single-function (no bluetooth), use the LED as power on/off indicator, and use the Mediatek default device ID. Basically, the adapters should say "Made for Linux". I've seen a lot of details that lead me to think that makers are paying attention to what works well with Linux... because we are telling them what we want at this site.

OscarAkaElvis, 2025-04-21:
Hi, I'm on the same boat as the OP. I bought (just for testing) the cheap Wifi 7 Fenvi BE6500 adapter (chipset Realtek RTL8912AU) from aliexpress (https://s.click.aliexpress.com/e/_ooILbML) and it works fine on W10 and W11, but no linux support yet.

Does anybody know if is a VIF capable chipset? you know, if it has the ability to split into two logical interfaces (very useful to perform evil twin attacks to do AP and DoS in monitor mode at the same time). I usually check the output of `sudo iw list | grep "Supported interface modes" -A 8` checking if "AP/VLAN" is appearing, but it is not working as there is no driver installed, so no output. Is there any other way to check this without having a driver installed or how to check this on Windows?

Let's cross fingers to have a Linux driver soon 🤞 . Thanks.

morrownr, 2025-04-21:
Hi @OscarAkaElvis 

> Does anybody know if is a VIF capable chipset?

If VIF support is important to you, then you really need to lean toward buying USB WiFi adapters that use Mediatek chips as the Mediatek drivers are Linux Wireless Standards (mac80211) compliant. The Mediatek based adapters and their drivers support VIF in a standards compliant manner. The Linux kernel already has a driver for the mt7925 (WiFi 7) USB chip. We are simply waiting on product to be available and that should happen soon as product is in pre production testing at this point.

If you look at one of the Realtek USB WiFi 6 drivers that I have up, https://github.com/morrownr/rtl8852bu-20240418 , you will see that Realtek is still supporting `concurrent mode` which is a sort of VIF capability but does not work with most software as it does not work in a standards compliant way. Realtek has been using that `concurrent mode` for years and all it does is cause Linux users to have problems.

If you buy product that is not what you want then you are sending the wrong signal to the makers and sellers. We have plenty of USB WiFi adapters that work very well with Linux these days and the options are growing. The Plug and Play List here at this site makes it easy to know what some of the best of those adapters are.

Regards

OscarAkaElvis, 2025-04-21:
Thanks for the advice @morrownr . I'm one of the authors of the [airgeddon](https://github.com/v1s1t0r1sh3r3/airgeddon) tool. I have dozens of Wi-Fi adapters and I know the ones that are worth (VIF capable) and others that are not good (mostly Realtek and others) 😺 , or at least not good for Wi-Fi hacking. Actually, I buy sometimes "random" adapters to test and extend the compatibility of the tool and to complete the whitelist and the blacklist of the adapters we have that you can check in our wiki: https://github.com/v1s1t0r1sh3r3/airgeddon/wiki/Cards%20and%20Chipsets

I bought this just for testing, to be sure before adding it to the blacklist. However, for now, I think I'll just omit it, since I can't confirm whether it's VIF-capable due to the lack of a driver for testing.

Thanks for your work and interest.

morrownr, 2025-04-21:
@OscarAkaElvis 

I knew Elvis was still alive! 

> I have dozens of Wi-Fi adapters and I know the ones that are worth (VIF capable) and others that are not good (mostly Realtek and others) 

I have a lot of adapters also. It kind of goes along with having the USB-WiFi github site. Something that might be of interest to you and would certainly help driver quality for the kernel:

In late 2023, Larry Finger (deceased in May 2024), Dubhater and myself started working on a project to improve the USB support in rtw88 which is for Realtek WiFi 5 chips. We are still at it but rtw88 been improved dramatically, we, mostly dubhater's coding, also added support for the popular rtl8821/11au, rtl8812au and rtl8814au chips. Those drivers have all gone into mainline at this point. Many people helped test but I am not sure if VIF received adequate testing. You don't have to use kernel 6.14 or 6.15 to test as you can install from the repo where the work is done:

https://github.com/lwfinger/rtw88

You are invited to test and report on VIF over at that repo. It is needed. I'm sure you have some adapters with those chips.

Regards

OscarAkaElvis, 2025-04-21:
Yeah, sure! I have some adapters using RTL8812AU, RTL8812BU and RTL8811AU and chipsets (that's for sure, maybe more RTL, not sure now). It will be a pleasure to test them using your driver. I can add it to our recommended list of drivers' repositories for these kind of adapters if the results are nice (I'm sure they will).

I will test all of them using the old 6.1 kernel as the present bug in modern kernels (>=6.2) regarding VIF is still present. I reported it on kernel bugtracker (https://bugzilla.kernel.org/show_bug.cgi?id=218763) . At the beginning I thought the bug was affecting only to Mediatek chipsets, but sadly, after a while we saw is affecting to all chipsets.

Anyway, despite the driver needs to support VIF, it is also a hardware limitation. And as far as I know, these chipsets are not VIF-capable (RTL8812AU, RTL8812BU and RTL8811AU). But I'll test anyway everything to see how they work also for normal injection.

I have also an adapter using RTL8188GU chipset but as your repo starts as 88 and not 81, I think is not relevant. I think I can't test this adapter with your driver.

I don't want to deviate from the topic of the thread. If you want, we can discuss this matter in private,my email is public. It's in my profile.

Cheers!

OscarAkaElvis, 2025-04-23:
Ok @morrownr , I've tested the drivers of this repo: https://github.com/lwfinger/rtw88

I must admit I'm impressed. Awesome work! 🚀 . These are the results of my tests (tested on kernel 6.1 as newer kernels have the VIF problem as stated before in my other post):

RTL8812AU -> it worked like a charm. Even the card is working flawlessly in VIF mode. Excelent!
RTL8812BU -> not worked. The card is detected on `lsusb` but is not appearing using `iw` command
RTL8811AU -> it worked very well. Same as RTL8812AU, VIF was also working well.

A proof of it. I've highlighted in red the chipset and the AP/VLAN which is the feature appearing when an adapter is VIF capable. But not only because of that I say this, I also tested it thoroughly and yeah... VIF is working fine:

[image]

I lost faith totally in RTL chipsets time ago, but now with this repo I feel like they are not trash and can be used again for Wi-Fi hacking. What to say but thank you and keep up the hard work. It's impressive. 👍

morrownr, 2025-04-23:
@OscarAkaElvis 

Thank you for testing.

> RTL8812BU -> not worked. The card is detected on lsusb but is not appearing using iw command

Can I get you to open an Issue over at :

https://github.com/lwfinger/rtw88

We need to investigate why this is not working.

> I also tested it thoroughly and yeah... VIF is working fine...

That is good to hear.

> no wireless extensions...

These drivers are fully Linux Wireless Standards (mac80211) compliant. They are in the Linux kernel as of kernel 6.14 for 8812au and 8821/11au and kernel 6.15 for 8814au. Duhater is the hero. He did most of the coding while I did a little coding and many admin things.

> I lost faith totally in RTL chipsets time ago,

I certainly recommend the Mediatek chips over Realtek due to Realtek's poor USB support but Realtek's chips are not the problem. The problem is their POOR USB support. Their driver support for cards is compliant these days and is working well. It is the Realtek team that does USB support that is the problem. I have multiple post warning Linux users about Realtek WiFi 6 and 7 USB adapters. Here is a direct link to where I spell it out:

https://github.com/morrownr/USB-WiFi/issues/449

If you could help spread the word, that would be appreciated.

I am interested in getting more information about the problem with VIF so I am going to include my email below. After this message is sent, I will edit and delete my email address. If you could contact me directly, I would appreciate that,

morrownr@gmail.com

morrownr, 2025-04-25:
@OscarAkaElvis 

I had time this morning while I have some hardcore testing going on with an AXML adapter on a consulting job. 

I reread some of the above discussion. I think we should take a look at the VIF problem. I think we can get this solved by working together. Getting kernel wireless devs to work on things can be a challenge these days unless you...well, hopefully I can help in that regard. How is the best way for me to reproduce the bug?

Actually, before you answer, could you start a new issue here in USB-WiFi to work this issue?

OscarAkaElvis, 2025-04-27:
yeah, I'll start another thread for this. I'll explain there how I reproduce the problem, chipsets affected and kernel versions affected. Thanks.

morrownr, 2025-04-30:
> RTL8812BU -> not worked. The card is detected on lsusb but is not appearing using iw command

What is the device ID of that rtl8812bu adapter:

$ lsusb

Let me check the source and make sure the device ID is in the source.

OscarAkaElvis, 2025-05-01:
@morrownr 

[image]

`Bus 003 Device 002: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]`

Hope it helps

morrownr, 2025-05-01:
@OscarAkaElvis 

Thanks for the info. That id is in the driver so that is not the issue. We can figure this out.

At this point, I'm going to move to the new issue regarding VIF.

duc5800, 2025-05-05:
Is that the answer you got from an AI chatbot?

Anyway, the information is not correct. The Realtek chipset for which support was added in Linux kernel 6.14 is the RTL8812AU, an older 2x2 Wi-Fi 5 chipset.

MR0000001, 2025-05-06:
Google AI, for sure, try to reply following the wish in the question with fake info....total garbage.
Dont trust AI in what you not already know but just forgot.

dubhater, 2025-05-08:
> ```
> 
> > The Realtek RTL8912AU is a chipset likely used in some WiFi 7 USB adapters,
> > such as the  [0]"COMFAST BE6500".
> 
> > Linux Driver Status:
> > Drivers for the RTL8912AU are reported to be included in the Linux kernel as of version 6.14.
> > This is significant as it suggests out-of-the-box support for newer Linux distributions.
> ```
> 
> [0] https://es.aliexpress.com/item/1005007791562123.html
> 
> Is this information correct?

Yes, it is correct, but it wasn't kernel 6.14, it was 6.13, and the driver is not for RTL8912AU but for RTL8192SU, and support wasn't added, it was removed. 😎

OscarAkaElvis, 2025-05-08:
> Thanks for the info. That id is in the driver so that is not the issue. We can figure this out.
> At this point, I'm going to move to the new issue regarding VIF.

@morrownr , I've tested my RTL8812BU adapter on another clean VM installing https://github.com/lwfinger/rtw88 drivers and the same result, the adapter is showin in lsusb command but is not appearing in iw or iwconfig command outputs. For some reason the drivers are not working. Let me know if there is a thread for that, or if we should open a new one, or how I can help you on this. Cheers.

ilikenwf, 2025-05-31:
I ordered a qualcomm atheros ath12k card I located...I still have high hopes for this one, but the driver's just not there yet.

dubhater, 2025-05-31:
@OscarAkaElvis Open a new issue at the rtw88 repository, please.

Kiwimannen, 2025-08-09:
Linux drivers for RTL8912AU are now available for testing at:

https://github.com/morrownr/rtw89

bjlockie, 2025-08-09:
Since the USB support was done by the community and is in the kernel hypothetically if none of the original people can maintain it for future kernels what is the timeline for it to be removed from future kernels?

Kiwimannen, 2025-08-09:
> Since the USB support was done by the community 

I'm not part of the development but I think some of the developers of rtw89 do have direct contact with representatives at realtek.

morrownr, 2025-08-09:
> what is the timeline for it to be removed from future kernels?

There is no timeline for the new driver to be removed. Below is a paste from the MAINTAINERS list:

```
REALTEK WIRELESS DRIVER (rtw88)
M:	Ping-Ke Shih <pkshih@realtek.com>
L:	linux-wireless@vger.kernel.org
S:	Maintained
T:	git https://github.com/pkshih/rtw.git
F:	drivers/net/wireless/realtek/rtw88/

REALTEK WIRELESS DRIVER (rtw89)
M:	Ping-Ke Shih <pkshih@realtek.com>
L:	linux-wireless@vger.kernel.org
S:	Maintained
T:	git https://github.com/pkshih/rtw.git
F:	drivers/net/wireless/realtek/rtw89/
```

Ping-Ke is the head Realtek wireless dev. Once he accepts the new drivers into the kernel, Realtek has ultimate responsibility. That is not to say that current devs cannot continue to work on the new drivers but that work does have to go through Ping-Ke.

Some usb bug reports on the rtw88 series have been reported and Realtek fixed them.

My best estimate right now is that by the time the the next LTS of Ubuntu, 26.04, is available, the Linux will have good drivers for all popular WiFi 5, 6 and 7 USB adapters. That will put Linux ahead of Win 11 as far as USB WiFi is concerned when looking at quality and overall functionality.

The last 7 years have totally change the USB WiFi situation on Linux and we have some really special people to thank for that.

OscarAkaElvis, 2025-08-20:
Another thing I noticed is that the Wifi7 adapter is not shown as wifi adapter in the `iwconfig` output. I know that `iwconfig` is deprecated, but it is weird to be shown like this:

[image]

Despite this output, it is working as I explained a couple of posts before. Just mentioning it.

morrownr, 2025-08-20:
@OscarAkaElvis 

> But what I really think it is important would be to make the new drivers to work with kernel 6.1...

That is not going to happen. I was working on a project that would be enhanced if I could backport mt76 from 6.7+ to 6.1 and I ran into some issues that made it a bridge too far for me. From kernel 6.2 to 6.7, many wireless API changes entered the kernel. They were necessary for added capability but they make it hard to backport from kernels 6.7+ back to 6.1-.

> this is absolutely necessary for wifi hacking.

It might be a good idea to speak of this in terms of functionality without using the word hacking or the like. 

> Some of us are still stuck using kernel 6.1...

I am aware of this. I have had a very difficult last few months due to family health issues so I have had little time to look at much. The bugzilla post shows the frustration. While the bisect pointed to a patch for the mt7921 driver, that actually does not pinpoint the specific patch that is the problem in this case as it could be a patch to a WiFi API. This appears to be supported by your testing with Ralink and Atheros drivers.

This problem is complicated and probably is not a driver problem. What do we know:

The problem started after kernel 6.1 and before the patch in the bisect. We could go to the wireless Patchworks site and go back to the bisect patch and work our way backwards until we see a suspect... then revert it and see what happens. That would need to be repeated until we have something solid. With a specific patch that is ultimately the cause we will know the author and we can report the problem to the person that caused it via linux-wireless.

Does that make sense? This will not be easy and it will likely never be fixed until we find the exact cause and who made the patch that caused it.

dubhater, 2025-08-20:
@morrownr I think it's enough to remind Johannes Berg about this problem when he returns from vacation. He can probably find the bug just from the commits which appear to break [mt7921](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers?id=41ac53c899bd1493a75ab9b52c8f76176b7419e1) and [rtw88](https://github.com/torvalds/linux/commit/0a44dfc070749514b804ccac0b1fd38718f7daa1).

Someone already reported it here: https://lore.kernel.org/linux-wireless/1935176921.584803.1751146076922@mail.yahoo.com/

OscarAkaElvis, 2025-08-20:
Weird... I tested it on kernels 6.2, 6.3, etc and it is also failing. On that link is said that the problem begins in 6.9.

Who knows... Maybe it's related.

OscarAkaElvis, 2025-09-10:
Hi @morrownr @dubhater , have you pinged Johannes Berg? Any updates?

morrownr, 2025-09-10:
I'm working on this. I am making contact with the OP of the linux-wireless bug report first. I am also going to start a new ISSUE here so as to keep things simple and provide a single point for all involved.

https://github.com/morrownr/USB-WiFi/issues/652

Please move the discussion on this Frame injection problem to the NEW issue. See you there.

hood177, 2025-11-01:
New WifiAutoInstall Setup 5102.24.126.6
https://files2.wavlink.com/drivers/driver/WAVLINK_695BE2_690BE2_5102.24.126.6.zip

hood177, 2026-01-01:
> Uninstall the drivers and reinstall.
> If that doesn't work, contact comfast to see if they can get later Windows drivers.

All of this has already been done and did not help. Support sends the same drivers that are stored in the internal flash memory of the adapters. Even installing previous builds of Windows 11 and Windows 10 from backups, in which these adapters were originally installed and worked perfectly for a long time, did not help. A similar situation occurred with the Wavlink WL-WN693X2F adapter, which stopped working in Win11 after more than a year of service, and now it only works in Win 11 with the driver from the Comfast CF-979AX (RTL8832CU) adapter, while in Linux it also works perfectly with drivers from morrownr. It seems that the adapters received some kind of silent update that disabled network detection in Windows. 
It would be great if someone could compare the drivers and help answer the question about the reasons for this phenomenon.

Kiwimannen, 2026-01-01:
Hood177 - try Asus Rog USB-BE92 drivers in Windows 11.

Install the drivers then you have to manually select the adapter and "force" Windows to use the Asus driver by choosing Asus from a vendor list.

Have been working perfect for me with a Comfast and Fenvi RTL8912AU adapter - even with Wifi7 MLO.

morrownr, 2026-01-09:
@hood177 

> So, if after purchasing a non-Chinese router, you lose all your networks, don't rush to throw away the adapter that refuses to connect out of anger, as I did ...

Making decisions while angry may not work out well. Though I understand the reasons that wifi can frustrate. Having wifi laws and regulations that differ greatly depending on your country can make coding for a router or for drivers and operating systems a challenge. The old style drivers from Realtek that I support here still default to pulling the regulatory information from the firmware on the adapter but that is not how it should be done so I have to change some code. The wireless driver should get regulatory information from the OS and the OS should get the regulatory information from the AP/Router and the AP/Router should get the regulatory information from the ISP. This system can break down if you have any old kit in any of those places or if poor programming has taken place. Generally there are ways to get things going without throwing hardware in the trash. Things are getting better.

dubhater, 2026-04-02:
The kernel is about to get support for RTL8912AU. Can someone test it with MLO, please?

You can compile just the driver from here: https://github.com/morrownr/rtw89

Or the kernel from the [rtw-next branch](https://github.com/pkshih/rtw) + [patches 02..12](https://patchwork.kernel.org/project/linux-wireless/list/?series=1073085&state=*).

dubhater, 2026-04-05:
@duc5800 By the way, what kernel version do you have there?

duc5800, 2026-04-08:
@dubhater  Ok. I uninstalled the rtw89 driver per the instructions given on the GitHub page, deleted the rtw89 folder and then installed the driver again from the GitHub. I didn't notice any difference. To me everything looks the same as before.

Below I also added a screenshot that shows the same output for a more established Intel BE200 card, for comparison.

What I cannot tell is if the MLO mode of the RTL8912AU (and the BE200, for that matter) actually provides anything useful, apart from "listening" on a second frequency band. (btw, the Wi-Fi router I have only supports MLO on two bands) 

[image]
BE200
[image]

[8 more replies]