morrownr/USB-WiFi #495: AWUS036AXM not a single-function dongle

Andy-45, 2024-08-31:
"Note: All USB WiFi adapters listed here are single-state (no Windows driver inside), single-function (no Bluetooth support)"

FYI  it has Bluetooth  -> AWUS036AXM is the WiFi 6/6E (802.11ax) 2x2 Tri-band 2.4 GHz / 5 GHz / 6 GHz and Bluetooth v5.2 high-performance USB adapter

kind regards

morrownr, 2024-08-31:
Hi @Andy-45 

Have you connected and used a bluetooth device with it?

Andy-45, 2024-09-08:
got the AWUS036AXM now. it has Wifi and Bluetooth, but same as with the AWUS036AXML ( https://github.com/morrownr/USB-WiFi/issues/378#issuecomment-2129877505 ) the BT part only works with Windows. 
All my linux boxes ( x86 and some Raspberry ) back to Kernel 5.1 get an Error hciX: Opcode 0x0c03 failed 

I don't need BT, but maybe it should be pointed out in the  "USB WiFi adapters that are supported with Linux in-kernel drivers" list

thank you for your work, morrownr !

Andy-45, 2024-10-18:
about the BT problem, I stumbled across this information, maybe it will be fixed soon:

https://lore.kernel.org/all/?q=MT7921AUN

morrownr, 2024-10-18:
@Andy-45 

Thank you so much for passing this on. I was aware that it had been reported but I only monitor linux-wireless so am not seeing anything about BT. I had suspected from early on that this was a BT problem. I need to read that thread in more detail. I certainly hope that this can be resolved ASAP.