morrownr/USB-WiFi #497: [Help]: AWUS036AXML kernel crash Raspberry PI OS / Debian promiscuous mode

mstekker, 2024-09-01:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux rpi0 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

### lsusb

Bus 001 Device 004: ID 0e8d:7961 MediaTek Inc. Wireless_Device Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

### rfkill

ID TYPE      DEVICE      SOFT      HARD  1 bluetooth hci0   unblocked unblocked  2 wlan      phy0   unblocked unblocked

### dkms

```text
bash: dkms: command not found
```

### iw

```text
phy#0
	Interface mon0
		ifindex 4
		wdev 0x2
		addr 00:c0:ca:b6:32:3a
		type monitor
		channel 5 (2432 MHz), width: 20 MHz (no HT), center1: 2432 MHz
		txpower 3.00 dBm
```

### What happened?

Hi,

When switching wifi channels with a mt7921u based wifi usb dongle (Alpha and others tested) the kernel crashes. "iw" is used to switch channels. The time between crashes is random. This happens on RPI 3b+ and RPI 4 devices. OS is fully up to date using a good PSU. Switching also to 6GHz channels (6GHz is supported in this dongles).

Details:

PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Linux rpi0 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

Aug 29 10:31:17 rpi0 kernel: x26: 0000000000000001 x25: ffffff800b89e000 x24: ffffff800b89e24c
Aug 29 10:31:17 rpi0 kernel: x29: ffffffc080dfbc80 x28: ffffff80172cbb00 x27: ffffff80172cbb00
Aug 29 10:31:17 rpi0 kernel: sp : ffffffc080dfbc80
Aug 29 10:31:17 rpi0 kernel: lr : mt7921_mac_tx_free+0x60/0x2d8 [mt7921_common]
Aug 29 10:31:17 rpi0 kernel: pc : 0x0
Aug 29 10:31:17 rpi0 kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Aug 29 10:31:17 rpi0 kernel: Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
Aug 29 10:31:21 rpi0 kernel: CPU: 2 PID: 19563 Comm: kworker/u8:1 Tainted: G D WC 6.6.31+rpt-rpi-v8 #1 Debian 1:6.6.31-1+rpt1
Aug 29 10:31:21 rpi0 kernel: Modules linked in: bnep mt7921u mt7921_common mt792x_lib mt76_connac_lib mt792x_usb mt76_usb mt76 mac80211 libarc4 cfg80211 btusb btrtl btintel btbcm bluetooth vc4 snd_soc_hdmi_codec i2c_mux ecdh_generic ecc drm_display_helper rfkill cec libaes drm_dma_helper drm_kms_helper binfmt_misc snd_soc_core bcm2835_isp(C) snd_compress snd_pcm_dmaengine bcm2835_codec(C) bcm2835_v4l2(C) raspberrypi_hwmon bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videodev snd_bcm2835(C) snd_pcm videobuf2_common snd_timer mc snd vc_sm_cma(C) raspberrypi_gpiomem uio_pdrv_genirq uio i2c_dev drm fuse dm_mod drm_panel_orientation_quirks backlight ip_tables x_tables ipv6 rtc_ds1307 regmap_i2c i2c_bcm2835
Aug 29 10:31:17 rpi0 kernel: Internal error: Oops: 0000000086000005 [#1] PREEMPT SMP
Aug 29 10:31:17 rpi0 kernel: [0000000000000000] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
Aug 29 10:31:17 rpi0 kernel: user pgtable: 4k pages, 39-bit VAs, pgdp=000000000d078000
Aug 29 10:31:17 rpi0 kernel: FSC = 0x05: level 1 translation fault
Aug 29 10:31:17 rpi0 kernel: EA = 0, S1PTW = 0
Aug 29 10:31:17 rpi0 kernel: SET = 0, FnV = 0
Aug 29 10:31:17 rpi0 kernel: EC = 0x21: IABT (current EL), IL = 32 bits
Aug 29 10:31:17 rpi0 kernel: ESR = 0x0000000086000005
Aug 29 10:31:17 rpi0 kernel: Mem abort info:
Aug 29 10:31:17 rpi0 kernel: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
Aug 29 10:31:17 rpi0 kernel: mt7921u 1-1.5.1:1.3 mon0: entered promiscuous mode
Aug 29 10:31:17 rpi0 kernel: mt7921u 1-1.5.1:1.3 mon0: left promiscuous mode
Aug 29 10:31:17 rpi0 kernel: mt7921u 1-1.5.1:1.3 mon0: entered promiscuous mode
Aug 29 10:31:17 rpi0 kernel: mt7921u 1-1.5.1:1.3 mon0: left promiscuous mode
Aug 29 10:31:17 rpi0 kernel: x26: 0000000000000001 x25: ffffff800b89e000 x24: ffffff800b89e24c
Aug 29 10:31:17 rpi0 kernel: x29: ffffffc080dfbc80 x28: ffffff80172cbb00 x27: ffffff80172cbb00
Aug 29 10:31:17 rpi0 kernel: sp : ffffffc080dfbc80
Aug 29 10:31:17 rpi0 kernel: lr : mt7921_mac_tx_free+0x60/0x2d8 [mt7921_common]
Aug 29 10:31:17 rpi0 kernel: pc : 0x0
Aug 29 10:31:17 rpi0 kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Aug 29 10:31:17 rpi0 kernel: Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
Aug 29 10:31:17 rpi0 kernel: CPU: 2 PID: 19563 Comm: kworker/u8:1 Tainted: G D WC 6.6.31+rpt-rpi-v8 #1 Debian 1:6.6.31-1+rpt1
Aug 29 10:31:17 rpi0 kernel: Modules linked in: bnep mt7921u mt7921_common mt792x_lib mt76_connac_lib mt792x_usb mt76_usb mt76 mac80211 libarc4 cfg80211 btusb btrtl btintel btbcm bluetooth vc4 snd_soc_hdmi_codec i2c_mux ecdh_generic ecc drm_display_helper rfkill cec libaes drm_dma_helper drm_kms_helper binfmt_misc snd_soc_core bcm2835_isp(C) snd_compress snd_pcm_dmaengine bcm2835_codec(C) bcm2835_v4l2(C) raspberrypi_hwmon bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videodev snd_bcm2835(C) snd_pcm videobuf2_common snd_timer mc snd vc_sm_cma(C) raspberrypi_gpiomem uio_pdrv_genirq uio i2c_dev drm fuse dm_mod drm_panel_orientation_quirks backlight ip_tables x_tables ipv6 rtc_ds1307 regmap_i2c i2c_bcm2835
Aug 29 10:31:17 rpi0 kernel: Internal error: Oops: 0000000086000005 [#1] PREEMPT SMP
Aug 29 10:31:17 rpi0 kernel: [0000000000000000] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
Aug 29 10:31:17 rpi0 kernel: user pgtable: 4k pages, 39-bit VAs, pgdp=000000000d078000
Aug 29 10:31:17 rpi0 kernel: FSC = 0x05: level 1 translation fault
Aug 29 10:31:17 rpi0 kernel: EA = 0, S1PTW = 0
Aug 29 10:31:17 rpi0 kernel: SET = 0, FnV = 0
Aug 29 10:31:17 rpi0 kernel: EC = 0x21: IABT (current EL), IL = 32 bits
Aug 29 10:31:17 rpi0 kernel: ESR = 0x0000000086000005
Aug 29 10:31:17 rpi0 kernel: Mem abort info:
Aug 29 10:31:17 rpi0 kernel: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000

Any idea's?

Best regards,

Mart

morrownr, 2024-09-01:
Hi @mstekker 

> Aug 29 10:31:17 rpi0 kernel: Unable to handle kernel NULL pointer dereference...

It would help if I had an exact series of commands that would reproduce this problem.

Next: I think I remember some recent patches that could be related:

https://patchwork.kernel.org/project/linux-wireless/list/

The following patch is to update the wifi firmware files for your mt7921au based adapters. It come about 2 months after the last update so maybe there was a sense of urgency. The files should be moved to the repo shown in the Firmware menu item on the Main Menu probably sometime this coming week:

https://patchwork.kernel.org/project/linux-wireless/list/?series=885153

I have also seen some patches going into mac80211 that could be involved and there are some patches related to mt7921 as well. The work to modernize the wifi stack to support WiFi 6 and 7 has led to issues that are slowly being fixed.

I am not seeing this so having the commands to duplicate it would help.

wellitsmez, 2024-11-19:
Quick update make sure you plugin your alfa adapter to usb2 interface but let it be for the vm settings as USB 3 have tried it everything for the past 10 hours didn't get any crash monitor mode and injection working perfect