morrownr/USB-WiFi #498: [Help]: EDUP EP-AX1672 is really unstable on opensuse tumbleweed

musdx, 2024-09-01:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux localhost.localdomain 6.10.5-1-default #1 SMP PREEMPT_DYNAMIC Fri Aug 16 11:15:41 UTC 2024 (30fd964) x86_64 x86_64 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 002: ID 04b3:310c IBM Corp. Wheel Mouse Bus 001 Device 006: ID 0e8d:7961 MediaTek Inc. Wireless_Device Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 005 Device 002: ID 320f:505b Evision RGB Keyboard Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

### rfkill

3: phy4: Wireless LAN 	Soft blocked: no 	Hard blocked: no

### dkms

```text
sudo: dkms: command not found(ig it do not exist on opensuse?)
```

### iw

```text
zsh: command not found: iw(same for this ig)
```

### What happened?

Hi everyone. I bought the EDUP EP-AX1672 recently. For some reason my wifi is really unstable my jitter and latency both is really high. Open website can sometime take minutes. The issues can be fix temporally by unplug the adapter and replug it. But even when i do that it is still super unstable. I try use my phone to tethering the wifi to my pc and the unstable issue didn't happens so i guess it is an issue with the adapter or maybe with my distro(Opensuse tumbleweed). I would appreciate any help thank!

morrownr, 2024-09-01:
Hi @musdx 

> I bought the EDUP EP-AX1672 recently.

I have one of these adapters. Really like it.

> ... my wifi is really unstable my jitter and latency both is really high. Open website can sometime take minutes.

The first thing I keyed on what kernel 6.10.5. I have been seeing lot of problem reports related to kernel 6.10. My first suggestion is to back off to kernel 6.6 LTS or if that is not easily done, try to use kernel 6.9 or earlier. I am not an opensure expert so you will have to figure out how to use an earlier kernel.

Report back when that is done and we can go from there.

FYI: A lot of modernization of the Linux wifi stack has been taking place over the last year to support WiFi 6 and 7 and it is ongoing. It seems that this has finally caught up with us with kernel 6.10. Things will get better but now is not a good time for wifi users to be on the bleeding edge.

musdx, 2024-09-01:
Hi @morrownr  Thank you a lot  for the help. I tried the LTS kernel. The performance is better now(still have latency issue compare to my phone wifi tethering). It is now fine for my web browsing stuffs tho.
[image]

morrownr, 2024-09-02:
Okay, that is improvement. If you decide to work on making it better, stop by again.