morrownr/USB-WiFi #501: [Help]: High latency with Mediatek mt7921 and Model: COMFAST CF-952AX

HuaZ, 2024-09-05:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux raspberrypi 5.19.17-v7l #1 SMP Wed Sep 20 12:06:15 CST 2023 armv7l GNU/Linux

### lsusb

Bus 001 Device 003: ID 3574:6211

### rfkill

0: phy0: Wireless LAN         Soft blocked: no         Hard blocked: no

### dkms

```text
null
```

### iw

```text
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr e0:e1:a9:36:26:df
                type managed
                channel 161 (5805 MHz), width: 80 MHz, center1: 5775 MHz
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
```

### What happened?

I used COMFAST Model:CF-952AX with MT7921u driver on the Raspberry Pi 4B

`ethtool -i wlan0
driver: mt7921u
version: 5.19.17-v7l
firmware-version: ____010000-20240826151030
expansion-rom-version:
bus-info: 1-1.3:1.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
`
`modinfo mt7921u
filename:       /lib/modules/5.19.17-v7l/kernel/drivers/net/wireless/mediatek/mt76/mt7921/mt7921u.ko.xz
license:        Dual BSD/GPL
author:         Lorenzo Bianconi <lorenzo@kernel.org>
firmware:       mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
firmware:       mediatek/WIFI_RAM_CODE_MT7961_1.bin
srcversion:     78A8A76A4A305A2938754ED
alias:          usb:v0E8Dp7961d*dc*dsc*dp*icFFiscFFipFFin*
depends:        mt76-connac-lib,mt76-usb,mt7921-common,mt76
intree:         Y
name:           mt7921u
vermagic:       5.19.17-v7l SMP mod_unload modversions ARMv7 p2v8
`
`sudo dmesg |grep 'mt'
[    7.897372] usbcore: registered new interface driver mt7921u
[    8.306440] mt7921u 1-1.3:1.0: HW/SW Version: 0x8a108a10, Build Time: 20240826150948a
[    8.594406] mt7921u 1-1.3:1.0: WM Firmware Version: ____010000, Build Time: 20240826151030
`

which works, but with really high latencies。 I have also tried some older firmware versions, and the situation is the same.

`pi@raspberrypi:~ $ ping 192.168.50.1
PING 192.168.50.1 (192.168.50.1) 56(84) bytes of data.
64 bytes from 192.168.50.1: icmp_seq=1 ttl=64 time=2.50 ms
64 bytes from 192.168.50.1: icmp_seq=2 ttl=64 time=328 ms
64 bytes from 192.168.50.1: icmp_seq=3 ttl=64 time=2.43 ms
64 bytes from 192.168.50.1: icmp_seq=4 ttl=64 time=375 ms
64 bytes from 192.168.50.1: icmp_seq=5 ttl=64 time=5.05 ms
64 bytes from 192.168.50.1: icmp_seq=6 ttl=64 time=319 ms
64 bytes from 192.168.50.1: icmp_seq=7 ttl=64 time=2.39 ms
64 bytes from 192.168.50.1: icmp_seq=8 ttl=64 time=365 ms
64 bytes from 192.168.50.1: icmp_seq=9 ttl=64 time=2.94 ms
64 bytes from 192.168.50.1: icmp_seq=10 ttl=64 time=311 ms
64 bytes from 192.168.50.1: icmp_seq=11 ttl=64 time=2.05 ms
64 bytes from 192.168.50.1: icmp_seq=12 ttl=64 time=357 ms
`

Any suggestions? thanks a lot.

gdluca, 2024-09-05:
What about a different channel? Is it the same with 40Mhz?

morrownr, 2024-09-05:
Hi @HuaZ 

Some questions:

What numbers are you seeing for latency and how are you testing latency?

You do realize that you are using the most modern version of the firmware with a version of the driver that is basically the very oldest version? 

What RasPi hardware are you using?

I see you are using a 32 bit version of the RasPi OS (Raspian at that time possibly) which may indicate a very old RasPi device. I have seen cases with very modern usb wifi adapters where the RasPi simply could not keep up with the adapter... CPU limitation.

HuaZ, 2024-09-06:
> What about a different channel? Is it the same with 40Mhz?

thanks, "192.168.50.1 is the gateway IP address which base on the Route Netgear RX80.

HuaZ, 2024-09-06:
Hi @morrownr  thanks
> Hi @HuaZ
> 
> Some questions:
> 
> What numbers are you seeing for latency and how are you testing latency?
to ping the gateway ip address  192.168.50.1,  the below is the result , one is 2.5ms, the next is 328 ms. 
I use my laptop(windows11 and intel AX211) to ping the same IP address. the times are <3ms.
```text
pi@raspberrypi:~ $ ping 192.168.50.1
PING 192.168.50.1 (192.168.50.1) 56(84) bytes of data.
64 bytes from 192.168.50.1: icmp_seq=1 ttl=64 time=2.50 ms
64 bytes from 192.168.50.1: icmp_seq=2 ttl=64 time=328 ms
64 bytes from 192.168.50.1: icmp_seq=3 ttl=64 time=2.43 ms
64 bytes from 192.168.50.1: icmp_seq=4 ttl=64 time=375 ms
64 bytes from 192.168.50.1: icmp_seq=5 ttl=64 time=5.05 ms
64 bytes from 192.168.50.1: icmp_seq=6 ttl=64 time=319 ms
64 bytes from 192.168.50.1: icmp_seq=7 ttl=64 time=2.39 ms
64 bytes from 192.168.50.1: icmp_seq=8 ttl=64 time=365 ms
64 bytes from 192.168.50.1: icmp_seq=9 ttl=64 time=2.94 ms
64 bytes from 192.168.50.1: icmp_seq=10 ttl=64 time=311 ms
64 bytes from 192.168.50.1: icmp_seq=11 ttl=64 time=2.05 ms
64 bytes from 192.168.50.1: icmp_seq=12 ttl=64 time=357 ms
```

> 
> You do realize that you are using the most modern version of the firmware with a version of the driver that is basically the very oldest version?

Yes, I followed the solution for a similar issue as described in the following link to upgrade the firmware, but it doesn't seem to have achieved the desired effect.
https://forums.debian.net/viewtopic.php?t=153490
> 
> What RasPi hardware are you using?
RasPI4 model B
> 
> I see you are using a 32 bit version of the RasPi OS (Raspian at that time possibly) which may indicate a very old RasPi device. I have seen cases with very modern usb wifi adapters where the RasPi simply could not keep up with the adapter... CPU limitation.

yes, it is armhf version.
dpkg --print-architecture
armhf

I also tried to disable the power-saving mode of the chip, but it doesn't seem to have any effect.
Are there any other troubleshooting methods you can suggest, or any recommended versions of the driver and firmware?
thanks

HuaZ, 2024-09-19:
Update:
Although the ping time is unstable, there doesn't seem to be any issues with the actual traffic.

morrownr, 2024-09-29:
@HuaZ 

If this becomes a problem, check back in with a report. Keep in mind that various things beyond the control of the driver can cause ping issues.

morrownr, 2024-10-19:
@HuaZ 

I ran across the following quoted from a user of a mt7922 card:

Throughput jitters on MT7922 cards. You might try to disable power management for that card by adding the line `options mt7921e disable_aspm=Y` to the file `/etc/modprobe.d/mt7921.conf`. This setting flattened out the valleys in my throughput
graphs.

Let us know what you find out.

flipreverse, 2024-11-13:
@HuaZ 
I'm experiencing a similar issue on my CF-953AX connected to a Pi4B.
I've just switched from an USB3 port to an USB2 one. Currently, the latencies are stable.

morrownr, 2024-11-13:
@HuaZ 

Some things to try:

Disabling scatter-gather could help for USB3. If you do not know how, let me know. I'm in a hurry right now.

Making sure you are not trying to use more power than the Pi can provide in its usb subsystem. You would need to provide a list of things that are plugged into usb ports if you want an opinion as to whether this could be a problem.

USB3 is not mankind's greatest invention. It can be problematic and RasPi has not helped by using a poor hub chipset and not providing spec power.