morrownr/USB-WiFi #504: [Help]: Netgear A6210 Slow Download Speed

43qcc2cn, 2024-09-10:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux {obscured} 6.6.46 #1 SMP Sun Aug 18 00:39:55 UTC 2024 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 002 Device 002: ID 8087:8000 Intel Corp. Integrated Rate Matching Hub Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub Bus 003 Device 002: ID 0846:9053 NetGear, Inc. A6210

### rfkill

1: phy1: wlan 	Soft blocked: no 	Hard blocked: no

### dkms

```text
-sh: dkms: not found
```

### iw

```text
phy#1
	Interface wlan1
		ifindex 5
		wdev 0x100000002
		addr bc:a5:11:ba:d2:57
		ssid {obscured}
		type managed
		channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 18.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
	Interface wlan0
		ifindex 4
		wdev 0x100000001
		addr bc:a5:11:ba:d2:57
		type managed
		txpower 18.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

### What happened?

I have a number of Netgear A6210 (mt7612u) devices.

Testing with iperf on various Linux based clients, against my router (Linksys MR8300 running OpenWrt) which is about seven meters away with a clean line of sight, shows a reasonable upload speed:

```
iperf -c {obscured}
Connecting to host {obscured}, port 5201
[  5] local {obscured} port 36056 connected to {obscured} port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  27.6 MBytes   232 Mbits/sec    6    223 KBytes       
[  5]   1.00-2.00   sec  24.5 MBytes   206 Mbits/sec    2    226 KBytes       
[  5]   2.00-3.00   sec  23.4 MBytes   196 Mbits/sec    2    214 KBytes       
[  5]   3.00-4.00   sec  24.5 MBytes   206 Mbits/sec    4    165 KBytes       
[  5]   4.00-5.00   sec  22.2 MBytes   187 Mbits/sec    2    187 KBytes       
[  5]   5.00-6.00   sec  25.6 MBytes   215 Mbits/sec    2    192 KBytes       
[  5]   6.00-7.00   sec  23.5 MBytes   197 Mbits/sec    2    208 KBytes       
[  5]   7.00-8.00   sec  24.6 MBytes   207 Mbits/sec    2    226 KBytes       
[  5]   8.00-9.00   sec  19.1 MBytes   160 Mbits/sec    4    191 KBytes       
[  5]   9.00-10.00  sec  22.4 MBytes   188 Mbits/sec    2    225 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   238 MBytes   199 Mbits/sec   28             sender
[  5]   0.00-10.01  sec   236 MBytes   198 Mbits/sec                  receiver

iperf Done.
```

although there are a number of retries, however the download speed is much less:

```
iperf -c {obscured} -R
Connecting to host {obscured}, port 5201
Reverse mode, remote host {obscured} is sending
[  5] local {obscured} port 34398 connected to {obscured} port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  1.38 MBytes  11.5 Mbits/sec                  
[  5]   1.00-2.00   sec  1.00 MBytes  8.39 Mbits/sec                  
[  5]   2.00-3.00   sec  1.62 MBytes  13.6 Mbits/sec                  
[  5]   3.00-4.00   sec  1.75 MBytes  14.7 Mbits/sec                  
[  5]   4.00-5.00   sec  1.25 MBytes  10.5 Mbits/sec                  
[  5]   5.00-6.00   sec  1.75 MBytes  14.7 Mbits/sec                  
[  5]   6.00-7.00   sec  1.62 MBytes  13.6 Mbits/sec                  
[  5]   7.00-8.00   sec  1.62 MBytes  13.6 Mbits/sec                  
[  5]   8.00-9.00   sec  1.25 MBytes  10.5 Mbits/sec                  
[  5]   9.00-10.00  sec  1.75 MBytes  14.7 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.11  sec  15.8 MBytes  13.1 Mbits/sec   80             sender
[  5]   0.00-10.00  sec  15.0 MBytes  12.6 Mbits/sec                  receiver

iperf Done.
```

Compared to the same clients using a TP-Link Archer T4U (rtl8812au) device:

```
iperf -c {obscured}
Connecting to host {obscured}, port 5201
[  5] local {obscured} port 49242 connected to {obscured} port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  34.8 MBytes   291 Mbits/sec    0   1.00 MBytes       
[  5]   1.00-2.00   sec  34.2 MBytes   287 Mbits/sec    0   1.08 MBytes       
[  5]   2.00-3.00   sec  33.6 MBytes   282 Mbits/sec    0   1.14 MBytes       
[  5]   3.00-4.00   sec  33.4 MBytes   280 Mbits/sec    0   1.42 MBytes       
[  5]   4.00-5.00   sec  33.6 MBytes   282 Mbits/sec    0   1.42 MBytes       
[  5]   5.00-6.00   sec  33.1 MBytes   278 Mbits/sec    0   1.42 MBytes       
[  5]   6.00-7.00   sec  33.9 MBytes   284 Mbits/sec    0   1.42 MBytes       
[  5]   7.00-8.00   sec  33.6 MBytes   282 Mbits/sec    0   1.42 MBytes       
[  5]   8.00-9.00   sec  33.5 MBytes   281 Mbits/sec    0   1.42 MBytes       
[  5]   9.00-10.00  sec  33.6 MBytes   282 Mbits/sec    0   1.42 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   337 MBytes   283 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   335 MBytes   281 Mbits/sec                  receiver

iperf Done.

iperf -c {obscured} -R
Connecting to host {obscured}, port 5201
Reverse mode, remote host {obscured} is sending
[  5] local {obscured} port 40434 connected to {obscured} port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  27.0 MBytes   226 Mbits/sec                  
[  5]   1.00-2.00   sec  28.5 MBytes   239 Mbits/sec                  
[  5]   2.00-3.00   sec  26.2 MBytes   220 Mbits/sec                  
[  5]   3.00-4.00   sec  29.8 MBytes   250 Mbits/sec                  
[  5]   4.00-5.00   sec  29.6 MBytes   249 Mbits/sec                  
[  5]   5.00-6.00   sec  30.9 MBytes   259 Mbits/sec                  
[  5]   6.00-7.00   sec  32.5 MBytes   273 Mbits/sec                  
[  5]   7.00-8.00   sec  32.0 MBytes   268 Mbits/sec                  
[  5]   8.00-9.00   sec  32.5 MBytes   273 Mbits/sec                  
[  5]   9.00-10.00  sec  32.8 MBytes   275 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec   305 MBytes   256 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   302 MBytes   253 Mbits/sec                  receiver

iperf Done.
```

where the upload and download speeds are similar and there are no retries.

I've tested these devices in various clients, using different Linux distros, with different kernel versions (5.x and 6.x), and the results are the same.

Testing using a different remote makes no difference.

Testing using a different Wi-FI channel makes no difference.

Testing using a different USB port makes no difference.

Does anyone know if this is simply expected behavior with this device type, if this is a known bug, or if there is any special configuration required to address the download speeds and number of retries?

Thanks in advance.

morrownr, 2024-09-10:
Here is what I am seeing:

Band 2, 5 GHz
2 walls, approximately meters distance
WiFi router running OpenWRT
Signal level: -55 dBm, not a perfect signal but not bad
Client kernel 6.6
Distro: Debian 12

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.236 port 43926 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  39.2 MBytes   329 Mbits/sec    0    675 KBytes       
[  5]   1.00-2.00   sec  35.0 MBytes   294 Mbits/sec    0    782 KBytes       
[  5]   2.00-3.00   sec  36.2 MBytes   304 Mbits/sec    0    822 KBytes       
[  5]   3.00-4.00   sec  32.5 MBytes   273 Mbits/sec    0    871 KBytes       
[  5]   4.00-5.00   sec  37.5 MBytes   315 Mbits/sec    0    871 KBytes       
[  5]   5.00-6.00   sec  33.8 MBytes   283 Mbits/sec    0    871 KBytes       
[  5]   6.00-7.00   sec  36.2 MBytes   304 Mbits/sec    0    871 KBytes       
[  5]   7.00-8.00   sec  33.8 MBytes   283 Mbits/sec    0    871 KBytes       
[  5]   8.00-9.00   sec  36.2 MBytes   304 Mbits/sec    0    915 KBytes       
[  5]   9.00-10.00  sec  33.8 MBytes   283 Mbits/sec    0    966 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   354 MBytes   297 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   352 MBytes   295 Mbits/sec                  receiver

iperf Done.

```

```
$ iperf3 -c 192.168.1.1 -R
Connecting to host 192.168.1.1, port 5201
Reverse mode, remote host 192.168.1.1 is sending
[  5] local 192.168.1.236 port 44610 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  27.8 MBytes   234 Mbits/sec                  
[  5]   1.00-2.00   sec  38.1 MBytes   320 Mbits/sec                  
[  5]   2.00-3.00   sec  33.6 MBytes   282 Mbits/sec                  
[  5]   3.00-4.00   sec  38.5 MBytes   323 Mbits/sec                  
[  5]   4.00-5.00   sec  34.5 MBytes   289 Mbits/sec                  
[  5]   5.00-6.00   sec  38.8 MBytes   326 Mbits/sec                  
[  5]   6.00-7.00   sec  34.1 MBytes   286 Mbits/sec                  
[  5]   7.00-8.00   sec  38.4 MBytes   322 Mbits/sec                  
[  5]   8.00-9.00   sec  37.7 MBytes   317 Mbits/sec                  
[  5]   9.00-10.00  sec  35.6 MBytes   299 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   359 MBytes   301 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   357 MBytes   300 Mbits/sec                  receiver

iperf Done.

```

This is about what I would expect given the details at my location. I have very little congestion but the signal is not perfect but is not too bad.

43qcc2cn, 2024-09-10:
Many thanks for the quick response. It is good to see what the results should look like.

I've just tested my three devices in a different client, a laptop running OpenSUSE Tumbleweed, which I've positioned a couple of meters away from the router. 5Ghz and -40 dBm.

I still get the same sort of results.

First device:

```
iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.105 port 40742 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  31.5 MBytes   264 Mbits/sec    6    230 KBytes       
[  5]   1.00-2.00   sec  29.2 MBytes   245 Mbits/sec    4    181 KBytes       
[  5]   2.00-3.00   sec  30.5 MBytes   256 Mbits/sec    2    238 KBytes       
[  5]   3.00-4.00   sec  29.4 MBytes   246 Mbits/sec    6    143 KBytes       
[  5]   4.00-5.00   sec  31.6 MBytes   265 Mbits/sec    2    184 KBytes       
[  5]   5.00-6.00   sec  28.1 MBytes   236 Mbits/sec    2    219 KBytes       
[  5]   6.00-7.00   sec  30.4 MBytes   255 Mbits/sec    2    235 KBytes       
[  5]   7.00-8.00   sec  29.1 MBytes   244 Mbits/sec    2    233 KBytes       
[  5]   8.00-9.00   sec  31.5 MBytes   264 Mbits/sec    2    257 KBytes       
[  5]   9.00-10.00  sec  29.6 MBytes   248 Mbits/sec    4    202 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   301 MBytes   252 Mbits/sec   32             sender
[  5]   0.00-10.01  sec   298 MBytes   250 Mbits/sec                  receiver

iperf Done.

iperf3 -c 192.168.1.1 -R
Connecting to host 192.168.1.1, port 5201
Reverse mode, remote host 192.168.1.1 is sending
[  5] local 192.168.1.105 port 58910 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  5.75 MBytes  48.2 Mbits/sec                  
[  5]   1.00-2.00   sec  8.25 MBytes  69.2 Mbits/sec                  
[  5]   2.00-3.00   sec  12.8 MBytes   107 Mbits/sec                  
[  5]   3.00-4.00   sec  16.9 MBytes   142 Mbits/sec                  
[  5]   4.00-5.00   sec  16.1 MBytes   135 Mbits/sec                  
[  5]   5.00-6.00   sec  11.1 MBytes  93.3 Mbits/sec                  
[  5]   6.00-7.00   sec  13.0 MBytes   109 Mbits/sec                  
[  5]   7.00-8.00   sec  8.38 MBytes  70.3 Mbits/sec                  
[  5]   8.00-9.00   sec  7.25 MBytes  60.8 Mbits/sec                  
[  5]   9.00-10.00  sec  5.88 MBytes  49.3 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   108 MBytes  90.1 Mbits/sec   51             sender
[  5]   0.00-10.00  sec   105 MBytes  88.4 Mbits/sec                  receiver

iperf Done.
```

Second device:

```
iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.159 port 34804 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  28.5 MBytes   239 Mbits/sec    6    229 KBytes       
[  5]   1.00-2.00   sec  24.5 MBytes   206 Mbits/sec    4    175 KBytes       
[  5]   2.00-3.00   sec  29.0 MBytes   243 Mbits/sec    2    209 KBytes       
[  5]   3.00-4.00   sec  23.2 MBytes   195 Mbits/sec    2    229 KBytes       
[  5]   4.00-5.00   sec  31.5 MBytes   264 Mbits/sec    0    318 KBytes       
[  5]   5.00-6.00   sec  30.1 MBytes   253 Mbits/sec    4    218 KBytes       
[  5]   6.00-7.00   sec  29.2 MBytes   245 Mbits/sec    2    215 KBytes       
[  5]   7.00-8.00   sec  30.2 MBytes   254 Mbits/sec    2    235 KBytes       
[  5]   8.00-9.00   sec  32.0 MBytes   268 Mbits/sec    2    245 KBytes       
[  5]   9.00-10.00  sec  29.1 MBytes   244 Mbits/sec    4    225 KBytes       
[... 57 lines cut ...]
[  5] local 192.168.1.190 port 50338 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  5.00 MBytes  41.9 Mbits/sec                  
[  5]   1.00-2.00   sec  5.38 MBytes  45.1 Mbits/sec                  
[  5]   2.00-3.00   sec  7.75 MBytes  65.0 Mbits/sec                  
[  5]   3.00-4.00   sec  7.00 MBytes  58.7 Mbits/sec                  
[  5]   4.00-5.00   sec  7.25 MBytes  60.8 Mbits/sec                  
[  5]   5.00-6.00   sec  6.50 MBytes  54.5 Mbits/sec                  
[  5]   6.00-7.00   sec  7.25 MBytes  60.8 Mbits/sec                  
[  5]   7.00-8.00   sec  5.75 MBytes  48.2 Mbits/sec                  
[  5]   8.00-9.00   sec  4.00 MBytes  33.6 Mbits/sec                  
[  5]   9.00-10.00  sec  6.25 MBytes  52.4 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec  63.6 MBytes  53.3 Mbits/sec  121             sender
[  5]   0.00-10.00  sec  62.1 MBytes  52.1 Mbits/sec                  receiver

iperf Done.
```

I'm at a complete loss as to why I'm not getting the same results as yourself.

The devices were purchased at different times from different suppliers so I can't believe they are all bad.

I get real problems trying to stream any kind of video (eg. online TV or video conferences) using the devices, which prompted my investigations, making them pretty much unusable to me.

43qcc2cn, 2024-09-10:
Thanks for following up.

- Are you using one of the usb cables that comes with the A6210?
I've tried with and without the cables supplied with the Netgear A6210. No difference.

- Do you know the approximate level of congestion at your site on the channel you are using?
My router has two 5 GHz radios. The first 5 GHz radio (QCA9888), is limited to ch. 100 and above (currently 144 5.720GHz). The second 5 GHz radio (IPQ4019) is limited to ch. 64 and below (currently 36 5.180Ghz). The first has no congestion, the second has lots. I'm in an apartment block. I've tested both, no difference.

  I also took my router out of the equation and tested using my provider's router, this resulted in awful figures for all connections so I am ignoring these results.

- Have you tried a difference channel?
Yes, although most of the channels used by the second radio have some sort of congestion. No difference.

- Have you tried different security settings? WPA2 vs WPA3?
I've had problems in the past using WPA3 so I'm only using WPA2.

- Are your clients all using Linux? If so, what kernel?
All Linux. All 6.x. eg 6.6.46, 6.10.8-1.

  I did also test using an old version of OpenSUSE Leap running a 5.x kernel.

- What driver is the rtl8812au adapter using?
OpenSUSE Leap - As provided by the distro.
OpenSUSE Tumbleweed - As provided by the distro.
LibreELEC - As provided by the distro.

- Are you testing different usb ports?
Yes, both USB2 and USB3 ports in difference machines. No difference.

- Are you using the rtl8812au adapter with a usb cable?
I've tried with and without the supplied cable supplied with the Netgear A6210. Both work fine.

- Have you tested the Netgear adapter directly plugged into the test clients?
I've tried with and without the supplied cable supplied with the Netgear A6210. No difference.

In summary, whatever client machine I use, whatever USB port I use, whether I use the USB extension cable or not, whether I use radio 1 or radio 2 on various different channels, the rtl8812au devices work fine every time and the mt7612u devices always results in much slower download speeds with lots of retries. Weird!

Can congestion affect one type of adapter but not another?

morrownr, 2024-09-10:
I don't have much time right  now to explain but I think I saw this on a laptop this morning:

Laptop is running Ubuntu 24.04 with kernel 6.08. I made a mistake earlier when I told you my main dev box here is running kernel 6.6... I actually had selected kernel 6.9 last time I had booted and forgot about it. I do a lot of testing so have many kernel installed.

So, I am not seeing this with kernel 6.9. I can test more kernels later but I did see if on a laptop with kernel 6.8. So, the investigation is underway... when I have time to work it again...

43qcc2cn, 2024-09-11:
Thanks again.

For info, I installed Debian on my laptop and tested using the following kernels:

6.1.0
6.6.8
6.6.15
6.7.7
6.7.12
6.8.9
6.8.12
6.9.7
6.9.12
6.10.3
6.10.9

The results were the same for every kernel.

morrownr, 2024-09-11:
Good info to add to what we have. One thing we need to keep in mind is that when testing we may need to take the systems down to a cold boot between tests.

I am busy but will continue as able. I do not think that it is your adapters that are faulty. My best guess at this point is that it could be something in the driver, wifi stack or usb stack. We have been seeing a lot of wifi problems lately and a lot of patches but it can take time for patches to work their way to us. I will be keeping an eye open and will continue testing. Keep me posted.

FYI: I have other adapters with the mt7612u so I can test them as well.

43qcc2cn, 2024-12-03:
A quick update just to say that, having replaced the ath10k-ct firmware with the standard ath10k firmware on my OpenWrt router, I am now getting a much improved connection (both up & down speed plus stability) on all my client devices & adapters. For example, using one of the Netgear A6210 (mt7612u) devices now shows:

```
iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.105 port 47626 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  34.0 MBytes   285 Mbits/sec    8    171 KBytes       
[  5]   1.00-2.00   sec  34.2 MBytes   287 Mbits/sec    2    228 KBytes       
[  5]   2.00-3.00   sec  33.2 MBytes   279 Mbits/sec    2    266 KBytes       
[  5]   3.00-4.00   sec  35.4 MBytes   297 Mbits/sec    4    208 KBytes       
[  5]   4.00-5.00   sec  30.5 MBytes   256 Mbits/sec    4    170 KBytes       
[  5]   5.00-6.00   sec  31.2 MBytes   262 Mbits/sec    2    208 KBytes       
[  5]   6.00-7.00   sec  33.2 MBytes   279 Mbits/sec    4    212 KBytes       
[  5]   7.00-8.00   sec  30.0 MBytes   252 Mbits/sec    4    173 KBytes       
[  5]   8.00-9.00   sec  37.0 MBytes   310 Mbits/sec    2    252 KBytes       
[  5]   9.00-10.00  sec  30.4 MBytes   255 Mbits/sec    4    223 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   329 MBytes   276 Mbits/sec   36             sender
[  5]   0.00-10.01  sec   328 MBytes   275 Mbits/sec                  receiver

iperf Done.

iperf3 -c 192.168.1.1 -R
Connecting to host 192.168.1.1, port 5201
Reverse mode, remote host 192.168.1.1 is sending
[  5] local 192.168.1.105 port 54250 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  27.6 MBytes   232 Mbits/sec                  
[  5]   1.00-2.00   sec  31.1 MBytes   261 Mbits/sec                  
[  5]   2.00-3.00   sec  29.1 MBytes   244 Mbits/sec                  
[  5]   3.00-4.00   sec  27.4 MBytes   230 Mbits/sec                  
[  5]   4.00-5.00   sec  31.4 MBytes   263 Mbits/sec                  
[  5]   5.00-6.00   sec  30.8 MBytes   258 Mbits/sec                  
[  5]   6.00-7.00   sec  32.6 MBytes   274 Mbits/sec                  
[  5]   7.00-8.00   sec  28.2 MBytes   237 Mbits/sec                  
[  5]   8.00-9.00   sec  31.1 MBytes   261 Mbits/sec                  
[  5]   9.00-10.00  sec  26.8 MBytes   224 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec   300 MBytes   251 Mbits/sec  368             sender
[  5]   0.00-10.00  sec   296 MBytes   248 Mbits/sec                  receiver

iperf Done.
```

Still seeing lots of retransmissions with the Netgear A6210 (mt7612u) devices but the actual usage experience is significantly better.

morrownr, 2024-12-03:
@43qcc2cn 

I have a ath10k based and my experience is the same as yours. The -ct version may have been better at some point in years past but it certainly is not these days. I have to wonder why OpenWRT does not make the default switch on many systems.