morrownr/USB-WiFi #507: [Help]: bluetoothctl scan on: No default controller available

HamzaHajeir, 2024-09-16:
### Checklist

- [X] I acknowledge that support is provided on a best-effort basis.
- [X] I acknowledge that the authors and contributors to this repository cannot be held responsible for the results of my use of any information contained in or linked from this repository.

### uname

Linux HamzaHajeir 5.15.153.1-microsoft-standard-WSL2+ #3 SMP Sun Sep 15 23:57:54 +03 2024 x86_64 x86_64 x86_64 GNU/Linux

### lsusb

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub Bus 001 Device 003: ID 0e8d:0608 MediaTek Inc. Wireless_Device Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

### rfkill

1: hci0: Bluetooth         Soft blocked: no         Hard blocked: no

### dkms

```text
[NO OUTPUT]
```

### iw

```text
nl80211 not found.
```

### What happened?

I have WSL and want to manage bluetooth from within it. 

I've followed [this instruction][1] to build the custom WSL, including all the required checks into .config file.

Here's the latest .config file content that built the latest WSL firmware: https://pastebin.com/5CBrBf7c

And have used usbipd to share and attach the bluetooth device through: 
- `usbipd bind --busid={BUSID}`
- `usbipd attach --wsl --busid={BUSID}`

Where both run successfully, and:
- `usbipd list` gives:
```
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-9    0e8d:0608  RZ608 Bluetooth(R) Adapter                                    Attached
1-10   0b05:1939  AURA LED Controller, USB Input Device                         Not shared
2-3    045e:07a5  USB Input Device                                              Not shared
2-4    045e:0745  Microsoft Hardware USB Keyboard, USB Input Device             Not shared
```

However, running `bluetoothctl scan on` result in `No default controller available`.

My finding concludes that btmtk module is missing within the system. And I don't know how to find/have/run it.

From my research I've registered many useful commands, here's the outputs of them: 

- `uname -a`:
```
Linux HamzaHajeir 5.15.153.1-microsoft-standard-WSL2+ #3 SMP Sun Sep 15 23:57:54 +03 2024 x86_64 x86_64 x86_64 GNU/Linux
```

- `lsb_release -a`
```
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04.1 LTS
Release:        24.04
Codename:       noble
[... 446 lines cut ...]
```
Sep 16 00:05:39 HamzaHajeir kernel: Bluetooth: Core ver 2.22
Sep 16 00:05:39 HamzaHajeir kernel: Bluetooth: Starting self testing
Sep 16 00:05:39 HamzaHajeir kernel: Bluetooth: ECDH test passed in 2653 usecs
Sep 16 00:05:39 HamzaHajeir kernel: Bluetooth: SMP test passed in 1100 usecs
Sep 16 00:05:39 HamzaHajeir kernel: Bluetooth: Finished self testing
Sep 16 00:05:39 HamzaHajeir kernel: NET: Registered PF_BLUETOOTH protocol family
Sep 16 00:05:39 HamzaHajeir kernel: Bluetooth: HCI device and connection manager initialized
Sep 16 00:05:39 HamzaHajeir kernel: Bluetooth: HCI socket layer initialized
Sep 16 00:05:39 HamzaHajeir kernel: Bluetooth: L2CAP socket layer initialized
Sep 16 00:05:39 HamzaHajeir kernel: Bluetooth: SCO socket layer initialized
Sep 16 00:05:39 HamzaHajeir kernel: bluetooth hci0: Direct firmware load for mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin failed with error -2
Sep 16 00:05:39 HamzaHajeir kernel: Bluetooth: hci0: Failed to load firmware file (-2)
Sep 16 00:05:39 HamzaHajeir kernel: Bluetooth: hci0: Failed to set up firmware (-2)
Sep 16 19:56:59 HamzaHajeir kernel: bluetooth hci0: Direct firmware load for mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin failed with error -2
Sep 16 19:56:59 HamzaHajeir kernel: Bluetooth: hci0: Failed to load firmware file (-2)
Sep 16 19:56:59 HamzaHajeir kernel: Bluetooth: hci0: Failed to set up firmware (-2)
```

What could be the issue? And how to solve it?

Note that I've tried running the bluetooth within Ubuntu from VMBox and it does work from within. Ofcourse when the bluetooth adapter not attached to WSL.

  [1]: https://github.com/dorssel/usbipd-win/wiki/WSL-support
  [2]: https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

morrownr, 2024-09-16:
Hi @HamzaHajeir 

You are doing some things that I have no experience with. A quick look and the only thing that stands out is the following:

```
BT_RAM_CODE_MT7961_1_2_hdr.bin
BT_RAM_CODE_MT7961_1_2_hdr.bin.zst
BT_RAM_CODE_MT7961_1a_2_hdr.bin
BT_RAM_CODE_MT7961_1a_2_hdr.bin.zst
WIFI_MT7961_patch_mcu_1_2_hdr.bin
WIFI_MT7961_patch_mcu_1_2_hdr.bin.zst
WIFI_RAM_CODE_MT7961_1.bin
WIFI_RAM_CODE_MT7961_1.bin.zst

```

That tells me you have probably updated the firmware but having two versions of each firmware file may or may not be bad but I have not tested it. You might want to delete the .zst files.

I'll ponder this issue.

HamzaHajeir, 2024-09-16:
@morrownr  Thanks for your quick response. 

AFAIK the zst files are just compressed files, not executable ones, therefore I don't see an issue (But maybe I'm wrong?). 

What I'm wondering is the question what triggers btmtk module? Is it these firmware files or another source? In the Ubuntu that runs over VMBox (And as I've seen in other issues while searching), btmtk presents when running `lsmod | grep -i bt`.

morrownr, 2024-09-16:
> AFAIK the zst files are just compressed files...

Yes, compressed binary files. The kernel reads compressed and non-compressed binary files. You have both. Tell me which one the kernel is using?

HamzaHajeir, 2024-09-17:
> Yes, compressed binary files. The kernel reads compressed and non-compressed binary files. You have both. Tell me which one the kernel is using?

I've just renamed the .bin files and here's:
- `ls -a | grep MT7961`
```
BT_RAM_CODE_MT7961_1_2_hdr-COPY-.bin
BT_RAM_CODE_MT7961_1_2_hdr.bin.zst
BT_RAM_CODE_MT7961_1a_2_hdr-COPY-.bin
BT_RAM_CODE_MT7961_1a_2_hdr.bin.zst
WIFI_MT7961_patch_mcu_1_2_hdr-COPY-.bin
WIFI_MT7961_patch_mcu_1_2_hdr.bin.zst
WIFI_RAM_CODE_MT7961_1-COPY-.bin
WIFI_RAM_CODE_MT7961_1.bin.zst
```

- `journalctl --dmesg --boot=-0 --grep blue`
```
Sep 17 20:55:10 HamzaHajeir kernel: Bluetooth: Core ver 2.22
Sep 17 20:55:10 HamzaHajeir kernel: Bluetooth: Starting self testing
Sep 17 20:55:10 HamzaHajeir kernel: Bluetooth: ECDH test passed in 2607 usecs
Sep 17 20:55:10 HamzaHajeir kernel: Bluetooth: SMP test passed in 882 usecs
Sep 17 20:55:10 HamzaHajeir kernel: Bluetooth: Finished self testing
Sep 17 20:55:10 HamzaHajeir kernel: NET: Registered PF_BLUETOOTH protocol family
Sep 17 20:55:10 HamzaHajeir kernel: Bluetooth: HCI device and connection manager initialized
Sep 17 20:55:10 HamzaHajeir kernel: Bluetooth: HCI socket layer initialized
Sep 17 20:55:10 HamzaHajeir kernel: Bluetooth: L2CAP socket layer initialized
Sep 17 20:55:10 HamzaHajeir kernel: Bluetooth: SCO socket layer initialized
Sep 17 20:55:10 HamzaHajeir kernel: bluetooth hci0: Direct firmware load for mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin failed with error -2
Sep 17 20:55:10 HamzaHajeir kernel: Bluetooth: hci0: Failed to load firmware file (-2)
Sep 17 20:55:10 HamzaHajeir kernel: Bluetooth: hci0: Failed to set up firmware (-2)
```

It seems the fail line is reduced by one :).

morrownr, 2024-09-17:
> Even when flipping the active one (renaming back the .bin files and renaming .zst files), produces the same error line.

The firmware issue I mentioned is likely not the source of your problem. I'll do some more research on that and update my firmware guide but let's drop that issue for now.

> I have WSL and want to manage bluetooth from within it.

I do not know what WSL is. Can you enlighten me?

FYI: This site does have heavy traffic but it is not a social media site so it can take some time for replies on specific topics... especially on a topic like this. Why? My history started with learning to program with FORTRAN, My history of small computer operating systems is: CP/M. PC-DOS, DR-DOS, OS/2 and Linux. I have almost no experience with Windows. There are a lot of very smart people, including Linux kernel devs, that stop by at times and some may have Windows experience but this may not be the optimal place for you to get help... but I don't know that for sure.

HamzaHajeir, 2024-09-18:
> I do not know what WSL is. Can you enlighten me?

It's Windows Subsystem for Linux. I'm also new into using this. 

I'm new to it. But the experience is too much better than VM.

I might consider immigrating into Linux sometime in future, not ready for that atm. Setup another OS? my memory is almost full :\.

> FYI: This site does have heavy traffic but it is not a social media site so it can take some time for replies on specific topics... especially on a topic like this. Why? My history started with [...]

I see your point, I appreciate your trying to help out. 

Bue me also don't know where the most suitable place to ask. [WSL2 repository](https://github.com/microsoft/WSL2-Linux-Kernel) have issues section disabled.

morrownr, 2024-09-19:
> But I also don't know where the most suitable place to ask.

I understand. Maybe this is a good place if you are not in a hurry.

Where would I go if I want to learn more about WSL?

> I might consider immigrating into Linux sometime in future...

I've been using Linux since 1994 so I have some experience... kidding. Something to consider: One of the best ways to learn Linux is to grab an old desktop or laptop that is still functional but not used anymore... those can usually be found found in many places for nothing... maybe in your closet. My main dev box here is 11 years old and it is faster than I need. My point being is that we can get you going with Linux on a system that would support Windows 11, 10 or ever 8... probably 7. That could save you the trouble of messing with WSL.

HamzaHajeir, 2024-09-19:
> I understand. Maybe this is a good place if you are not in a hurry.
> 
> Where would I go if I want to learn more about WSL?

Thank you for your patience and help. I'm not in a hurry.

Per anything new to me, I personally prefer to watch an introductory video, here's a top on search on Youtube: https://www.youtube.com/watch?v=qYlgUDKKK5A&t=9s . Ofcourse there's more. 

And here's an official document for WSL: https://learn.microsoft.com/en-us/windows/wsl/about. It also provides some videos.

> I've been using Linux since 1994 so I have some experience... kidding. Something to consider: One of the best ways to learn Linux is to grab an old desktop or laptop that is still functional but not used anymore... those can usually be found found in many places for nothing... maybe in your closet. My main dev box here is 11 years old and it is faster than I need. My point being is that we can get you going with Linux on a system that would not support Windows 11, 10 or ever 8... probably 7. That could save you the trouble of messing with WSL.

I see, I have another PC but I have one monitor now, that PC keeps an old state of files there :). I'll check if a too old laptop is working xD.

But per the project I work on, yet speed of compilation matters, but more importantly the apparently new Bluetooth standard (BLE) is supported.

morrownr, 2024-10-20:
What is your status? I have not run across anything that appears to be of use.

HamzaHajeir, 2024-10-20:
> What is your status? I have not run across anything that appears to be of use.

It should be something with WSL, however the assigned person closed the issue with unreasonable argument. https://github.com/microsoft/WSL/issues/12048

morrownr, 2024-10-21:
> Unfortunately that's not something that can be solved from the WSL side.

That means "This would take a lot of work and I don't feel like messing with it."