morrownr/USB-WiFi #508: Info: USB Ports can be a source of problems with USB WiFi adapters

morrownr, 2024-09-16:
Hello all,

I've been helping users of USB WiFi adapters for some years now and I have noticed that problems are generally first blamed on the driver in use. I get that. However, our USB WiFi adapters depend on support from multiple stacks (USB and WiFi) as well as hardware that needs to be compatible. As an example, here is a post in a Debian forum:

https://forums.debian.net/viewtopic.php?t=159402

Note that this user was able to stabilize his USB WiFi adapter after changing BIOS settings and switching to a different port.

What I have noticed over the last few years that contribute to problems for our adapters:

- USB3 gen2 ports can be problematic, especially with older adapter such as the one in the forum post above. The mt7612u is a favorite chip among Linux users but it was released about 10 years ago and could not be tested with modern USB3 gen2 ports because they did not exist at the time.

- USB ports wear over time and various things can happen if contacts are not solid.

- It is always a good idea to try a different USB port if you have a problem.

- There are problematic USB3 hub chips and problematic USB drivers.

- There are USB BIOS settings that can cause problems.

- When in doubt, if you have a USB2 port, give it a try. USB2 is simply more stable than USB3 and in most cases, it is fast enough for the use case.

- Powered USB hubs and extension cables can be problematic. Plug the adapter directly into a port to test if this could be a problem.

- Some USB subsystems to do not provide spec power leaving you in a low power condition. I am looking at you RasPi.

I hope this helps and I am going to pin this issue. Feel free to add your own ideas and experiences.

@morrownr

fhteagle, 2024-10-03:
Agreed with all of the above, and I'll add more detail about a couple of the cases on your list: 

If you're not getting the speed test results you are expecting, it may be that your USB adapter has fallen back to USB2 speeds despite both host and service being USB3+ capable. To check this, use ```lsusb -tv``` (not installed by default in all distros). Look for a speed to the right of your device, something like  "480M" or "5000M". Expect to maybe get 60% of whatever that number says as actual throughout. For example, I have a USB Ethernet NIC that fails down to 480M speeds, and I get a very solid 330Mbps on that NIC.

ea4eoz, 2024-12-12:
I can confirm this.

I use a MT7612U with a 2 meter long USB3 extension cable at my desktop computer to put the adapter in a good position with good signal. I had a ton of problems using it with the extension cable and the only cure was to plug the extension cable into a USB 2.0 port. It worked fine but with reduced throughput. 

After reading https://forums.debian.net/viewtopic.php?t=159402 and recognize some of the errors I went into the BIOS and configured the USB entries in this way:

XHCI Hand-off: Enabled
Legacy USB Support: Disabled

Now the adapter works just fine with this 2 meter long USB extension cable plugged into a USB 3.0 port. Also lsusb confirms it is working at USB 3.0 speeds.

Check your USB settings in your BIOS!

Thank you for the tip.

Andy-45, 2024-12-24:
Can confirm, loose contact to the usb port is a problem:

[image]

marvellz, 2025-01-04:
I also confirm this. My chinese adapter on MT7612U did not start on USB 3.2 ports neither in VB nor in LiveUSB Kali/ParrotOS.
[One of the tips](https://forums.kali.org/archived/showthread.php?128646-SOLVED-Alfa-AWUS036ACM-amp-Kali-2022-4-on-VB) also did not help.

morrownr, 2025-01-04:
Hi @marvellz 

>  maybe you could indicate in the table on which ports you test the adapters?

If you are talking about The Plug and Play List, I do have some of the adapters in each section but not all. If I test the adapter, I test it in USB2 and USB3 ports. I have a Mini PC and a laptop that do not have USB2 ports so I use a USB2 extension cable which ensures the adapter is working USB2 mode.

marvellz, 2025-01-05:
@morrownr, yes, now I see that you are specifying this. It would have been nice to [ask you who writes reviews what port they used](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md#chipset---realtek-rtl8812bu---supported-in-kernel-since-linux-kernel-62-2023-but-kernel-612-2024-or-later-is-recommended-due-to-numerous-stability-and-performance-enhancements).
Today I tested 8812BU/8811CU (although I should have received not 11, but 8812CU, but I think the result would have been the same with it), both of them it started for me (plug-and-play), but did not show network clients when intercepting a handshake in Airgeddon and I still couldn't catch the handshake, on ParrotOS (it is more convenient for me).
I think in a couple of years I will buy 8812BU again, if a driver for it is developed.
In the meantime I'll have to stick with my old RT5370 adapter 😥 it's reliable, keep in mind that it works great on USB 3.2 ports (in VB/LiveUSB).
P.S. I also want to take a chance and try 8811AU... But I'm not sure😬

morrownr, 2025-01-05:
Hi @marvellz 

>  It would have been nice to ask you who writes reviews what port they used.

Collecting some information can be problematic in that there are so many variables. Having enough information about USB ports to be useful is difficult because USB hardware is not all the same. It is made by many makers and the chips that run the ports is made by many makers and so on. I'll look to see if there is additional information that I can collect. There is also the option of stopping by here and asking for us to test specific situations.

> In the meantime I'll have to stick with my old RT5370 adapter...

That is a solid little chip and adapters are still available.

>  I also want to take a chance and try 8811AU... But I'm not sure...

Let me see, you mentioned using Kali and Parrot so I'll take a wild guess and say you are doing pen testing or security analysis.

What you did not mention is what kernel you are using. rtw88 has been improved greatly over the last year. I would recommend using kernel 6.12 or later for best results. Work is still in progress. You can install rtw88 from the following site if you want as this site is where the work is done:

https://github.com/lwfinger/rtw88

In case you are wondering, installing the above rtw88 blacklists the internal rtw88 automatically. Work is still underway but you should find many improvements.'

With that said, for security analysis, you might think about an adapter such as the  EDUP EP-AX1672 or some of the other adapters in the mt7921au section of The Plug and Play List. I have one of the 1672's and can say that it is an excellent adapter at a reasonable price. Of course the 1673, that is also in the same list, is reported to be a good adapter at a lower price. The Panda adapter in the same list gets really good reviews also. For what it is worth, I have tested my 1672 on systems with everything from USB2 to USB3.2 and have seen no problems. It is a particularly compatible adapter but the rest of the adapters in the Plug and Play List are pretty good as well. I heavily filter adapters for The Plug and Play List. The ones that make it onto the list are the best of the best.

marvellz, 2025-01-05:
@morrownr, hm, thx, but EDUP is too expensive for me. Considering that I'm just an amateur in this area.
Besides that, I know that many people even have separate laptops (netbooks) for wardriving. I myself had an old netbook on an old Atom CPU with USB 2.0 for a long time and had no problems with chinese adapters.
I only use them, and yes, I understand chinese adaptrs that it's a lottery.
As for the ParrotOS version (let's consider it, I use it more often, I checked the adapter on Kali just for fun), Parrot was the last one, more precisely and 6.0, 6.1, 6.2 (As you understand, these are very fresh kernels, above 6.5), in the latest version they added a lot of drivers, see [changelog](https://www.parrotsec.org/blog/2024-10-23-parrot-6.2-release-notes/):
> realtek-rtl8188eus-dkms 5.3.9 ~ git20230921.3fae723-0parrot2 [upgradable from: 5.3.9 ~ git20230921.3fae723-0parrot1]
realtek-rtl8814au-dkms 5.8.5.1 ~ git20240527.d8208c8-0parrot1 [upgradable from: 5.8.5.1 ~ git20230614.8a1908d-0parrot2]
realtek-rtl88xxau-dkms 5.6.4.2 ~ git20240510.b44d288-0parrot1 [upgradable from: 5.6.4.2 ~ git20231117.a3e0c0b-0parrot1]

And I'm also looking for convenience, so that it works "out of the box" and preferably in VB, I rarely use LiveUSB.
Size is also important, huge adapters-"transformers" are not for me 😁

10$ - price is reasonable, maybe it's worth the risk? 😁

[image]

bjlockie, 2025-01-05:
That looks like the Comfast one I have.
Go for it.

castillofrancodamian, 2025-01-05:
> @morrownr, hm, thx, but EDUP is too expensive for me. Considering that I'm just an amateur in this area. Besides that, I know that many people even have separate laptops (netbooks) for wardriving. I myself had an old netbook on an old Atom CPU with USB 2.0 for a long time and had no problems with chinese adapters. I only use them, and yes, I understand chinese adaptrs that it's a lottery. As for the ParrotOS version (let's consider it, I use it more often, I checked the adapter on Kali just for fun), Parrot was the last one, more precisely and 6.0, 6.1, 6.2 (As you understand, these are very fresh kernels, above 6.5), in the latest version they added a lot of drivers, see [changelog](https://www.parrotsec.org/blog/2024-10-23-parrot-6.2-release-notes/):
> 
> > realtek-rtl8188eus-dkms 5.3.9 ~ git20230921.3fae723-0parrot2 [upgradable from: 5.3.9 ~ git20230921.3fae723-0parrot1]
> > realtek-rtl8814au-dkms 5.8.5.1 ~ git20240527.d8208c8-0parrot1 [upgradable from: 5.8.5.1 ~ git20230614.8a1908d-0parrot2]
> > realtek-rtl88xxau-dkms 5.6.4.2 ~ git20240510.b44d288-0parrot1 [upgradable from: 5.6.4.2 ~ git20231117.a3e0c0b-0parrot1]
> 
> And I'm also looking for convenience, so that it works "out of the box" and preferably in VB, I rarely use LiveUSB. Size is also important, huge adapters-"transformers" are not for me 😁
> 
> 10$ - price is reasonable, maybe it's worth the risk? 😁
> 
> ![scr_ali_mt7921au](https://private-user-images.githubusercontent.com/48913829/400256325-65043fc0-4658-4eca-9396-1206449cdbb5.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxMTg1NzcsIm5iZiI6MTczNjExODI3NywicGF0aCI6Ii80ODkxMzgyOS80MDAyNTYzMjUtNjUwNDNmYzAtNDY1OC00ZWNhLTkzOTYtMTIwNjQ0OWNkYmI1LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA1VDIzMDQzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk4MjAyOGNhZjM5ODY1ZWY4NjFjZjYzMzAzYTk2NGYxMzQ4NWRmYjRhMDIzMTFiOGNjMDBhMzFjYzNlODFhZjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.NY4mqqn8RkfreRBvXuD5828n7BVIzbP4XlJcO5c5JIc)

It doesn't have good reception.

marvellz, 2025-01-05:
> That looks like the Comfast one I have.

Is everything bad?)))

> It doesn't have good reception.

This is one of the reasons why I never bought Alpha. I was always told that Alpha is more powerful, it shows more networks, but I don't need it, I don't need a network at 100-200 meters, so I think it's not critical for me. Or what do you mean by weak reception? If that adapter it sees network and catches handshake at 15 meters, that's enough for me.

morrownr, 2025-01-05:
@marvellz 

> 10$ - price is reasonable, maybe it's worth the risk?

It looks like a whitebox version of the Fenvi FU-AX1800. Not everyone needs an adapter with long range so a <=$10 adapter may work fine. Nobody should expect high quality amps and antennas in a low cost adapter.

sam8641, 2025-01-11:
I can confirm all USB ports on some of my USB controllers are bad with slow speed and stability issues. Used Realtek USB 2.5g ethernet (0bda:8156), USB wifi mt7921u, and USB HDMI 4K Cam Link capture device, all devices works fine on good USB controllers.

On AM4, 4 USB ports comes from CPU, the rest comes from chipset. Total number of USB controller is 2, 3, or 4; depends on CPU plus chipset. To determine which port is which controller: watch -n 0.2 lsusb -tv.

My bad USB controllers:
- Asus Prime B350 plus: 1 USB controller. Slow receive speed, USB timeout errors. With USB 2.0 Bluetooth BCM20702A0 plugged in on chipset USB port and reboot, then 20% chance for most or all USB ports connected to chipset to not work.
- Ryzen 2700x: 1 USB controller 4 ports. mt7921u slow 350 Mbps receive speed and 2.5g USB ethernet packets loss. 4 bad USB ports follows this CPU.
- ASM1042 chips on Asus M5A97 R2.0: The only SuperSpeed devices that worked for me is: USB flash drives. Now, motherboard stopped working and is thrown away.

My questionable stability of USB controllers:
- Renesas uPD720202 PCIe USB 2 ports: mt7921u works good. Using iperf --bidir on USB 2.5g ethernet for an hour have random USB controller fail that need reboot. Possible overheat, this PCIe card have no heatsink.
- MSI B450 Tomahawk: With USB ethernet plugged in, rare random USB controller fail that need reboot. Linux boot paremeter iommu.strict=1 appears to help avoid a complete Linux crash when USB fails.

My good USB controllers.
- Ryzen 3900x: Good, 1 USB controller.
- Ryzen 2400g: Good, 2 USB controllers, 1+3 SuperSpeed USB ports from CPU.
- Asrock Steel Legend X570 wifi: Good, 2 USB controllers.
- Asus TUF gaming X570 plus wifi: Good, 2 USB controllers.
- Asus P8H77-M, intel i7-2600: 1 USB controller, 3 of 4 SuperSpeed ports is good. One of the port on SuperSpeed header is completely dead.

All my bad USB controllers are more than 5 years old. This may be a possible old age chip degradation.

marvellz, 2025-01-11:
@sam8641, honestly? I don't care about speed. Such a chip MT7921 is taken not for speed, but to support monitor and injection modes.
So that it works on Linux without compiling drivers, out of the box.
Obviously, we are not taking it for the Internet.
But thx for the info.

morrownr, 2025-01-11:
@marvellz 

I think @sam8641 was adding to the original post that started this thread. I did not take it as an answer to you. Maybe I am wrong.

marvellz, 2025-01-21:
> USB3.2 and have seen no problems.

Thx, @morrownr, I received the adapter. It's great! I appreciate your work, thanks for updating the info!
P.S. I just had to look for a driver for Win, I took it from [Fenvi AX1800](https://www.fenvi.com/drive.html?keyword=FU-AX1800).

morrownr, 2025-01-21:
@marvellz 

Glad it worked out for you. Let us know how the adapter works for you over the longer term.

marvellz, 2025-01-23:
>  Let us know how the adapter works for you over the longer term.

Yes, there is a strange thing, in LiveUSB (ParrotOS v6.2-6.3.1) it works worse than in VB, which is strange... Usually it's the other way around.
Sees fewer networks, as if there is not enough power. But it’s not critical for me, I prefer VB.
Well, and a [well-known problem with the operation of PMKID](https://github.com/ZerBea/hcxdumptool/issues/376).