morrownr/USB-WiFi #509: Fenvi AX1800 (MT7921)

RikudouGoku, 2024-09-17:
Hopefully this is ok but I just wanted to share my first impressions on the Fenvi AX1800 (with MT7921) and how I got it working on Linux Mint 22 (kernel 6.8.0.41). 

At first, just plugging it in did not work and i could not see the wifi category in the network settings, however it did show up in lsusb under: ID 0e8d:7961 MediaTek Inc. Wireless_Device. 

The only thing I needed to do was follow section 3 from this guide: https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md
and download WIFI_MT7961_patch_mcu_1_2_hdr.bin and WIFI_RAM_CODE_MT7961_1.bin and then put them inside /lib/firmware/mediatek and then reboot. Then I could connect to wifi. 

As for performance
[image]
it performs better than my TP Link Archer T3U Plus.
[image]

Thanks @morrownr for all the work you have done on this massive list. 
(and please forgive me if I did something wrong with submitting this info as this is the first time I have done this.)

morrownr, 2024-09-17:
Hi @RikudouGoku 

> Hopefully this is ok but I just wanted to share my first impressions on the Fenvi AX1800...

Your report is very welcome. Getting information to Linux users is one of the core reasons that this site exist. Thank you.

> ... how I got it working on Linux Mint 22 (kernel 6.8.0.41).

This should not have happened so it would be good if we can figure out what happened. Mint 22 is based on Ubuntu 24.04. I have tested on Ubuntu 24.04 and did not see any problems with adapters that use the mt7921u driver. FYI: I use Mint Debian Edition myself as my workhorse distro and have not seen problems there. Hmmm...

> The only thing I needed to do was follow section 3 [of the Firmware guide]...

Hmmm... we don't know if the files were missing but I can check on that. Is Ubuntu compressing the files?

Run and post the results of the following:

$ ls -l /lib/firmware/mediatek | grep MT7961

Some distros are compressing the firmware files these days and that is leading to some confusion. I added a note to the firmware guide yesterday that brings this issue to the attention of users but I need to do some testing. What if users update their firmware in accordance with the guide and don't know if the existing files are compressed? They end up with 2 sets of files which leads to the question: Which files will the kernel use? I need to get to the bottom of this.

Thanks again for the report and enjoy your adapter. I have had several positive reports from users of that Fenvi adapter so the adapter remains in the Plug and Play List. In the future, after using the adapter for a while, feel free to come back here and add to this report.

RikudouGoku, 2024-09-17:
@morrownr 
[image]

Looks like there was one already there? 

Will update my thoughts on it in the future (if I remember), reason why I bought this in the first place is because for some reason the TP Link Archer T3U Plus had frequent disconnection problems so I needed to remove and plug it in again to make it work, this only happened in Linux Mint I had no such issues with it on my Windows 10. I have Windows 10 in my internal SSD drive and am currently running Linux Mint via an NVME SSD+Ugreen SSD case since I wanted to try to live with Linux for a while to see if I want to ditch Windows for my next pc I will be likely buying next year (this is 11 years old atm).

morrownr, 2024-09-18:
@RikudouGoku 

> I wanted to try to live with Linux for a while to see if I want to ditch Windows for my next pc I will be likely buying next year (this is 11 years old atm).

I like to keep my hardware in operation for a long time also. One of the really big benefits of Linux is being able to do exactly that. My printer and scanner are over 15 years old and as I look around, I have many things that are really old but still work fine with Linux. Many have no modern Windows drivers. To increase the odds of long term use, it can be a good idea to research the computer and components before buying. I've been using Linux since 1994 so if you need help, contact me here before you buy.

Back to topic:

> Looks like there was one already there?

Yes it does. Many distros are compressing their firmware these days as the kernel supports this. Unfortunately the kernel supports several methods of compression so the distros use various compression methods. Since the compressed file use different filenames, when you copy the uncompressed files to the directory, the compressed files are still there. I suspect the uncompressed files take precedence over the compressed files but I need to test this theory.

You could help test if you want:

Run the following and post the results:

ethtool -i <interface name> 

Note: You may need to install the ethtool and iw packages depending on your distro.

Then rename `WIFI_RAM_CODE_MT7961_1.bin` to `WIFI_RAM_CODE_MT7961_1-.bin` and rename `WIFI_MT7961_patch_mcu_1_2_hdr.bin` to `WIFI_MT7961_patch_mcu_1_2_hdr-.bin`, reboot and run ethtool again to see what version you get. Please post the results so we can see the difference.

RikudouGoku, 2024-09-18:
> I like to keep my hardware in operation for a long time also. One of the really big benefits of Linux is being able to do exactly that. My printer and scanner are over 15 years old and as I look [...]

Thank you, this is the potential build right now: https://www.inet.se/datorbygge/b1556134/beast-r5-7600-a750-v2
The GPU might or might not change when those new Intel Battlemage GPUs come. 

> You could help test if you want:
> 
> Run the following and post the results:
> 
> ethtool -i
> 
> Note: You may need to install the ethtool and iw packages depending on your distro.
> 
> Then rename `WIFI_RAM_CODE_MT7961_1.bin` to `WIFI_RAM_CODE_MT7961_1-.bin` and rename `WIFI_MT7961_patch_mcu_1_2_hdr.bin` to `WIFI_MT7961_patch_mcu_1_2_hdr-.bin`, reboot and run ethtool again to see what version you get. Please post the results so we can see the difference.

I will do it but I need to install it, can you share the code I need to use to do so? Step by step would be appreciated.

I will do a timeshift backup right now though (doesnt seem risky though).

morrownr, 2024-09-18:
> I will do it but I need to install it, can you share the code I need to use to do so? Step by step would be appreciated.

Sure. You are on Linux Mint so that is Debian based...

To bring up a terminal:

Ctrl-Alt-T

Next install the ethtool and iw packages:

$ sudo apt install ethtool iw

Next run iw:

$ iw dev

Note the name of your wifi interface. It should be something like `wlx0013ef6f0a98`.

Then run ethtool:

$ ethtool -i wlx0013ef6f0a98

...but replace wlx0013ef6f0a98 with your interface name.

Post the result back in this thread.

I need to run for now so we can continue later.

RikudouGoku, 2024-09-18:
> Post the result back in this thread.

[image]

dubhater, 2024-09-25:
> reason why I bought this in the first place is because for some reason the TP Link Archer T3U Plus had frequent disconnection problems so I needed to remove and plug it in again to make it work

Hey, @RikudouGoku . Can you tell me more about that? I assume it uses the RTL8812BU chip. How frequently was it disconnecting? Was the list of networks empty? Is that why you needed to plug it again?

RikudouGoku, 2024-09-25:
> Hey, @RikudouGoku . Can you tell me more about that? I assume it uses the RTL8812BU chip. How frequently was it disconnecting? Was the list of networks empty? Is that why you needed to plug it again?

I believe it disconnected almost every day at least once. By disconnecting, i mean that even with the wifi symbol showing it is connected, I could not actually connect to the internet and stuff like discord got disconnected and cant reload browser pages. I do not really remember if the list of wifi networks was empty though. But physically unplugging it and replugging it solved it every time. I believe this happened only when my pc was downloading though like steam or torrents. 

On that note, so far after around a week with this Fenvi AX1800 adapter, I have had ZERO issues.

dubhater, 2024-09-25:
> I believe this happened only when my pc was downloading though like steam or torrents.

What torrent client do you use? Are you sure Steam by itself was enough to break the wifi?

> On that note, so far after around a week with this Fenvi AX1800 adapter, I have had ZERO issues.

I think you ran into the same problem I have, which is likely in the USB side of rtw88, the Realtek wifi driver. Your Fenvi adapter uses a Mediatek driver, so you should not have that problem anymore.

RikudouGoku, 2024-09-25:
> What torrent client do you use? Are you sure Steam by itself was enough to break the wifi?
> 
> >

Qbittorrent. Yeah I believe the issue came when my speed got over like 5MB/s, thats when the risk of it disconnecting happened I think. Under that and I dont think it had a problem. None of that happened over on windows 10 btw.

> I think you ran into the same problem I have, which is likely in the USB side of rtw88, the Realtek wifi driver.

Yeah I suspect the driver as well, I did try this solution: https://gist.github.com/julianlam/0ce7692ca10fb91970b6693bc02587ce
Someone else with the EU version but it made no difference for me but seems to have worked for them: https://gist.github.com/julianlam/0ce7692ca10fb91970b6693bc02587ce?permalink_comment_id=4616884#gistcomment-4616884